libusb/libusb #1228: On Android platform, the packet receiving interval is not stable enough

atian123kiss, 2023-01-12:
On the Android platform, when I use libUsb to receive firmware packets, I find that sometimes the interval between the received timestamp and the timestamp in the packet cannot be aligned.
For example, the interval in the data packet is always 2 ms, but the actual interval I receive fluctuates between 6 ms and 1 ms. It seems that sometimes the data is blocked, and then the next packet is received quickly. Finally, the total data interval is consistent. I tested the packet interval on the computer platform and did not find this problem.
Our transmission protocol is CDC protocol, and the maximum transmission length of the device is 64 bytes at a time. The actual packet length is 31 bytes and is sent 500 times per second.
The demo I use is libusb example program for reading out USB descriptors on unrooted Android.
Do you have any suggestions for me to try to solve this problem

mcuee, 2023-01-13:
You may get better support from libusb-devel mailing list.

github is for issue reporting and not for technical support.

mcuee, 2023-01-13:
In any case, the protocol you use (CDC-ACM), the behavior is expected.

closed 2023-01-13 by mcuee