libusb/libusb #1232: Segmentation fault in libusb_claim_interface on macOS

edahlseng, 2023-01-22:
I recently started encountering a segmentation fault in `libusb_claim_interface` while trying to upgrade/version pin libusb. I'm working with a device that has custom firmware and I have a custom utility on the host side to transfer data to this device. The host utility was previously dynamically linking to who-knows-what-version of libusb installed on my computer who-knows-when. In effort to make the build process for the host utility more deterministic and consistent across colleagues' machines, I tried switching to statically linking to libusb v1.0.26, and have now encountered segmentation faults when calling `libusb_claim_interface`. The debug output for `libusb_claim_interface` when using `LIBUSB_DEBUG=4` is as follows (debug output for previous functions omitted for brevity, but I can share if it would be helpful):
```
$ LIBUSB_DEBUG=4 ./host-transfer ./data.txt
... (preceding log output omitted for brevity) ...
[776940.999746] [00638812] libusb: debug [libusb_claim_interface] interface 4
[776941.000600] [00638812] libusb: debug [get_endpoints] building table of endpoints.
[776941.000772] [00638812] libusb: debug [get_endpoints] interface: 4 pipe 1: dir: 0 number: 5
segmentation fault
```

Interestingly, I tried previous versions of libusb, and received the same result on v1.0.25. Version 1.0.23 and 1.0.24 did not seg fault, instead returning a `LIBUSB_ERROR_NO_DEVICE` error for `libusb_claim_interface`. None of the logic in the `host-transfer` utility has changed, however, and everything works as expected with whatever previous version of libusb is installed globally on my machine.

Does anyone know what might be going on?

mcuee, 2023-01-22:
Most likely the device FW has some issues. Please post the USB descriptors of the device. Thanks.

tormodvolden, 2023-01-22:
Please produce a full backtrace so that we see more in detail where the segfault is taking place, e.g. with:
`LIBUSB_DEBUG=4 gdb --args ./host-transfer ./data.txt`
then `run` on the _(gdb_) prompt, then `bt full` and `info locals` after the segfault. Make sure it has been compiled with debug symbols enabled (-g or even -ggdb), and disabling optimization with -O0 can be helpful.

And please also try latest git master. There have been small fixups since 1.0.26.

Please attach the full debug output. There is a lot of information in the debug log that saves us asking questions back and forth. Of course if there are thousands of repeated transfers in the log you may trim those away.

mcuee, 2023-02-17:
@edahlseng 
Any updates on this issue?

edahlseng, 2023-02-21:
Oh my, I'm so sorry for not circling back here to close the loop. Thanks for the bump @mcuee.

**tl;dr:** This is no longer a problem on my machine, so I will close the issue.

**Longer story:** Originally, I was using Cypress/Infineon's CYUSB library. On mac this is compiled from Cypress/Infineon's SDK to produce a dynamic library, and so when I originally wrote this issue I was trying to statically link a version-pinned release of `libusb` into the CYUSB dynamic library for later consumption by our application. When I saw the request for more debugging information I took a look at compiling the CYUSB dynamic library with debug symbols, but after seeing that it was just a really thin wrapper around `libusb`, I decided to remove the CYUSB library entirely, and just statically link `libusb` directly into our application. I was intending to then compile our application with debug symbols to post more information here when I discovered that after this change the problem went away... I still don't know what exactly caused the issue, but my suspicion is that maybe it was a linking issue. In any case, I'm all set now. Thank you for your assistance!

closed 2023-02-21 by edahlseng

mcuee, 2023-02-21:
Thanks for the updates and glad that you have sorted out the issue.