libusb/libusb #1233: Travis-CI for libusb

mcuee, 2023-01-22:
We have the link to travis-ci.org: https://travis-ci.org/github/libusb/libusb
And it says the following.
> Since June 15th, 2021, the building on [travis-ci.org](http://www.travis-ci.org/) is ceased. Please use [travis-ci.com](http://www.travis-ci.com/) from now on.

Shall we just disable it?

tormodvolden, 2023-01-23:
I suppose you mean the build status badge link in README? Yes, until someone sets up something on travis-ci.com we can remove that badge. The same goes for Coverity, it was only run 9 months ago.

Appveyor works fine, just a little annoying that it goes red in case one of the jobs experiences an internal error, which seems to  often happen during artifact uploading.

mcuee, 2023-01-23:
@tormodvolden 

Yes. I mean the build status badge link in README.
Since we already have github action for Linux and macOS, I think we can remove the badge link.

We can probably remove the configuration file as well.
https://github.com/libusb/libusb/blob/master/.travis.yml

As for Coverity, it is only triggered manually. I have invited you to be an owner of Coverity libusb project.

tormodvolden, 2023-01-24:
If it makes it easier for someone to get Travis-CI going again, I think we can leave the config file. I don't know if there are any advantages with Travis-CI, but it is good to have alternatives.

Thanks for the invite, but I have no plans to pick up on that. Maybe @benzea is interested?

mcuee, 2023-01-25:
@tormodvolden 
Okay, so we will just remove the badge for now.

mcuee, 2023-01-25:
@LudovicRousseau 

Just wondering if you can help to get Travis-CI going agin and help to submit for Coverity run. Thanks.

mcuee, 2023-01-25:
@benzea

Just wonder if you are interested in Coverity for libusb project. Thanks.

benzea, 2023-01-25:
@mcuee unfortunately I changed jobs, so I am not the RedHat downstream for the libusb package anymore.

LudovicRousseau, 2023-01-25:
I suggest to move away from Travis-CI and use github actions instead.

I just submitted a new Coverity scan: 1 bug fixed.
@mcuee  I think you are an admin for the Coverity libusb project. You can submit new builds yourself. No?

mcuee, 2023-01-26:
@benzea 

No problem. Thanks for the help for libusb projects.

mcuee, 2023-01-26:
@LudovicRousseau 

Thanks for the help. I have the account but I have not figured out the way to submit for Coverity scan yet. I will try it later.

tormodvolden, 2023-01-26:
> @mcuee unfortunately I changed jobs, so I am not the RedHat downstream for the libusb package anymore.

@benzea We hope you can stay involved with libusb, also as a volunteer! Your help is very much appreciated.

benzea, 2023-01-26:
I can try, feel free to CC me on issues for me to have a look (I have unwatched the repo as a whole, too much noise overall with all the non-linux backends).

LudovicRousseau, 2023-01-27:
To submit a new build to Coverity I use this `coverity.sh` script:
```bash
#!/bin/bash

set -e
set -x
source coverity.config

export LANG=C
VERSION=$(git log -1 | grep commit | cut --fields=2 --delimiter=' ')

# generate files
make clean
rm -rf {,coverity/}cov-int
cov-build --dir cov-int make $MAKEFLAGS
tar czvf project.tgz cov-int

curl \
	--form token=$TOKEN \
	--form email=$EMAIL \
	--form file=@project.tgz \
	--form version=$VERSION \
	--form description=$DESCRIPTION \
	https://scan.coverity.com/builds?project=$PROJECT
```
The file `coverity.conf` contains:
```
PROJECT=libusb-libusb
TOKEN=xxxxxxxx
EMAIL=your email for coverity
DESCRIPTION=libusb
```
You can see the project token at https://scan.coverity.com/projects/libusb-libusb?tab=project_settings

LudovicRousseau, 2023-01-27:
AFAIK Coverity does not build the software, you have to install a special toolchain (i.e. cov-build) to rebuild the software and submit it.
Maybe it is possible to do something else. It is just how I do it for "my" projects.

mcuee, 2023-01-28:
@LudovicRousseau 

Can we adopt the approach here from @pbatard for the rufus project for coverity scan?
https://github.com/pbatard/rufus/blob/master/.github/workflows/coverity.yml

LudovicRousseau, 2023-01-28:
@mcuee it should be possible.
Just do it :-)

I am not convinced we should submit a new Coverity check for *each* commit.

mcuee, 2023-01-28:
> @mcuee it should be possible. Just do it :-)
> 
> I am not convinced we should submit a new Coverity check for _each_ commit.

I agree. It should only be manually triggered.

mcuee, 2024-02-14:
hidapi has Coverity check for Windows, macOS and Linux.
https://github.com/libusb/hidapi/blob/master/.github/workflows/checks.yml

@Youw 

Just wondering if you can help here. Thanks.

Youw, 2024-02-15:
I can help with the advice/review/etc. if someone volunteers to get it done :)