libusb/libusb #1237: libusb_get_device_list() very slow on Windows

OpusElectronics, 2023-01-25:
The libusb_get_device_list() function is very slow on Windows. It typically takes in the order of a few seconds to complete.
On Linux, it's almost instant.

Is there a particular reason for this? Is this fixable?

mcuee, 2023-01-26:
@OpusElectronics 

Usually this is only one time effort. What is the use case which this will cause an issue?

mcuee, 2023-01-26:
For Linux, there is a very specific issue here.
* https://github.com/libusb/libusb/issues/237

And there is a Linux specific `libusb_wrap_sys_device()` to address potential use case for Linux.

OpusElectronics, 2023-01-26:
Any use of libusb_get_device_list() shows the issue on Windows. First thing I tested was testlibusb.c in your examples, and it's pretty slow. As slow in another program I wrote.

It is problematic in a number of use cases, for instance command-line tools of course. But also if you need to detect a device during execution. Especially since hotplug is not supported on Windows, and neither is libusb_wrap_sys_device().

On Linux, it's actually very fast, at least on several different machines I have - all on Arch Linux. So no problem at all with Linux as far as I'm concerned.

mcuee, 2023-01-26:
> It is problematic in a number of use cases, for instance command-line tools of course. But also if you need to detect a device during execution. Especially since hotplug is not supported on Windows, and neither is libusb_wrap_sys_device().

So if hotplug is implemented in the future for Windows (no idea when this will happen because of lack of Windows developers), do you still feel this is an issue?

mcuee, 2023-01-26:
BTW, just FYI, libusb-win32 and libusbK have different implementation for device enumeration -- they only enumerate device with supported drivers and not all USB devices. They are only for Windows so they can do that.

libusb under Windows try to follow the behaviors of libusb under Linux -- so it enumerate all USB devices. Same for libusb under macOS.

libusb under BSDs will only enumerate device with ugen driver.

OpusElectronics, 2023-01-26:
hotplug support in Windows would be useful, extending libusb_wrap_sys_device() would be useful as well but probably not as convenient, as finding the device path on Windows would probably be relatively cumbersome?

I have some use cases for enumerating all devices, so libusb in that regard is useful, but for this use case I can live with the longish device list execution.

But if libusbK is much faster for listing compatible devices and supports hotplug (as I think I've seen, but you can confirm?), then switching to libusbK looks like an option. Is the API compatible with libusb, and does it support WinUSB drivers (as I think it does?) If so, that would be the best option for the time being.

mcuee, 2023-01-26:
@OpusElectronics 

libusbk supports WinUSB, libusbk.sys and libusb0.sys. It also supports hotplug. But its API is totally different from libusb and it is Windows only.

OpusElectronics, 2023-01-26:
Thanks. I would prefer a common code base for both Windows and Linux, but I'll have a look at libusbK nevertheless.

tormodvolden, 2023-01-26:
I don't think the libusb Windows enumeration can be easily improved, seeing how some very competent people have developed it. However, I am also curious to why exactly it is so slow. There must be some fairly expensive system calls involved. Can you run some kind of profiler or system/library trace on it? Are other Windows programs that lists all USB devices (and their bus topology?) like USBViewer(?) so slow? Otherwise, how do they do it?

photron, 2023-01-26:
> It typically takes in the order of a few seconds to complete.

For me it takes 150 milliseconds to enumerate 8 devices on Windows 10 wih libusb 1.0.26.11755.

tormodvolden, 2023-01-26:
@OpusElectronics what OS revision and hardware do you have?

OpusElectronics, 2023-01-26:
I will try to investigate what system calls exactly take this long. I had noticed a bit of a long enumeration time with FTDI libraries on Windows too, which could well come from the same calls.

On this machine I have 15 USB devices total (including real devices, internal USB hubs, etc.)
The OS is Win 7/x64.

OpusElectronics, 2023-01-26:
Note that the USBView tool from the Windows SDK lists USB devices very fast on the same machine. (Probably under 100 ms)
Could be worth looking at the source code.
https://github.com/Microsoft/Windows-driver-samples/tree/main/usb/usbview

mcuee, 2023-01-27:
@OpusElectronics 

Please post the debug log for the example program `listdev.exe` under your system. 

My debug log for a Windows 11 laptop shows that it only takes 20ms to run `listdev.exe`. It has the 11th Gen Intel(R) Core(TM) i5-1135G7 CPU, 16GB RAM and 512GB SSD.

```
PS C:\work\libusb\libusb> $Env:LIBUSB_DEBUG=4
PS C:\work\libusb\libusb> .\build\v143\x64\Release\listdevs.exe >debug3.log 2>&1
PS C:\work\libusb\libusb> cat .\debug3.log
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000022] [00000460] libusb: debug [libusb_init_context] created default context
[ 0.000030] [00000460] libusb: debug [libusb_init_context] libusb v1.0.26.11778
[ 0.000035] [00000460] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000B4 events 0
[ 0.000042] [00000460] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000046] [00000460] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000B8 events 0
[ 0.000070] [00000460] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000073] [00000460] libusb: debug [htab_create] using 1021 entries hash table
[ 0.002153] [00000460] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005015] [00000460] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007493] [00000460] libusb: info [windows_init] UsbDk backend is not available
[ 0.007554] [00000460] libusb: debug [libusb_get_device_list]
[ 0.008352] [00003f78] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.015694] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.015743] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.015779] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.015817] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [8]
[ 0.016206] [00000460] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.016232] [00000460] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.016237] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.016333] [00000460] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.016357] [00000460] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.016364] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [20]
[ 0.016456] [00000460] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.016490] [00000460] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.016494] [00000460] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.016497] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [21]
[ 0.016672] [00000460] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.016710] [00000460] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.016715] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [44]
[ 0.016935] [00000460] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.016963] [00000460] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.016966] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [79]
[ 0.017059] [00000460] libusb: debug [get_api_type] driver(s): libusbK
[ 0.017085] [00000460] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.017088] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [7A]
[ 0.017182] [00000460] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017210] [00000460] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017213] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [7B]
[ 0.017318] [00000460] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017348] [00000460] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017351] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [7C]
[ 0.017448] [00000460] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017474] [00000460] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017477] [00000460] libusb: debug [winusb_get_device_list] allocating new device for session [7D]
[ 0.017682] [00000460] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.017717] [00000460] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.019475] [00000460] libusb: debug [winusb_get_device_list] found existing device for session [7D]
[ 0.019590] [00000460] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019609] [00000460] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 98 bytes)
[ 0.019626] [00000460] libusb: debug [init_device] (bus: 1, addr: 9, depth: 2, port: 4): 'USB\VID_CAFE&PID_4003\123456'
[ 0.019694] [00000460] libusb: debug [winusb_get_device_list] extra GUID: {D32EDB66-69FC-01DC-03DD-20936E7318C2}
[ 0.019706] [00000460] libusb: debug [winusb_get_device_list] found existing device for session [7A]
[... cut ...]
[ 0.032418] [00000460] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000B4
cafe:4003 (bus 1, device 9) path: 2.4
16c0:05dc (bus 1, device 6) path: 1
05e3:0612 (bus 2, device 2) path: 2
8087:0026 (bus 1, device 3) path: 10
04f2:b6dd (bus 1, device 4) path: 5
1c7a:0575 (bus 1, device 2) path: 7
03eb:2104 (bus 1, device 12) path: 2.3
8086:a0ed (bus 1, device 0)
05e3:0610 (bus 1, device 7) path: 2
046d:c52b (bus 1, device 1) path: 3
239a:8120 (bus 1, device 8) path: 2.1
0483:3748 (bus 1, device 11) path: 2.2
8086:9a13 (bus 2, device 0)
```

mcuee, 2023-01-27:
I have dual boot Ubuntu 20.04 on this laptop. Linux is indeed faster and only takes 5ms for the same USB devices.

```
mcuee@UbuntuSwift3:~/build/libusb$ export LIBUSB_DEBUG=4
mcuee@UbuntuSwift3:~/build/libusb$ ./examples/listdevs 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000039] [0000128b] libusb: debug [libusb_init_context] created default context
[ 0.000049] [0000128b] libusb: debug [libusb_init_context] libusb v1.0.26.11778
[ 0.000058] [0000128b] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000063] [0000128b] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000066] [0000128b] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000075] [0000128b] libusb: debug [get_kernel_version] reported kernel version is 5.15.10-051510-generic
[ 0.000088] [0000128b] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000091] [0000128b] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000096] [0000128b] libusb: debug [op_init] sysfs is available
[ 0.000207] [00001299] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004024] [0000128b] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004030] [0000128b] libusb: debug [linux_get_device_address] scan usb1
[ 0.004046] [0000128b] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004049] [0000128b] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004050] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004112] [0000128b] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004114] [0000128b] libusb: debug [linux_get_device_address] scan usb2
[ 0.004128] [0000128b] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004131] [0000128b] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004133] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004185] [0000128b] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004187] [0000128b] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004200] [0000128b] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004203] [0000128b] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004206] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004223] [0000128b] libusb: debug [linux_get_parent_info] dev 0x555fb51d75d0 (2-1) has parent 0x555fb51d76d0 (usb2) port 1
[ 0.004264] [0000128b] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004266] [0000128b] libusb: debug [linux_get_device_address] scan usb3
[ 0.004280] [0000128b] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004282] [0000128b] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004284] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004332] [0000128b] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.004335] [0000128b] libusb: debug [linux_get_device_address] scan 3-1
[ 0.004347] [0000128b] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004350] [0000128b] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004352] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004368] [0000128b] libusb: debug [linux_get_parent_info] dev 0x555fb51d7d30 (3-1) has parent 0x555fb51d7e70 (usb3) port 1
[ 0.004410] [0000128b] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004413] [0000128b] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004426] [0000128b] libusb: debug [linux_get_device_address] bus=3 dev=11
[ 0.004428] [0000128b] libusb: debug [linux_enumerate_device] busnum 3 devaddr 11 session_id 779
[ 0.004430] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 3/11 (session 779)
[ 0.004444] [0000128b] libusb: debug [linux_get_parent_info] dev 0x555fb51d7c70 (3-10) has parent 0x555fb51d7e70 (usb3) port 10
[ 0.004487] [0000128b] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.004489] [0000128b] libusb: debug [linux_get_device_address] scan 3-2
[ 0.004502] [0000128b] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.004504] [0000128b] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.004506] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.004521] [0000128b] libusb: debug [linux_get_parent_info] dev 0x555fb51d84f0 (3-2) has parent 0x555fb51d7e70 (usb3) port 2
[ 0.004565] [0000128b] libusb: debug [linux_get_device_address] getting address for device: 3-2.1 detached: 0
[ 0.004567] [0000128b] libusb: debug [linux_get_device_address] scan 3-2.1
[ 0.004582] [0000128b] libusb: debug [linux_get_device_address] bus=3 dev=5
[ 0.004584] [0000128b] libusb: debug [linux_enumerate_device] busnum 3 devaddr 5 session_id 773
[ 0.004585] [0000128b] libusb: debug [linux_enumerate_device] allocating new device for 3/5 (session 773)
[... cut ...]
[ 0.005655] [0000128b] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.005658] [0000128b] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.005661] [0000128b] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.005663] [0000128b] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.005671] [00001299] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.005707] [0000128b] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.005712] [0000128b] libusb: debug [usbi_remove_event_source] remove fd 3
```

mcuee, 2023-01-27:
Similar test under my Mac Mini M1 running macOS 13.2, with even more USB devices (same USB Type C hub, plus one USB 3.0 Hub and one USB 2.0 Hub). It takes only 8ms.

```
mcuee@mcuees-Mac-mini libusb % export LIBUSB_DEBUG=4
mcuee@mcuees-Mac-mini libusb % ./examples/listdevs  
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000017] [00004e5b] libusb: debug [libusb_init_context] created default context
[ 0.000022] [00004e5b] libusb: debug [libusb_init_context] libusb v1.0.26.11779
[ 0.000034] [00004e5b] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000072] [00004e5c] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000328] [00004e5c] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000472] [00004e5b] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8b54913
[ 0.000513] [00004e5b] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8b54913
[ 0.002343] [00004e5b] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002348] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002351] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002352] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002354] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002356] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002358] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002359] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002361] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002363] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002365] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002367] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002369] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002370] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002388] [00004e5b] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002391] [00004e5b] libusb: debug [process_new_device] allocating new device in context 0x12d7042f0 for with session 0x8b54913
[ 0.002396] [00004e5b] libusb: debug [process_new_device] found device with address 2 port = 6 parent = 0x0 at 0x60000118863a
[ 0.002455] [00004e5b] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8cce85a
[ 0.002477] [00004e5b] libusb: debug [darwin_get_cached_device] parent sessionID: 0x8b54913
[ 0.002479] [00004e5b] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x8cce85a/0x2610000 against cached device with sessionID/locationID 0x8b54913/0x2600000
[ 0.002482] [00004e5b] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8cce85a
[ 0.002681] [00004e5b] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002686] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002687] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002689] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002691] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002693] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002695] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002696] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002698] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002700] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002702] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002704] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002706] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002707] [00004e5b] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002719] [00004e5b] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002722] [00004e5b] libusb: debug [process_new_device] allocating new device in context 0x12d7042f0 for with session 0x8cce85a
[ 0.002724] [00004e5b] libusb: debug [process_new_device] found device with address 3 port = 1 parent = 0x600001b880c0 at 0x60000118883a
[ 0.002774] [00004e5b] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8dc16a5
[ 0.002794] [00004e5b] libusb: debug [darwin_get_cached_device] parent sessionID: 0x8cce85a
[ 0.002797] [00004e5b] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x8dc16a5/0x2611000 against cached device with sessionID/locationID 0x8cce85a/0x2610000
[... cut ...]
[ 0.007067] [00004e5b] libusb: debug [libusb_unref_device] destroy device 2.7
[ 0.007068] [00004e5b] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.007138] [00004e5b] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.007141] [00004e5b] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.007142] [00004e5b] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.007159] [00004e5c] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.008223] [00004e5b] libusb: debug [usbi_remove_event_source] remove fd 3
```

OpusElectronics, 2023-01-27:
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000011] [00002938] libusb: debug [libusb_init] created default context
[ 0.000017] [00002938] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000149] [00002938] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000024 events 0
[ 0.000155] [00002938] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000157] [00002938] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000028 events 0
[ 0.000159] [00002938] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.000160] [00002938] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001741] [00002938] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
[ 0.001831] [00002938] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.002674] [00002938] libusb: info [windows_init] UsbDk backend is not available
[ 0.002704] [00002938] libusb: debug [libusb_get_device_list]  
[ 0.002723] [00001d48] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.031680] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [22B8]
[ 0.031777] [00002938] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.031778] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [285C]
[ 0.031812] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [289C]
[ 0.031846] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [28D8]
[ 0.031881] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [2914]
[ 0.031915] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [2960]
[ 0.031948] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [29AC]
[ 0.031982] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [29F0]
[ 0.032016] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [2A3C]
[ 0.032054] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [2A8C]
[ 0.032150] [00002938] libusb: debug [get_api_type] driver(s): libusb0
[ 0.032205] [00002938] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.032206] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [8D4]
[ 0.032453] [00002938] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.032482] [00002938] libusb: debug [get_api_type] matched driver name against HID API
[ 0.032483] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [2300]
[ 0.032559] [00002938] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032587] [00002938] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032588] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [2350]
[ 0.032664] [00002938] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032692] [00002938] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032693] [00002938] libusb: debug [winusb_get_device_list] allocating new device for session [23B0]
[ 0.032834] [00002938] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1B21&DEV_1142&SUBSYS_85FD1043&REV_00\4&14F728CE&0&00E6' bus number 1
[ 0.032878] [00002938] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D26&SUBSYS_86001043&REV_05\3&11583659&0&E8' bus number 2
[ 0.032904] [00002938] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D2D&SUBSYS_86001043&REV_05\3&11583659&0&D0' bus number 3
[ 0.032931] [00002938] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D31&SUBSYS_86001043&REV_05\3&11583659&0&A0' bus number 4
[ 0.035555] [00002938] libusb: debug [init_root_hub] root hub 'USB\ASMEDIAUSBD_HUB\5&9B0B1EB&2&0' reports 4 ports
[ 0.035586] [00002938] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ASMEDIAUSBD_HUB\5&9B0B1EB&2&0'
[ 0.035854] [00002938] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&12773313&0' reports 2 ports
[ 0.035858] [00002938] libusb: debug [init_device] (bus: 3, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&12773313&0'
[ 0.036070] [00002938] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&35ABF192&0' reports 2 ports
[ 0.036073] [00002938] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&35ABF192&0'
[ 0.036139] [00002938] libusb: debug [winusb_get_device_list] found existing device for session [8D4]
[ 0.036220] [00002938] libusb: debug [init_device] found 6 configurations (current config: 1)
[ 0.258238] [00002938] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.479111] [00002938] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 34 bytes)
[ 0.703608] [00002938] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=1, 34 bytes)
[ 0.924734] [00002938] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=1, 34 bytes)
[ 1.144728] [00002938] libusb: debug [cache_config_descriptors] cached config descriptor 4 (bConfigurationValue=1, 34 bytes)
[... cut ...]
[ 2.545991] [00002938] libusb: debug [libusb_unref_device] destroy device 3.0
[ 2.545993] [00002938] libusb: debug [libusb_unref_device] destroy device 4.0
[ 2.545994] [00002938] libusb: debug [libusb_exit] destroying default context
[ 2.545999] [00001d48] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 2.546274] [00002938] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000028
[ 2.546298] [00002938] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000024

mcuee, 2023-01-27:
@OpusElectronics 

What is this device `0xb05:0x180a`? Is it recognized by Windows?

OpusElectronics, 2023-01-27:
I think we're onto something. I checked - this is an internal USB bluetooth adapter that I don't use, so I have disabled it in the device manager. Could it be that disabled devices would cause some kind of timeout in libusb?

mcuee, 2023-01-27:
> I think we're onto something. I checked - this is an internal USB bluetooth adapter that I don't use, so I have disabled it in the device manager. Could it be that disabled devices would cause some kind of timeout in libusb?

It could be. Please enable it and then run agin the `listdev.exe` to see if that helps.

OpusElectronics, 2023-01-27:
Yes, enabling it makes it faster - but it's still slow.
`[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000019] [000041fc] libusb: debug [libusb_init] created default context
[ 0.000025] [000041fc] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000156] [000041fc] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000024 events 0
[ 0.000162] [000041fc] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000164] [000041fc] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000028 events 0
[ 0.000166] [000041fc] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.000167] [000041fc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001754] [000041fc] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
[ 0.001847] [000041fc] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.002663] [000041fc] libusb: info [windows_init] UsbDk backend is not available
[ 0.002691] [000041fc] libusb: debug [libusb_get_device_list]  
[ 0.002714] [00000a98] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.033257] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [23C8]
[ 0.033353] [000041fc] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.033354] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [296C]
[ 0.033389] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [29AC]
[ 0.033423] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [29E8]
[ 0.033460] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2A24]
[ 0.033494] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2A70]
[ 0.033527] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2ABC]
[ 0.033561] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2B00]
[ 0.033595] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2B4C]
[ 0.033635] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2B9C]
[ 0.033732] [000041fc] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.033775] [000041fc] libusb: debug [get_api_type] lower filter driver(s): bcbtums;btwampfl
[ 0.033780] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [828]
[ 0.033870] [000041fc] libusb: debug [get_api_type] driver(s): libusb0
[ 0.033899] [000041fc] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.033900] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [920]
[ 0.034137] [000041fc] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.034166] [000041fc] libusb: debug [get_api_type] matched driver name against HID API
[ 0.034167] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2410]
[ 0.034248] [000041fc] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.034276] [000041fc] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.034277] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2460]
[ 0.034354] [000041fc] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.034382] [000041fc] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.034382] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [24C0]
[ 0.034529] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1B21&DEV_1142&SUBSYS_85FD1043&REV_00\4&14F728CE&0&00E6' bus number 1
[ 0.034573] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D26&SUBSYS_86001043&REV_05\3&11583659&0&E8' bus number 2
[ 0.034599] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D2D&SUBSYS_86001043&REV_05\3&11583659&0&D0' bus number 3
[ 0.034625] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D31&SUBSYS_86001043&REV_05\3&11583659&0&A0' bus number 4
[ 0.037434] [000041fc] libusb: debug [init_root_hub] root hub 'USB\ASMEDIAUSBD_HUB\5&9B0B1EB&2&0' reports 4 ports
[ 0.037466] [000041fc] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ASMEDIAUSBD_HUB\5&9B0B1EB&2&0'
[ 0.152218] [000041fc] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&12773313&0' reports 2 ports
[ 0.152225] [000041fc] libusb: debug [init_device] (bus: 3, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&12773313&0'
[ 0.152497] [000041fc] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&35ABF192&0' reports 2 ports
[ 0.152500] [000041fc] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&35ABF192&0'
[ 0.152574] [000041fc] libusb: debug [winusb_get_device_list] found existing device for session [920]
[ 0.152657] [000041fc] libusb: debug [init_device] found 6 configurations (current config: 1)
[ 0.374403] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.595276] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 34 bytes)
[... cut ...]
[ 1.615807] [000041fc] libusb: debug [libusb_unref_device] destroy device 3.0
[ 1.615808] [000041fc] libusb: debug [libusb_unref_device] destroy device 4.0
[ 1.615810] [000041fc] libusb: debug [libusb_exit] destroying default context
[ 1.615815] [00000a98] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 1.616142] [000041fc] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000028
[ 1.616167] [000041fc] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000024
`

mcuee, 2023-01-27:
The device seems to be still the problem. I am not so sure how to fix the issue though.

```
[ 0.152657] [000041fc] libusb: debug [init_device] found 6 configurations (current config: 1)
[ 0.374403] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.595276] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 34 bytes)
[ 0.814151] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=1, 34 bytes)
[ 1.036277] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=1, 34 bytes)
[ 1.255023] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 4 (bConfigurationValue=1, 34 bytes)
[ 1.476895] [000041fc] libusb: debug [cache_config_descriptors] cached config descriptor 5 (bConfigurationValue=1, 34 bytes)
```

I remember we had similar issues last time with Apple Bluetooth controller under macOS and the solution was to disable it for libusb.

mcuee, 2023-01-27:
> I remember we had similar issues last time with Apple Bluetooth controller under macOS and the solution was to disable it for libusb.

Past discussions for macOS.
https://libusb-devel.narkive.com/8n6a4ZTo/libusb-1-0-16rc2-available#post11
https://sourceforge.net/p/libusb/mailman/message/32272622/

Relevant commit

commit 0c4d180e971ebdbcce846902db0fff3aa0ee6447
Author: Nathan Hjelm <hj...@me...>
Date:   Fri Sep 28 21:18:36 2012 -0600
    Darwin: Don't retry or unsuspend Apple devices during enum

commit 4e31465ab7d83b5a5b68f5f9d27d6c77caa07b8b
Author: Nathan Hjelm <hj...@me...>
Date:   Fri Sep 28 21:18:08 2012 -0600
    Darwin: Reduce attempts to retrieve a device descriptor during enumeration

mcuee, 2023-01-27:
@OpusElectronics 

If you are keen, you can try to come out with the patch to ignore this device for Windows and see if others like the idea or not.

tormodvolden, 2023-01-27:
@OpusElectronics I extended your debug log to display "time since last timestamp" to more easily spot the slow bits  . Maybe you can sprinkle some more usbi_dbg's in the code to narrow it down further, unless you have some Windows profiler tools that does this better.

<details>
Conversion script: awk '{us=$2*1000000; printf "%7i us ", us-a; a=us; print $0}'

```
     19 us [ 0.000019] [000041fc] libusb: debug [libusb_init] created default context
      6 us [ 0.000025] [000041fc] libusb: debug [libusb_init] libusb v1.0.26.11724
    131 us [ 0.000156] [000041fc] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000024 events 0
      6 us [ 0.000162] [000041fc] libusb: debug [usbi_io_init] using timer for timeouts
      2 us [ 0.000164] [000041fc] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000028 events 0
      2 us [ 0.000166] [000041fc] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
      1 us [ 0.000167] [000041fc] libusb: debug [htab_create] using 1021 entries hash table
   1587 us [ 0.001754] [000041fc] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
     93 us [ 0.001847] [000041fc] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
    816 us [ 0.002663] [000041fc] libusb: info [windows_init] UsbDk backend is not available
     28 us [ 0.002691] [000041fc] libusb: debug [libusb_get_device_list]
     23 us [ 0.002714] [00000a98] libusb: debug [windows_iocp_thread] I/O completion thread started
  30543 us [ 0.033257] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [23C8]
     96 us [ 0.033353] [000041fc] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
      1 us [ 0.033354] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [296C]
     35 us [ 0.033389] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [29AC]
     34 us [ 0.033423] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [29E8]
     37 us [ 0.033460] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2A24]
     34 us [ 0.033494] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2A70]
     33 us [ 0.033527] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2ABC]
     34 us [ 0.033561] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2B00]
     34 us [ 0.033595] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2B4C]
     40 us [ 0.033635] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2B9C]
     97 us [ 0.033732] [000041fc] libusb: debug [get_api_type] driver(s): BTHUSB
     43 us [ 0.033775] [000041fc] libusb: debug [get_api_type] lower filter driver(s): bcbtums;btwampfl
      5 us [ 0.033780] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [828]
     90 us [ 0.033870] [000041fc] libusb: debug [get_api_type] driver(s): libusb0
     29 us [ 0.033899] [000041fc] libusb: debug [get_api_type] matched driver name against libusb0
      1 us [ 0.033900] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [920]
    237 us [ 0.034137] [000041fc] libusb: debug [get_api_type] driver(s): HidUsb
     29 us [ 0.034166] [000041fc] libusb: debug [get_api_type] matched driver name against HID API
      1 us [ 0.034167] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2410]
     81 us [ 0.034248] [000041fc] libusb: debug [get_api_type] driver(s): usbccgp
     28 us [ 0.034276] [000041fc] libusb: debug [get_api_type] matched driver name against Composite API
      1 us [ 0.034277] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [2460]
     77 us [ 0.034354] [000041fc] libusb: debug [get_api_type] driver(s): usbccgp
     28 us [ 0.034382] [000041fc] libusb: debug [get_api_type] matched driver name against Composite API
      0 us [ 0.034382] [000041fc] libusb: debug [winusb_get_device_list] allocating new device for session [24C0]
    147 us [ 0.034529] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1B21&DEV_1142&SUBSYS_85FD1043&REV_00\4&14F728CE&0&00E6' bus number 1
     44 us [ 0.034573] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D26&SUBSYS_86001043&REV_05\3&11583659&0&E8' bus number 2
     26 us [ 0.034599] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D2D&SUBSYS_86001043&REV_05\3&11583659&0&D0' bus number 3
     26 us [ 0.034625] [000041fc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8D31&SUBSYS_86001043&REV_05\3&11583659&0&A0' bus number 4
   2809 us [ 0.037434] [000041fc] libusb: debug [init_root_hub] root hub 'USB\ASMEDIAUSBD_HUB\5&9B0B1EB&2&0' reports 4 ports
     32 us [ 0.037466] [000041fc] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ASMEDIAUSBD_HUB\5&9B0B1EB&2&0'
 114752 us [ 0.152218] [000041fc] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&12773313&0' reports 2 ports
      7 us [ 0.152225] [000041fc] libusb: debug [init_device] (bus: 3, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&12773313&0'
[... cut ...]
      1 us [ 1.615808] [000041fc] libusb: debug [libusb_unref_device] destroy device 4.0
      2 us [ 1.615810] [000041fc] libusb: debug [libusb_exit] destroying default context
      5 us [ 1.615815] [00000a98] libusb: debug [windows_iocp_thread] I/O completion thread exiting
    327 us [ 1.616142] [000041fc] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000028
     25 us [ 1.616167] [000041fc] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000024
```
</details>

tormodvolden, 2023-01-27:
> Note that the USBView tool from the Windows SDK lists USB devices very fast on the same machine. (Probably under 100 ms) Could be worth looking at the source code. https://github.com/Microsoft/Windows-driver-samples/tree/main/usb/usbview

libusb is using IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION to retrieve the config descriptors, same as mentioned in the USBView README. I guess USBView will be equally slow if you use it to read the descriptors on the slow devices, and that the speed of the USBView listing is due to not reading anything from the devices themselves.

tormodvolden, 2023-01-27:
#1127 is also related, there the enumeration hangs for a minute with some devices, and the pull request is about making these DeviceIoControl IOCTL calls time out after a second.

OpusElectronics, 2023-01-27:
I'll have a look at the source code, which from what I've gathered is in the os/windows_winusb.c file.
Fixing the issue on a more general level rather than just "blacklisting" some devices seems to make more sense. If that is possible.

Yes, on Windows the only way of getting USB descriptors is to issue a IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION ioctl message to the device's parent hub. So that implies 1/ looking up the parent hub and 2/ issuing this message. I actually don't know how DeviceIoControl() reacts when the child device given in the message is not present - but I guess we have the answer, it does time-out after a while. I can make some minimal code test about this in my own code as I have written a minimal example to enumerate USB devices, but it doesn't include getting USB descriptors yet. It will allow narrowing it down more easily.

One general point though, that I'll check in windows_winusb.c: there is a way of enumerating only devices that are *present*, so it's definitely possible to ignore any device that would not be connected, or disabled, but it's an option flag. With default options, 
SetupDiGetClassDevs() will enumerate all USB devices that are installed, even if they are not currently connected, which I suspect is part of the issue there. I'll let you know.

tormodvolden, 2023-01-27:
Do such descriptor requests always go to the device or are some of them possibly returned from a cache in the USB stack? I read Microsoft's description of the OS enumeration process, and it says the hub driver caches the config descriptor. I don't know if it caches all or only the active though.

OpusElectronics, 2023-01-27:
Good question - I suspect it may depend on the version of the OS, and maybe the number of hubs (as the cache may be limited in size). There's a lot of internal hubs on my workstation which is a desktop one.

mcuee, 2023-01-28:
@tormodvolden and @hjelmn 

Right now `LIBUSB_OPTION_NO_DEVICE_DISCOVERY` is only valid under Linux. Do you think it is a good idea to extend this to Windows and macOS to sort out issues like this one? I tend to think it is not that difficult for macOS, but Windows may be a bit more complicated.

mcuee, 2023-01-28:
> Do such descriptor requests always go to the device or are some of them possibly returned from a cache in the USB stack? I read Microsoft's description of the OS enumeration process, and it says the hub driver caches the config descriptor. I don't know if it caches all or only the active though.

I believe only the active one. In any case, WinUSB (as well as libusbK) does not support setting the alternative configuration.

hjelmn, 2023-02-10:
Extending `LIBUSB_OPTION_NO_DEVICE_DISCOVERY` would probably have to be paired with extending `libusb_wrap_sys_device()`.

yyjdelete, 2023-08-16:
Try to open debug output, and found libusb will wait timout for about 1.3 for every single disable devices(I have 2 disabled devices and this will add an extra 10 second to run pyocd, which will call libusb_get_device_list 4 times during start).

I think libusb should just ignore disabled devices(if possible to check this) in windows instead of wait timeout again and again.

```
[ 1.340961] [00004f90] libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_0CF3&PID_E004\ALASKA_DAY_2006'
[ 1.341031] [00004f90] libusb: info [winusb_get_device_list] libusb will not be able to access it
[ 1.341151] [00004f90] libusb: debug [winusb_get_device_list] allocating new device for session [65]
[ 1.341322] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.397368] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.460360] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.522673] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.585788] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.648141] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.710258] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.773496] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.835586] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.897701] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 1.960857] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.024013] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.087078] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.149221] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.212196] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.275616] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.338446] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.400577] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.462720] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.524920] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.587935] [00004f90] libusb: debug [init_device] 0xcf3:0xe004 found 1 configurations (not configured)
[ 2.650066] [00004f90] libusb: info [init_device] 0xcf3:0xe004 found 1 configurations but device isn't configured, forcing current configuration to 1
[ 2.650532] [00004f90] libusb: info [cache_config_descriptors] could not access configuration descriptor 0 (dummy) for 'USB\VID_0CF3&PID_E004\ALASKA_DAY_2006': [31] 连到系统上的设备没有发挥作用。
[ 2.650673] [00004f90] libusb: debug [init_device] (bus: 2, addr: 6, depth: 2, port: 5): 'USB\VID_0CF3&PID_E004\ALASKA_DAY_2006'
```

```
[ 0.031186] [00004f90] libusb: debug [winusb_get_device_list] allocating new device for session [64]
[ 0.031389] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.082613] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.145281] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.208083] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.270601] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.333637] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.395847] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.458955] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.520028] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.583153] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.646206] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.709308] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.772496] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.834667] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.895767] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 0.958830] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 1.022084] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 1.085277] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 1.148453] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[... cut ...]
[ 1.273029] [00004f90] libusb: debug [init_device] 0xd28:0x204 found 1 configurations (not configured)
[ 1.336241] [00004f90] libusb: info [init_device] 0xd28:0x204 found 1 configurations but device isn't configured, forcing current configuration to 1
[ 1.337846] [00004f90] libusb: info [cache_config_descriptors] could not access configuration descriptor 0 (dummy) for 'USB\VID_0D28&PID_0204\7A42CDDB3E4509610BA9F01FD85953ED': [31] 连到系统上的设备没有发挥作用。
[ 1.338012] [00004f90] libusb: debug [init_device] (bus: 2, addr: 5, depth: 2, port: 4): 'USB\VID_0D28&PID_0204\7A42CDDB3E4509610BA9F01FD85953ED'
```

JoseGuilhermeCR, 2023-09-21:
I'm currently facing a similar problem in certain situations and the log is similar to @yyjdelete's log. I'll also try out #1270 and report my results.