libusb/libusb #1238: libusb Windows can not access second or later interface as part of an Interface group without IAD

mcuee, 2023-01-26:
With the merge of PR #965, the long standing issue of access the second interface or later interface inside an IAD group has been sorted out.

However, there are other interface groups without using IAD. libusb Windows will still have issues access the second interface or later interface in such case.

Example device:
Raspberry Pi Pico W CircuitPython FW.

I have used WinUSB driver to replace the Mass Storage driver (Interface 2), the CDC-ACM driver (Interface 0/1, with IAD) and the USB Audion Driver (Interface 4/5, no IAD). Interface 3 is using USB HID driver (USB HID Keyboard).

Claiming Interface 5 will fail as it is not using IAD -- therefore PR https://github.com/libusb/libusb/pull/965 will not help.

```
PS C:\work\libusb> .\libusb\examples\xusb.exe 239a:8120
Using libusb v1.0.26.11778

Opening device 239A:8120...
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 239A:8120
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 218
         descriptor length: 9
             nb interfaces: 6
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 10
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0040
          polling interval: 00
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 83
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 03
           max packet size: 0040
          polling interval: 00
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 84
           max packet size: 0040
          polling interval: 08
       endpoint[1].address: 04
           max packet size: 0040
          polling interval: 08
              interface[4]: id = 4
interface[4].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[5]: id = 5
interface[5].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 01.03.00
       endpoint[0].address: 05
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 85
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Kernel driver attached for interface 1: -12

Claiming interface 1...

Kernel driver attached for interface 2: -12

Claiming interface 2...

Kernel driver attached for interface 3: -12

Claiming interface 3...

Kernel driver attached for interface 4: -12

Claiming interface 4...

Kernel driver attached for interface 5: -12

Claiming interface 5...
   Failed.

Reading string descriptors:
   String (0x01): "Raspberry Pi"
   String (0x02): "Pico W"
   String (0x03): "E661410403559F31"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 1
      IAD 0:
            bFirstInterface: 0
            bInterfaceCount: 2
             bFunctionClass: 02
          bFunctionSubClass: 02
          bFunctionProtocol: 00
                  iFunction: 0

Reading Max LUN:
   Failed: Pipe error   Max LUN = 23

Sending Inquiry:
   sent 6 CDB bytes
   Operation timed out
   Input/Output Error

```

mcuee, 2023-01-26:
As of now, the only work-around is to use WinUSB (or libusbK driver or libusb0.sys driver) to act as the driver for the whole device (to repace USB composite parent driver).

mcuee, 2023-01-26:
`lsusb -vvv` info about the device.

```
mcuee@UbuntuSwift3:~$ sudo lsusb -vvv -d 239a:8120
[sudo] password for mcuee: 

Bus 003 Device 008: ID 239a:8120  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x239a 
  idProduct          0x8120 
  bcdDevice            1.00
  iManufacturer           1 Raspberry Pi
  iProduct                2 Pico W
  iSerial                 3 E66141040387422B
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00da
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 
      iInterface              4 CircuitPython CDC control
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x01
          call management
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 183 lines cut ...]
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       3
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

Novakov, 2023-01-26:
Can you share firmware for device you are testing with? I would like to take a look at this issue

mcuee, 2023-01-26:
> Can you share firmware for device you are testing with? I would like to take a look at this issue

The FW is supposed to be here but I have not figured out the exact codes. I believe it is not only for Raspberry Pi Pico and Pico W, but also for other CircuitPython boards, 
https://github.com/adafruit/circuitpython

More info:
https://learn.adafruit.com/customizing-usb-devices-in-circuitpython

mcuee, 2023-01-26:
The following is with CircuitPython FW for Raspberry Pi Pico which has two IADs for the two CDC-ACM IAD groups. Interface 6/7 are the Audio interface without IAD. libusb Windows will have issues accessing Interface 7.

IAD0 -- Interface 0/1, CDC-ACM
IAD1 -- Interface 2/3, CDC-ACM
Interface 4 -- USB Mass Storage
Interface 5 -- USB HID Keyboard
Interface 6/7 -- USB Audio Midi Inteface

```
PS C:\work\libusb\libusb> .\build\v143\x64\Release\xusb 239a:80f4
Using libusb v1.0.26.11778

Opening device 239A:80F4...
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 1
           VID:PID: 239A:80F4
         bcdDevice: 0100
   iMan:iProd:iSer: 2:3:1
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 284
         descriptor length: 9
             nb interfaces: 8
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 10
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0040
          polling interval: 00
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.00
       endpoint[0].address: 83
           max packet size: 0040
          polling interval: 10
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 04
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 84
           max packet size: 0040
          polling interval: 00
              interface[4]: id = 4
interface[4].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 85
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 05
           max packet size: 0040
          polling interval: 00
              interface[5]: id = 5
interface[5].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 86
           max packet size: 0040
          polling interval: 08
       endpoint[1].address: 06
           max packet size: 0040
          polling interval: 08
              interface[6]: id = 6
interface[6].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[7]: id = 7
interface[7].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 01.03.00
       endpoint[0].address: 07
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 87
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Kernel driver attached for interface 1: -12

Claiming interface 1...

Kernel driver attached for interface 2: -12

Claiming interface 2...

Kernel driver attached for interface 3: -12

Claiming interface 3...

Kernel driver attached for interface 4: -12

Claiming interface 4...

Kernel driver attached for interface 5: -12

Claiming interface 5...

Kernel driver attached for interface 6: -12

Claiming interface 6...

Kernel driver attached for interface 7: -12

Claiming interface 7...
   Failed.

Reading string descriptors:
   String (0x02): "Raspberry Pi"
   String (0x03): "Pico"
   String (0x01): "6E06892FB9E3C283"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 2
      IAD 0:
            bFirstInterface: 0
            bInterfaceCount: 2
             bFunctionClass: 02
          bFunctionSubClass: 02
          bFunctionProtocol: 00
                  iFunction: 0
      IAD 1:
            bFirstInterface: 2
            bInterfaceCount: 2
             bFunctionClass: 02
          bFunctionSubClass: 02
          bFunctionProtocol: 00
                  iFunction: 0

Reading Max LUN:
   Failed: Pipe error   Max LUN = 16

Sending Inquiry:
   sent 6 CDB bytes
   Operation timed out
   Input/Output Error
```

mcuee, 2023-01-27:
The comments in #85 is still valid.
* https://github.com/libusb/libusb/issues/85#issuecomment-131544141

For IAD or for the USB Audio devices or CDC-ACM or other USB interface groups, Windows will only create a PDO for the first interface. So it is not possible to use Zadig or libusbK-inf-wizard to assign WinUSB driver to the second interface or later interface.

https://www.mail-archive.com/libusbx-devel@lists.sourceforge.net/msg03059.html
From Xiaofan:
" I think there is a way to hack the inf-file generated by 
 Zadig to install WinUSB driver for each interface. 
 Interface 0 --> MI_00 
 Interface 1 --> MI_01 
 Interface 2 --> MI_02 "
From Tim Roberts:
"No, that won't work.  When you have an IAD that wraps those three 
interfaces, usbccgp only creates a PDO for MI_00.  You'll only get one 
driver, and that driver is assigned ownership of all three interfaces. 
There will never be an opportunity to load MI_01 and MI_02, because it 
won't create those PDOs. "

mcuee, 2023-10-10:
I believe #1307 is the same issue.
* #1307

mcuee, 2023-12-28:
Example device which show up in Windows as two separated devices.
A USB Headset: Interface 0/1/2 -- USB Audio Device (not using IAD), Interface 3 -- USB HID device.

libusb Windows can not claim Interface 1/2 even if I replace the USB Audio device driver using WinUSB. Only Interface 0 can be claimed.

```
C:\work\libusb\libusb-1.0.27-rc1\build\v142\x64\Release> .\xusb 047f:c056
Using libusb v1.0.27.11848

Opening device 047F:C056...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&36020D6F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842\0123456789ABCD'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\9&AF4BBD4&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\7&2452366F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E&MI_00\D&E651DD6&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_03EB&PID_2177&MI_00\6&315E345F&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056\BB305534D79EBD418B6757528EAC0C19'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_00\6&1BD5FF13&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_8153\111000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06F\C&1F76A113&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_03EB&PID_2177\BUR200973052'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_03EB&PID_2177&MI_01\6&315E345F&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD\200901010001'

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 047F:C056
         bcdDevice: 0210
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 288
         descriptor length: 9
             nb interfaces: 4
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.02.00
interface[1].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 01.02.00
       endpoint[0].address: 81
           max packet size: 00C8
          polling interval: 01
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.02.00
interface[2].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 01.02.00
       endpoint[0].address: 01
           max packet size: 00C0
          polling interval: 01
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 84
           max packet size: 0025
          polling interval: 01

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Kernel driver attached for interface 1: (not supported)

Claiming interface 1...
   Failed (error -12) Operation not supported or unimplemented on this platform

Kernel driver attached for interface 2: (not supported)

Claiming interface 2...
   Failed (error -12) Operation not supported or unimplemented on this platform

Kernel driver attached for interface 3: (not supported)

Claiming interface 3...

Reading string descriptors:
   String (0x01): "Plantronics"
   String (0x02): "Plantronics Blackwire 3220 Series"
   String (0x03): "BB305534D79EBD418B6757528EAC0C19"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Releasing interface 1...
Releasing interface 2...
Releasing interface 3...
Closing device...
```

mcuee, 2023-12-28:
No issues if IAD is used.

Example device: Microchip PICKit 4, libusb Windows can claim Interface 1 and Interface 2.
Interface 0 -- USB HID interface
Interface 1/2 -- USB CDC-ACM (using IAD) -- replace the driver (usbser.sys) with WinUSB. 

```
C:\work\libusb\libusb-1.0.27-rc1\build\v142\x64\Release> .\xusb 03eb:2177
Using libusb v1.0.27.11848

Opening device 03EB:2177...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&36020D6F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842\0123456789ABCD'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\9&AF4BBD4&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\7&2452366F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E&MI_00\D&E651DD6&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_03EB&PID_2177&MI_00\6&315E345F&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056\BB305534D79EBD418B6757528EAC0C19'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_00\6&1BD5FF13&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_8153\111000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06F\C&1F76A113&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_03EB&PID_2177\BUR200973052'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD\200901010001'

Reading device descriptor:
            length: 18
      device class: 239
               S/N: 3
           VID:PID: 03EB:2177
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [_hid_get_descriptor] unsupported
no descriptor

Reading first configuration descriptor:
              total length: 107
         descriptor length: 9
             nb interfaces: 3
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 01
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.00
       endpoint[0].address: 84
           max packet size: 0040
          polling interval: 10
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 83
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 05
           max packet size: 0200
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Kernel driver attached for interface 1: (not supported)

Claiming interface 1...

Kernel driver attached for interface 2: (not supported)

Claiming interface 2...

Reading string descriptors:
   String (0x01): "Microchip Technology Incorporated"
   String (0x02): "MPLAB PICkit 4 CMSIS-DAP"
   String (0x03): "BUR200973052"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 1
      IAD 0:
            bFirstInterface: 1
            bInterfaceCount: 2
             bFunctionClass: 02
          bFunctionSubClass: 02
          bFunctionProtocol: 00
                  iFunction: 5 (PICkit 4 Virtual COM Port)

Releasing interface 0...
Releasing interface 1...
Releasing interface 2...
Closing device...
```

mcuee, 2024-02-20:
Same thing with libusb-1.0.27 release. Here is another CircuitPython board with 6 interfaces.

Interface 0/1, using IAD, CDC-ACM, replaced the original driver with WinUSB
Interface 2, USB Mass Storage, replaced the original driver with WinUSB
Interface 3, USB HID
Interface 4/5, not using IAD, USB Audio, replaced the original driver with WinUSB

You can see that libusb-1.0.27 failed to claim Interface 5.

```
PS C:\work\libusb\libusb-1.0.27_bin\examples\bin64> .\xusb.exe 0x2e8a:0x102e
Using libusb v1.0.27.11882

Opening device 2E8A:102E...
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0612\5&1A00B1C&0&2'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2E8A&PID_102E&MI_03\7&16B7A8A2&0&0003'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2E8A&PID_102E\DE61A0D0C35D4A26'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0026\5&586B51A&0&10'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1EA7&PID_0064\5&586B51A&0&3'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&12C539F&0&0'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0610\5&586B51A&0&2'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&1A63DC7C&0&0'
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 2E8A:102E
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 218
         descriptor length: 9
             nb interfaces: 6
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 10
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0040
          polling interval: 00
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 83
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 03
           max packet size: 0040
          polling interval: 00
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 84
           max packet size: 0040
          polling interval: 08
       endpoint[1].address: 04
           max packet size: 0040
          polling interval: 08
              interface[4]: id = 4
interface[4].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[5]: id = 5
interface[5].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 01.03.00
       endpoint[0].address: 05
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 85
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Kernel driver attached for interface 1: (not supported)

Claiming interface 1...

Kernel driver attached for interface 2: (not supported)

Claiming interface 2...

Kernel driver attached for interface 3: (not supported)

Claiming interface 3...

Kernel driver attached for interface 4: (not supported)

Claiming interface 4...

Kernel driver attached for interface 5: (not supported)

Claiming interface 5...
   Failed (error -12) Operation not supported or unimplemented on this platform

Reading string descriptors:
   String (0x01): "VCC-GND Studio"
   String (0x02): "YD-RP2040"
   String (0x03): "DE61A0D0C35D4A26"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 1
      IAD 0:
            bFirstInterface: 0
            bInterfaceCount: 2
             bFunctionClass: 02
          bFunctionSubClass: 02
          bFunctionProtocol: 00
                  iFunction: 0

Reading Max LUN:
   Stalled, setting Max LUN to 0

Sending Inquiry:
   sent 6 CDB bytes
   Operation timed out
   Input/Output Error
```

mcuee, 2024-02-20:
@sonatique 

Just try my luck with the libusb windows hotplug PR but it does not help. This is expected though.

```
PS C:\work\libusb\libusb_hotplug\examples> .\xusb.exe 0x2e8a:0x102e
Using libusb v1.0.27.11882

Opening device 2E8A:102E...
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 2E8A:102E
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 218
         descriptor length: 9
             nb interfaces: 6
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 10
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0040
          polling interval: 00
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 83
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 03
           max packet size: 0040
          polling interval: 00
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 84
           max packet size: 0040
          polling interval: 08
       endpoint[1].address: 04
           max packet size: 0040
          polling interval: 08
              interface[4]: id = 4
interface[4].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[5]: id = 5
interface[5].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 01.03.00
       endpoint[0].address: 05
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 85
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Kernel driver attached for interface 1: (not supported)

Claiming interface 1...

Kernel driver attached for interface 2: (not supported)

Claiming interface 2...

Kernel driver attached for interface 3: (not supported)

Claiming interface 3...

Kernel driver attached for interface 4: (not supported)

Claiming interface 4...

Kernel driver attached for interface 5: (not supported)

Claiming interface 5...
   Failed (error -12) Operation not supported or unimplemented on this platform

Reading string descriptors:
   String (0x01): "VCC-GND Studio"
   String (0x02): "YD-RP2040"
   String (0x03): "DE61A0D0C35D4A26"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 1
      IAD 0:
            bFirstInterface: 0
            bInterfaceCount: 2
             bFunctionClass: 02
          bFunctionSubClass: 02
          bFunctionProtocol: 00
                  iFunction: 0

Reading Max LUN:
   Stalled, setting Max LUN to 0

Sending Inquiry:
   sent 6 CDB bytes
   Operation timed out
   Input/Output Error
```