libusb/libusb #1243: libusb_control_transfer would be stick on windows 11 with libusb0.sys 1.2.7.3

lujnan, 2023-02-06:
```
    rc = libusb_control_transfer(devh, LIBUSB_REQUEST_TYPE_VENDOR|LIBUSB_RECIPIENT_DEVICE, 0x03, 0xb0, 0, NULL, 0, 1000);
    if(rc == 0)
      break;
    LOGE("Control Transfer: {}, {}", libusb_error_name(rc), libusb_strerror(rc));
```

the function libusb_control_transfer no return, no timeout.

send libusb_control_transfer with zero wLength with libusb0.sys 1.2.7.3,
log as:

Control Transfer: LIBUSB_ERROR_IO, Input/Output Error

mcuee, 2023-02-07:
Do you know the communication protocol of the device? Does the device support this control transfer? Does it work with WinUSB driver?

mcuee, 2023-02-07:
You also need to provide more info.
1) What is the libusb version and Linux distro used?
2) You need to provide the snip of codes used to reproduce the issue.
3) You need to provide the debug log with LIBUSB_DEBUG=4.
4) You need to provide more info about the device.

More info:
https://github.com/libusb/libusb/wiki/Troubleshooting

lujnan, 2023-02-07:
> Do you know the communication protocol of the device? Does the device support this control transfer? Does it work with WinUSB driver?
This is a custom protocol, but it is normal on android, macosx and linux. The device is a composite device. There are three interfaces listed in zadig, the first(interface0) is audio with libusbsys0.sys 1.2.73, the second(interface2) is HidUsb (v10.0.22000.1), the last is usbser (v10.0.22000.1) 

On the linux system the same device will show as:
```
T:  Bus=01 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  5 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=04d2 ProdID=162f Rev= 2.00
S:  Manufacturer= XXXXXX.
S:  Product= XXXX
S:  SerialNumber=1501082252000162
C:* #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=100mA
A:  FirstIf#= 0 IfCount= 2 Cls=01(audio) Sub=00 Prot=00
A:  FirstIf#= 3 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
I:* If#= 0 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=00 Driver=snd-usb-audio
I:  If#= 1 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
I:* If#= 1 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
E:  Ad=83(I) Atr=09(Isoc) MxPS=  68 Ivl=1ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=cafe
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=02(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 3 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=01 Driver=(none)
E:  Ad=84(I) Atr=03(Int.) MxPS=   8 Ivl=16ms
I:* If#= 4 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=(none)
E:  Ad=01(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=81(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
```

lujnan, 2023-02-07:
When I use libusb-win32(v 1.2.6.0), The transmitted data can be received, but the IO error will be printed on the host

log like as:
Control Transfer: LIBUSB_ERROR_IO, Input/Output Error
Control Transfer: LIBUSB_ERROR_IO, Input/Output Error
Control Transfer: LIBUSB_ERROR_IO, Input/Output Error

So I used the libusb0.sys 1.2.73 version , but got stuck

mcuee, 2023-02-07:
@lujnan 
You need to set Environmental Variable `LIBUSB_DEBUG` to be equal to `4` and then run your application and then post the debug log.

You also need to provide more info.
1) What is the libusb-1.0 version and Windows version used?
2) You need to provide the snip of codes used to reproduce the issue.
3) You need to provide the debug log with LIBUSB_DEBUG=4.
4) You need to provide more info about the device. Do you know the communication protocol of the device? Does the device support this control transfer?
5) Does it work with WinUSB or libusbK driver? You can use Zadig to switch to WinUSB or libusbK driver. In terms of driver, it is better to use WinUSB driver or libusbK driver as there may be some issues with the support of libusb0.sys.

More info:
https://github.com/libusb/libusb/wiki/Troubleshooting

mcuee, 2023-02-13:
> That you get LIBUSB_ERROR_IO on 1.2.6 could /possibly/ be improved by #1018. However, that it is instead hanging with 1.2.7.3 sounds like a regression in libusb0.sys.

Indeed it sounds like a regression in libusb0.sys, but let's wait for the debug log to see what happens.

lujnan, 2023-02-17:
Sorry for replying so late, our custom protocol,  ```libusb_control_transfer(devh, LIBUSB_REQUEST_TYPE_VENDOR|LIBUSB_RECIPIENT_DEVICE, 0x03, 0xb0, 0, NULL, 0, 1000);```  this function is to switch from normal working mode to dfu mode, and the device side restarts after receiving this message, which is probably caused by the IO exception on the host side. I will try to fix the normal return packet on the device side first.

tormodvolden, 2023-02-17:
This is a Zero-Length Packet control transfer. They were broken in libusb0.sys 1.2.6 and would always return immediately with error without sending anything to the device (https://github.com/mcuee/libusb-win32/issues/21). In 1.2.7.3 these transfers should work, but it is possibly they will hang for a long time if the device doesn't respond.

lujnan, 2023-02-17:
> This is a Zero-Length Packet control transfer. They were broken in libusb0.sys 1.2.6 and would always return immediately with error without sending anything to the device (https://github.com/mcuee/libusb-win32/issues/21). In 1.2.7.3 these transfers should work, but it is possible there is an issue so it doesn't time out properly if the device doesn't respond.

You mean that even if the device restarts the device immediately after receiving the control packet, the IO on the host should return normally. Does the device not need to respond to this control transfer?

Which version of the log do you need, 1.2.6 or 1.2.7.3

tormodvolden, 2023-02-17:
Yes, the device should at least return an ACK on the control transfer before it restarts. I don't know how it is implemented in your firmware, but it would be reasonable to return ACK as on any successful control transfer and set a special reset flag in your USB call-back, then in your main loop check for this flag and reset, maybe after a small delay.

tormodvolden, 2023-02-17:
The 1.2.7.3 log is most important, but it could be interesting to see both.

lujnan, 2023-02-20:
there is the log of the 1.2.6 libusb0.sys, run on windows
```
.\xdemo.exe : [timestamp] [threadID] facility level [function call] <message>
所在位置 行:1 字符: 1
+ .\xdemo.exe 2>&1 > .\xdemo_1.2.6.log
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: ([timestamp] [th...call] <message>:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
--------------------------------------------------------------------------------
[ 0.000023] [00002d68] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000045] [00002d68] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000090 events 0
[ 0.000051] [00002d68] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000052] [00002d68] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000071] [00002d68] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000076] [00002d68] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001809] [00002d68] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.003452] [00002d68] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.005211] [00002d68] libusb: info [windows_init] UsbDk backend is not available
[ 0.005243] [00002d68] libusb: warning [libusb_init] installing new context as implicit default
[ 0.005244] [00002d68] libusb: debug [libusb_get_device_list]  
[ 0.005298] [000004dc] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.013969] [00002d68] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.014022] [00002d68] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.014406] [00002d68] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.014442] [00002d68] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.014447] [00002d68] libusb: debug [winusb_get_device_list] allocating new device for session [43]
[ 0.014560] [00002d68] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.014584] [00002d68] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.014585] [00002d68] libusb: debug [winusb_get_device_list] allocating new device for session [44]
[ 0.014698] [00002d68] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.014721] [00002d68] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.014721] [00002d68] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.014833] [00002d68] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_00000000&REV_20\3&11583659&0&A0' bus number 1
[ 0.014852] [00002d68] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_00000000&REV_01\3&11583659&0&68' bus number 2
[ 0.015390] [00002d68] libusb: debug [winusb_get_device_list] found existing device for session [45]
[ 0.015450] [00002d68] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.015458] [00002d68] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1026 bytes)
[ 0.015464] [00002d68] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 8): 'USB\VID_5986&PID_212B\5&31036E8E&0&8'
[ 0.015703] [00002d68] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3CE02D8&0&0' reports 5 ports
[ 0.015712] [00002d68] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3CE02D8&0&0'
[ 0.015857] [00002d68] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&C2333A7&0&0' reports 16 ports
[ 0.015880] [00002d68] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&C2333A7&0&0'
[ 0.016056] [00002d68] libusb: debug [winusb_get_device_list] found existing device for session [44]
[ 0.016096] [00002d68] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.016102] [00002d68] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 206 bytes)
[ 0.016106] [00002d68] libusb: debug [init_device] (bus: 1, addr: 15, depth: 1, port: 5): 'USB\VID_04D2&PID_162F\101E002245000013'
[ 0.016169] [00002d68] libusb: debug [winusb_get_device_list] extra GUID: {98E955ED-CF8D-465D-9D3A-6B09EF5C30B6}
[ 0.016230] [00002d68] libusb: debug [winusb_get_device_list] found existing device for session [43]
[ 0.016270] [00002d68] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.016289] [00002d68] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.016292] [00002d68] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&31036E8E&0&10'
[ 0.016431] [00002d68] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL01\6&1D7A9210&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.016478] [00002d68] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL02\6&1D7A9210&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.016496] [00002d68] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL03\6&1D7A9210&0&0002' (non USB HID, newly connected, etc.) - ignoring
[ 0.016508] [00002d68] libusb: debug [winusb_get_device_list] setting composite interface for [44]:
[... cut ...]
[ 0.018523] [00002d68] libusb: debug [libusb_close]  
[ 0.018539] [00002d68] libusb: debug [libusb_unref_device] destroy device 1.15
[ 0.018540] [00002d68] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.018544] [00002d68] libusb: debug [libusb_exit]  
[ 0.018549] [000004dc] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.018956] [00002d68] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000094
[ 0.018960] [00002d68] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000090

```

lujnan, 2023-02-20:
there is the log of the 1.2.7.3 libusb0.sys, run on windows 11 but not stuck
```
.\xdemo.exe : [timestamp] [threadID] facility level [function call] <message>
所在位置 行:1 字符: 1
+ .\xdemo.exe 2>&1 > .\xdemo_1.2.7.3.log
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: ([timestamp] [th...call] <message>:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
--------------------------------------------------------------------------------
[ 0.000062] [00000f28] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000078] [00000f28] libusb: debug [usbi_add_event_source] add HANDLE 000000000000009c events 0
[ 0.000083] [00000f28] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000083] [00000f28] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000a0 events 0
[ 0.000097] [00000f28] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000098] [00000f28] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001896] [00000f28] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.004835] [00000f28] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007587] [00000f28] libusb: info [windows_init] UsbDk backend is not available
[ 0.007631] [00000f28] libusb: warning [libusb_init] installing new context as implicit default
[ 0.007632] [00000f28] libusb: debug [libusb_get_device_list]  
[ 0.007964] [000010d8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.015984] [00000f28] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.016030] [00000f28] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.016398] [00000f28] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.016436] [00000f28] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.016441] [00000f28] libusb: debug [winusb_get_device_list] allocating new device for session [3A]
[ 0.016552] [00000f28] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.016577] [00000f28] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.016577] [00000f28] libusb: debug [winusb_get_device_list] allocating new device for session [3B]
[ 0.016687] [00000f28] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.016710] [00000f28] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.016711] [00000f28] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.016799] [00000f28] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_00000000&REV_
20\3&11583659&0&A0' bus number 1
[ 0.016817] [00000f28] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_00000000&REV_
01\3&11583659&0&68' bus number 2
[ 0.017348] [00000f28] libusb: debug [winusb_get_device_list] found existing device for session [3C]
[ 0.017404] [00000f28] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.017410] [00000f28] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1026
 bytes)
[ 0.017416] [00000f28] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 8): 'USB\VID_5986&PID_212B\5&31036
E8E&0&8'
[ 0.017635] [00000f28] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3CE02D8&0&0' reports 5 ports
[ 0.017650] [00000f28] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3CE02D8&0&0'
[ 0.017838] [00000f28] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&C2333A7&0&0' reports 16 ports
[ 0.017861] [00000f28] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&C2333A7&0&0'
[ 0.018066] [00000f28] libusb: debug [winusb_get_device_list] found existing device for session [3B]
[ 0.018104] [00000f28] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.018109] [00000f28] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 206 
bytes)
[ 0.018112] [00000f28] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_04D2&PID_162F\101E002
245000013'
[ 0.018172] [00000f28] libusb: debug [winusb_get_device_list] extra GUID: {664ADD4F-4362-4A4F-8372-D966B3F03F4C}
[ 0.018233] [00000f28] libusb: debug [winusb_get_device_list] found existing device for session [3A]
[ 0.018271] [00000f28] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.018290] [00000f28] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 
bytes)
[ 0.018293] [00000f28] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&3103
6E8E&0&10'
[ 0.018440] [00000f28] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL01\6&1D7A
9210&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.018497] [00000f28] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL02\6&1D7A
9210&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.018525] [00000f28] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL03\6&1D7A
9210&0&0002' (non USB HID, newly connected, etc.) - ignoring
[ 0.018585] [00000f28] libusb: debug [winusb_get_device_list] setting composite interface for [3B]:
[... cut ...]
[ 0.020842] [00000f28] libusb: debug [libusb_close]  
[ 0.020858] [00000f28] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.020859] [00000f28] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.020863] [00000f28] libusb: debug [libusb_exit]  
[ 0.020873] [000010d8] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.021338] [00000f28] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000a0
[ 0.021342] [00000f28] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000009c

```

lujnan, 2023-02-20:
there are the test codes.
```
lujnan@MacBook2025 xdemo % cat xdemo.cpp 
#include <cstdio>
#include "libusb-1.0/libusb.h"

static int switch_to_dfu(struct libusb_device_handle *devh){
  int rc;
  const uint8_t request = 0x03;
  const uint16_t value = 0xb0;
  const uint16_t index = 0;
  uint8_t bmRequest = LIBUSB_REQUEST_TYPE_VENDOR|LIBUSB_RECIPIENT_DEVICE;

  for (int retry = 0; retry < 1; retry++) {
    rc = libusb_control_transfer(devh, bmRequest, request, value, index, nullptr, 0, 1500);
    if (rc == 0)
      break;

    printf("%s libusb_control_transfer: %s, %s\n", __FUNCTION__, libusb_error_name(rc), libusb_strerror(rc));
#if 0
    if(rc == LIBUSB_ERROR_OVERFLOW)
      libusb_reset_device(devh);
    else if(rc == LIBUSB_ERROR_PIPE)
      libusb_clear_halt(devh, 0);
#endif
  }
  return rc;
}

libusb_device* get_device(libusb_context* ctx, int vid, int pid) {
  int rc;
  struct libusb_device *dev = nullptr;
  struct libusb_device **device_list;
  struct libusb_device_handle *handle = nullptr;

  auto cnt = libusb_get_device_list(ctx, &device_list);
  for (int i = 0; i < cnt; ++i) {
    struct libusb_device* device = device_list[i];
    struct libusb_device_descriptor desc;

    libusb_get_device_descriptor(device, &desc);	
    if ((vid == desc.idVendor) && (pid == desc.idProduct)) {
        dev = device;
        libusb_ref_device(dev);
        break;
    }
  }
  libusb_free_device_list(device_list, 1);
  return dev;
}

int main(int argc, char** argv) {
  int rc;
  libusb_context* ctx = nullptr;
  libusb_device_handle* devh = nullptr;
  libusb_device* dev = nullptr;

  putenv("LIBUSB_DEBUG=4");
  rc = libusb_init(&ctx);
  if (rc < 0) {
    printf("libusb init error.");
    goto lab_dfu_switch_error;
  }
  dev = get_device(ctx, 0x04d2, 0x162f);
  if (dev == nullptr) {
    libusb_exit(ctx);
    goto lab_dfu_switch_error;
  }
  rc = libusb_open(dev, &devh);
  if (rc || devh == nullptr) {
    libusb_unref_device(dev);
    goto lab_dfu_switch_error;
  }

  switch_to_dfu(devh);

lab_dfu_switch_error:
  printf("ready to exit\n");
  libusb_close(devh);
  libusb_unref_device(dev);
  libusb_exit(ctx);

  return 0;
}

```

tormodvolden, 2023-02-20:
Thanks for the logs and code. If you would apply PR #1018 you will probably see the more correct LIBUSB_ERROR_PIPE instead of LIBUSB_ERROR_IO. LIBUSB_ERROR_PIPE on a control transfer typically means the request is not supported by the device.

> there is the log of the 1.2.7.3 libusb0.sys, run on windows 11 but not stuck

I am not sure I understand what you mean by "stuck". Is it hanging forever, waiting to complete? In which case does it get stuck, if not in this test run?

lujnan, 2023-02-20:
there has a long time stuck between _hid_open_ and  _libusb_submit_transfer_.

```
D:\Downloads\LIDAN>xdemo.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.007106] [00000ed0] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.009998] [00000ed0] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.010437] [00000ed0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.010908] [00000ed0] libusb: debug [usbi_add_event_source] add HANDLE 000000000000009c events 0
[ 0.011360] [00000ed0] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.013748] [00000ed0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.015776] [00000ed0] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.020301] [00000ed0] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.023030] [00000ed0] libusb: info [windows_init] UsbDk backend is not available
[ 0.023502] [00000ed0] libusb: warning [libusb_init] installing new context as implicit default
[ 0.023884] [00002b78] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.023906] [00000ed0] libusb: debug [libusb_get_device_list]
[ 0.032034] [00000ed0] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.034121] [00000ed0] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.034910] [00000ed0] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.037707] [00000ed0] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.038118] [00000ed0] libusb: debug [winusb_get_device_list] allocating new device for session [3B]
[ 0.039057] [00000ed0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.039513] [00000ed0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.039989] [00000ed0] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.040160] [00000ed0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.040492] [00000ed0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.041005] [00000ed0] libusb: debug [winusb_get_device_list] allocating new device for session [3D]
[ 0.041154] [00000ed0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_00000000&REV_20\3&11583659&0&A0' bus number 1
[ 0.041471] [00000ed0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_00000000&REV_01\3&11583659&0&68' bus number 2
[ 0.042149] [00000ed0] libusb: debug [winusb_get_device_list] found existing device for session [3D]
[ 0.042543] [00000ed0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.042911] [00000ed0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1026 bytes)
[ 0.043279] [00000ed0] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 8): 'USB\VID_5986&PID_212B\5&31036E8E&0&8'
[ 0.043828] [00000ed0] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3CE02D8&0&0' reports 5 ports
[ 0.044326] [00000ed0] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3CE02D8&0&0'
[ 0.046614] [00000ed0] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&C2333A7&0&0' reports 16 ports
[ 0.046821] [00000ed0] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&C2333A7&0&0'
[ 0.047673] [00000ed0] libusb: debug [winusb_get_device_list] found existing device for session [3C]
[ 0.049761] [00000ed0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.052488] [00000ed0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 206 bytes)
[ 0.052941] [00000ed0] libusb: debug [init_device] (bus: 1, addr: 10, depth: 1, port: 5): 'USB\VID_04D2&PID_162F\101E002245000013'
[ 0.053576] [00000ed0] libusb: debug [winusb_get_device_list] extra GUID: {664ADD4F-4362-4A4F-8372-D966B3F03F4C}
[ 0.054027] [00000ed0] libusb: debug [winusb_get_device_list] found existing device for session [3B]
[ 0.054315] [00000ed0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.054926] [00000ed0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.054960] [00000ed0] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&31036E8E&0&10'
[ 0.055629] [00000ed0] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL01\6&1D7A9210&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.055990] [00000ed0] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL02\6&1D7A9210&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.056410] [00000ed0] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL03\6&1D7A9210&0&0002' (non USB HID, newly connected, etc.) - ignoring
[ 0.056944] [00000ed0] libusb: debug [winusb_get_device_list] setting composite interface for [3C]:
[ 0.056973] [00000ed0] libusb: debug [set_composite_interface] interface[2] = \\?\HID#VID_04D2&PID_162F&MI_02#7&1A7D4188&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.057367] [00000ed0] libusb: debug [get_api_type] driver(s): libusb0
[ 0.057711] [00000ed0] libusb: debug [get_api_type] matched driver name against libusb0
[... cut ...]
[16.116563] [00000ed0] libusb: debug [libusb_close]
[16.117255] [00000ed0] libusb: debug [libusb_unref_device] destroy device 1.10
[16.117645] [00000ed0] libusb: debug [libusb_unref_device] destroy device 1.0
[16.118823] [00000ed0] libusb: debug [libusb_exit]
[16.119589] [00002b78] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[16.120509] [00000ed0] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000009c
[16.120904] [00000ed0] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
```

lujnan, 2023-02-20:
@tormodvolden 
> I am not sure I understand what you mean by "stuck". Is it hanging forever, waiting to complete? In which case does it get stuck, if not in this test run?

Two weeks ago I needed to release a flashing tool to the factory, and I didn't have time to track this problem, now I look back at this problem, I can't reproduce it, it was stuck forever before. Now I fixed not replying Status transaction on device side. There are a few times now that it gets stuck for a while, which is not common.

tormodvolden, 2023-02-20:
At the end of hid_open() there are requests for Manufacturer, Product and SerialNumber strings. If each of those time out after 5 seconds, it would correspond to the 15 s gap in the log. Are string descriptor requests supported by the firmware?

tormodvolden, 2023-02-20:
It is very important which drivers are installed for the interfaces of the device. Maybe this was different two weeks ago.

lujnan, 2023-02-20:
Another log stuck for a little while, and encount IO error, but I already fixed Status Transaction replying.
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001001] [00001d14] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.001391] [00001d14] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.001479] [00001d14] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001566] [00001d14] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.001635] [00001d14] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001684] [00001d14] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003632] [00001d14] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005929] [00001d14] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.008771] [00001d14] libusb: info [windows_init] UsbDk backend is not available
[ 0.009194] [00001d14] libusb: warning [libusb_init] installing new context as implicit default
[ 0.009234] [00001d14] libusb: debug [libusb_get_device_list]
[ 0.009430] [00001350] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.017393] [00001d14] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.017636] [00001d14] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.018121] [00001d14] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.018470] [00001d14] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.018643] [00001d14] libusb: debug [winusb_get_device_list] allocating new device for session [3B]
[ 0.018997] [00001d14] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.019180] [00001d14] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.019314] [00001d14] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.019473] [00001d14] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.019649] [00001d14] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.019783] [00001d14] libusb: debug [winusb_get_device_list] allocating new device for session [3D]
[ 0.020016] [00001d14] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_00000000&REV_20\3&11583659&0&A0' bus number 1
[ 0.020186] [00001d14] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_00000000&REV_01\3&11583659&0&68' bus number 2
[ 0.020925] [00001d14] libusb: debug [winusb_get_device_list] found existing device for session [3D]
[ 0.021059] [00001d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.021244] [00001d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1026 bytes)
[ 0.021510] [00001d14] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 8): 'USB\VID_5986&PID_212B\5&31036E8E&0&8'
[ 0.022081] [00001d14] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3CE02D8&0&0' reports 5 ports
[ 0.022227] [00001d14] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3CE02D8&0&0'
[ 0.022553] [00001d14] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&C2333A7&0&0' reports 16 ports
[ 0.022613] [00001d14] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&C2333A7&0&0'
[ 0.022947] [00001d14] libusb: debug [winusb_get_device_list] found existing device for session [3C]
[ 0.023034] [00001d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.023136] [00001d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 206 bytes)
[ 0.023309] [00001d14] libusb: debug [init_device] (bus: 1, addr: 13, depth: 1, port: 5): 'USB\VID_04D2&PID_162F\101E002245000013'
[ 0.023660] [00001d14] libusb: debug [winusb_get_device_list] extra GUID: {664ADD4F-4362-4A4F-8372-D966B3F03F4C}
[ 0.023824] [00001d14] libusb: debug [winusb_get_device_list] found existing device for session [3B]
[ 0.023980] [00001d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.024127] [00001d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.024286] [00001d14] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&31036E8E&0&10'
[ 0.024629] [00001d14] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL01\6&1D7A9210&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.024743] [00001d14] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL02\6&1D7A9210&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.024887] [00001d14] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\VID_04F3&PID_0000&COL03\6&1D7A9210&0&0002' (non USB HID, newly connected, etc.) - ignoring
[ 0.025041] [00001d14] libusb: debug [winusb_get_device_list] setting composite interface for [3C]:
[ 0.025159] [00001d14] libusb: debug [set_composite_interface] interface[2] = \\?\HID#VID_04D2&PID_162F&MI_02#7&1A7D4188&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.025706] [00001d14] libusb: debug [get_api_type] driver(s): libusb0
[ 0.025775] [00001d14] libusb: debug [get_api_type] matched driver name against libusb0
[... cut ...]
[16.088555] [00001d14] libusb: debug [libusb_close]
[16.088735] [00001d14] libusb: debug [libusb_unref_device] destroy device 1.13
[16.088888] [00001d14] libusb: debug [libusb_unref_device] destroy device 1.0
[16.089049] [00001d14] libusb: debug [libusb_exit]
[16.089231] [00001350] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[16.089748] [00001d14] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088
[16.089949] [00001d14] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000094
```

tormodvolden, 2023-02-20:
> libusb v1.0.26.11724

This is the v1.0.26 release. We have fixed important issues for composite devices since then in git master. I'd suggest you check out the code from PR #1018 since you are using the libusb0.sys driver, I have rebased the PR to latest master.

lujnan, 2023-02-20:
> At the end of hid_open() there are requests for Manufacturer, Product and SerialNumber strings. If each of those time out after 5 seconds, it would correspond to the 15 s gap in the log. Are string descriptor requests supported by the firmware?

Yes, support for string descriptors.

It will be stuck for a little while, maybe I executed it too quickly, after inserting the device. Waiting for a few seconds after inserting is not easy to reproduce.

lujnan, 2023-02-20:
> It is very important which drivers are installed for the interfaces of the device. Maybe this was different two weeks ago.

The device is a composite device. There are three interfaces listed in zadig, the first(interface0) is audio with libusbsys0.sys 1.2.73, the second(interface2) is HidUsb (v10.0.22000.1), the last is usbser (v10.0.22000.1). The driver has not changed.

tormodvolden, 2023-02-20:
> It will be stuck for a little while, maybe I executed it too quickly, after inserting the device. Waiting for a few seconds after inserting is not easy to reproduce.

Maybe your device is too quick to announce itself on the bus. You should make sure the USB stack is ready before pulling the D+ line high.

mcuee, 2023-02-22:
@lujnan 

Just to debug the issue, you may want to write a test program using libusb-win32 API (which is basically the same as libusb-0.1 API plus some extra stuff). This may help to confirm this is an issue with libusb Windows backend, or with libusb0.sys driver.

mcuee, 2023-02-22:
@lujnan 

You can also try to use WinUSB drive to see if the issue is still present or not.

mcuee, 2023-02-23:
> @mcuee
> 
> > You can also try to use WinUSB drive to see if the issue is still present or not. This will also help to triage the issue to be libusb Windows related or your device related.
> 
> Use The WinUSB drive is good.

Great. Then that shows the device is okay and you have a work-around already.

To further debug the issue, please use PR #1018 to see if that helps. Thanks. You can easily build it using Visual C++ 2019/2022 or use MSYS2 mingw.

closed 2023-02-25 by lujnan

mcuee, 2023-02-25:
@lujnan 

Just wondering if you want to try PR #1018 or not with libusb0.sys. It is okay if you do not have time. Thanks.

mcuee, 2023-04-21:
@lujnan

Just wondering if you want to try PR https://github.com/libusb/libusb/pull/1018 or not with libusb0.sys. Thanks.