libusb/libusb #1244: libusb_exit() Segmentation fault in Linux

yuren0705, 2023-02-07:
Occasional segmentation fault occurs when I use libusb_exit()；
The following is my gdb error message：
#0  0xb6f7b738 in usbi_hotplug_exit () from /lib/libusb-1.0.so.0
(gdb) bt
#0  0xb6f7b738 in usbi_hotplug_exit () from /lib/libusb-1.0.so.0
#1  0xb6f78a2c in libusb_exit () at core.c:2448
#2  0x000ee260 in common_close_usb_device (state=0xb5b89b28, h_plate=0xb5b91428) at ../core/_ty_usb_m.c:2221
#3  0x000ec2e2 in free_h_plate (state=0xb5b89b28) at ../core/_ty_usb_m.c:1553

mcuee, 2023-02-07:
You need to provide more info.
1) What is the libusb version and Linux distro used?
2) You need to provide the snip of codes used to reproduce the issue.
3) You need to provide the debug log with LIBUSB_DEBUG=4.

More info:
https://github.com/libusb/libusb/wiki/Troubleshooting

yuren0705, 2023-02-07:
Thank you for your reply
1 Linux version 3.4.39 (root@ubuntu) (gcc version 4.6.3 20120201 (prerelease) (crosstool-NG linaro-1.13.1-2012.02-20120222 - Linaro GCC 2012.02) ) #117 SMP PREEMPT Wed Dec 16 16:08:44 CST 2020
2 libusbVersion = 1.0.26
[image]
[image]
[image]

yuren0705, 2023-02-07:
Sorry, because this problem is episodic, I have very little debugging information at the moment

mcuee, 2023-02-07:
@yuren0705 

From your gdb output, this is probably not a libusb problem, but rather the Linux kernel USB stack problem. You may want to upgrade your kernel to see if the issue goes away.

```
#2  0x000ee260 in common_close_usb_device (state=0xb5b89b28, h_plate=0xb5b91428) at ../core/_ty_usb_m.c:2221
#3  0x000ec2e2 in free_h_plate (state=0xb5b89b28) at ../core/_ty_usb_m.c:1553
```

You can also test with a normal x86_64 Linux machine to see if you can reproduce the issue.

yuren0705, 2023-02-07:
Thank you for the answer I will try to debug in this direction

yuren0705, 2023-02-07:
sorry excuse me again,Why is it doubtful Linux kernel USB stack problem?

mcuee, 2023-02-07:
> sorry excuse me again,Why is it doubtful Linux kernel USB stack problem?

This is because ` ../core/_ty_usb_m.c:1553` is apparently not from libusb, but rather some Linux kernel related file.

yuren0705, 2023-02-07:
_ty_usb_m.c Not at all Linux kernel related file，It's a program that I wrote, but it's named _ty_usb_m.c

yuren0705, 2023-02-07:
If there is a libusb_init() request failure,I did not judge its return value,Whether this is the case

mcuee, 2023-02-07:
Please provide more info.

1. Please provide the codes used to reproduce the issue. It is better to be complete minimum example codes
2. Please provide the debug log with the environment variable LIBUSB_DEBUG=4.
3. Please check the codes with a normal x86_64 Linux machine (say Ubuntu 20.04 or Ubuntu 22.04 or Debian 11). If the codes work there, then again it may have something to do with your embedded Linux distro.

More info:
https://github.com/libusb/libusb/wiki/Troubleshooting

mcuee, 2023-02-07:
> If there is a libusb_init() request failure,I did not judge its return value,Whether this is the case

Yes that will cause a problem.

hjelmn, 2023-02-10:
Yeah, you can't ignore a `libusb_init()` failure. No libusb functions can be used unless there is a working context.

mcuee, 2023-02-10:
@yuren0705 

I will close this issue for now. Please reopen if you found new issues after dealing with `libusb_init()` failure.

closed 2023-02-10 by mcuee