libusb/libusb #1249: Handling broken descriptors

robert-billing, 2023-02-23:
I have run into a device which has a broken implementation of USB. In the device descriptor bNumConfigurations is returned by the device as 2, but only one configuration descriptor exists. This causes "Entity not found" to be reported, and the device is inaccessible.

This patch makes missing configuration descriptors report as the first descriptor, and bypasses the problem. I now have the device working.
[conf.txt](https://github.com/libusb/libusb/files/10814269/conf.txt)

mcuee, 2023-02-23:
Please create a Pull Request for this issue. Thanks.

tormodvolden, 2023-02-23:
What device is this?

mcuee, 2023-08-16:
@robert-billing 

Please help to answer the question from @tormodvolden. Thanks. It will be good to know the problematic device to see if the patch is acceptable or not.

mcuee, 2023-12-21:
Reference:
* #1250