libusb/libusb #1254: libusb_bulk_transfer with zero length package with unexpected length

ezono-frankm, 2023-03-01:
I'm accessing a TI TMS320F2075 MCU using LibUSB v1.0.26 on Windows 11 22H2.
Under normal operation I'm continuously reading 256 bytes from the device using 
`libusb_bulk_transfer`. My endpoints max. packet size is 64, so I need to send a zero-length-package for 
correct operation.

I used a similar code as in #1029, to send the zero-length-package
```
int transferred = 0;
const auto res = libusb_bulk_transfer(handle, 0x82, nullptr, 0, &transferred, 1000);
```
This call was returning no error code but `transferred` was set to 256 and not as expected 0. 
All following reads where completely corrupted.

I found a solution to this by changing what I'm passing as `data` and `length` to
```
int transferred = 0;
std::uint8_t buffer = ~0;
const auto res = libusb_bulk_transfer(handle, 0x82, &buffer, 1, &transferred, 1000);
```
In this case the zero-length-packet is sent correctly, `transferred` is 0 after the call and following reads are not corrupted.

My solution to send the zero-length-packet feels counterintuitive, doing a transfer with length 1 to do a transfer of a zero-length-package. 
Or is this the intended way to send a zero-length-package?

mcuee, 2023-03-01:
Just wondering if the result is the same if using WinUSB driver. Thanks.

https://libusb.sourceforge.io/api-1.0/libusb_caveats.html
**Zero length packets**
libusb is able to send a packet of zero length to an endpoint simply by submitting a transfer of zero length.
The [LIBUSB_TRANSFER_ADD_ZERO_PACKET](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#gga1fb47dd0f7c209b60a3609ff0c03d56da26b66334b6ec0537c49841ca623d901f) flag is currently supported on Linux, Darwin and Windows (WinUSB).

ezono-frankm, 2023-03-01:
Thanks for your reply. 
Forgot to mention; I'm using WinUSB (through a custom INF that references winusb.inf)

As I understood the [LIBUSB_TRANSFER_ADD_ZERO_PACKET](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#gga1fb47dd0f7c209b60a3609ff0c03d56da26b66334b6ec0537c49841ca623d901f) flag is used with asynchronous I/O and `libusb_transfer` structure, which I'm not using.

I may try the other driver, thanks.

For completeness, see the debug messages from libusb when sending the zero-length-package below:
```
LLD/D: [ 9.107919] [0000a5f0] libusb: debug [libusb_submit_transfer] transfer 00000000045642B0
LLD/D: [ 9.107925] [0000a5f0] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
LLD/D: [ 9.107930] [0000a5f0] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 82 with interface 1
LLD/D: [ 9.107934] [0000a5f0] libusb: debug [winusbx_submit_bulk_transfer] reading 0 bytes
LLD/D: [ 9.107944] [0000a5f0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
LLD/D: [ 9.107947] [000079a0] libusb: debug [windows_iocp_thread] transfer 00000000045642B0 completed, length 256
LLD/D: [ 9.107951] [0000a5f0] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
LLD/D: [ 9.107957] [0000a5f0] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
LLD/D: [ 9.107963] [0000a5f0] libusb: debug [handle_event_trigger] event triggered
LLD/D: [ 9.107967] [0000a5f0] libusb: debug [windows_handle_transfer_completion] handling transfer 00000000045642B0 completion with errcode 0, length 256
LLD/D: [ 9.107971] [0000a5f0] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
LLD/D: [ 9.107977] [0000a5f0] libusb: debug [usbi_handle_transfer_completion] transfer 00000000045642B0 has callback 0000000070988790
LLD/D: [ 9.107981] [0000a5f0] libusb: debug [sync_transfer_cb] actual_length=256
LLD/D: [ 9.107985] [0000a5f0] libusb: debug [libusb_free_transfer] transfer 00000000045642B0 
```

mcuee, 2023-03-01:
Sorry but you do not need to try other driver if you are already using WinUSB. 

But you may want to post the debug log for the other one.

ezono-frankm, 2023-03-01:
I was assuming that as well, but it does not make a difference.

With this
```
int transferred = 0;
unsigned char buffer[1];
const auto res = libusb_bulk_transfer(handle, 0x82, buffer, 0, &transferred, 1000);
```

the log is as below

```
LLD/D: [14.353519] [00003054] libusb: debug [libusb_submit_transfer] transfer 000000000552CD40
LLD/D: [14.353522] [00003054] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
LLD/D: [14.353526] [00003054] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 82 with interface 1
LLD/D: [14.353531] [00003054] libusb: debug [winusbx_submit_bulk_transfer] reading 0 bytes
LLD/D: [14.353543] [00003054] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
LLD/D: [14.353546] [0000aa80] libusb: debug [windows_iocp_thread] transfer 000000000552CD40 completed, length 256
LLD/D: [14.353548] [00003054] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
LLD/D: [14.353563] [00003054] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
LLD/D: [14.353572] [00003054] libusb: debug [handle_event_trigger] event triggered
LLD/D: [14.353575] [00003054] libusb: debug [windows_handle_transfer_completion] handling transfer 000000000552CD40 completion with errcode 0, length 256
LLD/D: [14.353579] [00003054] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
LLD/D: [14.353583] [00003054] libusb: debug [usbi_handle_transfer_completion] transfer 000000000552CD40 has callback 0000000070988790
LLD/D: [14.353588] [00003054] libusb: debug [sync_transfer_cb] actual_length=256
LLD/D: [14.353591] [00003054] libusb: debug [libusb_free_transfer] transfer 000000000552CD40
```

ezono-frankm, 2023-03-01:
The same code snipped from my last comment - only with setting `length = 1` makes it work.

```
int transferred = 0;
unsigned char buffer[1];
const auto res = libusb_bulk_transfer(handle, 0x82, buffer, 1, &transferred, 1000);
```

Here is the log for the transfer:
```
LLD/D: [ 5.205183] [0000776c] libusb: debug [libusb_submit_transfer] transfer 0000000006DF8D40
LLD/D: [ 5.205186] [0000776c] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
LLD/D: [ 5.205190] [0000776c] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 82 with interface 1
LLD/D: [ 5.205194] [0000776c] libusb: debug [winusbx_submit_bulk_transfer] reading 1 bytes
LLD/D: [ 5.205229] [0000776c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
LLD/D: [ 5.205241] [0000776c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
LLD/D: [ 5.205306] [0000b40c] libusb: debug [windows_iocp_thread] transfer 0000000006DF8D40 completed, length 0
LLD/D: [ 5.205335] [0000776c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
LLD/D: [ 5.205375] [0000776c] libusb: debug [handle_event_trigger] event triggered
LLD/D: [ 5.205379] [0000776c] libusb: debug [windows_handle_transfer_completion] handling transfer 0000000006DF8D40 completion with errcode 0, length 0
LLD/D: [ 5.205392] [0000776c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
LLD/D: [ 5.205396] [0000776c] libusb: debug [usbi_handle_transfer_completion] transfer 0000000006DF8D40 has callback 0000000070988790
LLD/D: [ 5.205399] [0000776c] libusb: debug [sync_transfer_cb] actual_length=0
LLD/D: [ 5.205402] [0000776c] libusb: debug [libusb_free_transfer] transfer 0000000006DF8D40
```

tormodvolden, 2023-03-01:
Indeed, LIBUSB_TRANSFER_ADD_ZERO_PACKET would be for asynchronous transfers, and for host-to-device only (OUT). I understand you are reading _from_ the device here, although you haven't pasted the code for the actual transfer that you are interested in. What is the transaction protocol? Is the ZLP explicitly needed for IN transfers?

> This call was returning no error code but transferred was set to 256 and not as expected 0.

That is odd. Is the device firmware always using 256 byte transfers? (In which case a ZLP shouldn't be needed). Or where is the 256 coming from? Otherwise it could look like a bug in libusb. If you ask for x bytes you shouldn't get more than those.

ezono-frankm, 2023-03-02:
The firmware queries on request a number of sensors and provides the data, which happens to be 256 bytes, in a loop like this
```
while query_sensors_enabled
{
   write_transaction command_to_query_sensors
   read_transaction status_byte
   if status_byte indicates data_ready
   {
      write_transaction command_to_get_data
      read_transaction data_buffer
   }
   sleep
}
```

Not transferring a ZLP when reading data_buffer definitively causes data corruption.
Since I was not working on the firmware, I cannot tell much about the protocol and whether the ZLP is explicitly needed. I need to check with the colleague that was working on that.

tormodvolden, 2023-03-03:
Sending a ZLP is useful and used in many cases (often taken care of in intermediate layers), but _requesting_ a zero-length transfer is not common and maybe not a good practice. Note that e.g. WinUSB will just [discard such a request](https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/using-winusb-api-to-communicate-with-a-usb-device):

> Zero-length read requests complete immediately with success and aren't sent down the stack.

If this is low-bandwidth it could be better to use a control transfer instead, because there is a transfer size communicated (say 256 bytes) and the device would just return the 4 packets in one transfer, and stall if it was not ready.

Or add a vendor endpoint or interface control request to reset a packet counter on the device, that gets incremented for every bulk packet and a stall is returned if the host ever tries to get a fifth packet or data is not ready. The host would then just: send control_request; read 256 bytes; repeat if not success. The control request could alternatively return a ready value: while (control request returns not ready) { wait a bit} ; read 256 bytes. The reset-counter request could also be a (e.g. one-byte) OUT bulk transfer if that is easier to implement on the device.

mcuee, 2023-03-04:
> Sending a ZLP is useful and used in many cases (often taken care of in intermediate layers), 

https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/using-winusb-api-to-communicate-with-a-usb-device

**Write requests that contain zero-length data are forwarded down the USB stack. **

> but _requesting_ a zero-length transfer is not common and maybe not a good practice. Note that e.g. WinUSB will just [discard such a [...]

This makes sense, I tend to believe this should be the behavior for other OS as well -- but we may need to confirm.

Section 5.8.3 of the USB 2.0 Standard mentions the following.
https://www.usb.org/document-library/usb-20-specification
_A bulk transfer is complete when the endpoint does one of the following:
• Has transferred exactly the amount of data expected
• Transfers a packet with a payload size less than wMaxPacketSize or transfers a zero-length packet_

For Read, the host knows exactly what happens. Say it requests 256 bytes and the device sends 256B of data ( 4 packets, 64 Bytes each packet). So there is really no need to send another zero-length packet to signal the completion.

For write, the device can not differentiate whether there are more data or not after receiving 256 bytes of data. So the host has to send the zero-length packet to signal the end of the transfer.

mcuee, 2023-03-04:
> This makes sense, I tend to believe this should be the behavior for other OS as well -- but we may need to confirm.

I believe it is the same for Linux as well.
https://www.kernel.org/doc/html/v6.2/driver-api/usb/usb.html

_Data transfer on USB is packetized; each endpoint has a maximum packet size. Drivers must often be aware of conventions such as flagging the end of bulk transfers using “short” (including zero length) packets.

Bulk URBs may use the URB_ZERO_PACKET transfer flag, indicating that bulk OUT transfers should always terminate with a short packet, even if it means adding an extra zero length packet._

I take that the above means Bulk IN does not need the zero length packet and I beleieve there will be no activity on the bus even if the host sends the Bulk IN request with zero length -- but this will need confirmation again from the source code.

tormodvolden, 2023-03-05:
> The reset-counter request could also be a (e.g. one-byte) OUT bulk transfer if that is easier to implement on the device.

Taking this further, it could also be a zero-byte OUT bulk transfer, generating a ZLP. But importantly this would be an OUT transfer, not an IN transfer.

Reading the original issue description, I almost wonder if this the intention in the device protocol. You say that you need to "send" a ZLP, but in your code you are "requesting" one instead.

You would need to send to e.g. OUT endpoint 0x02 (instead of IN endpoint 0x82) and the device firmware would need to check for received ZLP, not check for requests of zero size.

ezono-frankm, 2023-03-06:
We are using accessing the device also from a Linux host using libusb v1.0.21. 
The code is similar with the on on Windows, sending the ZLP during read transactions. On Linux the package is transferred and handled as expected.

One more thing, the `normal` way to access the device under Windows would be as CDC through usbser.sys . There are practical reasons for me to use WinUsb driver and libusb to access the device.

> ** Zero-length read requests complete immediately with success and aren't sent down the stack.**

I guess that explains why I have to send one byte to do the 'ZLP', still it leaves open why the number of bytes returned differs from the requested amount.

ezono-frankm, 2023-03-06:
> > This call was returning no error code but `transferred` was set to 256 and not as expected 0.
> 
> Can you please apply this patch and attach a new debug log? Trying to figure out why this happens. [issue1254_sync_length.patch.txt](https://github.com/libusb/libusb/files/10898762/issue1254_sync_length.patch.txt)

Sure, please allow me some time to give you the logs.

aouchabane, 2023-03-07:
Hello,
please let me add that the firmware is configured as USB CDC, after 256 bytes of IN bulk transfer, the device sends ZLP to the host to notify him that the transmission is complete. So the host has to request to read the ZLP otherwise this ZLP will remain in the IN endpoint of the device, and the next IN bulk read will get the previous ZLP. and the communication will get corrupted.

This is why after reading the 256 bytes, a request to read the ZLP is needed to free the device IN endpoint.

tormodvolden, 2023-03-07:
@aouchabane Thanks for this information!

Maybe it would work out of the box if the host code simply requests 257 bytes. The USB stack will receive 4 full packets, then the ZLP and return the 256 bytes to the user.

mcuee, 2023-03-07:
> Hello, please let me add that the firmware is configured as USB CDC, after 256 bytes of IN bulk transfer, the device sends ZLP to the host to notify him that the transmission is complete. 

Firstly this is not needed. The host knows that it requests 256Bytes from the IN endpoint and therefore it will stop. But this is not wrong either. Once the host received the ZLP, it will not ask for more data. There is no need to for the host to send another ZLP READ request.

> So the host has to request to read the ZLP otherwise this ZLP will remain in the IN endpoint of the device, and the next IN bulk read will get the previous ZLP. and the communication will get [...]

I believe the above is not correct. The host does not need to request to read the ZLP. Please remove this request to see if it is working properly or not.

Section 5.8.3 of the USB 2.0 Standard mentions the following.
https://www.usb.org/document-library/usb-20-specification

```
An endpoint must always transmit data payloads with a data field less than or equal to the endpoint’s
reported wMaxPacketSize value. When a bulk IRP involves more data than can fit in one maximum-sized
data payload, all data payloads are required to be maximum size except for the last data payload, which will
contain the remaining data. A bulk transfer is complete when the endpoint does one of the following:
• Has transferred exactly the amount of data expected
• Transfers a packet with a payload size less than wMaxPacketSize or transfers a zero-length packet

When a bulk transfer is complete, the Host Controller retires the current IRP and advances to the next IRP. 
If a data payload is received that is larger than expected, all pending bulk IRPs for that endpoint will be
aborted/retired.
```

mcuee, 2023-03-07:
@aouchabane and @ezono-frankm 

Just repeat Section 5.8.3 of the USB 2.0 Standard which mentions the following.
https://www.usb.org/document-library/usb-20-specification

```
An endpoint must always transmit data payloads with a data field less than or equal to the endpoint’s
reported wMaxPacketSize value. When a bulk IRP involves more data than can fit in one maximum-sized
data payload, all data payloads are required to be maximum size except for the last data payload, which will
contain the remaining data. A bulk transfer is complete when the endpoint does one of the following:
• Has transferred exactly the amount of data expected
• Transfers a packet with a payload size less than wMaxPacketSize or transfers a zero-length packet

**When a bulk transfer is complete, the Host Controller retires the current IRP and advances to the next IRP.
If a data payload is received that is larger than expected, all pending bulk IRPs for that endpoint will be
aborted/retired. **
```

>When a bulk transfer is complete, the Host Controller retires the current IRP and advances to the next IRP.

I then to believe that is what has happened. The host received 256 Bytes and then it will advances to next IRP. That is why you receive 256 Bytes again.

I see potential solution as following.
1) Do not send the ZLP in the FW if sending the same length of bytes requested by the host (say 256).
2) Do not send the ZLP read request in the host software.

aouchabane, 2023-03-07:
Sending the ZLP is done by USB lib, if I prevent it, the issue is solved and no ZLP read request is needed in the host.

mcuee, 2023-03-07:
> Sending the ZLP is done by USB lib, if I prevent it, the issue is solved and no ZLP read request is needed in the host.

I am not sure what you mean by `the ZLP is done by the USB lib`.

aouchabane, 2023-03-07:
> > Sending the ZLP is done by USB lib, if I prevent it, the issue is solved and no ZLP read request is needed in the host.
> 
> I am not sure what you mean by `the ZLP is done by the USB lib`.
> 
I had to disable sending of the ZLP
> In any case, have you tried to remove the ZLP from the FW side?
Yes

mcuee, 2023-03-07:
Google seems to say the 

> > > Sending the ZLP is done by USB lib, if I prevent it, the issue is solved and no ZLP read request is needed in the host.
> > 
> > 
> > I am not sure what you mean by `the ZLP is done by the USB lib`.
> 
> I had to disable sending of the ZLP

In the host code?

> > In any case, have you tried to remove the ZLP from the FW side?
> 
> Yes

Is it working now?

mcuee, 2023-03-07:
It seems to me that the standard may be ambiguous and it may be driver or upper level protocol specific.

1) I found my own post in Microchip forum here back in 28-March-2008. Looks like at that time Windows `usbser.sys` did ask for a ZLP from the device. So it is not wrong to send the ZLP in the FW.

+++++++++++++++++++++
https://www.microchip.com/forums/m325655.aspx
```
From Jan Axelson's post in the old USBIF Forum (no longer exists now).

A driver can mandate that a transfer that is an exact multiple of wMaxPacketSize terminates in a ZLP to indicate end of transfer.

If wMaxPacketSize = 512 (or 64) and you request to send 511 bytes, the transfer ends with a short packet (<wMaxPacketSize), so there's no need for a ZLP to indicate end of transfer.

The Windows CDC virtual COM-port driver requires a ZLP if the transfer is an exact multiple of wMaxPacketSize because there is no other way to detect the end of the transfer.

And the HID spec says:

All reports except the longest which exceed wMaxPacketSize for the endpoint must terminate with a short packet. The longest report does not require a short packet terminator.

Jan Axelson
++++++++++++++++++++++++++
```
2) In the above forum thread, I quoted another post.
```
Re: [KMDF USB] Who is responsible for zero-length packet termination of bulk write? by Tim

Mon Sep 04 16:07:15 CDT 2006

"Maxim S. Shatskih" wrote:
>
>> For a bulk write, there is no defined method of indicating the end of a USB
>> bulk transfer that ends on a MaxPacketSize boundary. That behavior is class
>> and device specific.
>
>Am I wrong that in this case ZLP is required by the spec?

It depends. A bulk transfer that ends NORMALLY at a MaxPacketSize boundary
does not need a ZLP. But if you need to end a bulk transfer early, and the
actual size is a multiple of MaxPacketSize, then you send a ZLP to signal
the early end of the transfer.

The host controller driver can't know this, so the client has to do it.
--
- Tim Roberts,
Providenza & Boekelheide, Inc.
```

mcuee, 2023-03-07:
More Google search found one of my favority USB related post from Tsuneo in the Cygnal Forum (now part of Silabs) back in 2006. Unfortunately the code was gone. He did add ZLP from the device side for the Bulk IN endpoint two years later in 2008.
https://community.silabs.com/s/question/0D51M00007xeFihSAE/usb-cdc-implementation-for-f32x?language=en_US
```
 v1.5 (May 10, 2008)
 Rewrote most of the source code to fit to composite device
 Added ZLP to terminate the bulk IN transfer
 Fixed bug of FIFO_Read_generic()
 Revised the INF file to support Vista
```

aouchabane, 2023-03-07:
> In the host code?
>
>
> Is it working now?

Yes, it is working.

aouchabane, 2023-03-08:
> @aouchabane Thanks for this information!
> 
> Maybe it would work out of the box if the host code simply requests 257 bytes. The USB stack will receive 4 full packets, then the ZLP and return the 256 bytes to the user.

Exactly this is what I am doing.

aouchabane, 2023-03-08:
> > > > Sending the ZLP is done by USB lib, if I prevent it, the issue is solved and no ZLP read request is needed in the host.
> > > 
> > > 
> > > I am not sure what you mean by `the ZLP is done by the USB lib`.
> > 
> > 
> > I had to disable sending of the ZLP
> 
> In the host code?
In the device

mcuee, 2023-03-08:
> > @aouchabane Thanks for this information!
> > Maybe it would work out of the box if the host code simply requests 257 bytes. The USB stack will receive 4 full packets, then the ZLP and return the 256 bytes to the user.
> 
> Exactly this is what I am doing.

Does this workaround work for you without changing the FW (still sending ZLP from the device side)?

aouchabane, 2023-03-08:
> > Yes, it is working.
> 
> Glad to hear that it is working now.
> 
> So the fix for you is to remove the ZLP in the FW side and remove the ZLP request in the host code, right? Just want to confirm Thanks.

Removing the ZLP on the FW side and removing the ZLP request in the host code was for testing.

I am keeping the FW as it was and I am doing in the host the following:

ZLPSize  = 0
if ( (datasize % 64 ) == 0)
{
    ZLPSize = 1;
}

usb_bulk_transfer(hd, 0x82, buffer, size+ZLPSize, bytesRead, timeouts)

> 
> But now your device may not work with Windows usbser.sys driver if the behavior of usbser.sys is the same as last time (require ZLP for the IN request from the device FW side). Just wondering why you want to use libusb and not the normal Windows Serial API for your application.

The device is still compatible with the Windows Serial driver since the FW still sends the ZLP.

aouchabane, 2023-03-08:
> Does this workaround work for you without changing the FW (still sending ZLP from the device side)?

yes

mcuee, 2023-03-08:
In this case, I will close this issue. Thanks for the updates.

tormodvolden, 2023-03-09:
Let's keep this open at least until we have the logs I asked about. If the user asks for 0 bytes (whether reasonable or not) the function shouldn't return 256 bytes (of possible garbage).

ezono-frankm, 2023-03-20:
> Can you please apply this patch and attach a new debug log? Trying to figure out why this happens. [...]

I just managed to get the logs with a your patch applied

```
[ 3.704024] [00000600] libusb: debug [libusb_submit_transfer] transfer 00000000004E2E80
[ 3.704049] [00000600] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 3.704074] [00000600] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 82 with interface 1
[ 3.704092] [00000600] libusb: debug [winusbx_submit_bulk_transfer] reading 0 bytes
[ 3.704148] [00000600] libusb: debug [winusbx_submit_bulk_transfer] !FM1! synchronous completion, length 0; 0
[ 3.704167] [00000600] libusb: debug [winusbx_submit_bulk_transfer] !FM2! synchronous completion, length 0; 0
[ 3.704186] [00000600] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.704181] [00000f24] libusb: debug [windows_iocp_thread] transfer 00000000004E2E80 completed, length 256
[ 3.704203] [00000600] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 3.704241] [00000600] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 3.704260] [00000600] libusb: debug [handle_event_trigger] event triggered
[ 3.704276] [00000600] libusb: debug [windows_handle_transfer_completion] handling transfer 00000000004E2E80 completion with errcode 0, length 256
[ 3.704295] [00000600] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 3.704315] [00000600] libusb: debug [usbi_handle_transfer_completion] transfer 00000000004E2E80 has callback 000007FEED448C80
[ 3.704332] [00000600] libusb: debug [sync_transfer_cb] actual_length=256
[ 3.704350] [00000600] libusb: debug [libusb_free_transfer] transfer 00000000004E2E80
```

Please note, that with your original patch I was not getting any extra output in the logs, so I added `else` clauses (the two lines starting with `!FM`) after both `if` in `windows_winusb.c` line 3317 and up

```c++
	if (ret) {
		/* successful without overlapped pending */
		usbi_dbg(TRANSFER_CTX(transfer), "synchronous completion, length %lu", len_sync);
	}
	else
	{
		usbi_dbg(TRANSFER_CTX(transfer), "!FM1! synchronous completion, length %lu; %d", len_sync, ret);
	}

	if (!ret && GetLastError() != ERROR_IO_PENDING) {
		usbi_err(TRANSFER_CTX(transfer), "ReadPipe/WritePipe failed: %s", windows_error_str(0));
		return LIBUSB_ERROR_IO;
	}
	else
	{
		usbi_dbg(TRANSFER_CTX(transfer), "!FM2! synchronous completion, length %lu; %d", len_sync, ret);
	}
```

ezono-frankm, 2023-03-20:
> 
> What I would suggest is clamping this value to the original length requested, in this case zero. Can you please rerun with this new patch? It should give a warning that the transfer was truncated to zero length in your case. [issue1254_clamp_length.patch.txt](https://github.com/libusb/libusb/files/11016763/issue1254_clamp_length.patch.txt)

```
[ 5.298749] [00000438] libusb: debug [libusb_submit_transfer] transfer 0000000000522C10
[ 5.298768] [00000438] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 5.298789] [00000438] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 82 with interface 1
[ 5.298806] [00000438] libusb: debug [winusbx_submit_bulk_transfer] reading 0 bytes
[ 5.298839] [00000438] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 5.298857] [00000438] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 5.298922] [00000474] libusb: debug [windows_iocp_thread] transfer 0000000000522C10 completed, length 256
[ 5.299117] [00000438] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 5.299142] [00000438] libusb: debug [handle_event_trigger] event triggered
[ 5.299159] [00000438] libusb: debug [windows_handle_transfer_completion] handling transfer 0000000000522C10 completion with errcode 0, length 256
[ 5.299177] [00000438] libusb: warning [windows_handle_transfer_completion] transfer 0000000000522C10 truncated to requested length 0
[ 5.299195] [00000438] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 5.299214] [00000438] libusb: debug [usbi_handle_transfer_completion] transfer 0000000000522C10 has callback 000007FEE4418C80
[ 5.299231] [00000438] libusb: debug [sync_transfer_cb] actual_length=0
[ 5.299248] [00000438] libusb: debug [libusb_free_transfer] transfer 0000000000522C10
```

The warning is logged and the `actual_length` matches the requested amount of data

mcuee, 2023-04-12:
@tormodvolden 

Just wondering if you have the pull request ready or not. Thanks.

mcuee, 2023-08-15:
@tormodvolden 

Ping. Thanks.