libusb/libusb #1255: Windows question: why is libusbK used in place of libusb0 despite SUB_API_LIBUSB0 being defined?

sonatique, 2023-03-01:
And does this mean that, even if a given device is "officially" using libusb0 (i.e. libusb0.sys, libusb0.dll reported by Device Manager), then in the end all functions of winusb_interface are backed by libusbK implementation and none of libusb0 code will be called? 

I am having this understanding based on this code snippet from windows_winsub.c, funciton winusbx_init, line 2505:
```
// NB: The below for loop works because the sub_api value for WinUSB
// is a higher value than that of libusbK and libusb0
for (; sub_api < SUB_API_WINUSB; sub_api++) {
    libusbK_Set(sub_api, AbortPipe, true);
    libusbK_Set(sub_api, ControlTransfer, true);
    libusbK_Set(sub_api, FlushPipe, true);
    libusbK_Set(sub_api, Free, true);
    libusbK_Set(sub_api, GetAssociatedInterface, true);
    libusbK_Set(sub_api, Initialize, true);
    libusbK_Set(sub_api, ReadPipe, true);
    libusbK_Set(sub_api, ResetPipe, true);
    libusbK_Set(sub_api, SetCurrentAlternateSetting, true);
    libusbK_Set(sub_api, SetPipePolicy, true);
    libusbK_Set(sub_api, WritePipe, true);
//etc.
```
 `sub_api` takes value from 0 to 1, i.e. `SUB_API_LIBUBSK `and `SUB_API_LIBUSB0`, meanig that function pointers corresponding to libusb0 are actually pointing to libusbK.

Thank you in advance for your help understanding this.
Best regards.

sonatique, 2023-03-01:
Hello, thank you for your prompt response. Sorry that I missed the Wiki FAQ.
Yes I got my answers, thank you.

The reason behind it is that I have an issue with libusb_clear_halt: when called to a libusb0.sys device, it sometimes freeze completely, making the whole process unkillable. I am trying to understand where in the call it freezes exactly and started by looking at libusb sources.

But then I guess I shall have a look a libusbK so start with.

tormodvolden, 2023-03-01:
Interesting. However buggy the device is, this shouldn't freeze the program or the machine if a timeout is given.

There are some parts in libusb0.sys, especially the error paths, that I am unsure about. Do you have the WDK so that you can build and test patched drivers? Can you reproduce this in a VM (would be nice for such tests)?

sonatique, 2023-03-02:
Thank you for your interest in my issue. "However buggy the device is, this shouldn't freeze the program": yes, that's my point, I happy that we agree.
As for now I am working on finding a workaround at the application level, that's the priority.
But in parallel I will swap driver to better understand who is responsible for this behavior, and in the end decide whether I could try to fix it or rather change driver for good. I should be able to build patched libusb0.sys version if it happens that this code is where the freeze occurs.

For reference: my device is made of a FX3 connected to a Zynq. The issue happen when the Zynq is power-cycled while the FX3 is kept alive. It occurs with a probabily of 1/5 approximately, so there is probably some incidence of the exact timing of the events.

Also, I just notive I have the same issue with other devices when I hiibernate my computer, remove the devices, then wake it up: it seems that some part of the library / driver do not realize what just happened and go on with the task they were doing. If I issue a clear_halt at this point, it will freeze 100% of the times.

sonatique, 2023-03-05:
I am wondering whether the "freezes" I experience are somehow related to what is described here in https://github.com/libusb/libusb/pull/1127 . I will try with the code proposed in this PR as well.

mcuee, 2023-08-15:
@sonatique 

Any updates on this issue? Thanks.

sonatique, 2023-08-22:
@mcuee : Regarding the "freeze": I should soon be able to reproduce it and study it more in depth. I'll keep you posted

sonatique, 2023-09-19:
@mcuee : So I have been able to reproduce a similar issue, quite easily actually.
I made a "while(1)" loop issuing `libusb_control_transfer `of type Out|Vendor|Device to a previously open device. The detail of the transfer doesn't really matter but it corresponds to a command that is supported by the physical device (FX2).
In this loop I also print the return value if < 0.

I start the program, I can see that the control transfer out are correctly transmitted and received, then I simply disconnect the USB cable. 

The behavior that occurs depends on the driver used. With libusb0.sys 1.2.7.3, the next transfer will fail with error code LIBUSB_ERROR_IO, but the next one will completely freeze the program. 
Since I am launching my program from Visual Studio I can break. I see that the program is stuck in the call at line 3157 of windows_winusb.c 
`if (!WinUSBX[sub_api].ControlTransfer(winusb_handle, *setup ...`
and I can see that it is somewhere in libusbK.dll.

From there I have an unkillable process that will stay there until I reboot the OS.

Using kernel debugger I can see that it is frozen somewhere within libusb0:

`0: kd> !process 0 7 sam3u_benchmark.exe
PROCESS ffffb38cbaec2080
    SessionId: 1  Cid: 6adc    Peb: e03fc78000  ParentCid: 5924
    DirBase: 6b7b63002  ObjectTable: ffff9d02c8118cc0  HandleCount:   2.
    Image: sam3u_benchmark.exe
    VadRoot ffffb38cbaddf300 Vads 53 Clone 0 Private 324. Modified 259. Locked 0.
    DeviceMap ffff9d02a64ea090
    Token                             ffff9d02c9093060
    ElapsedTime                       01:14:28.309
    UserTime                          00:00:00.046
    KernelTime                        00:00:00.046
    QuotaPoolUsage[PagedPool]         107520
    QuotaPoolUsage[NonPagedPool]      7456
    Working Set Sizes (now,min,max)  (1532, 50, 345) (6128KB, 200KB, 1380KB)
    PeakWorkingSetSize                1885
    VirtualSize                       4183 Mb
    PeakVirtualSize                   4188 Mb
    PageFaultCount                    2017
    MemoryPriority                    BACKGROUND
    BasePriority                      8
    CommitCharge                      535

        THREAD ffffb38cbaac1080  Cid 6adc.4274  Teb: 000000e03fc79000 Win32Thread: ffffb38cbadd08f0 WAIT: (Executive) KernelMode Non-Alertable
            ffffae0f7001b5f0  NotificationEvent
        IRP List:
            ffffb38cba1b6a20: (0006,0508) Flags: 00060030  Mdl: 00000000
        Not impersonating
        DeviceMap                 ffff9d02a64ea090
        Owning Process            ffffb38cbaec2080       Image:         sam3u_benchmark.exe
        Attached Process          N/A            Image:         N/A
        Wait Start TickCount      402061
        Context Switch Count      7847           IdealProcessor: 2
        UserTime                  00:00:00.250
        KernelTime                00:00:00.078
        Win32 Start Address 0x00007ff6c81234f4
        Stack Init ffffae0f7001bc90 Current ffffae0f7001b1d0
        Base ffffae0f7001c000 Limit ffffae0f70016000 Call 0000000000000000
        Priority 8 BasePriority 8 PriorityDecrement 0 IoPriority 2 PagePriority 5
        Child-SP          RetAddr           : Args to Child                                                           : Call Site
        ffffae0f`7001b210 fffff802`4fc1bca0 : 00000000`00000008 00000000`ffffffff 00000000`00000000 ffffb38c`a9187158 : nt!KiSwapContext+0x76
        ffffae0f`7001b350 fffff802`4fc1b1cf : 00000801`000005fd 00000000`00000000 ffffae0f`7001b510 00000000`00000000 : nt!KiSwapThread+0x500
        ffffae0f`7001b400 fffff802`4fc1aa73 : ffffae0f`00000000 00000000`00000000 00000000`00000300 ffffb38c`baac11c0 : nt!KiCommitThreadWait+0x14f
        ffffae0f`7001b4a0 fffff802`8e363f2a : ffffae0f`7001b5f0 00000000`00000000 ffffae0f`7001b600 00000000`00000000 : nt!KeWaitForSingleObject+0x233
        ffffae0f`7001b590 fffff802`8e36713e : 001f0003`ffff0000 fffff802`8e369b50 ffffae0f`7001b7e0 00000000`00000000 : libusb0+0x3f2a
        ffffae0f`7001b620 fffff802`8e362bc0 : ffffb38c`ba8ce1a0 00000000`00000000 ffffb38c`ba8ce1a0 ffffb38c`9bfeca00 : libusb0+0x713e
        ffffae0f`7001b720 fffff802`8e3614b9 : 00000000`00000002 ffffb38c`ba1b6a20 00000000`00000000 00000000`00000000 : libusb0+0x2bc0
        ffffae0f`7001b7d0 fffff802`4fc10665 : ffffb38c`b9034500 00000000`00000000 ffffae0f`20206f49 00000000`00000020 : libusb0+0x14b9
        ffffae0f`7001b800 fffff802`50000fec : 00000000`00000002 00000000`00000000 ffffb38c`b9034500 ffffb38c`baec2080 : nt!IofCallDriver+0x55
        ffffae0f`7001b840 fffff802`50000c41 : ffffb38c`ba1b6e98 ffffae0f`7001bb80 00000000`00040005 ffffb38c`ba1b6e98 : nt!IopSynchronousServiceTail+0x34c
        ffffae0f`7001b8e0 fffff802`4fffffb6 : 00000000`00000000 00000000`000002fc 00000000`00000000 000000e0`3feff890 : nt!IopXxxControlFile+0xc71
        ffffae0f`7001ba20 fffff802`4fe105f8 : ffffb38c`baac1080 000000e0`3feff7b8 ffffae0f`7001baa8 00000000`00000000 : nt!NtDeviceIoControlFile+0x56
        ffffae0f`7001ba90 00007ffc`d950cf34 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nt!KiSystemServiceCopyEnd+0x28 (TrapFrame @ ffffae0f`7001bb00)
        000000e0`3feff788 00000000`00000000 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : 0x00007ffc`d950cf34

0: kd> !processirps ffffb38cbaec2080
**** PROCESS ffffb38cbaec2080 (Image: sam3u_benchmar) ****

Checking threads for IRPs.

  Thread ffffb38cbaac1080:

[... cut ...]
Checking file objects for IRPs.`

I tried to have a look at libusb-win32 source code, but so far I didn't understand exactly what is the issue, and I doubt I will.

Using Zadig to replace driver by either WinUSB or libusbK and trying the same experiment result in no freeze at all: after I unplugged the USB cable, `libusb_control_transfer `simply returns LIBUSB_ERROR_NOT_FOUND .

So, it is not exactly the issue with `libusb_clear_halt`, but it shows that there is a problem with libusb0.sys, I guess

tormodvolden, 2023-09-19:
I don't think 1018 will help for the kernel hang, but it is in any case appreciated if you test that PR. I notice your kernel dump is stuck in KeWaitForSingleObject() which is called from the libusb.sys parts that I mentioned above. From looking at the code I had a suspicion that the IRP locking is not done properly and there is a chance of races or deadlocks. I started working on some patches (based on the MS documentation) but haven't been able to test them myself. I could compile it with the WDK but didn't figure out the installation part. I can push this to an experimental libusb.sys PR and hope someone can look at it or test it.

sonatique, 2023-09-19:
@tormodvolden : thank you. I can try to build a patched libusb0.sys and try again, or better just try to replace mine with your built libusb0.sys. I'll do this soon.

sonatique, 2023-09-19:
@tormodvolden : would you mind attaching a built libusb0.sys containing your latest patch somewhere? Thanks in advance!

mcuee, 2023-09-19:
@dontech
Please help to look at this issue and https://github.com/mcuee/libusb-win32/pull/38, when you got the time. Thanks.

sonatique, 2023-09-19:
> I don't think 1018 will help for the kernel hang, but it is in any case appreciated if you test that PR....

Ok, no, it doesn't help with kernel hang, but I get 2 control transfer fails (with ERROR_IO) before the freeze ;-)

I was hoping that the `windows_force_sync_completion` that you added in case of failure could help putting the driver back on track. Maybe it does a little bit but not enough

Apart from this I see no difference under normal behavior in between master and your PR. I guess this is good..

tormodvolden, 2023-09-19:
@sonatique : This should be the patched libusb.sys. I also included the other build artifacts (from running make_all.bat in libusb/ddk_make in a "Windows Win7 x64 Free Build Environment" window).
[libusb-win32-PR38.zip](https://github.com/libusb/libusb/files/12659204/libusb-win32-PR38.zip)

sonatique, 2023-09-19:
@tormodvolden : so I tested your patched libusb0.sys by replacing my current one with yours (and of course rebooting with disabled driver signature enforcement). I then repeated the exact same procedure as described above:
instead of kernal hang now I get:
1) at the instant when I disconnect the cable: ERROR_IO once (like before)
2) then ERROR_NO_DEVICE 5 times
3) then OK again.

So very good improvement since no hang at all, but still strange behavior I think:
While 1 and 2 seems OK to me, perfectly making sens with respect to the situation, I am puzzled by the fact that calls resume to returning OK when there is no longer a cable connected to any devices. Worse is that the returned value is 8, i.e. the expected written byte count of my request (that I correctly get as well before disconnecting).

I thank you very much for this attempt, and for sure it goes into the right direction, but I think it is not usable as is since it could fool the application into thinking everything is fine when there are no longer any device attached.

Anyway this "disable driver signature enforcement" thing makes me think we are doomed: we cannot any longer use driver that we built ourselves (I am yes we could use them but not distribute them, etc.), so basically we are stuck with staying with driver we built before the change in requirements made by MS, or switch to using WinUSB.... This basically makes any futur special driver development meaningless, expect maybe for niche/hobby things. What do you think?

tormodvolden, 2023-09-19:
Thanks for testing this. I am very glad that the hang went away. I also wonder why "then OK again" happens. Can you please attach debug logs?

I wish I could provide a debug build of libusb0.sys so that we can see driver debug output with DebugView but I haven't figured out what to do in in the ddk_build folder to enable this. Maybe @dontech can help.

For signed libusb0.sys updates, this is still possible because dontech has the appropriate certifications to sign it.

tormodvolden, 2023-09-19:
Here is what should have been a debug build, I gave it version number 1.2.7.101. The size is larger, however I don't see anything in DebugView from it. (Built in CMD.exe window with `CMD /C make.cmd "arch=x64" "outdir=.\x64" "DEBUGMODE=true" "app=driver"`)
[libusb0_sys_PR38_dbg.zip](https://github.com/libusb/libusb/files/12664417/libusb0_sys_PR38_dbg.zip)

dontech, 2023-09-20:
New snapshots: https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20230920/

sonatique, 2023-09-20:
@tormodvolden : OK here are some results with your debug build (was tricky because of the overwhelming amount of log produced by the very fast looping).
Yes my last message was still with 1018 applied, you were right.
I test with and without it.
My procedure is as follow:
I start with libusb debug log disabled but DebugView enabled
I start the vendor request write loop and remove the cable as promptly as possible (after 50-150 loops).
In my program when I ses the first non-OK error returned, I enable libusb debug logging.
I then allow 5 non-error returns, then exit the loop (otherwise it will loop forever)
Basically what we see is that the driver correctly report 1 or 2 failed requests (I guess this is during the removal of the cable) then says " SURPRISE_REMOVAL" and basically doesn't do anything anymore after it (which is correct I guess).
However libusb-1.0 happily continues to return "8 bytes written" forever after a few error messages.
The behavior is slightly different with or with PR 1018 applied, but not fundamentally. 
I think that while you definitively fixed something in the driver (congratulation!) this also shows an issue in libusb-1.0 that shall be fixed sooner or later.

Here are some logs:

### Current master, commit 6bf2db6 (without PR 1018)
Output from my modified sam3u_benchmark (a few printf and libusb-1.0 debug-level logs):
(each initial dot represent a successful write request before cable removal, debug logs enable after first error, you can see 5 times "Write Success: 8" after cable removal that shouldn't be there in my opinion)
```
................................................................................................................................Write Error: LIBUSB_ERROR_IO
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 8.419228] [00005030] libusb: debug [libusb_submit_transfer] transfer 000001C8A73B6ED0
[ 8.419306] [00005030] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 8.419374] [00005030] libusb: debug [libusb_claim_interface] interface 0
[ 8.419484] [00005030] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 8.419585] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 01 to interface 0
[ 8.419694] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 8.419798] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 8.419901] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 82 to interface 0
[ 8.420005] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 04 to interface 0
[ 8.420090] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 84 to interface 0
[ 8.420174] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 06 to interface 0
[ 8.420256] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 86 to interface 0
[ 8.420331] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 08 to interface 0
[ 8.420406] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 88 to interface 0
[ 8.420488] [00005030] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 8.420496] [0000022c] libusb: debug [windows_iocp_thread] ignoring overlapped 000000570ADDF3B0 for handle 000001C8A73AC060
[ 8.420555] [00005030] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 8.420635] [0000022c] libusb: debug [windows_iocp_thread] ignoring overlapped 000000570ADDF3B0 for handle 000001C8A73AC060
[ 8.420759] [0000022c] libusb: debug [windows_iocp_thread] ignoring overlapped 000000570ADDF3B0 for handle 000001C8A73AC060
[ 8.420811] [00005030] libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [433] A device which does not exist was specified.
[ 8.420826] [0000022c] libusb: debug [windows_iocp_thread] ignoring overlapped 000000570ADDF3B0 for handle 000001C8A73AC060
[ 8.420888] [00005030] libusb: debug [libusb_release_interface] interface 0
[ 8.420954] [0000022c] libusb: debug [windows_iocp_thread] ignoring overlapped 000000570ADDF490 for handle 000001C8A73AC060
[ 8.421014] [00005030] libusb: debug [auto_release] auto-released interface 0
[ 8.421132] [00005030] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 8.421195] [00005030] libusb: debug [libusb_free_transfer] transfer 000001C8A73B6ED0
Write Error: LIBUSB_ERROR_IO
[ 8.421305] [00005030] libusb: debug [libusb_submit_transfer] transfer 000001C8A73B6250
[ 8.421365] [00005030] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 8.421431] [00005030] libusb: debug [libusb_claim_interface] interface 0
[ 8.421543] [00005030] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 8.421607] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 01 to interface 0
[ 8.421671] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 8.421735] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 8.421799] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 82 to interface 0
[ 8.421863] [00005030] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 04 to interface 0
[... cut ...]
libusb0-sys:err [vendor_class_request] request failed: status: 0xc000000e, urb-status: 0x80000300
libusb0-sys:[dispatch_pnp] IRP_MN_SURPRISE_REMOVAL: is-filter=N usb\vid_1500&pid_0008&rev_0100
libusb0-sys:[set_filter_interface_key] updated interface registry with LUsb0 direct-access symbolic link. id=-1
libusb0-sys:[dispatch_pnp] IRP_MN_REMOVE_DEVICE: is-filter=N usb\vid_1500&pid_0008&rev_0100
libusb0-sys:[dispatch_pnp] deleting device #2 usb\vid_1500&pid_0008&rev_0100
libusb0-sys:[unload] [unloading-driver] v1.2.7.101
```

sonatique, 2023-09-20:
@tormodvolden @dontech : I just tried with 1.2.7.4 snapshot of libusb0.sys driver (thanks for it) and got the same results (minus the driver logs obviously)

sonatique, 2023-09-20:
@dontech : side question: any plan to compile libusb0.sys for ARM64, by luck?

mcuee, 2023-09-20:
> @dontech : side question: any plan to compile libusb0.sys for ARM64, by luck?

You can add your voice here.
https://github.com/mcuee/libusb-win32/issues/29

tormodvolden, 2023-09-20:
> I just tried with 1.2.7.4 snapshot of libusb0.sys driver (thanks for it) and got the same results (minus the driver logs obviously)

libusb-win32-bin-debug-1.2.7.4.zip should give you driver debug output.

tormodvolden, 2023-09-20:
It seems from the debug logs, if I have correlated the libusb and kernel logs correctly, that there is a good match between the kernel call_usbd_ex() status and the libusb transfer status as long as the driver is loaded. As soon as the PNP management unloads the driver, the false transfer success appears. I am not sure what happens with device objects and ioctl's and so forth in this case, it is possible that this needs special care by the library/DLL ("Whenever a client opens a handle to a pnp device, it should register for pnp notification and close the handle when the system notifies the device is being removed" [1]).

We can see that the unload routine is being called, but it is also not clear from the log whether the driver is successfully unloaded ("The fact that unload-routine is called doesn't mean the driver unload is successful" [1]). This can maybe be checked through other means.

[1] https://techcommunity.microsoft.com/t5/microsoft-usb-blog/why-doesn-t-my-driver-unload/ba-p/270680

dontech, 2023-09-20:
> @dontech : side question: any plan to compile libusb0.sys for ARM64, by luck?

Yes. I can probably get it to compile, but I don't currently have a windows device to compile with.

And yes, we also need to port to new SDK first, as the current one does not support ARM at all.

sonatique, 2023-09-20:
> In theory yes, but it could also be remaining bugs in libusb.sys or libusb-1.0. Is there a debug version of libusbK.dll that reports something in DebugView? EDIT: There is -debug.7z here https://sourceforge.net/projects/libusbk/files/libusbK-release/3.1.0.0/
@tormodvolden : I will give it a try asap and post logs.

@dontech thank you for your reply regarding ARM.

I have a question for both of you, sorry if it appears rude with respect to your hard work, but: for somebody using libusb-1.0 (or simply directly libsubK.dll): what would be the benefit of trying to stick with libusb0.sys as opposed to switching to libusbK.sys ? 

Both project are in maintenance mode, but one (libusbK) seem to have less bugs. Why not encourage people to move to libusbK.sys from libusb0.sys ? This would even allow to phase out libusb0.sys support inside libusbK.dll let's say for "version 4".

I am asking because I currently cannot find something that is better with libusb0.sys in term of feature, performance, etc. While I find it admirable, I don't get what is your motivation for still trying to fix libsub0.sys. What am I missing here? Are there many people directly using libusb0.sys API?

sonatique, 2023-09-21:
> In theory yes, but it could also be remaining bugs in libusb.sys or libusb-1.0. Is there a debug version of libusbK.dll that reports something in DebugView? EDIT: There is -debug.7z here [...]
@tormodvolden : I don't see anything from libusbK in DebugView... is there a trick?

mcuee, 2023-09-21:
> @tormodvolden : I don't see anything from libusbK in DebugView... is there a trick?
> 
> EDIT: well I had a quick look a the code: if it is simply compiled in Debug I don't see a lot of debug output in the code, but maybe I am missing something...

Try the debug version of libusbK here.
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0 (download and run libusbK-3.1.0.0-setup-debug.exe).

mcuee, 2023-09-21:
> I have a question for both of you, sorry if it appears rude with respect to your hard work, but: for somebody using libusb-1.0 (or simply directly libsubK.dll): what would be the benefit of trying to stick with libusb0.sys as opposed to switching to libusbK.sys ?
> 
> Both project are in maintenance mode, but one (libusbK) seem to have less bugs. Why not encourage people to move to libusbK.sys from libusb0.sys ? This would even allow to phase out libusb0.sys support inside libusbK.dll let's say for "version 4".
> 
> I am asking because I currently cannot find something that is better with libusb0.sys in term of feature, performance, etc. While I find it admirable, I don't get what is your motivation for still trying to fix libsub0.sys. What am I missing here? Are there many people directly using libusb0.sys API?
> 
> Thank you in advance for your reply.

I am the co-admin of both libusb-win32 project and libusbK project.

A few benefits for libusb0.sys:
1) libusb0.sys is based on the older WDM and it may work for some devices which do not work with libusbk.sys and WinUSB.
2) libusb0.sys supports setting USB configuration (not libusbk.sys or WinUSB).
3) libusb-win32 is still much widely used compared to libusbK, including by quite some companies. Take note libusbK API is Windows only. libusb-win32 API is basically a superset of libusb-0.1 API.

You can compare the SourceForge download 

4) @dontech has the necessary driver signing certificate now so there will be further development of libusb0.sys. There will be no future signed libusbk.sys version (current libusbk.sys is signed in 2014). Focus of libusbK project will shift to improving the support of WinUSB.
Ref: https://github.com/mcuee/libusbk/issues/26

There may be still some issues with libusb0.sys but I think it is pretty stable. I tend to think most of the issues reported for libusb0.sys under libusb-1.0 are libusb-1.0 bugs. 

Of course, @tormodvolden has fixed many of the issues (Kudos to him). If not libusb0.sys was actually not usable with libusb-1.0.24 version.

And of course there are some advantages of libusbk.sys as well.
1) Better isochronous transfer support. 
2) It is based on KMDF. As per Travis (current developer of libusbK project and previous developer of libusb-win32 project), it is a bit easier to develop and maintain. 

But then both he and I think the future of generic Windows USB driver is WinUSB, so he has shift the focus to the libusbK library and he improved the support of WinUSB in version 3.1.0.0.

sonatique, 2023-09-21:
@mcuee : thank you very much for the detailed answer, very interesting.
Now, here is a few things about my side of the thing:
I have been using libusb0.sys for years now. I noticed a few bugs that I workarounded in the application code. I did not reported them nor tried to fix them because I understood the development was stall.

A few month ago a reworked some parts of my application and for reasons that are still unclear to me, I started to observe more and more kernel hang in various scenario. I have been looking for cause and solution to this because it is impacting my customers.

As of today, even corrected libusb0.sys is not a solution (because of the wrong behavior after the cable removal). LibusbK however seems to be a nice working solution, I didn't see any drawback so far.

I can use signed version of 3.1.0.0 to distribute this as the new driver for my customers.

But, in the meantime, I also need a solution for Windows ARM. As of now I am using WinUSB for ARM. The issue is that WinUSB has the limitation of not permitting more than one device Open. 

Moreover, my ideal feature set would be : mutliple Opens, but one single Claim Interface. Since no driver is currently offering this, And I don't think that this feature could be implemented in libusbK.dll + WinUSB.sys, but maybe I am wrong.

That is way I am seriously thinking about adding this feature to libusbK. I selected libusbK over libusb0 because I have been able to successfully build it and use it for ARM ( https://github.com/mcuee/libusbk/pull/52 ), libusbK doesn't have the "kernel hang" bug and is directly usable with current libusb-1.0, and libubsK is based on most recent kernel driver architecture which seems good for futur efforts.

The only thing that is preventing me for going in this direction is the signing question. In the past I was able to sign everything by myself, but since then it became a nightmare. I am not sure I will be able to do everything that is needed for being able to sing my own code.

@dontech : do you have any pointer for the simplest way to get the appropriate certificate? Alternatively: would you be OK with building and signing libusbK.sys for ARM64 based on what I did in my branch?

Thank you very much in advance for your replies.

mcuee, 2023-09-21:
@sonatique 

Thanks a lot for the PR to add ARM64 support for libusbK. Hopefully Travis will have some time to review. He is pretty busy.

As for driver signing, the most important thing is that you need to get a EV certificate. Usually you need to be from a corporate in order to get the EV certificate. So that poses quite some difficulties for individual developer.  

Reference:
https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-cert-manage#get-or-renew-a-code-signing-certificate

After that it is really not that difficult to go for attestation signing (no need to go for more difficult WHQL route).

Reference:
https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-attestation

Limitation: Windows 7 may not be easy to sign now.
* https://github.com/mcuee/libusb-win32/issues/26

sonatique, 2023-09-21:
@mcuee I still have a questions regarding what you wrote:
> Focus of libusbK project will shift to improving the support of WinUSB.

If I understood correctly: given that libubsK API is 1:1 , i.e. given the chain look like this:
libusb-1.0.dll --> WinUSB.dll --> WinUSB.sys (if libusbK.dll is not present)
libusb-1.0.dll --> libusbK.dll --> WinUSB.dll --> WinUSB.sys

what is the added value of libusbK.dll when used with WinUSB.dll? I looks like in this case its role is basically to issue 1:1 all forwards. So: what can you do to "improve the support of WinUSB"? Do you mean staying up-to-date with possible development? Or adding features not found in WinUSB?

In other words, what is the benefit of using
"libusb-1.0.dll --> libusbK.dll --> WinUSB.dll --> WinUSB.sys"
 over 
"libusb-1.0.dll --> WinUSB.dll --> WinUSB.sys"

Did you add features that can only be used in the first case and not the second one? If not: is the first chain simply equivalent to the second one, maybe with the addition of a little overhead?

Thanks in advance

tormodvolden, 2023-09-21:
> As of now, I would say that unfortunately current libusb-1.0 is not usable with libusb0.sys:: with version <= 1.2.7.3 we get a kernel hang and with >= 1.2.7.4 we incorrectly get OK status for write request that can not physically be completed successfully...

To me it seems from the logs above that with libusb0.sys 1.2.7.4 and libusb-1.0 with PR 1018 you can deal with it like this: If you ever receive a LIBUSB_ERROR_NO_DEVICE error, do not try to access the device any longer.

sonatique, 2023-09-22:
> To me it seems from the logs above that with libusb0.sys 1.2.7.4 and libusb-1.0 with PR 1018 you can deal with it like this: If you ever receive a LIBUSB_ERROR_NO_DEVICE error, do not try to access the device any longer.

Yes, indeed, this would be a simple solution for a simple program. However, as soon as you have various "modules" in a complex application that needs to query various information on the USB device, you cannot assume that every module got the whole history of returned values (unless you explicitly add code for sharing the state based on the history of all commands issues by all modules).

mcuee, 2023-09-22:
> @mcuee I still have a questions regarding what you wrote:
>
>
> If I understood correctly: given that libubsK API is 1:1 , i.e. given the chain look like this: libusb-1.0.dll --> WinUSB.dll --> [...]

When I say that `Focus of libusbK project will shift to improving the support of WinUSB`, I mean that libusbK project (not libusb project) will shift the focus to the library part (libusbK.dll) and not kernel driver part (libusbK.sys). You can see that from the change log of libusbK version 3.1.0.0. Most important change is the "Added isochronous support for WinUSB".
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0

As for the use of libusbK.dll in libusb Windows backend, I think the main goal was to add the support of libusbK.sys and libusb0.sys, in a unified way. 

Since libusbK.dll also supports WinUSB, Pete, the developer at that time, decided to make  "libusb-1.0.dll --> libusbK.dll --> WinUSB.dll --> WinUSB.sys" the default for WinUSB as well when libusbK.dll is present in the system.

Personally I myself do not really see the benefit of using "libusb-1.0.dll --> libusbK.dll --> WinUSB.dll --> WinUSB.sys" over "libusb-1.0.dll --> WinUSB.dll --> WinUSB.sys". In fact, I will welcome PRs to change "libusb-1.0.dll --> WinUSB.dll --> WinUSB.sys" to be the default for WinUSB even if libusbK.dll is present.

sonatique, 2023-09-22:
@mcuee thank you very much for the time spent explaining this to me.

I dont want to seem over-picky or to annoy you more than needed, but I really have one more iteration to make, for things to be completely clear to me:

If "focus of libusbK is the library part (dll)" and also somehow "support of WinUSB" (i.e  not support of libusbK.sys) and if you don't think libusbK.dll is required at all in the "libusb-1.0.dll --> WinUSB.dll --> WinUSB.sys" chain, I tend to conclude that:

1) libubsK support for libusbK.sys could as well be removed in future libusbK version
2) libusbK.dll support for WinUSB is useless with libusb-1.0 (you want to ignore it for WinUSB "as well when libusbK.dll is present in the system"
3) libusbK api is 1:1 with WinUSB API so any application currently using libusbK.dll could switch to WinUSB.dll.

So: what are we left with for future development of libusbK?: 
1) not the support of libusbK.sys
2) not the support of WinUSB

I think we're left with the support of libsubo.sys, such that people using the libusbK.dll / WinUSB.dll API could switch from WnUSB.sys to libusb0.sys without changing their API.

That is the only reason I can figure out for keeping libusbK alive if libusbK.sys dev is stopped.

Did I miss something?

Thanks.

pbatard, 2023-09-22:
@mcuee, I'm racking my brains to remember what the arguments were, for hooking into libusbK when present rather than WinUSB, when we actually had a working direct WinUSB implementation at the time, but I have nothing.

I believe it came from a consensus not to overcomplicate the code, as a single indirection from libusbK.dll into WinUSB.dll is not something that will hurt performance, and this way, we could avoid having explicit `if (driver type) then (call this API) else (call this other API)` all over the place in the libusb source. I don't really remember being the one pushing for this though but the thing is, since it's pure C, you're never going to eat enough CPU cycles over this extra hook into libusbK to affect perf. So if the idea here is that doing away with the extra indirection may improve speed or reduce latency, I don't think the effort will be worth it, and it may very well make libusb maintenance harder in the long run.

Maybe we could see clearer as to how this actually came about if the #11 and #12 issues that are referenced in commit (f2b19e4a32c1445f4fea06beca61e8c7e7757950) could be located, since they aren't from this repo but most likely from the old libusbx repo (of which I can no longer find a trace of)...

sonatique, 2023-09-22:
@pbatard : thank you for your message.

Actually I have a breaking news for you: libsub-1.0 doesn't load libusbK for WinUBS any longer since February 2020, commit 9c28ad21 ( https://github.com/libusb/libusb/commit/9c28ad219b654011783a42ec888ca87dbda704a6#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4 )

While thinking about since a few days it I had a doubt because I didn't remember seeing anything like this when I last studied the code, and I remembered this comment in windows_winusb.c
```
// NB: The below for loop works because the sub_api value for WinUSB
// is a higher value than that of libusbK and libusb0
```

Tonight I then used blame and just found out that this comment has been introduce at commit 9c28ad21 precisely for removing libusb-1.0 -> libusbK.dll -> WinUSB.dll -> WinUSB.sys, as explained in the commit text:

```
Finally, the initialization code for the WinUSB backend has been
reworked to load the WinUSB DLL independent of the libusbK DLL.
Previously when the libusbK DLL was present, all requests to devices
using WinUSB would first be sent through the libusbK DLL where
they would then be forwarded to the WinUSB DLL. This is slightly
inefficient but is also limiting when using Windows 8.1 or later because
support for isochronous transfers through WinUSB will be lost.
```

The WikI FAQ https://github.com/libusb/libusb/wiki/FAQ#user-content-libusbwin32_libusbK_and_libusb_project_what_about_openusb should be updated, I guess.

So well, @pbatard, @mcuee  if you had such difficulties remembering a good reason why libusbK was used as an intermediate dll with WinUSB, in might be because there was no good reason actually ;-)

And in the end, this confirms, I think, that libusbK.dll no longer has to interact with WinUSB.dll since 2020. Its only reason to exist (which is a good one) is therefore to provide the "WinUSB API" for libusb0.sys and libusbK.sys .

For instance releasing it for ARM64 only makes sens if libusbK.sys and/or libusb0.sys are also released for ARM64.
Similarly, if libusb0.sys and libusbK.sys are no longer considered relevant alternative to WinUSB.sys , or if both remains in bugfix/maintenance mode, then there is no reasons to keep on developing libubsK.dll.

Well, basically I am thinking out loud because I need to decide on which I am going to walk from now. Feel free to tell me what you think about it and about the futur of USB support under Windows., thanks.

mcuee, 2023-09-23:
> @pbatard : thank you for your message.
> 
> Actually I have a breaking news for you: libsub-1.0 doesn't load libusbK for WinUBS any longer since February 2020, commit [9c28ad2](https://github.com/libusb/libusb/commit/9c28ad219b654011783a42ec888ca87dbda704a6) ( [9c28ad2#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4](https://github.com/libusb/libusb/commit/9c28ad219b654011783a42ec888ca87dbda704a6#diff-bd7e9ed3e7d8a4a69d0664da0f6b8dbbff0b2c608d8dd63de02c379c98cf88b4) )
> 
> While thinking about since a few days it I had a doubt because I didn't remember seeing anything like this when I last studied the code, and I remembered this comment in windows_winusb.c
> 
> ```
> // NB: The below for loop works because the sub_api value for WinUSB
> // is a higher value than that of libusbK and libusb0
> ```
> 
> Tonight I then used blame and just found out that this comment has been introduce at commit [9c28ad2](https://github.com/libusb/libusb/commit/9c28ad219b654011783a42ec888ca87dbda704a6) precisely for removing libusb-1.0 -> libusbK.dll -> WinUSB.dll -> WinUSB.sys, as explained in the commit text:
> 
> ```
> Finally, the initialization code for the WinUSB backend has been
> reworked to load the WinUSB DLL independent of the libusbK DLL.
> Previously when the libusbK DLL was present, all requests to devices
> using WinUSB would first be sent through the libusbK DLL where
> they would then be forwarded to the WinUSB DLL. This is slightly
> inefficient but is also limiting when using Windows 8.1 or later because
> support for isochronous transfers through WinUSB will be lost.
> ```
> 
> The WikI FAQ https://github.com/libusb/libusb/wiki/FAQ#user-content-libusbwin32_libusbK_and_libusb_project_what_about_openusb should be updated, I guess.
> 
> So well, @pbatard, @mcuee if you had such difficulties remembering a good reason why libusbK was used as an intermediate dll with WinUSB, in might be because there was no good reason actually ;-)

Thanks a lot for the information. Somehow I missed the changes of this important commit.

Wiki has been updated.

mcuee, 2023-09-23:
> @mcuee thank you very much for the time spent explaining this to me.
> 
> I dont want to seem over-picky or to annoy you more than needed, but I really have one more iteration to make, for things to be completely clear to me:
> 
> If "focus of libusbK is the library part (dll)" and also somehow "support of WinUSB" (i.e not support of libusbK.sys) and if you don't think libusbK.dll is required at all in the "libusb-1.0.dll --> WinUSB.dll --> WinUSB.sys" chain, I tend to conclude that:
> 
> 1. libubsK support for libusbK.sys could as well be removed in future libusbK version
> 2. libusbK.dll support for WinUSB is useless with libusb-1.0 (you want to ignore it for WinUSB "as well when libusbK.dll is present in the system"
> 3. libusbK api is 1:1 with WinUSB API so any application currently using libusbK.dll could switch to WinUSB.dll.
> 
> So: what are we left with for future development of libusbK?:
> 
> 1. not the support of libusbK.sys
> 2. not the support of WinUSB
> 
> I think we're left with the support of libsub0.sys, such that people using the libusbK.dll / WinUSB.dll API could switch from WnUSB.sys to libusb0.sys without changing their API.
> 
> That is the only reason I can figure out for keeping libusbK alive if libusbK.sys dev is stopped.
> 
> Did I miss something?

Hmm, I think you have quite some mis-understanding of libusbK project, which is independant project from libusb.

1) libusbK library API supports libusbK.sys, libusb0.sys and WinUSB.sys (through WinUSB.dll). This will not change. We will for sure not remove the support of any of them.

2) libusbK library/API is quite different from WinUSB or libusb-1.0 or libusb-win32. The user base is not big but there are some users as it provides some unique APIs like streaming API and advanced isochronous transfer API.
https://libusbk.sourceforge.net/UsbK3/group__stmk.html
https://libusbk.sourceforge.net/UsbK3/group__isochk.html

3) libusbK.sys was developed to address the shortcomings of WinUSB, but WinUSB has since catched up for most use case. And there is a certificate issue so we mention that the focus will be shift towards the library part, including to bring WinUSB support to be on par with libusbK.sys in terms of isochronous transfer support (3.1.0.0 release).

4) But there are still some advatages of libusbK.sys compared to WinUSB. So if you hit into the limitation, you may want to use libusbK.sys when using libusb-1.0 API and not WinUSB. Improvemt of libusbK.sys will continue, but we need to sort out the signing certificate issue in order to release to the general public.
https://github.com/libusb/libusb/wiki/Windows#user-content-Known_Restrictions

More info: libusbK documentation including API
https://libusbk.sourceforge.net/UsbK3/index.html

BTW, for Windows ARM64 support, there is still a missing piece in terms of driver installer.libwdi/Zadig do not work well under Windows 11 for ARM64 yet.
https://github.com/pbatard/libwdi/issues/289

mcuee, 2023-09-23:
> And in the end, this confirms, I think, that libusbK.dll no longer has to interact with WinUSB.dll since 2020. Its only reason to exist (which is a good one) is therefore to provide the "WinUSB API" for libusb0.sys and libusbK.sys .

Not "WinUSB API". libusbK API is unique.

> For instance releasing it for ARM64 only makes sens if libusbK.sys and/or libusb0.sys are also released for ARM64. Similarly, if libusb0.sys and libusbK.sys are no longer considered relevant [...]

As mentioned, libusbK.dll supports WinUSB as well and it has unique API. Even if we do not have ARM64 version of libusbK.sys and libusb0.sys, it still makes sense to have ARM64 version of libusbK.dll, as libusbK API is unique.

libusb0.sys and libusbK.sys will still be maintained for the users who are still using them. They still have some advantages compared to WinUSB. So it is still relevant. It is just that it will be less important.
https://github.com/libusb/libusb/wiki/Windows#user-content-Known_Restrictions

> Well, basically I am thinking out loud because I need to decide on which I am going to walk from now. Feel free to tell me what you think about it and about the futur of USB support under Windows., thanks.

My recommendations:
1) If you need cross-platform API, libusb-1.0 is the library to go. I believe this is your use case. There are still quite some users of libusb-0.1/libusb-win32 API (through legacy libusb-0.1, or libusb-compat-0.1, or libusb-win32 under Windows). There are less users of libusbK API (Windows only) but still there are some users.

2) If you do not hit into the limitation of WinUSB (WinUSB.dll+WinUSB.sys), just use WinUSB.
https://github.com/libusb/libusb/wiki/Windows#user-content-Known_Restrictions

3) If you hit into the limitation of WinUSB.sys, consider using libusbK.sys.

4) If you need to support set_configuration, then the only choice is libusb0.sys.

mcuee, 2023-09-23:
@sonatique 

BTW, it is also possible for libusb Windows backend to directly talk to libusb0.sys and libusbK.sys through the driver API (IOCTLs), without using libusbK.dll. But I am not sure if there are any real benefits other than one less dependancy. But if users using Zadig or libusbK-inf-Wizard to install the libusb0.sys/libusbK.sys driver, libusbK.dll will be present in the system anyway.

libusb0.sys driver API/IOCTLs
https://github.com/mcuee/libusb-win32/blob/master/libusb/src/driver/driver_api.h

libusbK.sys driver API/IOCTLs
https://github.com/mcuee/libusbk/blob/master/libusbK/src/drv_api.h

You can see that libusbK.sys driver API is very much a superset of libusb0.sys driver API. That is why libusb-win32 library (libusb0.dll) can support libusbK.sys without any mentioning of libusbK.sys.

pbatard, 2023-09-23:
@sonatique, I'm just going to point out that I have stopped being involved in libusb for at least the past 10 years, so, yeah, please don't rely on me to provide accurate information about the _current_ codebase... And, just to set expectations straight, please don't expect me to have any involvement with this effort either, as I was just answering a question that had been directed to me about how some very old code might have come to be.

mcuee, 2023-09-23:
> @sonatique, I'm just going to point out that I have stopped being involved in libusb for at least the past 10 years, so, yeah, please don't rely on me to provide accurate information about the [...]

Don't worry. That was my fault -- I missed the commit so that I did not update the WiKi.
Thanks a lot for answering my ping. I do not expect you to help out on the current development. And you are really helpful to answer my questions.

tormodvolden, 2023-09-26:
@sonatique I understand libusb0.sys may be of less importance for you if you move to libusbk.dll, however I'd like to fix this bug and it is probably a small thing - once we figure out where - so I would be grateful for any help.

Is it possible to reproduce the bug with a USB-serial/uart adapter like FT232 and sending CDC ACM class requests?

> Could the "return success after cable removal problem" caused by libusbK.dll? And not by libusb-1.0 nor libusb0.sys? Given the results this is what seems to be the most probable to me...

The problem arises after the libusb0.sys driver instance has been unloaded, so it could seem like the bug is elsewhere, and I haven't been able to spot anything suspicious in the libusb-1.0 code or logs, so I tend to agree with you.

> I don't understand why the "removal during transfer" would end-up in a different state than "removal while idle" except if there is a bug in libusbK.dll in the specific code that interact with libusb0.sys

Does that mean that you don't see this issue if you slow down your request loop, and you pull out the USB cable between two transfers? The libusb device is opened only once at the start, so a single libusb handle is being used, right?

sonatique, 2023-09-27:
@tormodvolden : sure! I will help you as soon as possible. Sorting through all of this is one of my priorities. I'm still interested in libusb0.sys, given that libusbK.sys is giving me a BSOD

I'll come back to you.

mcuee, 2023-09-27:
> @tormodvolden : sure! I will help you as soon as possible. Sorting through all of this is one of my priorities. I'm still interested in libusb0.sys, given that libusbK.sys is giving me a BSOD
> 
> I'll come back to you.

If you got sometime, give  a try to libusb-win32 git main, now ARM64 is supported (not so sure if works or not). I am asking Peter for a signed snapshot release.
* https://github.com/mcuee/libusb-win32/pull/40

dontech, 2023-10-01:
> @dontech : side question: any plan to compile libusb0.sys for ARM64, by luck?

@sonatique, could you possibly try the ARM64 build now that it builds?

sonatique, 2023-10-02:
@dontech : yes, with pleasure, thanks a lot for providing this.. I'll really do this ASAP, I promise, it's on my short term todo list, but it's not empty unfortunately.

tormodvolden, 2023-10-02:
@sonatique Maybe you can use the driver verifier I/O verification and IRP logging and see if something is detected? https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/i-o-verification and https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/irp-logging

mcuee, 2023-10-05:
@sonatique 

FYI libusb-win32 snapshot release 1.3.0.0 has been released by @dontech. Please give it a try when you have the time, especially under Windows 11 on ARM64. Thanks.

sonatique, 2023-10-10:
@mcuee :
> FYI libusb-win32 snapshot release 1.3.0.0 has been released by @dontech.
I cannot find anything beyond 1.2.7.4 looking around  https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20230920/ or github.
Shall I look somewhere else? Thanks

mcuee, 2023-10-10:
> @mcuee :
> 
> > FYI libusb-win32 snapshot release 1.3.0.0 has been released by @dontech.
> 
> I cannot find anything beyond 1.2.7.4 looking around https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20230920/ or github. Shall I look somewhere else? Thanks

Ah I see. It has not been uploaded to Sourceforge yet. You can use github for now.
https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.3.0.0

mcuee, 2023-10-10:
@sonatique and @dontech 
I have also uploaded libusb-win32 1.3.0.0 snapshot release to Sourceforge.
https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20231003/

Somehow the installer ([libusb-win32-devel-filter-1.3.0.0.exe](https://github.com/mcuee/libusb-win32/releases/download/snapshot_1.3.0.0/libusb-win32-devel-filter-1.3.0.0.exe) was flagged as virus and I can not upload it.

tormodvolden, 2023-10-10:
> Somehow the installer ([libusb-win32-devel-filter-1.3.0.0.exe](https://github.com/mcuee/libusb-win32/releases/download/snapshot_1.3.0.0/libusb-win32-devel-filter-1.3.0.0.exe) was flagged as virus and I can not upload it.

You can try uploading with sftp instead of the web interface, I think it will go through then.

mcuee, 2023-10-10:
> You can try uploading with sftp instead of the web interface, I think it will go through then.

Thanks for the tip. Now even my local Windows defender is flagging it as virus and delete the file. So I will be a bit careful and not to upload it for now.

sonatique, 2023-10-10:
@mcuee :Thanks.

I am now trying to get more insight about the remaining issue which is "libusb-1.0 + libusbK + libusb0.sys" return successful fir control transfers after device has been disconnected" (see https://github.com/libusb/libusb/issues/1255#issuecomment-1727376483)

To this extent I installed libusbK debug (from https://github.com/libusb/libusb/issues/1255#issuecomment-1729339705) and I verified that my system files are really those coming from this installer.

I am using libusb0.sys from libusb-win32-bin-debug-1.3.0.0.zip .

I have DebugView up and running with all Capture items checked, but I cannot see any debug lines from netiher libusbK.dll nor libusb0.DLL when I run my test program.

I can see the behavior described above: after cable removal there are a few errors, but then I start received "OK" for subsequent control transfer (that should obviously fail since there is no longer any device attached).

I was hoping to get some libubsK DebugView output in order to understand where this problem lies, given we already saw that it is probably not within libusb0.sys (I saw it being unloaded in logs before I still get successful transfers).

What am I missing to get some DebugView compatible output from libusbK? Does the fact I am not seeing anything indicate that I am not actually using it?

mcuee, 2023-10-10:
@sonatique 

Have you tried the debug version of libusbK?
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0
https://github.com/mcuee/libusbk/releases/download/V3.1.0.0/libusbK-3.1.0.0-setup-debug.exe

mcuee, 2023-10-10:
@dontech 

Do you know if libusb-win32 1.3.0.0 debug version works with DebugView or not?

mcuee, 2023-10-10:
@sonatique 

To remove the effect of libusbK.dll and libusb-1.0.dll, can you try simple applcation using just libusb-win32.dll? That would confirm the issue is with libusb-win32 or not. Thanks.

tormodvolden, 2023-10-10:
I know from previous reports that the libusbk.dll is not very verbose, but you should see something, e.g. https://github.com/libusb/libusb/issues/94#issuecomment-954379510

[29 more replies]