libusb/libusb #1259: [winusb_get_device_list

xfayh, 2023-03-09:
Using libusb v1.0.26.11784
[[winusb_get_device_list] could not retrieve port number for device](libusb: warning [winusb_get_device_list] could not retrieve port number for device 'USBESTUB\DEVICES\0000')

windows 11

mcuee, 2023-03-09:
You have to provide more details about your device.

Most likely we can not help you much as it is a problem for the `'USBESTUB` driver, which seems to be from Eltima (https://www.eltima.com/products/usb-over-ethernet/).

mcuee, 2023-03-09:
But this is just a warning. Are you facing real problems if you ignore the warning?

mcuee, 2023-03-09:
Ref:
https://learn.microsoft.com/en-us/samples/microsoft/windows-driver-samples/usbview-sample-application/

Most likely the 3rd party driver failed to respond properly to `IOCTL_USB_GET_NODE_CONNECTION_INFORMATION` request.

tormodvolden, 2023-03-09:
Yes, I just saw this in the commit message of 71a779d

    Finally, the query for the port number has been moved to the GENeric
    pass at the point where the devices are actually initialized. This query
    operation has been relaxed to allow failure, since some virtual USB
    devices don't properly implement this query in their drivers.

mcuee, 2023-03-27:
@xfayh 

I will close this issue for now. Please reopen if you have new information to say there is an issue with libusb.

closed 2023-03-27 by mcuee