libusb/libusb #1260: priv->sub_api not set in winusb_set_option and blocks access to LIBUSB_OPTION_WINUSB_RAW_IO

iacg68, 2023-03-09:
When activating the LIBUSB_OPTION_WINUSB_RAW_IO option, LIBUSB_ERROR_ACCESS is always returned despite correct configuration of the libusb_set_option(...) call. This is caused by the fact that the value 0 is always entered in the winusb_set_option(...) function under priv->sub_api and thus the test on WinUSBX[sub_api].hDll rightly fails. However, under priv->usb_interface[0].sub_api the correct value 2 is present and WinUSBX[2] contains all driver calls to the DLL.

With the following addition before the call of CHECK_WINUSBX_AVAILABLE(..) the activation of the WINUSB_RAW_IO works correctly:
`sub_api = priv->usb_interface[current_interface].sub_api;`
However, this cannot be the solution, because both sub_api and CHECK_WINUSBX_AVAILABLE(...) are used in many places in the code. So question is why and where the correct value from usb_interface[current_interface].sub_api is not transferred to the priv-struct.

Platforms: Windows 10 using both MSYS/MinGW and VC2019, compiling to 64-bit libraries, WinUSB installed for the device
libusb, 1.0.26, 2abb2a18
WinUSB.dll: 10.0.19041.1 (WinBuild.160101.0800)

Code to reproduce:
`#include <iostream>
#include <libusb.h>
int main(int, char *[])
{
    struct libusb_context *ctx = nullptr;
    libusb_init(&ctx);
    libusb_set_option(ctx, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG);

    // -------------------------------------------------
    // device is a FT2232H with custom pid
    auto dev = libusb_open_device_with_vid_pid(ctx, 0x0403, 0x6B59);
    const auto errClaim = libusb_claim_interface(dev, 0);

    const unsigned int enable = 1;
    unsigned int maxTransferSize = 0;
    const auto errRawIO = libusb_set_option(ctx
                                       , LIBUSB_OPTION_WINUSB_RAW_IO
                                       , dev
                                       , 0x81       // from FT_2232's point of view
                                       , enable
                                       , &maxTransferSize);
    if (errRawIO)
    {
        std::cerr << "Error RAW_IO " << errRawIO << '\n';
        return -1;
    }

    // .. do more stuff
    libusb_release_interface(dev, 0);
    libusb_close(dev);
    libusb_exit(ctx);
    return 0;
}
`

[log_and_libusb_locals@interface_by_endpoint.txt](https://github.com/libusb/libusb/files/10932557/log_and_libusb_locals%40interface_by_endpoint.txt)

containes:
- log from libusb debug output up to breakpoint in windows_winusb.c:2217
- local variables in interface_by_endpoint(...)
- hotfix assigning the local sub_api from usb_interface
- log from libusb debug output (with positive message on completion, thanks to the hot fix)

mcuee, 2023-03-10:
No issues with Singtel interface FT232R. I will try to test FT2232H later.

```
$ cat test1260.cpp
#include <iostream>
#include <libusb.h>
int main(int, char *[])
{
    struct libusb_context *ctx = nullptr;
    libusb_init_context(&ctx, NULL, 0);
    libusb_set_option(ctx, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_INFO);

    // -------------------------------------------------
    // device is a FT232R with default vid/pid
    auto dev = libusb_open_device_with_vid_pid(ctx, 0x0403, 0x6001);
    const auto errClaim = libusb_claim_interface(dev, 0);

    const unsigned int enable = 1;
    unsigned int maxTransferSize = 0;
    const auto errRawIO = libusb_set_option(ctx
                                       , LIBUSB_OPTION_WINUSB_RAW_IO
                                       , dev
                                       , 0x81       // from FT_2232's point of view
                                       , enable
                                       , &maxTransferSize);
    if (errRawIO)
    {
        std::cerr << "Error RAW_IO " << errRawIO << std::endl;
        return -1;
    }
        else
        {
                std::cout << "Setting RAW IO successful" << std::endl;
        }

    // .. do more stuff
    libusb_release_interface(dev, 0);
    libusb_close(dev);
    libusb_exit(ctx);
    return 0;
}

$ g++ -I /mingw64/include/libusb-1.0 test1260.cpp -o test1260 -lusb-1.0

$ ./test1260.exe
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
Setting RAW IO successful
```

mcuee, 2023-03-10:
For USB Composite Device FT4232H (0403:6011), then it depends on the driver situation,

1) WInUSB driver is only installed for the first interface
```
$ ./test1260_FT4232H.exe
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
Segmentation fault
```

2) Use WinUSB driver as the driver for the whole USB composite device.
```
$ ./test1260_FT4232H.exe
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
Setting RAW IO successful
```

iacg68, 2023-03-13:
Confirm that the problem no longer occurs with WinUSB as parent driver. Even if this is a temporary solution, in the future we really need different drivers on the different interfaces, e.g. to make a COM driver or a standard programming interface usable. 

Looking forward to the BugFix :-)

mcuee, 2023-05-03:
Please also refer to #1274.
* #1274

mcuee, 2023-12-21:
Take note #1208 was reverted in https://github.com/libusb/libusb/commit/dac541dc75370cf7b6051cc6c163af1f468b19ed.
* #1208

mcuee, 2023-12-21:
@tormodvolden 
Since #1208 was reverted, I think we can close this issue. Any concerns?

tormodvolden, 2023-12-21:
It looks like raw IO will be implemented a little bit differently, but this issue is still relevant and is referenced in the new PR at least as background information. It can be kept open as a reminder that these cases should be tested also for the new implementation.

tormodvolden, 2024-01-19:
During enumeration there is a sub_api set for the device by get_api_type() and for composite devices each interface will have sub_api set as well by set_composite_interface(). In the case above the value was not the same for the device (what CHECK_WINUSBX_AVAILABLE will look up if SUB_API_NOTSET is passed to it) as for the interface in question (belonging to the endpoint being configured for raw IO). So the above hotfix of first setting sub_api from the interface makes sense. If there are different sub_api among the interfaces there must naturally be a difference between device and _some_ of the interfaces. And the other workaround of setting WinUSB as the parent driver made the sub_api of the device and the interface in question match as well.

So I think the "hotfix" was what was missing from the (now reverted) code in commit 2abb2a18 and there is no bug elsewhere. Just that CHECK_WINUSBX_AVAILABLE is a bit treacherous because it opaquely looks up the variable "priv".

mcuee, 2024-02-02:
This is no longer a bug so I remove the label.

martinling, 2024-07-25:
The latest proposal for RAW_IO support is in #1512, and specifically addresses the issues with composite devices.

So I think this issue can be closed now, and any remaining problems discussed on #1512.

closed 2024-07-25 by mcuee