libusb/libusb #1263: Crash due to use-after-free caused by race condition

Erlkoenig90, 2023-03-30:
I'm experiencing an access violation within libusb while it is used from a multi-threaded application. The application calls `libusb_bulk_transfer` simultaneously from 3 threads (all IN transfers) to continuously transfer data on multiple endpoints (one thread per endpoint). There is only one device and one context, no further libusb activity is happening in parallel (no concurrent calls to e.g. `libusb_open`/`libusb_close`). I'm having trouble finding the exact cause. Here is what I found out so far:

System: Windows 10 x64, CPU i7-8550U
Build Mode: Both Debug and Release, libusb compiled as DLL
Compiler: MSVC 17.5.3 (both libusb and application)
libusb Version: Both 1.0.26 and `master` (commit: `fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a`) have the issue

The crash happens randomly within minutes to hours after the application start and is hard to reproduce. Both application and device are proprietary and I'm unable to construct a minimal example that triggers this bug. The bug appears to be very sensitive to timing; just adding a few assertions in some places stops it from happening. I think it's a race conditions between multiple calls to `libusb_bulk_transfer` in parallel.

The crash always happens in [`sync_transfer_cb`](https://github.com/libusb/libusb/blob/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a/libusb/sync.c#L38) when attempting to write to `*completed`. The memory pointed to by `transfer` just contains `0xDD` which [indicates](https://stackoverflow.com/a/127404/4730685) that the memory has been `free`d. Accessing the `user_data` pointer which then points to `0xDDDDDDDDDDDDDDDD` causes the access violation. So it's a use-after-free on `struct libusb_transfer`. The backtrace for this is (on [master](https://github.com/libusb/libusb/tree/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a)):

```
libusb-1.0.dll!sync_transfer_cb(libusb_transfer * transfer) Line 39
	at libusb\sync.c (39)
libusb-1.0.dll!usbi_handle_transfer_completion(usbi_transfer * itransfer, libusb_transfer_status status) Line 1726
	at libusb\io.c (1726)
libusb-1.0.dll!windows_handle_transfer_completion(usbi_transfer * itransfer) Line 855
	at libusb\os\windows_common.c (855)
libusb-1.0.dll!handle_event_trigger(libusb_context * ctx) Line 2155
	at libusb\io.c (2155)
libusb-1.0.dll!handle_events(libusb_context * ctx, timeval * tv) Line 2261
	at libusb\io.c (2261)
libusb-1.0.dll!libusb_handle_events_timeout_completed(libusb_context * ctx, timeval * tv, int * completed) Line 2368
	at libusb\io.c (2368)
libusb-1.0.dll!libusb_handle_events_completed(libusb_context * ctx, int * completed) Line 2467
	at libusb\io.c (2467)
libusb-1.0.dll!sync_transfer_wait_for_completion(libusb_transfer * transfer) Line 49
	at libusb\sync.c (49)
libusb-1.0.dll!do_sync_bulk_transfer(libusb_device_handle * dev_handle, unsigned char endpoint, unsigned char * buffer, int length, int * transferred, unsigned int timeout, unsigned char type) Line 196
	at libusb\sync.c (196)
libusb-1.0.dll!libusb_bulk_transfer(libusb_device_handle * dev_handle, unsigned char endpoint, unsigned char * data, int length, int * transferred, unsigned int timeout) Line 280
	at libusb\sync.c (280)
```

By storing the [current thread ID](https://learn.microsoft.com/en-us/windows/win32/api/processthreadsapi/nf-processthreadsapi-getcurrentthreadid) in `struct libusb_transfer` I determined that each transfer gets allocated, started, completed and free'd "cleanly" all by the same thread. I added an assertion to `libusb_free_transfer` to check whether the transfer gets freed by the same thread as allocated, which never got triggered. I also found that `libusb_free_transfer` always gets called from `do_sync_bulk_transfer` and not somewhere else.

I'm pretty sure the transfer that causes the crash somehow gets "stolen" by the current thread from one of the other threads. The other thread starts and completes the transfer normally and then frees it in `do_sync_bulk_transfer`, but it simultaneously also somehow ends up in the current thread's call chain leading to `sync_transfer_cb`. I suspected that there might be a race condition in [`handle_event_trigger`](https://github.com/libusb/libusb/blob/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a/libusb/io.c#L2150) which sends the completed transfer to both threads, but couldn't find anything wrong there (`event_data_lock` is used, and `libusb_handle_events_timeout_completed` also prevents this from being entered by two threads simultaneously).

Slight remark: [`libusb_handle_events_timeout_completed` calls](https://github.com/libusb/libusb/blob/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a/libusb/io.c#L2364)  [libusb_lock_events](https://github.com/libusb/libusb/blob/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a/libusb/io.c#L1843) which sets `event_handler_active = 1`. However, there is no synchronization primitive with release semantics after that, so in theory, other threads might not see this change, and enter [this](https://github.com/libusb/libusb/blob/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a/libusb/io.c#L2384) if-block needlessly and got to `retry`. However I don't think this causes any issue besides a few useless iterations, and I don't think it's the cause for my problem. Perhaps accesses to `event_handler_active` need to be protected by `event_waiters_lock`?

Does anyone have an idea what could be causing the "transfer stealing" and subsequent access violation? Thanks!

mcuee, 2023-03-30:
@Erlkoenig90 
Just wonder if you encounter the issue in other OS or not, say Linux or macOS?

mcuee, 2023-03-30:
BTW, by right what you do should work. Still just wonder if you can change your code to use asynchronous API and see if the issue goes away. Thanks.

Ref: https://libusb.sourceforge.io/api-1.0/libusb_io.html

_The main advantage of this model ( synchronous I/O interface) is simplicity: you did everything with a single simple function call.

However, this interface has its limitations. Your application will sleep inside libusb_bulk_transfer() until the transaction has completed. If it takes the user 3 hours to press the button, your application will be sleeping for that long. Execution will be tied up inside the library - the entire thread will be useless for that duration.

Another issue is that by tying up the thread with that single transaction there is no possibility of performing I/O with multiple endpoints and/or multiple devices simultaneously, unless you resort to creating one thread per transaction.

Additionally, there is no opportunity to cancel the transfer after the request has been submitted._

tormodvolden, 2023-03-31:
So you don't experience the issue if you run without debug logging?

mcuee, 2023-04-03:
@tormodvolden 

Looks like we need to change this line as mentioned by Niklas: https://github.com/libusb/libusb/blob/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a/libusb/core.c#L2435

mcuee, 2023-04-04:
@Erlkoenig90 

I think it is good to create a seperate PR and mention that it may replace PR #1264.

hjelmn, 2023-04-04:
LGTM. But if order is the issue we can insert a memory barrier instead. The only issue there is not all supported Windows compilers have C11 atomics support so we would probably have to use `asm volatile("" ::: "memory")` which is a bit ugly. Depends on the performance hit of taking the lock.

Erlkoenig90, 2023-04-04:
`asm volatile("" ::: "memory")` is only an optimization barrier, we'd also need a `atomic_thread_fence` after the write access, because otherwise and currently there is really no guarantee the write access to `completed` is visible to other threads at all. Also in theory, the write access could be completed partially before the other thread reads it (in practice 32bit-integers are always written as-whole on x86/x86_64 and ARM IIRC).

Making `completed` atomic would probably be correct too. Atomics *do* work in C (not just C++) with recent versions of MSVC now. We'd have to make sure applications using `libusb_handle_events_timeout_completed` directly use an atomic integer as well, which would change the API and make the application depend on atomic support.

closed 2023-04-12 by hjelmn