libusb/libusb #1271: Usage of LibUsb

edoardolotti, 2023-04-18:
HI, in the past we developed a library interface based on libusb for a Mass Storage.  This was a library for an FPGA board with Xilinx Zynq7020 with all the related instructions.
Now I need to change the Class code and SubClass code because this should not be a Mass Storage but an Audio interface or HID interface.
We are having problems to reuse this library (based on libusb) and we have not found the way to interconnect the same device but with the different Class Code and SubClass code).
Can you help on this?

Thanks a lot.
Best regards Edo.

mcuee, 2023-04-19:
You should use custom class and the use Windows OS descriptor for USB devices so that Windows will automatically use WinUSB driver for your device.

Please refer to the following web sites for more info.
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors
https://github.com/pbatard/libwdi/wiki/WCID-Devices

mcuee, 2023-04-19:
As this github is not for technical support but for issue reporting., I will close the issue. If you need further help, please subscribe to libusb-devel mailing list and ask there.

mcuee, 2023-04-19:
FAQ entry added.
https://github.com/libusb/libusb/wiki/FAQ#user-content-What_USB_class_should_I_choose_for_my_USB_device_I_am_developing_the_firmware_if_I_want_to_use_it_with_libusb