libusb/libusb #1272: libusb.device-hotplug:signal SIGABRT in MacOS

xumoxiao, 2023-04-20:
### Environment

- OS : MacOS 13.2.1
- Chip: Apple Silicon M2
- USB Device: 2 x Cyusb3014 with Private firmware，
- Protocol：usb 3.0.5 gbps

### Problem Description

The main thread scans how many cyusb devices there are, and generates multiple threads according to the location of the usb device. Each thread runs the following process. After all threads run, repeat this process. This problem may occur if I repeat this process several times

The flow of each sub-thread is as follows.

### 1. connect device

1. libusb_init
2. libusb_set_option
3. libusb_get_device_list
4. libusb_open  all usb handle
5. libusb_claim_interface the chose one

### 2. reset device

1. libusb_control_transfer  
    
     **cool reboot cyusb chip**
    
2. libusb_release_interface release chose handle
3. libusb_close close all handle
4. libusb_free_device_list
5. libusb_exit

### 3. Connect the device again

same as step 1

### 4. read-write device

- libusb_bulk_transfer

### 5. Release device

1. libusb_release_interface 
    
    release chose handle
    
2. libusb_close 
    
    close all handle
    
3. libusb_free_device_list
4. libusb_exit
[image]

[image]

```prolog
[42.063402] [001164ed] libusb: debug [usbi_handle_transfer_completion] transfer 0x11fe195a0 has callback 0x1051e099c
[42.063403] [001164ed] libusb: debug [sync_transfer_cb] actual_length=28
[42.063405] [001164ed] libusb: debug [libusb_free_transfer] transfer 0x11fe195a0
[42.063406] [001164ed] libusb: debug [libusb_submit_transfer] transfer 0x11fe195a0
[42.063407] [001164ed] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[42.063408] [001164ed] libusb: debug [ep_to_pipeRef] pipe 3 on interface 0 matches
[42.063423] [001164ed] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[42.063425] [001164ed] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[42.063531] [001164ed] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[42.063490] [00116643] libusb: debug [darwin_async_io_callback] an async io operation has completed
[42.063538] [001164ed] libusb: debug [usbi_wait_for_events] poll() returned 1
[42.063540] [001164ed] libusb: debug [handle_event_trigger] event triggered
[42.063541] [001164ed] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type bulk with kernel status 0
[... 57 lines cut ...]
[42.079470] [001164ed] libusb: debug [libusb_close]  
[42.079486] [001164ed] libusb: debug [libusb_close]  
[42.079496] [001164ed] libusb: debug [libusb_exit]  
[42.079509] [00116643] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[42.079566] [001164ed] libusb: debug [libusb_unref_device] destroy device 0.1
[42.079631] [001164ed] libusb: debug [libusb_unref_device] destroy device 1.2
[42.079666] [001164ed] libusb: debug [libusb_unref_device] destroy device 1.3
[42.079702] [001164ed] libusb: debug [libusb_unref_device] destroy device 1.1
[42.079753] [001164ed] libusb: debug [usbi_remove_event_source] remove fd 6
[4840456336]cyusb_close
SN:2030360005H22304140037
[4697955840]Cyusb2Axi
[4697955840]BusInit
[4697955840]EnumUsbDev
[4697955984]cyusb_open 0
SN:2030360005D22210240021
[4697955984]cyusb_open 1
JCPF(24259,0x16f0c7000) malloc: *** error for object 0xffffffff: pointer being freed was not allocated
JCPF(24259,0x16f0c7000) malloc: *** set a breakpoint in malloc_error_break to debug
JCPF(24259,0x16f0c7000) malloc: *** error for object 0xffffffff: pointer being freed was not allocated
```

From the order of the log, it stays in libusb_init every time, but from the perspective of SIGABRA, it may be related to the hot plug caused by the cold start of the usb device.

May I ask what is the reason for this, and is there any way to fix it?

mcuee, 2023-04-20:
Sorry but I do not quite understand what you are trying to do. It will be much easier you post the codes to demonstrate the problem. For example, are you using a new contect for each thread?

If you want to ask for the specific question of how to reset the CYUSB3014, you may want to ask in libusb-devel mailing list.

Please make sure you follow the guidelines mentioned in the FAQ. 
https://github.com/libusb/libusb/wiki/FAQ#what-are-the-differences-between-synchronous-and-asynchronous-device-io

1) Please take note that ALL transfers MUST be freed before libusb_exit() is called.
2) Please take note that ALL libusb objects MUST be released before libusb_exit() is called.
3) Please also take note that there is no way to cancel Synchronous transfer after the request has been submitted.

xumoxiao, 2023-04-20:
Sorry, I didn't describe it clearly.
1.   version : 1.0.26
2.  **libusb_control_transfer, cool reboot cyusb chip**  .The meaning of this sentence is to execute the following code
   `libusb_control_transfer(h, bmRequestType & 0x7F, bRequest, wValue, wIndex, data, wLength, timeout)`
    After running, the CYUSB chip will be reset, and CYUSB will restart to clear possible internal errors. The Mac will disconnect from cyUSB and then immediately connect to it again.
3. You can ignore the issue of multithreading for now, as I have found that single threading also has issues.
4. I will organize a streamlined code, as the current code is too complex

The problem can be that the errors caused by USB fast hot swapping are likely caused by process issues I executed. Simplify the problem.
1. How to determine if the USB is being occupied by other processes? Is using libUSB_ claim_ Does the interface return 0.
2. After fast hot swapping, can I restore the link (without using hotplug_callback) and directly release, close, exit and then init, open, and claim?

xumoxiao, 2023-04-23:
@mcuee After I simplified the code, I found that the error disappeared, because it was caused by my wrong use of multiple contexts. Still thanks for your reply, I will close this question

closed 2023-04-23 by xumoxiao

mcuee, 2023-04-23:
Thanks for the updates.