libusb/libusb #1273: Is there a problem with the reap_for_handle function in linux usbfs

lujnan, 2023-04-21:
Here there are two threads A and B, A is calling libusb_control_transfer, B uses asynchronous interrupt transfer

mcuee, 2023-04-21:
Does the following libusb documentation answer your question?
https://libusb.sourceforge.io/api-1.0/libusb_mtasync.html

lujnan, 2023-04-21:
I'will close this issue, #1267 fix this issue.

closed 2023-04-21 by lujnan

mcuee, 2023-04-21:
@lujnan 
Another reference:
https://github.com/libusb/libusb/issues/182
https://marc.info/?l=libusb-devel&m=147808669913635&w=2

From the answer above by @jwrdegoede, I think the recommendation is not to mix synchronous and asynchronous APIs in multithreaded application.

++++++++++++++++++++++++++++++++++++++++++++++++++
List:       [libusb-devel](https://marc.info/?l=libusb-devel&r=1&w=2)
Subject:    [Re: [libusb] Wake up synchronous calls on close?](https://marc.info/?t=147808429500002&r=1&w=2)
From:       [Hans de Goede <hdegoede () redhat ! com>](https://marc.info/?a=122354014900025&r=1&w=2)
Date:       [2016-11-02 11:37:42](https://marc.info/?l=libusb-devel&r=1&w=2&b=201611)

Hi,

On 02-11-16 11:56, Romain Vimont wrote:
> Hi,
>
> I observed that closing a device handle does not wake up any synchronous
> call blocking on it.
>
> I just wrote a sample to reproduce this:
> https://github.com/rom1v/libusb-sample
>
> Is it the expected behavior?

Yes, your supposed to clean up any outstanding transfers
before closing. If you use synchronous calls, then I would
expect your apps to be single threaded, use timeouts
to detect errors, and close after timeout. If you've
a multi-threaded app you really should not be using
the synchronous API.

Regards,

Hans

lujnan, 2023-04-21:
like the camera and IMU application，mix the sync and async API is a nice programming pattern.

mcuee, 2023-04-21:
> like the camera and IMU application，mix the sync and async API is a nice programming pattern.

I understand. It is just you need to be very careful when mixing the sync and async API in multithreaded application.

lujnan, 2023-04-24:
My issue is the same as #1077

lujnan, 2023-04-24:
the ioctl(hpriv->fd, IOCTL_USBFS_REAPURBNDELAY, &urb) function will take out an urb， after that, no matter whether it is locked or not, if it is taken out in a different thread, it will cause disorder.

lujnan, 2023-04-24:
This issue is easy to appear on embedded devices such as mobile phones, and it is not easy to reproduce on PCs. It may be that the USB controller or CPU on the PC is more powerful.

mcuee, 2023-04-26:
@lujnan 
I do not think you can conclude your issue is the same as #https://github.com/libusb/libusb/issues/1077 yet. It is better not to mix the two before confirmation as of now.

mcuee, 2023-08-15:
@lujnan 

Any updates on this topic? Have you tried with Windows or macOS? 

I tend to believe this is most likely an application issue and not libusb issue.

lujnan, 2023-09-13:
Yes, It's a libusb issue, The current architecture cannot guarantee that it will work properly in application or third-party libraries

mcuee, 2023-09-13:
> Yes, It's a libusb issue, The current architecture cannot guarantee that it will work properly in application or third-party libraries

You will need to provide sample codes and debug log to demonstrate that it is a libusb library issue.