libusb/libusb #1274: usbi_backend.set_option(...) is called with invalid va_list

iacg68, 2023-05-02:
Hello,

I have compiled the libusb with MSVC2019 and found that in the function _libusb_set_option(...)_ the access to _va_list ap_ is terminated too early. This leads to the fact that in the cases LIBUSB_OPTION_USE_USBDK, LIBUSB_OPTION_NO_DEVICE_DISCOVERY and LIBUSB_OPTION_WINUSB_RAW_IO when calling _usbi_backend.set_option(..., ap)_ the already deleted ap-structure is passed. This leads to a crash if later in the backend (e.g. winusb_set_option(..., ap)) a nullptr is accessed via va_arg(ap, ...).

mcuee, 2023-05-02:
Are you using the latest version of libusb git master? Do you have a code snippet to demonstrate the issue?

iacg68, 2023-05-02:
Issue is visible in the source code, cloned from 
- v1.0.23-rc1-428-gc5aec6bd (c5aec6bd678cddfab87df35de2c133f51bc8cce9) and also in 
- v1.0.23-rc1-419-g2abb2a18 (2abb2a186e6c085396654851f7d18aa20ce2e753)

Use the same code sample as mentioned in https://github.com/libusb/libusb/issues/1260

tormodvolden, 2023-05-02:
Yes, you are right. va_end(ap) must be done after calling usbi_backend.set_option(..., ap).

"After the call va_end(ap) the variable ap is undefined."

mcuee, 2023-05-18:
@tormodvolden 

Since we have not heard from @iacg68, just wondering if you can create a PR for this issue. Thanks.

closed 2023-07-06 by tormodvolden

tormodvolden, 2023-07-06:
For the record this was possibly an issue since version 1.0.25 (commit a524555).

commit 76df571c (2023-07-06): core: Avoid crash due to premature va_end

After va_end(ap) the variable ap is undefined and cannot be
accessed.

Fixes #1274
Closes #1284