libusb/libusb #1276: In the Windows environment, use libUSB_ interrupt_ The transfer failed to read mouse events. What is the reason for this?

liutt1312, 2023-05-10:
In the Windows environment, use libUSB_ interrupt_ The transfer failed to read mouse events. What is the reason for this?
int nRet = libusb_interrupt_transfer(m_devHandle, 0x81, pData, 0x4, &nLength, 1000);
cout << libusb_strerror(nRet) << endl;

Input/Output Error
Read Error
[ 1.557828] [0000043c] libusb: debug [libusb_submit_transfer] transfer 000002224F8BEF80
[ 1.558441] [0000043c] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 1.560129] [0000043c] libusb: debug [hid_submit_bulk_transfer] matched endpoint 81 with interface 0
[ 1.561331] [0000043c] libusb: debug [hid_submit_bulk_transfer] reading 5 bytes (report ID: 0x00)
[ 1.561991] [0000043c] libusb: error [hid_submit_bulk_transfer] HID transfer failed: [5] 拒绝访问。
[ 1.562661] [0000043c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.563341] [0000043c] libusb: debug [libusb_free_transfer] transfer 000002224F8BEF80
Input/Output Error
Read Error
[ 2.066713] [0000043c] libusb: debug [libusb_submit_transfer] transfer 000002224F8BEF80
[ 2.066947] [0000043c] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 2.067521] [0000043c] libusb: debug [hid_submit_bulk_transfer] matched endpoint 81 with interface 0
[ 2.067800] [0000043c] libusb: debug [hid_submit_bulk_transfer] reading 5 bytes (report ID: 0x00)
[ 2.068049] [0000043c] libusb: error [hid_submit_bulk_transfer] HID transfer failed: [5] 拒绝访问。
[ 2.068291] [0000043c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 2.068514] [0000043c] libusb: debug [libusb_free_transfer] transfer 000002224F8BEF80
Input/Output Error
Read Error

mcuee, 2023-05-10:
https://github.com/libusb/libusb/wiki/Windows#user-content-Known_Restrictions
HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.

mcuee, 2023-05-10:
Basically libusb (or even HIDAPI) is not the right library for your task Please use Windows provided function like the following.
https://learn.microsoft.com/en-us/windows/win32/inputdev/user-input (Mouse Input and RawInput).

closed 2023-05-10 by mcuee