libusb/libusb #1278: macos m2 libusb_control_transfer() perfomace

Maxsheng, 2023-05-11:
I am writing a C program on Mac that calls the libUSB API. However, I found some performance differences between the same programs on MAC x64 and MAC M2 systems. Program on Mac m2 call libusb_control_transfer() function is much slower than programs on Mac x64.

For example, running a program on Mac x64 takes 3 seconds to complete a function, while on Mac m2 it takes 10 seconds. I try to count the number of times a program calls libusb_control_transfer() and the time it takes each time, and finally calculate the average time it takes to call libusb_control_transfer().

The difference in average time spent calling a function has reached over 10 times.

Is this performance gap caused by the USB driver or the libusb library? Or is it a problem with my libusb usage?

Thanks.

mcuee, 2023-05-11:
You need to provide more details like the device and your codes and debug log.

Please read the following troubleshooting guide and post the relevant details. Thanks.
https://github.com/libusb/libusb/wiki/Troubleshooting

mcuee, 2023-05-25:
Close this for now. Please re-open if you can provide more detail to show it is a potential issue of libusb.

closed 2023-05-25 by mcuee