libusb/libusb #1280: [WinUSB] Incorrect handling of composite devices with more than 9 interfaces

afeinman-snap, 2023-05-16:
https://github.com/libusb/libusb/blob/07441f54244991af55df158b61fd69ca95b39662/libusb/os/windows_winusb.c#L1390-L1391
The logic above fails to account for the fact that the MI_XX is a hex number. So on a device with 10 interfaces it is MI_0A. As a result the interface is being ignored.

mcuee, 2023-05-16:
Just curious what is this device which has more than 9 interfaces. It will be great to post the USB descriptors for such device. Thanks.

mcuee, 2023-05-17:
The issue comes from pyocd.
* https://github.com/pyocd/pyOCD/issues/1556

afeinman-snap, 2023-05-17:
> The issue comes from pyocd.
> 
> * [Unable to force cmsis-dap v1 when flashing pyocd/pyOCD#1556](https://github.com/pyocd/pyOCD/issues/1556)

The issue was discovered via pyocd by me and initially blamed on pyocd.

afeinman-snap, 2023-05-17:
[usbdesc.txt](https://github.com/libusb/libusb/files/11493079/usbdesc.txt)
USB descriptor details. This is a custom CMSIS-DAP -based debug interface extended to provide 4 virtual UARTs. 2 interfaces per UART + 1 CMSIS-DAP v1 + 1 CMSIS-DAP v2. 
Regardless, the logic in the quoted code is wrong. For the moment I have replaced it with 
```
if (mi_str != NULL) {
   interface_number = strtoul(mi_str + 3, NULL, 16);
```

mcuee, 2023-05-17:
@afeinman-snap 

Thanks. Please help to create a PR to address the issue.

afeinman-snap, 2023-05-17:
Done: https://github.com/libusb/libusb/pull/1281

closed 2023-05-19 by tormodvolden