libusb/libusb #1283: libusb error with PS3 Eye cam when USB-C hub is connected

chapinei, 2023-05-17:
I'm trying to use a PS3 Eye Camera with opentrack 2023.1.0 and the bundled libusb0 v1.2.6.0 (06/17/2021). As soon as I connect a Dell DA305 USB-C Hub, libusb (and therefore opentrack) can no longer access the camera. 

Opentrack displays "PS3 Eye driver error: Unknown error" and logs the following to cmd:
```
[ INFO:0@8.110] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
libusb: info [init_device] 0xb95:0x1790 found 3 configurations but device isn't configured, forcing current configuration to 1
libusb: info [cache_config_descriptors] could not access configuration descriptor 1 (dummy) for 'USB\VID_0B95&PID_1790\00000000000001': [31] A device attached to the system is not functioning.
libusb: info [cache_config_descriptors] could not access configuration descriptor 2 (dummy) for 'USB\VID_0B95&PID_1790\00000000000001': [31] A device attached to the system is not functioning.
libusb: info [init_device] 0xb95:0x1790 found 3 configurations but device isn't configured, forcing current configuration to 1
libusb: info [cache_config_descriptors] could not access configuration descriptor 1 (dummy) for 'USB\VID_0B95&PID_1790\00000000000001': [31] A device attached to the system is not functioning.
libusb: info [cache_config_descriptors] could not access configuration descriptor 2 (dummy) for 'USB\VID_0B95&PID_1790\00000000000001': [31] A device attached to the system is not functioning.
```

I'm running Windows 11 with the latest patches.
The camera is plugged into a USB port on the Mainboard and I don't touch it when plugging in the USB-C hub.
No peripherals are connected to the USB hub. The hub has a HDMI and Display Port, but they are (currently) not used.
Rebooting before/after connecting the DA305 does not make a difference.
When I unplug the USB hub and then restart opentrack, everything works fine again.
I don't know whether this is of help, but I attached a screenshot from InfWizard with all plugged-in USB devices.

Is this a problem with libusb directly and you guys can help me? Or do I need to raise the issue with the opentrack team because of how they configured (?) and bundled libusb with opentrack?

Any help is greatly appreciated.

[image]

tormodvolden, 2023-05-17:
Can you please try the libusb utilities like listdevs.exe and xusb.exe to keep openview out of the equation? At first look it seems like a bug in the operating system and its drivers. Can openview access it through the WinUSB device driver (you must switch driver for the device using e.g. zadig)?

chapinei, 2023-05-18:
Thanks for trying to help. This is the output from listdevs (no USB-C hub):

```
046d:c548 (bus 2, device 4) path: 3
1415:2000 (bus 1, device 1) path: 6
174c:2074 (bus 2, device 1) path: 2
17ef:609c (bus 2, device 3) path: 4
05e3:0610 (bus 1, device 5) path: 11
174c:3074 (bus 2, device 2) path: 6
1022:149c (bus 2, device 0)
26ce:01a2 (bus 1, device 2) path: 12
1022:43ee (bus 1, device 0)
```

listdevs with USB-C hub plugged in:

```
046d:c548 (bus 2, device 4) path: 3
1415:2000 (bus 1, device 1) path: 6
2109:2822 (bus 1, device 19) path: 5
174c:2074 (bus 2, device 1) path: 2
0b95:1790 (bus 1, device 20) path: 1.3
17ef:609c (bus 2, device 3) path: 4
05e3:0610 (bus 1, device 5) path: 11
174c:3074 (bus 2, device 2) path: 6
1022:149c (bus 2, device 0)
2109:8884 (bus 1, device 21) path: 5.5
26ce:01a2 (bus 1, device 2) path: 12
1022:43ee (bus 1, device 0)
2109:0822 (bus 1, device 18) path: 1
```

This is the output from xusb 1415:2000 with the USB-C hub connected. The only difference to running xusb without the USB-C hub is the new line at the beginning where it states that it found 3 configurations but device isn't configured (line 4):

```
Using libusb v1.0.26.11724

Opening device 1415:2000...
libusb: info [init_device] 0xb95:0x1790 found 3 configurations but device isn't configured, forcing current configuration to 1

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 1415:2000
         bcdDevice: 0200
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: libusb: error [libusb_get_bos_descriptor] failed to read BOS (-2)
no descriptor

Reading first configuration descriptor:
              total length: 142
         descriptor length: 9
             nb interfaces: 3
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 0A
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 0
[... 252 lines cut ...]

          ===>Endpoint Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x05
bEndpointAddress:                  0x84  -> Direction: IN - EndpointID: 4
bmAttributes:                      0x05  -> Isochronous Transfer Type, Synchronization Type = Asynchronous, Usage Type = Data Endpoint
wMaxPacketSize:                  0x0300 = 1 transactions per microframe, 0x300 max bytes
wInterval:                       0x0004
bSyncAddress:                      0x00

          ===>Audio Streaming Class Specific Audio Data Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x25 (CS_ENDPOINT)
bDescriptorSubtype:                0x01 (EP_GENERAL)
bmAttributes:                      0x01
                                   (Sampling Frequency control)
bLockDelayUnits:                   0x00 (Undefined)
wLockDelay:                      0x0000
```

Where in this does it state which configuration is needed?

Finally, it certainly is a good thing that there does not seem to be a bug in libusb. However, if it really is a bug in the OS or it's drivers, I doubt that I will be able to convince Microsoft to fix this... :-/

I hope I could provide you with everything you asked for. Any suggestions what I should do next?

mcuee, 2023-05-18:
It is not a bug in libusb. To libusb Windows backend, an unconfigured device is basically equivalent to an non-existent device.

There is actually a PR for it.
https://github.com/libusb/libusb/pull/1270

Maybe it is not a bug in the OS as well, but rather there is a limitation in this setup.

closed 2023-05-18 by mcuee