libusb/libusb #1287: libusb_free_device_list(ptr, 1) does not unref all devices when the device list has a null entry

diablodale, 2023-05-31:
`libusb_free_device_list(ptr, 1)` does not handle the scenario of having `null` entries in the list. It will errantly stop its unref walk of the list thinking it has reached the terminating `null`.

### Setup

* all os and platforms
* libusb at `master`

### Repro

Found in code review as it is hard to repro.

`libusb_get_device_list()` returns both a `count` and a `libusb_device**` which points to the allocated list of devices `libusb_device*`.
The `count` is the allocated `listSize - 1`. The list is null terminated as per libusb docs.
In some scenarios, it is possible for the allocated list to have `null` entries.

`libusb_free_device_list(list, 1)` accepts the pointer to the allocated list of devices. No count parameter.
This api's code does not handle the scenario of having `null` entries in the list. It will errantly stop its walk of the list thinking it has reached the terminating `null`.

https://github.com/libusb/libusb/blob/cc498ded18fb2c6e4506c546d0351c4ae91ef2cc/libusb/core.c#L895-L901

### Workaround

Don't use `libusb_free_device_list(ptr, 1)`. Instead, write custom code to walk the known `count` size of the list and unref any devices that are not null.

Youw, 2023-06-01:
I do not believe it is possible to fix this w/o introducing something like `libusb_free_device_list_n(libusb_device ** list, size_t device_count, int unref_devices)` and it is up to the core team to decide is this is needed at all.

The real question is: how do you come up with `NULL` devices in between your list? I believe the second parameter in the `libusb_free_device_list` is designed to include cases like this - when a user code decides to modify the list itself.

mcuee, 2023-06-01:
> The api libusb_get_device_list() makes no guarantee that every pointer it returns will be non-null. 

I believe it will not contain a null entry by definition. In a system libusb is expected to work, there will be USB devices (eg: at least the root hub).

As far as I know, a lot of libusb APIs expect you to call them in pair.

[libusb_get_device_list()](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gac0fe4b65914c5ed036e6cbec61cb0b97)
> You are expected to unreference all the devices when you are done with them, and then free the list with [libusb_free_device_list()](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gad3b8561d064bb3e1b8851ddeed3cd7d6). Note that [libusb_free_device_list()](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gad3b8561d064bb3e1b8851ddeed3cd7d6) can unref all the devices for you. Be careful not to unreference a device you are about to open until after you have opened it.

mcuee, 2023-06-01:
@diablodale 

I tend to think there are issues inside libusb when dealing with reference count and in the past similar issues have been fixed. But I think this issue is not one of them.

I see that you referenced two other libusb issues #988 and #1165 here:
* https://github.com/luxonis/XLink/issues/62#issuecomment-1570779452

Based on my investigation, it seems to me #998 is more likely a libusb-compat-0.1 issue and not a libusb issue.
* https://github.com/libusb/libusb/issues/988

I am not so sure about #1165 as of now. I tend to believe that is more an application issue.
* https://github.com/libusb/libusb/issues/1165

diablodale, 2023-06-01:
I hope there is never a `libusb_device* == null` also. However, I see no guarantee in the docs or in the code comments that makes the guarantee that all returned pointers in the list within count are not null.

The code for all backends and platforms on all os versions would need to handle all error conditions, hotplugs, device reconfigures, perfect refcount handling, no bugs, etc. while `libusb_get_device_list()` is running...such that it can make the guarantee that it will never return a null pointer in the list.

Perhaps libusb code already does that. If that is true then 🎉party-time I recommend the API docs be updated with the guarantee and then this issue is resolved.

I have only looked briefly into libusb code; not the corpus of every backend to check for this handling. I see in the code a temporary linked list is populated with found devices and eventually that is copied into the contiguous block of memory returned via the api's pointer. All the code which does the population on all platforms and the code which does the convert from list->block would need to guarantee that it never has a null device pointer.

Separately, yes, I am investigating hubs not having their refs released on libusb shutdown. That issue led me into the libusb code itself where I discovered this separate issue.

fyi, I'm not discussing anything about the need to unref devices in the list. The api doc is clear the list's devices need to be unref() by someone's code.

mcuee, 2023-06-01:
> fyi, I'm not discussing anything about the need to unref devices in the list. The api doc is clear the list's devices need to be unref() by someone's code.

I think most of the libusb applications do not need to call  `libusb_unref_device` and `libusb_ref_device` at all. They do need to call `libusb_get_device_list` and l `libusb_free_device_list` in pair, and `libusb_open` and `libusb_close` in pair.

diablodale, 2023-06-01:
You are going off topic in your discussion on ref and unref. That is not the topic of this issue.

The topic of this issue is libusb_free_device_list() not handling a list which contains a null pointer.

mcuee, 2023-06-02:
> The topic of this issue is libusb_free_device_list() not handling a list which contains a null pointer.

Which I believe will not happen. If you think the documentation needs to be improved, please raise a PR. If not I will close this issue.

diablodale, 2023-06-02:
That's great to hear. Then this is a doc clarification issue.

Happy to make a PR. The doc https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html looks like doxygen so I will look through the code comments to find clarifying text to add. Likely will be in: libusb_free_device_list, libusb_get_device_list, and somewhere in the long Nutshell discussion.

tormodvolden, 2023-07-05:
> In some scenarios, it is possible for the allocated list to have `null` entries.

What scenarios would that be? As far as I can see, that would be a bug.

libusb_get_device_list() returns a list of devices structures. It refcounts each of the devices. Maybe some of the real/underlying devices that they are associated with don't exist any longer after the call, but the devices structures still exist as long as they are referenced.

libusb_get_device_list() works perfectly in pair with libusb_free_device_list(, 1) because the former has refcounted all the devices in the list, and the latter unrefs them. Unless something else in between has messed up the reference counting, all devices will be released.

diablodale, 2023-07-06:
When libusb functions perfectly always and never writes a null into any place in the list except at the "end", and the contiguous list of pointers is not interntionally/accidentially manipulated by a user, then it is true that a contiguous non-null list terminated by a null would work correctly.

Unfortunately, libusb isn't perfect, users manipulate lists, etc. And so using this API will unsuccessfully and prematurely release only a subset of the originally ref'd devices.

Personally, I check for null and don't use the "1" option so that I correctly and completely free all devices. This was purely for the community so that the documentation was updated to be clear and declare its dependency on a non-null contiguous list.
As this PR was closed, the doc won't be as clear. But this PR and the related issue will serve as a bridge for those that do searches. Cheers.

tormodvolden, 2023-07-07:
Please provide more information on the case where libusb_get_device_list() returns a list shorter than the return value says. We really want to fix this. I haven't seen any reports on it before.

If a user manipulates the list by truncating it, it cannot be expected that libusb_free_device_list(ptr, 1) will be able to do its work correctly.

diablodale, 2023-07-07:
This issue I documented well in OP with repo and code. It is focused on `libusb_free_device_list()` API's lack of solid parameter validation and boundary checks. It will behave errantly due to the errant code in the scenarios I describe in the OP by: stopping its iteration early, leak the resources, and will not return an error code.

I offered a discussion on code changes -and- an alternative PR to only document the lack/gap in this API. Maintainers chose to deny both -- fully their choice. ✌️ As I wrote above...

> Personally, I check for null and don't use the "1" option so that I correctly and completely free all devices. This was purely for the community so that the documentation was updated to be clear and declare its dependency on a non-null contiguous list.
As this PR was closed, the doc won't be as clear. But this PR and the related issue will serve as a bridge for those that do searches. Cheers.

This issue is not about `libusb_get_device_list()` and I will not engage in a discussion on that separate API at this time. If you want to open your own issue on that API and want my input on it, please mention me so I know about it.

tormodvolden, 2023-07-07:
I can't see there is a lack in the API here. As documented for libusb_free_device_list():

> Frees a list of devices previously discovered using libusb_get_device_list()

There is nothing there suggesting that the user should be able to modify the list in any way. Not in the libusb_get_device_list() section either:

> output location for a list of devices. Must be later freed with libusb_free_device_list().

It also clearly points out that it is a null-terminated list. So if the user truncates the list by writing a null element, it is obvious that libusb_free_device_list(x, 1) will only operate on what is left of that list. There is no hidden structure keeping track of this other than the list itself, and the list is not modified or free'd by libusb other than with libusb_free_device_list(). I believe you can copy the list and hand the copy to libusb_free_device_list() after modifying it but I wouldn't suggest going in that direction - the original list would leak. Whatever you are trying to accomplish should be done in another way than modifying this list.

If there isn't any indication that it ever can happen that libusb_get_device_list() would write a list of different length than its return value, I guess there is not much to do here.

Your documentation suggestion was insisting on "non-null contiguous list" but that is given by it being a null-terminated list.

> This issue is not about libusb_get_device_list() and I will not engage in a discussion on that separate API at this time. If you want to open your own issue on that API and want my input on it, please mention me so I know about it.

This is maybe the fundamental misunderstanding. These two functions go together, as documented by both of them. They are not "separate APIs".

diablodale, 2023-07-08:
Again for the 4th time, I am highlighting and offering two solutions to solidify parameter validation and boundary checks of `libusb_free_device_list` for the community. That's how open source works. When someone (me!) isolates and fixes an issue, they (me!) contribute the fix back for the benefit of the community.

I have solidified `libusb_free_device_list` parameter checking and boundary checks myself in my own code. My code protects itself from poor behavior by libusb in scenarios where it generates device list data structures which have nulls in the device list.

> Whatever you are trying to accomplish should be done in another way than modifying this list.

I'm not engaging in the above. You are going offtrack into a discussion unknown to me and unrelated to this OP issue.

While you have the opinion that "these two functions go together", that is false and perhaps the core of your misunderstanding. I don't fault you. Your lens is through libusb and libusb coding patterns. Said another way, when you have a hammer, everything looks like a nail.

There is no technical need to call `libusb_get_device_list` and then call `libusb_free_device_list` after that. Their implementation code in libusb `core.c` is not interdependent (I reviewed all of code.c and linked to part of it in OP). They also do not own or maintain private state/data between themselves (again linked in OP). Its a simple heap malloc+free for trival POD array of pointers. There is no magic and nothing difficult or advanced here. To use your words, there is nothing "scary". Due to these APIs being independent, I focused my OP issue on only `libusb_free_device_list`.

Unfortunately, you keep attempting to drag in `libusb_get_device_list` and for the 5th time I have nothing to discuss regarding `libusb_get_device_list` at this time. You have needlessly complicated the discussion; unconsciously collapsing two things/APIs which they are instead distinct likely due to your libusb lens.

I appreciate libusb's value to the community. ✌️ It is not perfect and has known and readily found issues like not validating its parameters (which I referenced in this issue and its PR) and UNsafe thread APIs like `libusb_get_device_list`. Due to the codebase's age, it is stuck in legacy C code and a 3000 line implementation file.

And...all that is ok 🕊️  Why?

...Because many of libusb's faults are documented. However, not all are documented.

> Sidebar: `libusb_get_device_list()` on Windows is NOT thread-safe. The first sentence of https://libusb.sourceforge.io/api-1.0/libusb_mtasync.html and safety section of https://libusb.sourceforge.io/api-1.0/libusb_caveats.html are false. Two+ threads can not call `libusb_get_device_list()` at the same time with the same or default context. It results in crashes/seg violations, invalid reference counts, missing/invalid devices, and other unpredictable behavior.
> 
> I reviewed the 3000 lines of `core.c` to see if I could quickly fix these threading bugs and submit a PR. Nope...I don't have the priorities and free time to fix manual C critical sections and goto statements. Instead, I wrap my code's call to `libusb_get_device_list` with its own mutex to workaround the unsafe `libusb_get_device_list` api.
> 
> While reviewing `core.c` code, I isolated independent problems with `libusb_free_device_list` and I therefore opened this OP issue. I may in the future open the threading bug with `libusb_get_device_list` if I have the priority and time.

In the spirit of open source, I used this OP issue and PR to add helpful text to this well-intending libusb documentation to caution users on the use of `libusb_free_device_list`. This proposed doc addition would be a substitute since maintainers did not want to fix the API code.

> To maintain API compat would likely need counts stored in a `struct{size_t count; void* user_pointer}` for `libusb_get_device_list(list)` to return `list = user_pointer` and `libusb_free_device_list(list)` walk backwards from `list` param to get the actual `count` and correct address for `free(addressof(struct))`. Might need a magic. Or could redefine the list pointer to point to the same struct which then points to the list of devices. Best would be to have a new safe `libusb_free_device_list(libusb_device** list, ssize_t count, int unref_devices)` since this would work with today's `libusb_get_device_list`.

libusb maintainers have denied my text addition and a code fix. And for the 2nd time, that is fully the maintainers choice. ✌️

This is my last post since I have achieved my goal to fix and/or document the OP issues of parameter validation and boundary checks. This issue will serve as the documentation. We both keep saying the same things to each other with no progress; this discussion is not productive and it is time to move on. Cheers ✌️🕊️

closed 2023-07-08 by diablodale