libusb/libusb #1288: Is there a problem with slow speed when using libUSB to control type transfer on the MAC m2 architecture?

Maxsheng, 2023-06-08:
Why is using control type transfer on the MAC m2 architecture slower than other platforms?

The following is my test code, which was tested using the same USB device in Mac m2, Mac x64, and Ubuntu x64 environments.

```c
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <libusb.h>
#include <sys/time.h>

uint64_t get_us()
{
    uint64_t time = 0;
    struct timeval tv;
    gettimeofday(&tv, NULL);
    time = tv.tv_sec * 1000 * 1000 + tv.tv_usec;

    return time;
}

int main()
{
    // open usb handle
    int ret;
    ret = libusb_init(NULL);
    libusb_device_handle *usb_handle; 
    usb_handle = libusb_open_device_with_vid_pid(NULL, 0x248a, 0x8266);
    if (usb_handle == NULL) {
        exit(1);
    }
    ret = libusb_set_auto_detach_kernel_driver(usb_handle, 1);
    ret = libusb_claim_interface(usb_handle, 0);

    uint64_t start = get_us();
    unsigned char buf[256];

    ret = libusb_control_transfer(usb_handle, 0xC1, 0x02, 0x8014, 0, buf, 12, 1000);
    ret = libusb_control_transfer(usb_handle, 0xC1, 0x02, 0x8014, 0, buf, 12, 1000);
    ret = libusb_control_transfer(usb_handle, 0xC1, 0x02, 0x8014, 0, buf, 12, 1000);
    ret = libusb_control_transfer(usb_handle, 0xC1, 0x02, 0x8014, 0, buf, 12, 1000);

    memset(buf, 0, 256);
    
    ret = libusb_control_transfer(usb_handle, 0x41, 0x02, 0x9548, 0, buf, 16, 1000);
    ret = libusb_control_transfer(usb_handle, 0x41, 0x02, 0x9548, 0, buf, 16, 1000);
    ret = libusb_control_transfer(usb_handle, 0x41, 0x02, 0x9548, 0, buf, 16, 1000);
    ret = libusb_control_transfer(usb_handle, 0x41, 0x02, 0x9548, 0, buf, 16, 1000);
    uint64_t end = get_us();

    printf("spent time %ld(us)\n", end - start);

    libusb_close(usb_handle);
    libusb_exit(NULL);

    return 0;
}
```

Finally, my conclusion is to call libusb_control_transfer() function on the MAC m2 architecture takes about **10 times slower** than the average in x64 architecture.

The following are the software running records and wireshark software packet capture records on the **MAC m2** architecture.

```
MacBook-Pro % ./libusb_test
spent time 22949(us)
```
[image]

The following are the running records and wireshark software packet capture records on **Ubuntu20 X64**.

```
ubuntu20:~$ ./libusb_test 
spent time 1246(us)
```
[image]

From the data captured by Wireshark software, it can be seen that on the MAC m2 architecture, a control transmission takes about **3ms** from submission to reply. On the x64 architecture, only around **200us** is required.

I also tested bulk type transmission, and the performance on the MAC m2 architecture is consistent with that on the x64 architecture.

Is this the reason for the MAC m2 architecture USB driver? Is there a solution to this problem?

mcuee, 2023-06-08:
This is unlikely a libusb issue. You may want to check with Apple.

Link:
https://developer.apple.com/bug-reporting/

Maxsheng, 2023-06-09:
My macos system is the latest(ventura 13.4). As I mentioned above, the performance of the macos x64 control type has not been affected. My USB device protocol is 1.1.
I have discovered an interesting phenomenon. I use the Beagle USB packet grabber to insert the device into the Beagle packet grabber. Then use the data center to click capture and run my libUSB program. At this point, the performance of USB control transmission is normal.
I'm not sure what data center software is used to develop USB programs. But it seems like after clicking capture, it's like configuring the system, which makes the control performance different.

mcuee, 2023-06-09:
@Maxsheng 

In this case, it could be your device does not work well with the root hubs from macOS M2 machine. By inserting the Beagle, I assue that is a full speed or high speed device, your device may work better now.

You can carry out an experiment by adding a high speed hub before your device to see if that helps.

mcuee, 2023-08-15:
@Maxsheng 

Apples has updated macOS and you may want to try the latest macOS version to see if this is fixed. Thanks.

mcuee, 2023-10-03:
No update. I will close this issue now.

Please re-open if you still have the issue. Thanks.

closed 2023-10-03 by mcuee