libusb/libusb #1290: on macos , use libusb_control_transfer return LIBUSB_ERROR_PIPE(Pipe error)

pengxianheng-nreal, 2023-06-12:
On macos 13.X,  I use libusb to communicate with usb devices。 but sometimes call libusb_control_transfer  return -9(LIBUSB_ERROR_PIPE)。 before call  libusb_control_transfer，libusb_claim_interface is success。the code is roughly as follows

`libusb_init(&ctx_)
libusb_get_device_list(ctx_, &devs)
libusb_get_device_descriptor(dev, &desc);
libusb_open(usb_dev_, &dev_handle_);
libusb_free_device_list(devs, 1);

ret = libusb_detach_kernel_driver(dev_handle_, idx);

if (ret == LIBUSB_SUCCESS || ret == LIBUSB_ERROR_NOT_FOUND || ret == LIBUSB_ERROR_NOT_SUPPORTED) {
    DRIVER_LOG_TRACE("claiming interface {}", idx);
    if (!( ret = libusb_claim_interface(dev_handle_, idx))) {
        claimed_ |= ( 1 << idx );
    }
} else {
    DRIVER_LOG_TRACE("not claiming interface {}: unable to detach kernel driver ({})", idx, ret);
}

int ret = libusb_control_transfer(
        dev_handle_,
        0xa1, 0x85,
        0x0200,
        0x0400,
        data,
        2,
        0);  `

the libusb log shows as blow:

	行  17: [ 0.031844] [00bcaa69] libusb: debug [libusb_get_device_list]  
	行  19: [ 0.031886] [00bcaa69] libusb: debug [libusb_get_device_descriptor]  
	行  20: [ 0.031927] [00bcaa69] libusb: debug [libusb_get_device_descriptor]  
	行  21: [ 0.031942] [00bcaa69] libusb: debug [libusb_open] open 20.9
	行  22: [ 0.032035] [00bcaa69] libusb: debug [darwin_open] device open for access
	行  23: [ 0.032072] [00bcaa69] libusb: debug [libusb_get_config_descriptor] index 0
	行  24: [ 0.032090] [00bcaa69] libusb: debug [parse_configuration] skipping descriptor 0xb
	行  25: [ 0.032102] [00bcaa69] libusb: debug [parse_endpoint] skipping descriptor 0x25
	行  26: [ 0.032113] [00bcaa69] libusb: debug [parse_endpoint] skipping descriptor 0xb
	行  27: [ 0.032128] [00bcaa69] libusb: debug [libusb_detach_kernel_driver] interface 0
	行  28: [ 0.032139] [00bcaa69] libusb: debug [darwin_detach_kernel_driver] attempting to detach kernel driver from device
	行  33: [ 0.032179] [00bcaa69] libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device
	行  34: [ 0.032292] [00bcaa69] libusb: debug [darwin_reenumerate_device] darwin/reenumerate_device: restoring state...
	行  35: [ 0.032319] [00bcaa69] libusb: debug [darwin_open] device open for access
	行  36: [ 0.032330] [00bcaa69] libusb: debug [darwin_restore_state] darwin/restore_state: reclaiming interfaces
	行  37: [ 0.032340] [00bcaa69] libusb: debug [darwin_restore_state] darwin/restore_state: device state restored
	行  38: [ 0.032363] [00bcaa69] libusb: debug [libusb_claim_interface] interface 0
	行  40: [ 0.033020] [00bcaa69] libusb: debug [get_endpoints] building table of endpoints.
	行  41: [ 0.033053] [00bcaa69] libusb: debug [get_endpoints] interface: 0 pipe 1: dir: 1 number: 6
	行  42: [ 0.033092] [00bcaa69] libusb: debug [darwin_claim_interface] interface opened
	行  43: [ 0.033119] [00bcaa69] libusb: debug [libusb_submit_transfer] transfer 0x7fbcdc6043f0
	行  48: [ 0.033181] [00bcaa69] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
	行  49: [ 0.033234] [00bcaa69] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
	行  50: [ 0.033241] [00bcaa69] libusb: debug [handle_events] event sources modified, reallocating event data
	行  51: [ 0.033250] [00bcaa69] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
	行  52: [ 0.033339] [00bcaa6b] libusb: debug [darwin_async_io_callback] an async io operation has completed
	行  53: [ 0.033376] [00bcaa69] libusb: debug [usbi_wait_for_events] poll() returned 1
	行  54: [ 0.033386] [00bcaa69] libusb: debug [handle_event_trigger] event triggered
	行  55: [ 0.033393] [00bcaa69] libusb: debug [darwin_handle_transfer_completion] handling transfer completion type control with kernel status -536854449
	行  56: [ 0.033399] [00bcaa69] libusb: debug [darwin_transfer_status] transfer error: pipe is stalled
	行  57: [ 0.033404] [00bcaa69] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fbcdc6043f0 has callback 0x10f0c8ba0
	行  58: [ 0.033410] [00bcaa69] libusb: debug [sync_transfer_cb] actual_length=0
	行  59: [ 0.033418] [00bcaa69] libusb: debug [libusb_free_transfer] transfer 0x7fbcdc6043f0

the log is showing:  transfer error: pipe is stalled

Any one get any ideas? Thanks a lot.

mcuee, 2023-06-12:
One more question, are you using the 1.0.26 release version or git?

There is a fix in current git after 1.0.26 release which may help on this issue.
* https://github.com/libusb/libusb/pull/1117

pengxianheng-nreal, 2023-06-13:
@mcuee yeah, the above code works fine on linux ,android, and most macos。 Even if the macOS which cause this problem is not alway present。Sometimes it works, sometimes it doesn't。On macOS 12.X and 13.X  also  encountered this problem。I use the lasted master branch code.

pengxianheng-nreal, 2023-06-22:
@mcuee  I found the solve way from the issue:  https://github.com/libusb/libusb/issues/838.    it seems recover normal , when I cause this error ,I call libusb_reset_device and retry again, it will ok . why call libusb_reset_device can solve this problem?

mcuee, 2023-06-22:
> @mcuee I found the solve way from the issue: #838. it seems recover normal , when I cause this error ,I call libusb_reset_device and retry again, it will ok . why call libusb_reset_device can solve this problem?

Most likely the device goes to a bad state (eg: firmware crashed, or Data Toggle Error) and the libusb_reset_device will return the device into a normal state.

tormodvolden, 2023-07-07:
From the given information, my guess is that it is a device firmware bug, and that the different results between the tested environments are due to small timing differences.

mcuee, 2023-08-15:
@pengxianheng-nreal 

Any updates on this topic? 

I intend to close this issue after two weeks if no further updates are provided.

pengxianheng-nreal, 2023-08-29:
@mcuee no updates. you can close this issue. Thanks~

pengxianheng-nreal, 2023-08-29:
> From the given information, my guess is that it is a device firmware bug, and that the different results between the tested environments are due to small timing differences.

yeah. probably. but we use the third device fireware. can not detect the firmware problem. now use libusb_reset_device  can works for me.

closed 2023-08-29 by mcuee