libusb/libusb #1293: Linux: Downstream to Downstream connections stall libusb_open related calls.

Mitch9192, 2023-06-27:
We have a very simple application that prints out all string descriptors for all devices connected to a computer (basically examples/listdevs.c but with open and close calls).   This action is preformed every ~5 seconds or so and executing on both Linux and Mac.

It has come to our attention that there is a special case which can occur from 2x downstream devices being connected together.  When this occurs the following calls take ~30-60 seconds each to preform on Linux machines, however, Mac is unaffected.

- libusb_open
- libusb_get_string_descriptor_ascii

The easiest way to reproduce is to connect two Type C ports together via CtoC cable.  Although this test configuration has no real world value the possible condition does exist between dual role devices.  For instance phones which can be both host and clients. 

Simply adding a libusb_open and libusb_close to the basic listdevs.c example will reproduce the issue pending a CtoC cable is connected to 2x downstream facing ports (DFP's).

mcuee, 2023-06-27:
@Mitch9192 

I tend to think this is not a libusb issue but rather related to the Linux kernel USB subsystem. What is the host and Linux distro and kernel version used?

Please also post the connection scheme, test code and debug log (with env variable LIBUSB_DEBUG=4). Thanks.

Mitch9192, 2023-06-29:
@mcuee 

Please find the requested information.

As to your test: 
In most cases you will not run into this.  The phone would need to preform a Data Role Swap (become the host.  common implementations are for something like CarPlay or Android DeX). With that said this is not very common for host computers as they are typically not capable of doing a Data Role Swap.  It is fair to say that the phone should ASK before swapping, but I'm sure you are aware that not all devices adhere to specification.  Especially as it gets more complicated. 

Again, I admit this is a rare case, but the company I work for (Acroname Inc) makes equipment for testing USB devices. We can manually request and override all sorts of things like this; however, in this case the easiest way is to simulate it is by using a cable to connect 2x downstream ports together. 

FYI: Libusb was built from the top of the tree (commit: 8450cc93f6c8747a36a9ee246708bf650bb762a8).  tag 1.0.26 also exhibits the same result. 

System Stuff:
[SystemInfo.txt](https://github.com/libusb/libusb/files/11908001/SystemInfo.txt)
[OS_info.txt](https://github.com/libusb/libusb/files/11908002/OS_info.txt)

Output results:
[libusb_no_loopback_success.txt](https://github.com/libusb/libusb/files/11908005/libusb_no_loopback_success.txt)
[libusb_loopback_fail.txt](https://github.com/libusb/libusb/files/11908006/libusb_loopback_fail.txt)

Program: (renamed from .c to .txt)
[main.txt](https://github.com/libusb/libusb/files/11908010/main.txt)
[makefile.txt](https://github.com/libusb/libusb/files/11908015/makefile.txt)

Scheme
[image]

Let me know if I can provide anything else.

Mitch9192, 2023-06-30:
Interesting Debugging Notes:  

1. Our application uses libusb in 2x places: 1: device communication 2: descriptor print out.  Originally these 2x cases were in the same thread.  Obviously when we hit the issue reported it would take down our device communication.  As a stopgap we moved these items into separate threads thinking we could keep our device communication up.  We were wrong.  Hitting the reported issue in one thread took down the other libusb thread.  Currently we are using the default context.  Modifying our code to use a user defined context allowed the device thread to continue while the descriptor thread was blocked.

2. The issue appears to be isolated to USB 3.0.  USB 2.0 loopback does not cause issues.

FYI: The issue can be repeated on a USB 3.0 Type A connectors; however, this is a very RARE cable (USB3.0 Type-A to Type-A).

mcuee, 2023-07-01:
@Mitch9192 

Thanks for the detailed info. Please help to provide the output of `lsusb -t` and `lsusb -vvv` info as well. Thanks.

At the same time, I will suggest you post the question to linux-usb mailing list as well since I tend to believe this is most likely an issue with Linux kernel USB stack. I think you need to subscribe in order to post.
http://vger.kernel.org/vger-lists.html#linux-usb

mcuee, 2023-07-01:
@jwrdegoede 

Just wondering if you can help here as well. Thanks.

tormodvolden, 2023-07-07:
It may be worth doing a strace as well. libusb_open() isn't doing much more than a plain open() on the e.g. /dev/bus/usb/002/004 device node, and a IOCTL_USBFS_GET_CAPABILITIES. Can you take libusb out of the equation by just doing this in a test program?

tormodvolden, 2023-07-07:
It shouldn't be very complicated (famous last words :) ). Just look at op_open() in libusb/os/linux_usbfs.c. Try this:
```
#include <unistd.h>
#include <fcntl.h>
#include <linux/types.h>
#include <sys/ioctl.h>
#include <linux/usbdevice_fs.h>

int main (int argc, char **argv)
{
    int fd, caps, r = 0;

    fd = open(argv[1], O_RDWR);
    /* r = ioctl(fd, USBDEVFS_GET_CAPABILITIES, &caps); */
    if (fd >= 0)
        close(fd);
 
    return fd < 0 || r != 0;
}
```
and run the program with the device node, e.g
`./a.out /dev/bus/usb/002/004`
Then run it with strace. And if it doesn't hang with open() try uncommenting the ioctl to see if that one will hang.

Mitch9192, 2023-07-09:
@tormodvolden Thanks for the code and great idea.  I think you have successfully proved that it is NOT libusb.  Please see attached strace file (I have also attached the other requested files for completeness.)

Not sure how I didn't notice this before, but "lsusb" even hangings when the loopback is present. 

[open_device_strace_time.txt](https://github.com/libusb/libusb/files/11994087/open_device_strace_time.txt)
[lsusb_vvv.txt](https://github.com/libusb/libusb/files/11994098/lsusb_vvv.txt)
[lsusb_t.txt](https://github.com/libusb/libusb/files/11994099/lsusb_t.txt)

tormodvolden, 2023-07-09:
For timing documentation you can use the strace -tt or -T options as well.

I'll close this ticket for now, but please post a reference to the Linux bug report once it is filed.

closed 2023-07-09 by tormodvolden