libusb/libusb #1302: libusb open fails returns error not supported

Janarthanan-S-03, 2023-08-29:
while trying to open a hub with vid:0x424 and pid:0x2906 in windows 10 x64 processor

logs:-
Application Code Example Version: 1.0.0
DEBUGINFO:libusb_open_device_with_vid_pid HUB VID:0x0424, PID:0x2906
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.024401] [00005ea8] libusb: debug [libusb_get_device_list]
[ 0.027186] [00004c74] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.039438] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [3D]
[ 0.039571] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [3E]
[ 0.039649] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [3F]
[ 0.039751] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [40]
[ 0.039825] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [43]
[ 0.039902] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [44]
[ 0.040020] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.040328] [00005ea8] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.040488] [00005ea8] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.040510] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.040842] [00005ea8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.040906] [00005ea8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.040926] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [41]
[ 0.041079] [00005ea8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.041120] [00005ea8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.041142] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [42]
[ 0.041418] [00005ea8] libusb: debug [get_api_type] driver(s): rtump64x64
[ 0.041476] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.041636] [00005ea8] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.041677] [00005ea8] libusb: debug [get_api_type] matched driver name against HID API
[ 0.041695] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.041872] [00005ea8] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.041910] [00005ea8] libusb: debug [get_api_type] matched driver name against HID API
[ 0.041928] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.042147] [00005ea8] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.042327] [00005ea8] libusb: debug [get_api_type] matched driver name against HID API
[ 0.042460] [00005ea8] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.043298] [00005ea8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_0B201028&REV_01\3&11583659&1&68' bus number 1
[ 0.047806] [00005ea8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_0B201028&REV_20\3&11583659&1&A0' bus number 2
[ 0.049775] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [42]
[ 0.049945] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.050123] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1011 bytes)
[ 0.050445] [00005ea8] libusb: debug [init_device] (bus: 2, addr: 2, depth: 1, port: 6): 'USB\VID_1BCF&PID_28C4\01.00.00'
[ 0.050804] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [41]
[ 0.051140] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.052055] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 235 bytes)
[ 0.052265] [00005ea8] libusb: debug [init_device] (bus: 2, addr: 30, depth: 3, port: 1): 'USB\VID_046D&PID_0A38\7&D4EFE21&0&1'
[ 0.052737] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [45]
[ 0.052872] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.053323] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.053783] [00005ea8] libusb: debug [init_device] (bus: 2, addr: 31, depth: 1, port: 7): 'USB\VID_0424&PID_2906\5&20218183&0&7'
[ 0.054032] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [3E]
[ 0.054346] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.054448] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.054831] [00005ea8] libusb: debug [init_device] (bus: 1, addr: 12, depth: 1, port: 2): 'USB\VID_0BDA&PID_0487\5&E0F0CD&0&2'
[ 0.056890] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [44]
[ 0.057297] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.057375] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.057737] [00005ea8] libusb: debug [init_device] (bus: 1, addr: 15, depth: 2, port: 3): 'USB\VID_0BDA&PID_0413\6&355B9095&0&3'
[ 0.057986] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [5A]
[ 0.058494] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.061577] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.062229] [00005ea8] libusb: debug [init_device] (bus: 2, addr: 29, depth: 3, port: 3): 'USB\VID_03F0&PID_134A\7&D4EFE21&0&3'
[ 0.062635] [00005ea8] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&380AFD4&0&0' reports 16 ports
[ 0.062739] [00005ea8] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&380AFD4&0&0'
[ 0.063033] [00005ea8] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&C7086AB&0&0' reports 5 ports
[ 0.063232] [00005ea8] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&C7086AB&0&0'
[ 0.063465] [00005ea8] libusb: debug [winusb_get_device_list] found existing device for session [40]
[ 0.063792] [00005ea8] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.063902] [00005ea8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.064217] [00005ea8] libusb: debug [init_device] (bus: 2, addr: 26, depth: 1, port: 2): 'USB\VID_0BDA&PID_5487\5&20218183&0&2'
[... 40 lines cut ...]
[ 1.398632] [00005ea8] libusb: debug [set_hid_interface] interface[0] = \\?\HID#VID_413C&PID_B06E#7&E8BEB33&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 1.399020] [00005ea8] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\DELL0B20&COL04\5&EB44C66&0&0003' (non USB HID, newly connected, etc.) - ignoring
[ 1.400289] [00005ea8] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\CONVERTEDDEVICE&COL01\5&2204AA2D&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 1.400931] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.0
[ 1.402165] [00005ea8] libusb: debug [libusb_get_device_descriptor]
[ 1.402590] [00005ea8] libusb: debug [libusb_get_device_descriptor]
[ 1.402891] [00005ea8] libusb: debug [libusb_get_device_descriptor]
[ 4.853956] [00005ea8] libusb: debug [libusb_open] open 2.31
[ 6.864877] [00005ea8] libusb: debug [winusb_open] unsupported API call for 'open' (unrecognized device driver)
[ 6.864954] [00005ea8] libusb: debug [libusb_open] open 2.31 returns -12
[ 6.866000] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.2
[ 6.867383] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.30
[ 6.867699] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.31
[ 6.868051] [00005ea8] libusb: debug [libusb_unref_device] destroy device 1.15
[ 6.868412] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.29
[ 6.869313] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.4
[ 6.869778] [00005ea8] libusb: debug [libusb_unref_device] destroy device 1.14
[... cut ...]
[ 6.870275] [00005ea8] libusb: debug [libusb_unref_device] destroy device 1.0
[ 6.870446] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.32
[ 6.870756] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.28
[ 6.870940] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.27
[ 6.871268] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.26
[ 6.871430] [00005ea8] libusb: debug [libusb_unref_device] destroy device 2.0
DEBUGINFO:libusb_open_device_with_vid_pid: Fail

mcuee, 2023-08-29:
Please read here: did you install a  supported driver?
https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows

Janarthanan-S-03, 2023-08-30:
[image]
Installed the driver same issue exists.

mcuee, 2023-08-31:
What is the version of libusb are you using? Please use libusb-1.0.26 or git. 

I have a similar smart HUB on Windows 10 x64 and I can not reproduce the issue.
```
C:\work\libusb\libusb_bin\libusb-1.0.26-binaries\VS2015-x64\examples> .\xusb 0424:2740
Using libusb v1.0.26.11724

Opening device 0424:2740...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0424:2740
         bcdDevice: 0200
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: 1 caps
    USB 2.0 extension:
      attributes             : A21E

Reading first configuration descriptor:
              total length: 18
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.FF.FF

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Microchip Tech"
   String (0x02): "Hub Controller"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
   Failed: Pipe error
Releasing interface 0...
Closing device...
```

mcuee, 2023-08-31:
@Janarthanan-S-03 

Please post the full debug log of `xusb -d 0424:2906`. Thanks.

Janarthanan-S-03, 2023-08-31:
Can please try getting the handle for the hub.

Attaching the logs:-

Aplication Code Example Version: 1.0.0
DEBUGINFO:libusb_open_device_with_vid_pid HUB VID:0x0424, PID:0x2906
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.066622] [00003f98] libusb: debug [libusb_get_device_list]
[ 0.073770] [000011c8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.100183] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [48]
[ 0.100433] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [4F]
[ 0.100608] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [50]
[ 0.100773] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.101224] [00003f98] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.101394] [00003f98] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.101453] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [1]
[ 0.102232] [00003f98] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.102356] [00003f98] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.102410] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [38]
[ 0.102832] [00003f98] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.102952] [00003f98] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.103005] [00003f98] libusb: debug [winusb_get_device_list] allocating new device for session [39]
[ 0.103396] [00003f98] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C31&SUBSYS_05CB1028&REV_04\3&11583659&1&A0' bus number 1
[ 0.103524] [00003f98] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C26&SUBSYS_05CB1028&REV_04\3&11583659&1&E8' bus number 2
[ 0.105112] [00003f98] libusb: debug [winusb_get_device_list] found existing device for session [38]
[ 0.105386] [00003f98] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.105467] [00003f98] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 687 bytes)
[ 0.105566] [00003f98] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 4): 'USB\VID_0C45&PID_64D2\5&512A3B2&0&4'
[ 0.106098] [00003f98] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3967BCE8&0&0' reports 13 ports
[ 0.106239] [00003f98] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3967BCE8&0&0'
[ 0.107676] [00003f98] libusb: debug [winusb_get_device_list] found existing device for session [39]
[ 0.108736] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.172551] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.226037] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.309271] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.373299] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.426498] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.496159] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.558989] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.611136] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.723321] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.819742] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.897299] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.979950] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.071909] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.206539] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.301806] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.362724] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.449323] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.537879] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.650649] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.733273] [00003f98] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.800386] [00003f98] libusb: info [init_device] 0xa5c:0x5800 found 1 configurations but device isn't configured, forcing current configuration to 1
[ 1.815692] [00003f98] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=0, 139 bytes)
[ 1.825385] [00003f98] libusb: debug [init_device] (bus: 2, addr: 3, depth: 2, port: 5): 'USB\VID_0A5C&PID_5800\0123456789ABCD'
[ 1.829790] [00003f98] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&1B07C554&0' reports 2 ports
[... cut ...]
[ 1.939246] [00003f98] libusb: debug [libusb_unref_device] destroy device 2.3
[ 1.940819] [00003f98] libusb: debug [libusb_unref_device] destroy device 1.21
[ 1.941238] [00003f98] libusb: debug [libusb_unref_device] destroy device 1.0
[ 1.942875] [00003f98] libusb: debug [libusb_unref_device] destroy device 2.2
[ 1.943679] [00003f98] libusb: debug [libusb_unref_device] destroy device 2.1
[ 1.944882] [00003f98] libusb: debug [libusb_unref_device] destroy device 2.0
DEBUGINFO:libusb_open_device_with_vid_pid: Fail

Janarthanan-S-03, 2023-08-31:
Can you please send the code your using.

Thanks,
Janarthanan

Janarthanan-S-03, 2023-08-31:
Attaching the Logs

C:\Perforce\Karpagam_P4V\depot_dsg\Apps Group\src\Application Code Example\USB249xx\Vs2015\libusb-1.0.23\Win32\Debug\examples>xusb -d 0424:2734
Using libusb v1.0.23.11470

 

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000009] [00007f10] libusb: debug [libusb_init] created default context
[ 0.002658] [00007f10] libusb: debug [libusb_init] libusb v1.0.23.11470
[ 0.003374] [00007f10] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.003577] [00007f10] libusb: debug [windows_init_clock] hires timer frequency: 10000000 Hz
[ 0.003695] [00007f10] libusb: debug [htab_create] using 1021 entries hash table
[ 0.030939] [00007f10] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.032747] [00007f10] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.037288] [00007f10] libusb: info [windows_init] UsbDk backend is not available
[ 0.037460] [00007f10] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 0424:2734...
[ 0.039568] [00007f10] libusb: debug [libusb_get_device_list]
[ 0.052296] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [3B]
[ 0.052494] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [3D]
[ 0.052886] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [60]
[ 0.054421] [00007f10] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.054540] [00007f10] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.054846] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [A]
[ 0.055228] [00007f10] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.055313] [00007f10] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.055603] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.056091] [00007f10] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.056200] [00007f10] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.056497] [00007f10] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.056537] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [4B]
[ 0.056980] [00007f10] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.057230] [00007f10] libusb: debug [get_api_type] matched driver name against HID API
[ 0.057434] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [4C]
[ 0.058092] [00007f10] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.058264] [00007f10] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.058476] [00007f10] libusb: debug [winusb_get_device_list] allocating new device for session [4D]
[ 0.058917] [00007f10] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9DED&SUBSYS_507517AA&REV_30\3&11583659&1&A0' bus number 1
[ 0.059725] [00007f10] libusb: debug [winusb_get_device_list] found existing device for session [4C]
[ 0.059891] [00007f10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.060084] [00007f10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.060470] [00007f10] libusb: debug [init_device] (bus: 1, addr: 25, depth: 1, port: 1): 'USB\VID_413C&PID_301A\5&253C&0&1'
[ 0.060827] [00007f10] libusb: debug [winusb_get_device_list] extra GUID: {B35924D6-3E16-4A9E-9782-5524A4B79BAC}
[ 0.060950] [00007f10] libusb: debug [winusb_get_device_list] found existing device for session [4D]
[ 0.061272] [00007f10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.061493] [00007f10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 18 bytes)
[ 0.061764] [00007f10] libusb: debug [init_device] (bus: 1, addr: 19, depth: 2, port: 5): 'USB\VID_0424&PID_49A0\6&D5AE27&1&5'
[ 0.062239] [00007f10] libusb: debug [winusb_get_device_list] found existing device for session [3D]
[ 0.062387] [00007f10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.062607] [00007f10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.062849] [00007f10] libusb: debug [init_device] (bus: 1, addr: 18, depth: 1, port: 2): 'USB\VID_0424&PID_4926\5&253C&0&2'
[ 0.063183] [00007f10] libusb: debug [winusb_get_device_list] found existing device for session [3C]
[ 0.063365] [00007f10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.063583] [00007f10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1026 bytes)
[ 0.063677] [00007f10] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 7): 'USB\VID_04CA&PID_7070\5&253C&0&7'
[ 0.064036] [00007f10] libusb: debug [winusb_get_device_list] found existing device for session [4B]
[ 0.064212] [00007f10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.064439] [00007f10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.064698] [00007f10] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 8): 'USB\VID_06CB&PID_00BE\79A422D1E505'
[ 0.065071] [00007f10] libusb: debug [winusb_get_device_list] extra GUID: {EC11E711-0B8D-458C-A879-289C367A3419}
[... cut ...]
[ 0.077910] [00007f10] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.078121] [00007f10] libusb: debug [libusb_unref_device] destroy device 1.18
[ 0.078397] [00007f10] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.078609] [00007f10] libusb: debug [libusb_unref_device] destroy device 1.0
  Failed.
[ 0.079083] [00007f10] libusb: debug [libusb_exit]
[ 0.079347] [00007f10] libusb: debug [libusb_exit] destroying default context
[ 0.079567] [00007f10] libusb: debug [usbi_remove_pollfd] remove fd 0

Janarthanan-S-03, 2023-09-01:
@mcuee  Installed and tried using libusb-1.0.26

C:\jana>xusb -d 0424:2906
Using libusb v1.0.26.11724

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000213] [00003f90] libusb: debug [libusb_init] created default context
[ 0.000346] [00003f90] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000453] [00003f90] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000DC events 0
[ 0.000555] [00003f90] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000642] [00003f90] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000D8 events 0
[ 0.000742] [00003f90] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000850] [00003f90] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006079] [00003f90] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.067029] [00003f90] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.073136] [00003f90] libusb: info [windows_init] UsbDk backend is not available
Opening device 0424:2906...
[ 0.073433] [00003f90] libusb: debug [libusb_get_device_list]
[ 0.074713] [00002d1c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.087802] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.088009] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.088166] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.088311] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.088946] [00003f90] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.089122] [00003f90] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.089204] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [1]
[ 0.089954] [00003f90] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.090068] [00003f90] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.090134] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [41]
[ 0.091302] [00003f90] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.091696] [00003f90] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.094115] [00003f90] libusb: debug [winusb_get_device_list] allocating new device for session [42]
[ 0.095485] [00003f90] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C31&SUBSYS_05CB1028&REV_04\3&11583659&1&A0' bus number 1
[ 0.097741] [00003f90] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C26&SUBSYS_05CB1028&REV_04\3&11583659&1&E8' bus number 2
[ 0.100361] [00003f90] libusb: debug [winusb_get_device_list] found existing device for session [41]
[ 0.100816] [00003f90] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.101170] [00003f90] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 687 bytes)
[ 0.101761] [00003f90] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 4): 'USB\VID_0C45&PID_64D2\5&512A3B2&0&4'
[ 0.103059] [00003f90] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3967BCE8&0&0' reports 13 ports
[ 0.103332] [00003f90] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3967BCE8&0&0'
[ 0.104356] [00003f90] libusb: debug [winusb_get_device_list] found existing device for session [42]
[ 0.104785] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.169588] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.232460] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.290066] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.348127] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.404057] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.465422] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.526499] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.588009] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.648783] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.706113] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.768589] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.830485] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.891780] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 0.953872] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.015800] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.078005] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[ 1.140677] [00003f90] libusb: debug [init_device] 0xa5c:0x5800 found 1 configurations (not configured)
[... cut ...]
[ 1.417045] [00003f90] libusb: debug [libusb_unref_device] destroy device 2.1
[ 1.417499] [00003f90] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 1.418627] [00003f90] libusb: debug [libusb_exit] destroying default context
[ 1.423312] [00002d1c] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 1.425120] [00003f90] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000D8
[ 1.425198] [00003f90] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000DC

The same issue exists.
Thanks,
Janarthanan

mcuee, 2023-09-01:
Now this is strange. 
Let's see if others have the idea how to debug further.

mcuee, 2023-09-01:
@Janarthanan-S-03 

For troubleshooting, can you use Zadig to install WinUSB driver to see if that helps? Thanks. You can also try libusbK driver.

Zadig 2.8 download:
https://github.com/pbatard/libwdi/releases/tag/v1.5.0

Janarthanan-S-03, 2023-09-01:
@mcuee tried using libusb-win32 driver,libusbk driver but the issue still exists.

Janarthanan-S-03, 2023-09-06:
Hi @mcuee  Any update on this issue>

mcuee, 2023-09-06:
> Hi @mcuee Any update on this issue?

Sorry but I have no ideas why this happens. Hopefully others can have better ideas.

In any case, why do you want to talk to this Hub? What is your goal?

Janarthanan-S-03, 2023-09-12:
In any case, why do you want to talk to this Hub? What is your goal?
--> To perform Register Read/Writes.

The issue still exists

tormodvolden, 2023-10-04:
Looking at the debug output I cannot tell which driver the device (1.5 is bus 1 address 5, from session 51) is detected as using (or is it just recognized as a hub early in the enumeration?) and I guess the lack of a suitable driver is why it cannot be opened, but I am also not sure what /should/ happen here. @mcuee can you post your debug log from the similar smart hub where it works? @sonatique can you spot from the log what goes wrong? This is one example where I find our debug output is a bit difficult to parse and generally lacking.

sonatique, 2023-10-04:
@tormodvolden 
Hi,
Here is what I can say as of now:
1) The sentence “unsupported API call for” is generic (I mean: statically defined in a constant string) and will be printed every time a given function pointer is null (her the “open” function pointer is null) (see windows_winusb.h line 96). Therefore this is not an reliable indication that something is wrong specifically with the driver, but only that this function is not supported by the “sub api” we assigned to this device during enumeration

2) This device was assigned the “USB_API_HUB” api value, because it is a hub. As you can see at line 2475 of file windows_winusb.c, “open” function pointer is NULL in this case. Basically it seems that the Windows backend does not provide any implementation for any function for HUB. In fact, even if “open” was implemented, it would be possible to do anything with the hub device, since all functions are NULL. I have to say that I have no idea why. Linux implementation is simpler and seems to provide “open” and other functions regardless of the type of device

I would say that maybe it would worth experimenting with what would happen if lines 2515 to 2529 would be used to replace lines 2473 to 2487 (i.e. using the same function pointers for a hub than for a regular device). Or maybe should this hub be considered as composite device?

mcuee, 2023-10-05:
> @mcuee can you post your debug log from the similar smart hub where it works?

@tormodvolden 
The Dell USB Type-C docking station is not with me now but I should be able to test this next Monday.

mcuee, 2023-10-05:
> This device was assigned the “USB_API_HUB” api value, because it is a hub. 

I think that should not be correct for this device.

Related to USB Hub under Windows but not the same issue -- for external USB HUB.
* https://github.com/libusb/libusb/issues/391

tormodvolden, 2023-10-05:
Thanks @sonatique, point 1 was relatively easy and point 2 was close to my guesses, but I couldn't see it explicitly in the log that the USB_API_HUB was assigned (and that it was not reassigned later). If you know the enumeration code well enough it is maybe obvious. But if the WinUSB.sys driver is installed for the device, as per the screenshot, why isn't the winusbx api assigned to it?

Janarthanan-S-03, 2023-10-05:
Do I need to try anything?

tormodvolden, 2023-10-05:
@Janarthanan-S-03 Can you compile libusb yourself (from git)? One small thing to get a bit more debug information is to uncomment `//#define ENUM_DEBUG`
in libusb/os/windows_winusb.c

Does the device show up in the Device Manager?

Janarthanan-S-03, 2023-10-05:
@tormodvolden  I have complied with the source. What operation do I need to perform?

Does the device show up in the Device Manager?
Yes 
[image]

tormodvolden, 2023-10-05:
> @tormodvolden I have complied with the source. What operation do I need to perform?

You can edit the file libusb/os/windows_winusb.c and on the line where you have:
`//#define ENUM_DEBUG`
(around line number 1570 somewhere) remove the two slashes. Then recompile and run xusb -d again.

> Does the device show up in the Device Manager? Yes

Yes, that was a stupid question. What I meant to ask was, where in Device Manager does it show up, e.g. under Hub devices or under WinUSB devices (or whatever they will call it)?

Janarthanan-S-03, 2023-10-05:
C:\jana>xusb -d 0424:2906
Using libusb v1.0.26.11724

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000238] [00003ba0] libusb: debug [libusb_init] created default context
[ 0.000378] [00003ba0] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000522] [00003ba0] libusb: debug [usbi_add_event_source] add HANDLE 000000FC events 0
[ 0.000661] [00003ba0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000786] [00003ba0] libusb: debug [usbi_add_event_source] add HANDLE 00000100 events 0
[ 0.000937] [00003ba0] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001054] [00003ba0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003584] [00003ba0] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.003828] [00003ba0] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.006427] [00003ba0] libusb: info [windows_init] UsbDk backend is not available
Opening device 0424:2906...
[ 0.006758] [00003ba0] libusb: debug [libusb_get_device_list]
[ 0.007939] [00001bdc] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.022123] [00003ba0] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.022788] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_2906\5&512A3B2&0&6
[ 0.022972] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.023117] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_8087&PID_8000\5&DE4DCAF&0&1
[ 0.023284] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.023398] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB20\4&1B07C554&0
[ 0.023753] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.026017] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\4&3967BCE8&0&0
[ 0.026213] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.026320] [00003ba0] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.026448] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_8087&PID_07DC\6&29D86918&0&3
[ 0.027664] [00003ba0] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.027969] [00003ba0] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.028361] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [1]
[ 0.029536] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0C45&PID_64D2\5&512A3B2&0&4
[ 0.030082] [00003ba0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.030283] [00003ba0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.030815] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [41]
[ 0.031421] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0A5C&PID_5800\0123456789ABCD
[ 0.032304] [00003ba0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032644] [00003ba0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.033185] [00003ba0] libusb: debug [winusb_get_device_list] allocating new device for session [42]
[ 0.033954] [00003ba0] libusb: debug [winusb_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.034498] [00003ba0] libusb: debug [winusb_get_device_list] PRO: PCI\VEN_8086&DEV_9C31&SUBSYS_05CB1028&REV_04\3&11583659&1&A0
[ 0.034940] [00003ba0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C31&SUBSYS_05CB1028&REV_04\3&11583659&1&A0' bus number 1
[ 0.035585] [00003ba0] libusb: debug [winusb_get_device_list] PRO: PCI\VEN_8086&DEV_9C26&SUBSYS_05CB1028&REV_04\3&11583659&1&E8
[ 0.036082] [00003ba0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C26&SUBSYS_05CB1028&REV_04\3&11583659&1&E8' bus number 2
[ 0.037080] [00003ba0] libusb: debug [winusb_get_device_list] #### PROCESSING GENs
[ 0.037784] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0A5C&PID_5800&MI_01\7&66DE6C9&0&0001
[ 0.038132] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0A5C&PID_5800&MI_00\7&66DE6C9&0&0000
[ 0.038583] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0C45&PID_64D2\5&512A3B2&0&4
[ 0.039270] [00003ba0] libusb: debug [winusb_get_device_list] found existing device for session [41]
[ 0.043320] [00003ba0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.043810] [00003ba0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 687 bytes)
[ 0.044665] [00003ba0] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 4): 'USB\VID_0C45&PID_64D2\5&512A3B2&0&4'
[ 0.045286] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\4&3967BCE8&0&0
[ 0.046512] [00003ba0] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&3967BCE8&0&0' reports 13 ports
[ 0.046683] [00003ba0] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&3967BCE8&0&0'
[ 0.047214] [00003ba0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0A5C&PID_5800\0123456789ABCD
[ 0.048024] [00003ba0] libusb: debug [winusb_get_device_list] found existing device for session [42]
[... cut ...]
[ 1.425800] [00003ba0] libusb: debug [libusb_unref_device] destroy device 2.1
[ 1.426358] [00003ba0] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 1.427361] [00003ba0] libusb: debug [libusb_exit] destroying default context
[ 1.427939] [00001bdc] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 1.429278] [00003ba0] libusb: debug [usbi_remove_event_source] remove HANDLE 00000100
[ 1.429388] [00003ba0] libusb: debug [usbi_remove_event_source] remove HANDLE 000000FC

Janarthanan-S-03, 2023-10-05:
[image]

tormodvolden, 2023-10-05:
So it shows up under "Universal Serial Bus devices" but as "USB2906 Smart Hub", thanks.

mcuee, 2023-10-09:
> > @mcuee can you post your debug log from the similar smart hub where it works?
> 
> @tormodvolden The Dell USB Type-C docking station is not with me now but I should be able to test this next Monday.

This is what I have.
```
C:\work\libusb\libusb [master ≡ +3 ~0 -0 !]> .\examples\xusb 0424:2740
Using libusb v1.0.26.11807

Opening device 0424:2740...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0424:2740
         bcdDevice: 0200
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: 1 caps
    USB 2.0 extension:
      attributes             : A21E

Reading first configuration descriptor:
              total length: 18
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.FF.FF

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Microchip Tech"
   String (0x02): "Hub Controller"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
   Failed: Pipe error
Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
C:\work\libusb\libusb [master ≡ +3 ~0 -0 !]> .\examples\xusb -d 0424:2740
Using libusb v1.0.26.11807

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000262] [00004db0] libusb: debug [libusb_init_context] created default context
[ 0.000355] [00004db0] libusb: debug [libusb_init_context] libusb v1.0.26.11807
[ 0.000441] [00004db0] libusb: debug [usbi_add_event_source] add HANDLE 000000000000009c events 0
[ 0.000526] [00004db0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000602] [00004db0] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.000700] [00004db0] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.000772] [00004db0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.009462] [00004db0] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.017370] [00004db0] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.025000] [00004db0] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[... 373 lines cut ...]
[ 0.162531] [00004db0] libusb: debug [libusb_release_interface] interface 0
[ 0.162600] [00004db0] libusb: debug [auto_release] auto-released interface 0
[ 0.162722] [00004db0] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.162795] [00004db0] libusb: debug [usbi_handle_transfer_completion] transfer 00000166a061ec80 has callback 00007ff91e359430
[ 0.162887] [00004db0] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.162956] [00004db0] libusb: debug [libusb_free_transfer] transfer 00000166a061ec80
   Failed: Pipe error
Reading interface association descriptors (IADs) for first configuration:
[ 0.171910] [00004db0] libusb: debug [libusb_get_interface_association_descriptors] IADs for config index 0
    nb IADs: 0

Releasing interface 0...
[ 0.174275] [00004db0] libusb: debug [libusb_release_interface] interface 0
Closing device...
[ 0.176200] [00004db0] libusb: debug [libusb_close]
[ 0.176301] [00004db0] libusb: debug [libusb_unref_device] destroy device 3.21
[ 0.176375] [00004db0] libusb: debug [libusb_unref_device] destroy device 3.8
[ 0.176451] [00004db0] libusb: debug [libusb_unref_device] destroy device 3.6
[ 0.176518] [00004db0] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.176589] [00004db0] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.176675] [00004db0] libusb: debug [libusb_exit] destroying default context
[ 0.176755] [0000573c] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.178305] [00004db0] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
[ 0.178402] [00004db0] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000009c
```

mcuee, 2023-10-09:
> [image]

My device is on the external Dell Type C docking station.

What is your device? Is it built-in with the computer or an external device?

tormodvolden, 2023-10-09:
@mcuee So yours is under "Universal Serial Bus devices" as a "Hub Controller". What is listed under the Drivers tab? I see that in the enumeration the usbccgp driver is detected and the Composite API is assigned.

mcuee, 2023-10-09:
> @mcuee So yours is under "Universal Serial Bus devices" as a "Hub Controller". What is listed under the Drivers tab? I see that in the enumeration the usbccgp driver is detected and the Composite API is assigned.

[image]

tormodvolden, 2023-10-09:
This is how I read the debug log, backwards.

The device that is opened:
```
[ 0.095376] [00004db0] libusb: debug [libusb_open] open 3.21
```
This is bus 3 and addr 21 and matches your VID:PID:
```
[ 0.076721] [00004db0] libusb: debug [winusb_get_device_list] found existing device for session [8D]
[ 0.076898] [00004db0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.076987] [00004db0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 18 bytes)
[ 0.077090] [00004db0] libusb: debug [init_device] (bus: 3, addr: 21, depth: 4, port: 5): 'USB\VID_0424&PID_2740\D&127B8A6A&0&5'
```
This matches the session ID:
```
[ 0.068687] [00004db0] libusb: debug [winusb_get_device_list] allocating new device for session [8D]
[ 0.069194] [00004db0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.069479] [00004db0] libusb: debug [get_api_type] matched driver name against Composite API
```
But I might read it wrongly.

mcuee, 2023-10-09:
@tormodvolden 

Hmm, I think you are right. This is very strange...

mcuee, 2023-10-09:
> If you select View -> "Devices by connection", or use another tool like USB Device Viewer, does it belong to a composite device?

Unlikely. But I can only test this on Wednesday as I do not have acces to the docking station right now.

tormodvolden, 2023-10-09:
The ENUM_DEBUG helped to see what is going on. I think we should add it by default.

What seems to happen is that the api is first detected as hub, then in the GEN pass the extra GUID is discovered and added as an EXT pass. In this EXT pass the api is detected as WinUSB, however it never gets set in the dev-priv, because that would only happen in the previous HUB and DEV passes (see "// Setup device").

tormodvolden, 2023-10-09:
@mcuee and @Janarthanan-S-03 please try these patches (or simply run the included xusb.exe which is statically linked with patched libusb). It is probably not an ideal fix, but it might do something.

[libusb-issue1302-v1.zip](https://github.com/libusb/libusb/files/12850735/libusb-issue1302-v1.zip)

mcuee, 2023-10-09:
> The ENUM_DEBUG helped to see what is going on. I think we should add it by default.

I agree. 

> What seems to happen is that the api is first detected as hub, then in the GEN pass the extra GUID is discovered and added as an EXT pass. In this EXT pass the api is detected as WinUSB, however it never gets set in the dev-priv, because that would only happen in the previous HUB and DEV passes (see "// Setup device").
>
> @mcuee and @Janarthanan-S-03 please try these patches (or simply run the included xusb.exe which is statically linked with patched libusb). It is probably not an ideal fix, but it might do something.
> 
> [libusb-issue1302-v1.zip](https://github.com/libusb/libusb/files/12850735/libusb-issue1302-v1.zip)

I will try this tommorrow when I have access to the docking station.

mcuee, 2023-10-10:
Since I do not have the docking station with me today, I use an USB 2.0 hub and change the driver to WinUSB. Interestingly libusb git main, libusb-1.0.22 release onwards failed to open the device. libusb-1.0.21 release and libusb-1.0.20 release are able to open the device.

1) libusb git HEAD debug log.

<details>
<summary>git HEAD debug log</summary>

```
PS C:\work\libusb\libusb\build\v143\x64\Release> .\xusb.exe -d 0x1a40:0x0101
Using libusb v1.0.26.11791

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000278] [00002e38] libusb: debug [libusb_init_context] created default context
[ 0.000442] [00002e38] libusb: debug [libusb_init_context] libusb v1.0.26.11791
[ 0.000579] [00002e38] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000AC events 0
[ 0.000717] [00002e38] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000870] [00002e38] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000B0 events 0
[ 0.001032] [00002e38] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001175] [00002e38] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003288] [00002e38] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005414] [00002e38] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007412] [00002e38] libusb: info [windows_init] UsbDk backend is not available
Opening device 1A40:0101...
[ 0.007734] [00002e38] libusb: debug [libusb_get_device_list]
[ 0.008016] [00004500] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.018143] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.018329] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.018514] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.018685] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [8]
[ 0.018857] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [9]
[ 0.019517] [00002e38] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.019710] [00002e38] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.019819] [00002e38] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.019860] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [27]
[ 0.020074] [00002e38] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.020191] [00002e38] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.020280] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [4B]
[ 0.020493] [00002e38] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.020601] [00002e38] libusb: debug [get_api_type] matched driver name against HID API
[ 0.020702] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [75]
[ 0.020814] [00002e38] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.020919] [00002e38] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.021026] [00002e38] libusb: debug [winusb_get_device_list] allocating new device for session [76]
[ 0.021292] [00002e38] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.021439] [00002e38] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.022160] [00002e38] libusb: debug [winusb_get_device_list] found existing device for session [5]
[ 0.022360] [00002e38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.022458] [00002e38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.022588] [00002e38] libusb: debug [init_device] (bus: 2, addr: 3, depth: 1, port: 2): 'USB\VID_05E3&PID_0612\5&1A00B1C&0&2'
[ 0.022778] [00002e38] libusb: debug [winusb_get_device_list] found existing device for session [4B]
[ 0.022911] [00002e38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.023000] [00002e38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.023071] [00002e38] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.023208] [00002e38] libusb: debug [winusb_get_device_list] found existing device for session [75]
[ 0.023294] [00002e38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.023331] [00002e38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.023373] [00002e38] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 3): 'USB\VID_1EA7&PID_0064\5&586B51A&0&3'
[ 0.023457] [00002e38] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.023493] [00002e38] libusb: debug [winusb_get_device_list] found existing device for session [76]
[ 0.023552] [00002e38] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.023588] [00002e38] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.023628] [00002e38] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[... cut ...]
[ 0.020123] [00003e4c] libusb: debug [libusb_unref_device] destroy device 1.28
[ 0.020123] [00003e4c] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.020123] [00003e4c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.020123] [00003e4c] libusb: debug [libusb_exit]
[ 0.020123] [00003e4c] libusb: debug [libusb_exit] destroying default context
[ 0.020123] [00003e4c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.020123] [000006b4] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

</details>

mcuee, 2023-10-10:
BTW, libusbK klist is able to open the device.
```
PS C:\libusbK-dev-kit_release> .\kList.exe

Loading USB ID's maintained by Stephen J. Gowdy <linux.usb.ids@gmail.com>..

 1. USB 2.0 Hub (TERMINUS TECHNOLOGY INC.) [Connected]
    Service              : WinUSB
    ClassGUID            : {88BAE032-5A81-49F0-BC3D-A4FF138216D6}
    DeviceID             : USB\VID_1A40&PID_0101\5&586B51A&0&1
    DeviceInterfaceGUID  : {2A34E4EF-F044-1A2F-11F6-19A7CCE144AA}
    SymbolicLink         : \\?\usb#vid_1a40&pid_0101#5&586b51a&0&1#{2a34e4ef-f044-1a2f-11f6-19a7cce144aa}
    DevicePath           : \\?\usb#vid_1a40&pid_0101#5&586b51a&0&1#{2a34e4ef-f044-1a2f-11f6-19a7cce144aa}
    SerialNumber         : 5&586B51A&0&1
    BusNumber            : 0
    DeviceAddress        : 1

Select device (1-1) :1

Loading driver api..
Getting descriptors..

-Device:
  bLength             :18
  bDescriptorType     :0x01
  bcdUSB              :0x0200
  bDeviceClass        :0x09 (Hub)
  bDeviceSubClass     :0x00 (Unused)
  bDeviceProtocol     :0x01 (Single TT)
  bMaxPacketSize0     :64
  idVendor            :0x1A40 (Terminus Technology Inc.)
  idProduct           :0x0101 (Hub)
  bcdDevice           :0x0111
  iManufacturer       :0
  iProduct            :1
  iSerialNumber       :0
  bNumConfigurations  :1
!End Device
-Configuration:
  bLength             :9
  bDescriptorType     :0x02
  wTotalLength        :25
  bNumInterfaces      :1
  bConfigurationValue :0x01
  iConfiguration      :0
  bmAttributes        :0xE0 (Self-Powered)
  MaxPower            :50 (100ma)
 -Interface:
    bLength             :9
    bDescriptorType     :0x04
    bInterfaceNumber    :0x00
    bAlternateSetting   :0x00
    bNumEndpoints       :1
    bInterfaceClass     :0x09 (Hub)
    bInterfaceSubClass  :0x00 (Unused)
    bInterfaceProtocol  :0x00 (Full speed (or root) hub)
    iInterface          :0
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x81
      bmAttributes        :0x03 (Interrupt)
      wMaxPacketSize      :1
      bInterval           :0x0C
   !End Interface
 !End Configuration
```

mcuee, 2023-10-10:
The commit already failed.

<details>
<summary>libusb-1.0.21.11255 debug log</summary>

```
 MINGW64 /c/work/libusb/libusb
$ ./examples/xusb.exe -d 0x1a40:0x0101
Using libusb v1.0.21.11255

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [0000243c] libusb: debug [libusb_init] created default context
[ 0.000000] [0000243c] libusb: debug [libusb_init] libusb v1.0.21.11255
[ 0.000000] [0000243c] libusb: debug [get_windows_version] Windows 11 or later 64-bit
[ 0.000000] [0000243c] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.010582] [0000243c] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.010582] [0000243c] libusb: debug [winusbx_init] libusbK version: 3.1.0.0
[ 0.010582] [0000243c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.010582] [0000243c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.010582] [0000243c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.010582] [0000243c] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.010582] [0000243c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.010582] [0000243c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.010582] [0000243c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 1A40:0101...
[ 0.010582] [0000243c] libusb: debug [libusb_get_device_list]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_05E3&PID_0612\5&1A00B1C&0&2
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] found new PnP enumerator string 'USB'
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [5]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\ROOT_HUB30\4&12C539F&0&0
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [6]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\ROOT_HUB30\4&1A63DC7C&0&0
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [7]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_05E3&PID_0610\5&586B51A&0&2
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [8]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_1A40&PID_0101\5&586B51A&0&1
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [9]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_1C7A&PID_0575\077E2F9A
[ 0.020111] [0000243c] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.020111] [0000243c] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.020111] [0000243c] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [27]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_8087&PID_0026\5&586B51A&0&10
[ 0.020111] [0000243c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.020111] [0000243c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [4B]
[ 0.020111] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_1EA7&PID_0064\5&586B51A&0&3
[ 0.030119] [0000243c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.030119] [0000243c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [75]
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_04F2&PID_B6DD\0001
[ 0.030119] [0000243c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.030119] [0000243c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] allocating new device for session [76]
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] PRO: PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0
[ 0.030119] [0000243c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] PRO: PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68
[ 0.030119] [0000243c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] #### PROCESSING GENs
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] PRO: USB\VID_05E3&PID_0612\5&1A00B1C&0&2
[ 0.030119] [0000243c] libusb: debug [windows_get_device_list] found existing device for session [5]
[ 0.030119] [0000243c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[... cut ...]
[ 0.040127] [0000243c] libusb: debug [libusb_unref_device] destroy device 1.56
[ 0.040127] [0000243c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.040127] [0000243c] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 0.040127] [0000243c] libusb: debug [libusb_exit]
[ 0.040127] [0000243c] libusb: debug [libusb_exit] destroying default context
[ 0.040127] [0000243c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.040127] [0000478c] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

mcuee, 2023-10-10:
This commit is still good: https://github.com/libusb/libusb/commit/0ccf7eaaae893f24162ab97f7e8f244407bf8237

```
MINGW64 /c/work/libusb/libusb
$ ./examples/xusb.exe -d 0x1a40:0x0101
Using libusb v1.0.21.11199

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000037b4] libusb: debug [libusb_init] created default context
[ 0.000000] [000037b4] libusb: debug [libusb_init] libusb v1.0.21.11199
[ 0.000000] [000037b4] libusb: debug [windows_init] Windows 11 or later 64-bit
[ 0.000000] [000037b4] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.009967] [000037b4] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.009967] [000037b4] libusb: debug [winusbx_init] libusbK version: 3.1.0.0
[ 0.009967] [000037b4] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.009967] [000037b4] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.009967] [000037b4] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.009967] [000037b4] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.009967] [000037b4] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.009967] [000037b4] libusb: debug [htab_create] using 1021 entries hash table
[ 0.009967] [000037b4] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 1A40:0101...
[ 0.009967] [000037b4] libusb: debug [libusb_get_device_list]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\PCI#VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20#3&11583659&0&A0
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [343]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\PCI#VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01#3&11583659&0&68
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [3D8]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2
[ 0.009967] [000037b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009967] [000037b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [159]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#ROOT_HUB30#4&12C539F&0&0
[ 0.009967] [000037b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009967] [000037b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [1FB]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#ROOT_HUB30#4&1A63DC7C&0&0
[ 0.009967] [000037b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009967] [000037b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [2DC]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0610#5&586B51A&0&2
[ 0.009967] [000037b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009967] [000037b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [35C]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_1A40&PID_0101#5&586B51A&0&1
[ 0.009967] [000037b4] libusb: debug [get_api_type] driver(s): libusbK
[ 0.009967] [000037b4] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [F9]
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] #### PROCESSING GENs
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] found existing device for session [159] (0.0)
[ 0.009967] [000037b4] libusb: debug [init_device] got bus number from ancestor #2
[ 0.009967] [000037b4] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009967] [000037b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.009967] [000037b4] libusb: debug [init_device] (bus: 2, addr: 4, depth: 1, port: 2): '\\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2'
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_8087&PID_0026#5&586B51A&0&10
[ 0.009967] [000037b4] libusb: debug [windows_get_device_list] allocating new device for session [3F1]
[ 0.009967] [000037b4] libusb: debug [init_device] got bus number from ancestor #2
[ 0.009967] [000037b4] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009967] [000037b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.009967] [000037b4] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 10): '\\.\USB#VID_8087&PID_0026#5&586B51A&0&10'
[... cut ...]
[ 0.049957] [000037b4] libusb: debug [libusb_unref_device] destroy device 1.57
[ 0.049957] [000037b4] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.049957] [000037b4] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.049957] [000037b4] libusb: debug [libusb_exit]
[ 0.049957] [000037b4] libusb: debug [libusb_exit] destroying default context
[ 0.049957] [000037b4] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.049957] [00002520] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

mcuee, 2023-10-10:
I think I can quickly zoom down to the commits which breaks the external hub support. But I am not so sure whether it is easy to add it back. Let's see.

mcuee, 2023-10-10:
This commit is good: https://github.com/libusb/libusb/commit/0b3d4c68d87a4ffda2a4e0ff86983f69599048ce

<details>
<summary>libusb-1.0.21.11239 debug log</summary>

```
$ ./examples/xusb.exe -d 0x1a40:0x0101
Using libusb v1.0.21.11239

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00004704] libusb: debug [libusb_init] created default context
[ 0.000000] [00004704] libusb: debug [libusb_init] libusb v1.0.21.11239
[ 0.000000] [00004704] libusb: debug [get_windows_version] Windows 11 or later 64-bit
[ 0.000000] [00004704] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.009999] [00004704] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.009999] [00004704] libusb: debug [winusbx_init] libusbK version: 3.1.0.0
[ 0.009999] [00004704] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.009999] [00004704] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.009999] [00004704] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.009999] [00004704] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.009999] [00004704] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.009999] [00004704] libusb: debug [htab_create] using 1021 entries hash table
[ 0.009999] [00004704] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 1A40:0101...
[ 0.009999] [00004704] libusb: debug [libusb_get_device_list]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\PCI#VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20#3&11583659&0&A0
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [343]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\PCI#VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01#3&11583659&0&68
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [3D8]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2
[ 0.009999] [00004704] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009999] [00004704] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [159]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#ROOT_HUB30#4&12C539F&0&0
[ 0.009999] [00004704] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009999] [00004704] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [1FB]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#ROOT_HUB30#4&1A63DC7C&0&0
[ 0.009999] [00004704] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009999] [00004704] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [2DC]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0610#5&586B51A&0&2
[ 0.009999] [00004704] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009999] [00004704] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [35C]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_1A40&PID_0101#5&586B51A&0&1
[ 0.009999] [00004704] libusb: debug [get_api_type] driver(s): libusbK
[ 0.009999] [00004704] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [F9]
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] #### PROCESSING GENs
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] found existing device for session [159] (0.0)
[ 0.009999] [00004704] libusb: debug [init_device] got bus number from ancestor #2
[ 0.009999] [00004704] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009999] [00004704] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.009999] [00004704] libusb: debug [init_device] (bus: 2, addr: 4, depth: 1, port: 2): '\\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2'
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_8087&PID_0026#5&586B51A&0&10
[ 0.009999] [00004704] libusb: debug [windows_get_device_list] allocating new device for session [3F1]
[ 0.009999] [00004704] libusb: debug [init_device] got bus number from ancestor #2
[ 0.009999] [00004704] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009999] [00004704] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.009999] [00004704] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 10): '\\.\USB#VID_8087&PID_0026#5&586B51A&0&10'
[... cut ...]
[ 0.050786] [00004704] libusb: debug [libusb_unref_device] destroy device 1.57
[ 0.050786] [00004704] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.050786] [00004704] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.050786] [00004704] libusb: debug [libusb_exit]
[ 0.050786] [00004704] libusb: debug [libusb_exit] destroying default context
[ 0.050786] [00004704] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.050786] [000005c0] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

</details>

mcuee, 2023-10-10:
> Further zoom down: [8d248ab](https://github.com/libusb/libusb/commit/8d248ab899e286aefa73c5700a23b3585d17b916) is bad already. Maybe this is the one. I will check later.
> 
> libusb-1.0.21.11243 debug log

Indeed that paticular commit is the culprit.

The one commit before it (libusb-1.0.21.11242) is good.

<details>
<summary>libusb-1.0.21.11242 debug log</summary>

```
$ ./examples/xusb.exe -d 0x1a40:0x0101
Using libusb v1.0.21.11242

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000029a8] libusb: debug [libusb_init] created default context
[ 0.000000] [000029a8] libusb: debug [libusb_init] libusb v1.0.21.11242
[ 0.000000] [000029a8] libusb: debug [get_windows_version] Windows 11 or later 64-bit
[ 0.000000] [000029a8] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.010036] [000029a8] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.010036] [000029a8] libusb: debug [winusbx_init] libusbK version: 3.1.0.0
[ 0.010036] [000029a8] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.010036] [000029a8] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.010036] [000029a8] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.010036] [000029a8] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.010036] [000029a8] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.010036] [000029a8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.010036] [000029a8] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 1A40:0101...
[ 0.010036] [000029a8] libusb: debug [libusb_get_device_list]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\PCI#VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20#3&11583659&0&A0
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [343]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\PCI#VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01#3&11583659&0&68
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [3D8]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2
[ 0.010036] [000029a8] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.010036] [000029a8] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [159]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#ROOT_HUB30#4&12C539F&0&0
[ 0.010036] [000029a8] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.010036] [000029a8] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [1FB]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#ROOT_HUB30#4&1A63DC7C&0&0
[ 0.010036] [000029a8] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.010036] [000029a8] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [2DC]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0610#5&586B51A&0&2
[ 0.010036] [000029a8] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.010036] [000029a8] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [35C]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_1A40&PID_0101#5&586B51A&0&1
[ 0.010036] [000029a8] libusb: debug [get_api_type] driver(s): libusbK
[ 0.010036] [000029a8] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [F9]
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] #### PROCESSING GENs
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] found existing device for session [159] (0.0)
[ 0.010036] [000029a8] libusb: debug [init_device] got bus number from ancestor #2
[ 0.010036] [000029a8] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.010036] [000029a8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.010036] [000029a8] libusb: debug [init_device] (bus: 2, addr: 4, depth: 1, port: 2): '\\.\USB#VID_05E3&PID_0612#5&1A00B1C&0&2'
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] PRO: \\.\USB#VID_8087&PID_0026#5&586B51A&0&10
[ 0.010036] [000029a8] libusb: debug [windows_get_device_list] allocating new device for session [3F1]
[ 0.010036] [000029a8] libusb: debug [init_device] got bus number from ancestor #2
[ 0.010036] [000029a8] libusb: debug [init_device] found 1 configurations (active conf: 1)
[... cut ...]
[ 0.050008] [000029a8] libusb: debug [libusb_unref_device] destroy device 1.57
[ 0.050008] [000029a8] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.050008] [000029a8] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.050008] [000029a8] libusb: debug [libusb_exit]
[ 0.050008] [000029a8] libusb: debug [libusb_exit] destroying default context
[ 0.050008] [000029a8] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.050008] [0000212c] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

</details>

tormodvolden, 2023-10-10:
Thanks for bisecting this!

Trying to understand why my patch didn't help for you: There is an extra GUID 2A34E4EF* found, but I cannot see any EXT pass for it. Is the libusb-issue1302-v1 debug log above truncated?

tormodvolden, 2023-10-10:
> Is the libusb-issue1302-v1 debug log above truncated?

No, I think xusb was crashing. Here is a new version.

[libusb-issue1302-v2.zip](https://github.com/libusb/libusb/files/12861590/libusb-issue1302-v2.zip)

mcuee, 2023-10-10:
> > Is the libusb-issue1302-v1 debug log above truncated?
> 
> No, I think xusb was crashing. Here is a new version: [libusb-issue1302-v2.zip](https://github.com/libusb/libusb/files/12861934/libusb-issue1302-v2.zip)

Great. This new version works fine.

<details>
<summary>libusb-issue1302-v2 debug log</summary>

```
PS C:\work\libusb\libusb-issue1302v2> .\xusb.exe -d 0x1a40:0x0101
Using libusb v1.0.26.11807

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000543] [00001b40] libusb: debug [libusb_init_context] created default context
[ 0.000787] [00001b40] libusb: debug [libusb_init_context] libusb v1.0.26.11807_issue1302_v2
[ 0.001043] [00001b40] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000C0 events 0
[ 0.001290] [00001b40] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001522] [00001b40] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000C4 events 0
[ 0.001786] [00001b40] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001945] [00001b40] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004114] [00001b40] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.021854] [00001b40] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.024314] [00001b40] libusb: info [windows_init] UsbDk backend is not available
Opening device 1A40:0101...
[ 0.024930] [00001b40] libusb: debug [libusb_get_device_list]
[ 0.024964] [0000364c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.035506] [00001b40] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.035712] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_1A40&PID_0101\5&586B51A&0&1
[ 0.035897] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.036039] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_05E3&PID_0612\5&1A00B1C&0&2
[ 0.036196] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.036339] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\4&12C539F&0&0
[ 0.036495] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.036614] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\4&1A63DC7C&0&0
[ 0.036770] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [8]
[ 0.036909] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_05E3&PID_0610\5&586B51A&0&2
[ 0.037127] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [9]
[ 0.037575] [00001b40] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.037908] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_1C7A&PID_0575\077E2F9A
[ 0.038143] [00001b40] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.038308] [00001b40] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.038427] [00001b40] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.038548] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [28]
[ 0.038784] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_8087&PID_0026\5&586B51A&0&10
[ 0.039032] [00001b40] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.039259] [00001b40] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.039477] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [4C]
[ 0.039801] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_1EA7&PID_0064\5&586B51A&0&3
[ 0.039987] [00001b40] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.040100] [00001b40] libusb: debug [get_api_type] matched driver name against HID API
[ 0.040204] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [76]
[ 0.040357] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_04F2&PID_B6DD\0001
[ 0.040538] [00001b40] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.040647] [00001b40] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.040769] [00001b40] libusb: debug [winusb_get_device_list] allocating new device for session [77]
[ 0.040941] [00001b40] libusb: debug [winusb_get_device_list] #### PROCESSING HCDs {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.041232] [00001b40] libusb: debug [winusb_get_device_list] PRO: PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0
[ 0.041296] [00001b40] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.041362] [00001b40] libusb: debug [winusb_get_device_list] PRO: PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68
[ 0.041419] [00001b40] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.041660] [00001b40] libusb: debug [winusb_get_device_list] #### PROCESSING GENs
[ 0.042158] [00001b40] libusb: debug [winusb_get_device_list] PRO: USB\VID_05E3&PID_0612\5&1A00B1C&0&2
[ 0.042308] [00001b40] libusb: debug [winusb_get_device_list] found existing device for session [6]
[ 0.042425] [00001b40] libusb: debug [init_device] 0x5e3:0x612 found 1 configurations (current config: 1)
[... cut ...]
[ 0.058795] [0000364c] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.059254] [00001b40] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000C4
[ 0.059323] [00001b40] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000C0
[ 0.059372] [00001b40] libusb: warning [libusb_exit] device 1.1 still referenced
[ 0.059418] [00001b40] libusb: warning [libusb_exit] device 1.0 still referenced
[ 0.059463] [00001b40] libusb: warning [libusb_exit] device 1.62 still referenced
```

</details>

mcuee, 2023-10-10:
@Janarthanan-S-03 

Please help to give it a try on your device. Thanks.

mcuee, 2023-10-11:
> > If you select View -> "Devices by connection", or use another tool like USB Device Viewer, does it belong to a composite device?
> 
> Unlikely. But I can only test this on Wednesday as I do not have access to the docking station right now.

Hmm, I have the docking station with me now but it is not on the docking station. It may be on the external monitor. But I will not have the access to the monitor until next Monday.

tormodvolden, 2023-10-11:
> does it belong to a composite device?

I had previously a suspicion the issue affected composite devices, but it turned out to affect all hubs - once the hub API was assigned, it wouldn't be reassigned after a driver or GUID match. But I am still curious to see if the patch works for your "smart hub" as well.

mcuee, 2023-10-11:
> I had previously a suspicion the issue affected composite devices, but it turned out to affect all hubs - once the hub API was assigned, it wouldn't be reassigned after a driver or GUID match. But I [...]

I will carry out the test early next week.

mcuee, 2023-10-17:
> Hmm, I have the docking station with me now but it is not on the docking station. It may be on the external monitor. But I will not have the access to the monitor until next Monday.

@tormodvolden 

It is not part of the USB Composite Device.
[image]

Here is the debug log using your patched xusb.exe. There are some leaks in the end of the debug log. Please take a look.

<details>
<summary> debug log for the smart hub </summary>

```
C:\work\libusb\libusb-issue1302> .\xusb -d 0424:2740
Using libusb v1.0.26.11807

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000573] [000042f4] libusb: debug [libusb_init_context] created default context
[ 0.000852] [000042f4] libusb: debug [libusb_init_context] libusb v1.0.26.11807_issue1302_v2
[ 0.001132] [000042f4] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.001386] [000042f4] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001615] [000042f4] libusb: debug [usbi_add_event_source] add HANDLE 000000000000009C events 0
[ 0.001817] [000042f4] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001893] [000042f4] libusb: debug [htab_create] using 1021 entries hash table
[ 0.012182] [000042f4] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.021450] [000042f4] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.029710] [000042f4] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.030009] [000042f4] libusb: info [windows_init] UsbDk backend is not available
Opening device 0424:2740...
[ 0.030576] [000042f4] libusb: debug [libusb_get_device_list]
[ 0.030631] [000025d8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.065884] [000042f4] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.066778] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_5413\B&2EAF716D&0&3
[ 0.067113] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [BA]
[ 0.067360] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\9&AF4BBD4&0&0
[ 0.067666] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [BB]
[ 0.067873] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_5487\A&46F7C3F&0&2
[ 0.068118] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [BC]
[ 0.068285] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_0413\B&1637EEB3&0&3
[ 0.068520] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [BD]
[ 0.068746] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\4&36020D6F&0&0
[ 0.068977] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [BE]
[ 0.069179] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\7&2452366F&0&0
[ 0.069396] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [BF]
[ 0.069575] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_0487\A&46F7C3F&0&4
[ 0.069789] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [C0]
[ 0.070035] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_5744\C&1936BD58&0&2
[ 0.070291] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [C1]
[ 0.070448] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_2744\C&1F76A113&0&2
[ 0.070617] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [C2]
[ 0.070785] [000042f4] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.070998] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_8087&PID_0AAA\5&E9F3E45&0&14
[ 0.071426] [000042f4] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.071639] [000042f4] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.071811] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [1C]
[ 0.072258] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_402E\200901010001
[ 0.072793] [000042f4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.072968] [000042f4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.073080] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [90]
[ 0.073277] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_2717&PID_5013\5&E9F3E45&0&3
[ 0.073603] [000042f4] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.073737] [000042f4] libusb: debug [get_api_type] matched driver name against HID API
[ 0.073819] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [91]
[ 0.073926] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_046D&PID_C077\D&127B8A6A&0&4
[ 0.074355] [000042f4] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.074516] [000042f4] libusb: debug [get_api_type] matched driver name against HID API
[ 0.074664] [000042f4] libusb: debug [winusb_get_device_list] allocating new device for session [92]
[ 0.074859] [000042f4] libusb: debug [winusb_get_device_list] PRO: USB\VID_413C&PID_B06E\B&2EAF716D&0&5
[... cut ...]
[ 0.158672] [000042f4] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000098
[ 0.158876] [000042f4] libusb: warning [libusb_exit] device 3.15 still referenced
[ 0.159030] [000042f4] libusb: warning [libusb_exit] device 3.8 still referenced
[ 0.159176] [000042f4] libusb: warning [libusb_exit] device 3.4 still referenced
[ 0.159281] [000042f4] libusb: warning [libusb_exit] device 3.0 still referenced
[ 0.159421] [000042f4] libusb: warning [libusb_exit] device 3.6 still referenced
```

</details>

mcuee, 2023-10-17:
> I had previously a suspicion the issue affected composite devices, but it turned out to affect all hubs - once the hub API was assigned, it wouldn't be reassigned after a driver or GUID match. But I am still curious to see if the patch works for your "smart hub" as well.

Yes it still works. Please refer to the debug log posted above, especially the "device still referenced" warning messages.

The debug log from git HEAD is also attached here for reference.

```
C:\work\libusb\libusb [master ≡ +3 ~0 -0 !]> git diff
diff --git a/libusb/os/windows_winusb.c b/libusb/os/windows_winusb.c
index 93f2a4e8..e20bc285 100644
--- a/libusb/os/windows_winusb.c
+++ b/libusb/os/windows_winusb.c
@@ -1568,7 +1568,7 @@ static int winusb_get_device_list(struct libusb_context *ctx, struct discovered_
        }

        for (pass = 0; ((pass < nb_guids) && (r == LIBUSB_SUCCESS)); pass++) {
-//#define ENUM_DEBUG
+#define ENUM_DEBUG
 #if defined(ENABLE_LOGGING) && defined(ENUM_DEBUG)
                const char * const passname[] = {"HUB", "DEV", "HCD", "GEN", "HID", "EXT"};
                usbi_dbg(ctx, "#### PROCESSING %ss %s", passname[MIN(pass, EXT_PASS)], guid_to_string(guid_list[pass], guid_string));
```

<details>
<summary> debug log from git HEAD </summary>

```
C:\work\libusb\libusb [master ≡ +3 ~1 -0 !]> .\examples\xusb -d 0424:2740
Using libusb v1.0.26.11807

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000299] [00003b8c] libusb: debug [libusb_init_context] created default context
[ 0.000710] [00003b8c] libusb: debug [libusb_init_context] libusb v1.0.26.11807
[ 0.000804] [00003b8c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000a4 events 0
[ 0.000959] [00003b8c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001054] [00003b8c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000ac events 0
[ 0.001151] [00003b8c] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001231] [00003b8c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.011251] [00003b8c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.022747] [00003b8c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.030619] [00003b8c] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.030957] [00003b8c] libusb: info [windows_init] UsbDk backend is not available
O[ 0.031697] [000039d0] libusb: debug [windows_iocp_thread] I/O completion thread started
pening device 0424:2740...
[ 0.034341] [00003b8c] libusb: debug [libusb_get_device_list]
[ 0.066417] [00003b8c] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.067319] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_5413\B&2EAF716D&0&3
[ 0.067709] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [BA]
[ 0.068017] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\9&AF4BBD4&0&0
[ 0.068289] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [BB]
[ 0.068523] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_5487\A&46F7C3F&0&2
[ 0.068777] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [BC]
[ 0.068995] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_0413\B&1637EEB3&0&3
[ 0.069227] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [BD]
[ 0.069332] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\4&36020D6F&0&0
[ 0.069449] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [BE]
[ 0.069548] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\ROOT_HUB30\7&2452366F&0&0
[ 0.069667] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [BF]
[ 0.069766] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0BDA&PID_0487\A&46F7C3F&0&4
[ 0.069893] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [C0]
[ 0.069986] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_5744\C&1936BD58&0&2
[ 0.070115] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [C1]
[ 0.070206] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_2744\C&1F76A113&0&2
[ 0.070334] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [C2]
[ 0.070418] [00003b8c] libusb: debug [winusb_get_device_list] #### PROCESSING DEVs {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.070605] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_8087&PID_0AAA\5&E9F3E45&0&14
[ 0.070971] [00003b8c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.071123] [00003b8c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.071203] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [1C]
[... 481 lines cut ...]
[ 0.197593] [00003b8c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.197679] [00003b8c] libusb: debug [usbi_handle_transfer_completion] transfer 000002346f59ec80 has callback 00007ffbbbcf9430
[ 0.197769] [00003b8c] libusb: debug [sync_transfer_cb] actual_length=0
[... cut ...]
[ 0.205094] [00003b8c] libusb: debug [libusb_unref_device] destroy device 3.4
[ 0.205161] [00003b8c] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.205259] [00003b8c] libusb: debug [libusb_exit] destroying default context
[ 0.205540] [000039d0] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.207844] [00003b8c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000ac
[ 0.207958] [00003b8c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000a4
```

</details>

tormodvolden, 2023-10-17:
Thanks! So your smart hub 0424:2740, which always have worked, is not detected in the HUB pass, but detected with WinUSB in the DEV pass, and then again in its EXT pass. Since it detected correctly on the first shot, it doesn't matter if later matches reassigns the api or not.

The debug log is a bit easier to read (and I don't get confused about any composite driver involved!):
DEV:
```
[ 0.078888] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_2740\D&127B8A6A&0&5
[ 0.079506] [00003b8c] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.079822] [00003b8c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.080079] [00003b8c] libusb: debug [winusb_get_device_list] allocating new device for session [99]
```
GEN:
```
[ 0.088326] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_2740\D&127B8A6A&0&5
[ 0.088761] [00003b8c] libusb: debug [winusb_get_device_list] extra GUID: {b35924d6-3e16-4a9e-9782-5524a4b79bac}
[ 0.089053] [00003b8c] libusb: debug [winusb_get_device_list] found existing device for session [99]
[ 0.089395] [00003b8c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.089568] [00003b8c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 18 bytes)
[ 0.089750] [00003b8c] libusb: debug [init_device] (bus: 3, addr: 19, depth: 4, port: 5): 'USB\VID_0424&PID_2740\D&127B8A6A&0&5'
```
EXT:
```
[ 0.115329] [00003b8c] libusb: debug [winusb_get_device_list] #### PROCESSING EXTs {B35924D6-3E16-4A9E-9782-5524A4B79BAC}
[ 0.115835] [00003b8c] libusb: debug [winusb_get_device_list] PRO: USB\VID_0424&PID_2740\D&127B8A6A&0&5
[ 0.116304] [00003b8c] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.116414] [00003b8c] libusb: debug [get_api_type] matched driver name against WinUSB
```

tormodvolden, 2023-10-17:
I am not surprised by the leaks seen in the patched version. It is not a clean patch, it is just to see what can be done. We have some pending merge requests on the surrounding code, so I will rather continue on this after they have been merged and after having adding some more log verbosity, and hopefully find a proper way to fix it then.

It would be good if @Janarthanan-S-03 can test the patched version too.

mcuee, 2023-12-28:
Another log: the patched version works fine for a external USB hub using WinUSB driver (replaced the existing USB Hub driver with WinUSB using Zadig)

```
C:\work\libusb\libusb_issue1302\libusb-issue1302> .\xusb 05e3:0608
Using libusb v1.0.26.11807

Opening device 05E3:0608...

Reading device descriptor:
            length: 18
      device class: 9
               S/N: 0
           VID:PID: 05E3:0608
         bcdDevice: 8536
   iMan:iProd:iSer: 0:1:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 25
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 09.00.00
       endpoint[0].address: 81
           max packet size: 0001
          polling interval: 0C

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "USB2.0 Hub"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
libusb: warning [libusb_exit] device 1.0 still referenced
libusb: warning [libusb_exit] device 1.11 still referenced
```

git main or libusb-1.0.27-rc1 will fail.
```
C:\work\libusb\libusb-1.0.27-rc1\build\v142\x64\Release> .\xusb 05e3:0608
Using libusb v1.0.27.11848

Opening device 05E3:0608...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
[... 354 lines cut ...]
[ 0.091651] [00005320] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.091699] [00005320] libusb: debug [libusb_unref_device] destroy device 1.6
[ 0.091751] [00005320] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.091803] [00005320] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.091849] [00005320] libusb: debug [libusb_unref_device] destroy device 1.12
[ 0.091897] [00005320] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.091951] [00005320] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.091998] [00005320] libusb: debug [libusb_unref_device] destroy device 2.7
[ 0.092043] [00005320] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.092087] [00005320] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.092144] [00005320] libusb: debug [libusb_unref_device] destroy device 1.10
[ 0.092189] [00005320] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.092232] [00005320] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.092276] [00005320] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.092337] [00005320] libusb: debug [libusb_unref_device] destroy device 2.11
[ 0.092385] [00005320] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.092432] [00005320] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 0.092527] [00005320] libusb: debug [libusb_exit] destroying default context
[ 0.092618] [00002bdc] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.094014] [00005320] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000A0
[ 0.094070] [00005320] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000009C
```

</details>

mcuee, 2024-02-15:
@tormodvolden 

My Dell WD19TB Thunderbolt Docking Station supports 10Gbps. Unfortunately I can not test PR #1428 under Windows because of this issue, even after I change the driver to WinUSB. Just wondering if you can help to see how I can apply your draft patch on top of PR #1428 (or better PR #1428 rebase to git).
* https://github.com/libusb/libusb/pull/1428#issuecomment-1931199388