libusb/libusb #1303: linux/android libusb_bulk_transfer return -11(LIBUSB_ERROR_NO_MEM), v1.0.26

gongjianbo, 2023-09-01:
libusb_bulk_transfer define:
`
int LIBUSB_CALL libusb_bulk_transfer(libusb_device_handle *dev_handle,
	unsigned char endpoint, unsigned char *data, int length,
	int *actual_length, unsigned int timeout);

`
or using libusb_fill_bulk_transfer + libusb_submit_transfer.
If I set the length to 18,374,720, it will return -11 on Ubuntu or Android. My memory is enough.
If executed on MacOS, data can be successfully read.

gongjianbo, 2023-09-01:
[ 9.628684] [00003af0] libusb: debug [libusb_submit_transfer] transfer 0x562adb4362f0
[ 9.628701] [00003af0] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 9.628722] [00003af0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 9.628724] [00003af0] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 9.632381] [00003af0] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 9.632436] [00003af0] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=1
[ 9.632443] [00003af0] libusb: debug [handle_control_completion] handling completion status 0
[ 9.632456] [00003af0] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 9.632469] [00003af0] libusb: debug [usbi_handle_transfer_completion] transfer 0x562adb4362f0 has callback 0x7f38b82c23a0
[ 9.632480] [00003af0] libusb: debug [sync_transfer_cb] actual_length=1
[ 9.632490] [00003af0] libusb: debug [libusb_free_transfer] transfer 0x562adb4362f0
[ 9.632509] [00003af0] libusb: debug [libusb_submit_transfer] transfer 0x562adb4362f0
[ 9.632518] [00003af0] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 18374720
[ 9.632528] [00003af0] libusb: debug [submit_bulk_transfer] first URB failed, easy peasy
[ 9.632534] [00003af0] libusb: debug [libusb_free_transfer] transfer 0x562adb4362f0

libusb_fill_bulk_transfer return -11, Insufficient memory

mcuee, 2023-09-01:
This may be a Linux/Android kernel version limitation.

What is the Ubuntu Linux version used? Are you using an x86/x64 machine or an embedded Linux machine? What is the Android device and the kernel version used?

gongjianbo, 2023-09-01:
> This may be a Linux/Android kernel version limitation.
> 
> What is the Ubuntu Linux version used? Are you using an x86/x64 machine or an embedded Linux machine? What is the Android device and the kernel version used?

Ubuntu18.04 desktop, and Android 11.

mcuee, 2023-09-01:
Ubuntu 18.04 desktop may be a bit old so you may want to try a more recent version to see if the issue is still there or not (say Ubuntu 22.04).

gongjianbo, 2023-09-01:
> Ubuntu 18.04 desktop may be a bit old so you may want to try a more recent version to see if the issue is still there or not (say Ubuntu 22.04).

Ubuntu22.04 is the same result.

mcuee, 2023-09-01:
There are some codes in libusb Linux usbfs backend which may be the reason.
https://github.com/libusb/libusb/blob/6bf2db6feaf3b611c9adedb6c4962a07f5cb07ae/libusb/os/linux_usbfs.c#L1921-L1974

@tormodvolden or @jwrdegoede if you are available.
Please help here as well. Thanks.

mcuee, 2023-09-01:
There are transfer size limit for Windows as well.
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-bandwidth-allocation

<html>
<body>

Transfer pipe | Windows 8.1, Windows 8 | Windows 7, Windows Vista | Windows XP, Windows Server 2003 | Windows 2000
-- | -- | -- | -- | --
Control | 64K for SuperSpeed and high speed (xHCI)4K for full and low speed (xHCI, EHCI, UHCI, OHCI)For UHCI, 4K on the default endpoint; 64K on non-default control pipes | 64K for high speed (EHCI)4K for full and low speed (EHCI, UHCI, OHCI)For UHCI, 4K on the default endpoint; 64K on non-default control pipes (UHCI) | 64K for high speed (EHCI)4K for full and low speed (EHCI, UHCI, OHCI)For UHCI, 4K on the default endpoint; 64K on non-default control pipes (UHCI) | 4K on the default endpoint; 64K on non-default control pipes (OHCI)
Interrupt | 4MB for SuperSpeed, high, full, and low speed (xHCI, EHCI, UHCI, OHCI) | 4MB for high, full, and low speed (EHCI, UHCI, OHCI) | Unlimited | Undetermined(OHCI)
Bulk | 32MB for SuperSpeed (xHCI)4MB for high and full speed (xHCI)4MB for high and full speed (EHCI and UHCI)256K full speed (OHCI) | 4MB for high and full speed (EHCI, UHCI)256K for full speed (OHCI) | 3MB for high and full speed (EHCI)Undetermined (UHCI)256K for full speed (OHCI) | Undetermined(OHCI)
Isochronous | 1024 * wBytesPerInterval for SuperSpeed (xHCI)(see USB_SUPERSPEED_ENDPOINT_COMPANION_DESCRIPTOR)1024 * MaximumPacketSize for high speed (xHCI, EHCI)256 * MaximumPacketSize for full speed (xHCI, EHCI)64K for full speed (UHCI, OHCI) | 1024 * MaximumPacketSize for high speed (EHCI)256 * MaximumPacketSize for full-speed (EHCI)64K for full speed (UHCI, OHCI) | 1024 * MaximumPacketSize for high-speed (EHCI)256 * MaximumPacketSize for full speed (EHCI)64K for full speed (UHCI, OHCI) | 64K for full speed (OHCI)

</body>
</html>
Transfer pipe	Windows 8.1, Windows 8	Windows 7, Windows Vista	Windows XP, Windows Server 2003	Windows 2000
Control	64K for SuperSpeed and high speed (xHCI)

4K for full and low speed (xHCI, EHCI, UHCI, OHCI)

For UHCI, 4K on the default endpoint; 64K on non-default control pipes	64K for high speed (EHCI)

4K for full and low speed (EHCI, UHCI, OHCI)

For UHCI, 4K on the default endpoint; 64K on non-default control pipes (UHCI)	64K for high speed (EHCI)

4K for full and low speed (EHCI, UHCI, OHCI)

For UHCI, 4K on the default endpoint; 64K on non-default control pipes (UHCI)	4K on the default endpoint; 64K on non-default control pipes (OHCI)
Interrupt	4MB for SuperSpeed, high, full, and low speed (xHCI, EHCI, UHCI, OHCI)	4MB for high, full, and low speed (EHCI, UHCI, OHCI)	Unlimited	Undetermined(OHCI)
Bulk	32MB for SuperSpeed (xHCI)

4MB for high and full speed (xHCI)

4MB for high and full speed (EHCI and UHCI)

256K full speed (OHCI)	4MB for high and full speed (EHCI, UHCI)

256K for full speed (OHCI)	3MB for high and full speed (EHCI)

Undetermined (UHCI)

256K for full speed (OHCI)	Undetermined(OHCI)
Isochronous	1024 * wBytesPerInterval for SuperSpeed (xHCI)
(see [USB_SUPERSPEED_ENDPOINT_COMPANION_DESCRIPTOR](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/usbspec/ns-usbspec-_usb_superspeed_endpoint_companion_descriptor))

1024 * MaximumPacketSize for high speed (xHCI, EHCI)

256 * MaximumPacketSize for full speed (xHCI, EHCI)

64K for full speed (UHCI, OHCI)	1024 * MaximumPacketSize for high speed (EHCI)

256 * MaximumPacketSize for full-speed (EHCI)

64K for full speed (UHCI, OHCI)	1024 * MaximumPacketSize for high-speed (EHCI)

256 * MaximumPacketSize for full speed (EHCI)

64K for full speed (UHCI, OHCI)	64K for full speed (OHCI)

mcuee, 2023-09-01:
@gongjianbo 

So in the end, it is probably not a good idea to submit large transfer size like 18,374,720. You may want to limit the size to be below 4MB.

gongjianbo, 2023-09-01:
> @gongjianbo
> 
> So in the end, it is probably not a good idea to submit large transfer size like 18,374,720. You may want to limit the size to be below 4MB, or even lower value to be compatible with different OS.

Can I complete the transfer of large size by calling libusb_bulk_transfer multiple times? I will test it next week.

mcuee, 2023-09-02:
> Can I complete the transfer of large size by calling libusb_bulk_transfer multiple times? I will test it next week.

Yes you can.

It will be even better you try to use the async API.
https://libusb.sourceforge.io/api-1.0/libusb_io.html

gongjianbo, 2023-09-04:
> > Can I complete the transfer of large size by calling libusb_bulk_transfer multiple times? I will test it next week.
> 
> Yes you can.
> 
> It will be even better you try to use the async API. https://libusb.sourceforge.io/api-1.0/libusb_io.html

By calling libusb_bulk_transfer( or libusb_fill_bulk_transfer ) multiple times, the large size can be successfully read. Thank you very much for your help!   

`int MyUSB::readBulk(unsigned char requestCode, unsigned char *bufPtr, int bufLen, unsigned short argValue, unsigned short argIndex)
{
    if (!mHandle) {
        return 0;
    }
    unsigned char arg_buf[1] = {0};
    const unsigned char arg_type = 0x00 | LIBUSB_REQUEST_TYPE_VENDOR;
    int usb_ret = 0;
    usb_ret = libusb_control_transfer(mHandle, arg_type, requestCode, argValue, argIndex, arg_buf, 1, 2000);
    if (usb_ret < 0) {
        LOGD("libusb_control_transfer error. %d, %s", usb_ret, libusb_strerror(usb_ret));
    }
    int actual = 0;
    int pack_len = 1024 * 1024 * 4;
    int pack_count = (bufLen + pack_len - 1) / pack_len;
    for (int i = 0; i < pack_count; i++)
    {
        if (i == pack_count - 1) {
            pack_len = bufLen - pack_len * i;
        }
        if (pack_count > 1) {
            LOGD("libusb transfer loop. %d/%d. has %d, request %d.", i, pack_count, actual, pack_len);
        }
        int once = 0;
        usb_ret = libusb_bulk_transfer(mHandle, 0x82, bufPtr + actual, pack_len, &once, 0);
        if (usb_ret < 0) {
            LOGD("libusb_bulk_transfer error. %d, %s", usb_ret, libusb_strerror(usb_ret));
            actual = 0;
            return 0;
        }
        actual += once;
    }
    return actual;
}`

closed 2023-09-04 by gongjianbo