libusb/libusb #1304: Detach event not making it from netlink to poll when using hotplug on Linux

joelreymont, 2023-09-06:
I see that libusb fails to detect device removal on Linux from time to time. 

This is what I see when it happens 

```
60.075687] [000004d2] libusb: debug [usbi_wait_for_events] poll() returned 0
[726.202826] [000004d3] libusb: debug [linux_netlink_parse] unknown device action unbind
[726.267862] [000004d3] libusb: debug [linux_netlink_parse] unknown device action unbind
[726.284230] [000004d3] libusb: debug [linux_netlink_parse] unknown device action unbind
[726.296603] [000004d3] libusb: debug [linux_netlink_parse] unknown device action unbind
[726.296860] [000004d3] libusb: debug [linux_netlink_read_message] netlink hotplug found device busnum: 1, devaddr: 34, sys_name: 1-1.3, removed: yes
```

The code just hangs at this point and does not react when I attach or detach the device again, save for netlink hotplug messages. Apparently, it's stuck in `poll`. The code does return when libusb _notices_ device removal, e.g.

```
[20.893889] [0000051a] libusb: debug [linux_netlink_read_message] netlink hotplug found device busnum: 1, devaddr: 36, sys_name: 1-1.3, removed: yes
[20.894221] [00000519] libusb: debug [usbi_wait_for_events] poll() returned 1
[20.894283] [00000519] libusb: debug [handle_event_trigger] event triggered
[20.894316] [00000519] libusb: debug [handle_event_trigger] hotplug message received
[20.894389] [00000519] libusb: debug [libusb_get_device_descriptor]
```

This is happening on my custom-built Yocto image for the Raspberry Pi

```
Linux coolbox 6.1.20-v8 #1 SMP PREEMPT Wed Mar 22 16:44:37 UTC 2023 aarch64 GNU/Linux
```

How can I troubleshoot this?

mcuee, 2023-09-07:
Does this happen in the normal Raspberry Pi OS?

mcuee, 2023-10-18:
Closing this as there is no update. 

Please re-open if you find issues with libusb. Thanks.

closed 2023-10-18 by mcuee