libusb/libusb #1306: Assessment of the difficulty in porting CPU architecture for libusb

wangyuliu, 2023-09-11:
Hello everyone! I am working on implementing a tool to assess the complexity of CPU architecture porting. It primarily focuses on RISC-V architecture porting. In fact, it can be estimated on average across various architectures.My focus is on the overall workload and difficulty of transplantation in the past and future.As part of my dataset, I have collected the **libusb** project. **I would like to gather community opinions to support my assessment. I appreciate your help and response!** Based on scanning tools, the porting complexity is determined to be simple, with a small amount of code related to the CPU architecture in the project. (referring to the overall workload from adapting the project to a specific architecture to achieving full functionality on that architecture). Is this assessment accurate? I look forward to your help and response.

mcuee, 2023-09-11:
If you have Linux USB stack working, and if you have libudev available (or you can use netlink using `--disable-udev`), then there should be no real porting efforts required.

For new Linux platforms, usually it is the issue with the Linux kernel USB stack and not libusb issue, based on the previous reports.

mcuee, 2023-09-19:
Close this issue for now.

If you got real issue during the porting, you can re-open this issue.

closed 2023-09-19 by mcuee

wangyuliu, 2023-12-07:
> If you have Linux USB stack working, and if you have libudev available (or you can use netlink using ), then there should be no real porting efforts required. There should be NO CPU architecture codes required.`--disable-udev`
> 
> For new Linux platforms, usually it is the issue with the Linux kernel USB stack and not libusb issue, based on the previous reports.

Thank you for your reply. I think the problem you raised is about the order of transplantation on the new CPU architecture, but the difficulty of transplanting libusb itself is very low. Can I understand it this way?