libusb/libusb #1307: libusb_open() returns LIBUSB_ERROR_NOT_FOUND

pazourek, 2023-09-12:
I am using a full-speed device with own firmware, 0x2846:1780. The device is a composite device with 4 simplified CDC-ACM and one DFU interface. I implemented WCID (Extended compat ID only) that automatically installs WinUSB driver for all device interfaces. That works for Win10 as well for Win11. I can see WinUSB of version 10.0.22621.608 is installed. 

I am able to read a device descriptor calling the libusb_get_device_descriptor(), but every attempt to open the device (libusb_open()) fails with an error LIBUSB_ERROR_NOT_FOUND (-5). I have also tried to disable WCID and install custom drivers that left the CDC-ACM interface to use default driver (usbser.sys), the DFU interface is used with libusb0 driver. Neither this configuration worked as expected. Version of Libusb is 1.0.26.11724, both x86 and amd64 binaries behave the same way.

Other devices using the WinUSB seems to work well.

[230912_WCID_beagle.txt](https://github.com/libusb/libusb/files/12586484/230912_WCID_beagle.txt)
[230912_WCID_open.txt](https://github.com/libusb/libusb/files/12586485/230912_WCID_open.txt)
[230912_WCID_treeview.txt](https://github.com/libusb/libusb/files/12586486/230912_WCID_treeview.txt)

pazourek, 2023-09-12:
I also tried to modify registry records (_Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_2846&PID_1780\01\Service_). I replaced the usbccgp driver by the WinUSB, so the WinUSB became be used on the top level parent node. This fixed the issue, now I can open the device. But it seem that WCID is not able to do such assignation. So using separate interfaces for automatic driver installation remains the only way.

mcuee, 2023-09-13:
I tend to think this is a duplicate of https://github.com/libusb/libusb/issues/1238.
* https://github.com/libusb/libusb/issues/1238

pazourek, 2023-10-10:
Digging around this issue I found out, the libusb is able to open any WinUsb interface of the composite device only if the device driver provides GUID. This also works with the automatic driver installation (WCID) if the device implements "Extended Properties OS Feature Descriptor" with one GUID. If the device reports more GUIDs, opening the device fails with the error LIBUSB_ERROR_NOT_FOUND (-5). 

The issue seems to be in reading the multiple GUIDs in the winusb_get_device_list() function. Each GUID is 39 bytes long, but storing two GUIDs require at least 79 bytes. But the internal buffer is only 40 bytes long.

Youw, 2023-10-10:
> After fixing that

Anything you can contribute in a PR?

mcuee, 2023-10-10:
@pazourek 

Thanks for the updates. I have re-opened the issue and removed the duplicate lable. It would be greatly appreciated that you can provide more details of your fix. A PR will be very welcome.

pazourek, 2023-10-11:
* The internal buffer has been increased to be able to handle 5 GUIDs.
* Updated checking whether the GUIDs are properly terminated by null characters - if the GUID string is longer than the internal buffer, the buffer stays unfilled, ERROR_MORE_DATA is returned by pRegQueryValueExA() and the necessary size is passed to an internal variable 'size'. The original code failed here if buffer reserved space for 5 GUIDs, but the device reported only 2.

tormodvolden, 2023-10-11:
Is 5 an arbitrary number here? Should we maybe instead use dynamic allocation with the ERROR_MORE_DATA mechanism?

pazourek, 2023-10-11:
@tormodvolden, yes, 5 is an arbitrary number here. It would be definitely better to use dynamic allocations, but I wanted not to do complex changes.

tormodvolden, 2023-10-11:
That's fine, as long as we know why :) It can be improved later. This whole GUID extraction is quite complex and should also be moved to a separate function to make it all more readable.

tormodvolden, 2023-11-06:
What platform is your device? Is the firmware something you can share?

pazourek, 2023-11-08:
I am using non-open devices based on LPC1766 (ARM Cortex M3) and ATSAMD21G (ARM Cortex M0) MCUs. But I would be probably able to create an open project for the ATSAMD21G based on [libopencm3 ](https://libopencm3.org) USB stack.

tormodvolden, 2023-11-27:
A test firmware based on libopencm3 would be nice, I might be able to port that to stm32's I have laying around.

commit fdab67b1 (2023-11-29): windows: Allow device GUIDs missing terminating characters or separators

Allow REG_MULTI_SZ strings with multiple GUIDs, and use the first GUID
for enumerating the device interfaces.

This solves issues with e.g. a composite device with four WinUSB
interfaces, where the device implements "Extended Properties OS Feature
Descriptor" with multiple GUIDs. An attempt to open the device fails
with an error LIBUSB_ERROR_NOT_FOUND (-5).

Fixes #1307
closed 2023-11-29 by tormodvolden