libusb/libusb #1310: HID not recognized

1369841815, 2023-09-18:
Hello, I want to ask my HID device that the port that was just plugged in can be recognized, but it cannot be recognized when I change the port, but the device manager can recognize it, I use a cloud desktop, window7 system.

mcuee, 2023-09-19:
Sorry but this has nothing to do with libusb.

1369841815, 2023-09-19:
When I look at the source code to pass 4, I get the GUID of the HID device, but it does not seem to be loaded into the device list. When I check the device VID,PID does not exist. The function I use is libusb_get_device_list.

1369841815, 2023-09-19:
I added it myself. Thank you！

1369841815, 2023-09-20:
Again to bother you, I still want to know, the HID device in obtaining the device list did not, but the computer device manager has, I follow the source code found that the DeviceIoControl function to obtain device information failed, but I changed to another port of the usb hub works, other ports have failed, The scanned device list is not displayed.

mcuee, 2023-09-20:
Please post the full debug log using the example xusb. `xxxx:yyyy` is the USB VID/PID of your USB HID device.
```
xusb -d xxxx:yyyy
```

Take note we recommend the user to use hidapi and not libusb for HID devices. You may want to give hdiapi a try.
hidapi: https://github.com/libusb/hidapi

1369841815, 2023-09-20:
I tested hidapi today, but this also did not show any data sent. When I used libusb earlier, the DeviceIoControl of parent_priv returned invalid parameters. I only changed the port of the inserted USB hub, and the address above was the same

1369841815, 2023-09-20:
[image]

1369841815, 2023-09-20:
[image]
[image]

1369841815, 2023-09-20:
我今天测试用hidapi的时候一直提示发送失败

1369841815, 2023-09-20:
[image]

1369841815, 2023-09-20:
[image]

mcuee, 2023-09-21:
@1369841815 

As mentioned, please post the debug log of `xusb -d 0483:5710`. Thanks. This will help to debug if there is a real issue with libusb or not.

**Please do not use screen shot. Just copy and paste the full debug log.**

As for the output of hidapi, that is normal. The test program carries out generic test and then a specific test. The specific test will not work for your device. So HIDAPI is working for your device.

1369841815, 2023-09-21:
D:\09_test\libusb-master\build\v142\Win32\Debug>xusb -d 0483:5710 Using libusb v1.0.2611807
[timestamp] [threadID] facility level [function call] <message>
[0.000299] [00006a44] libusb: debug [libusb_init_context] created default conte
0.000453][00006a44] libusb:debug [1ibusb_init_context] libusb v1.0.26.11807
0.000561]	[00006a44] 1ibusb:debug [usbi_add_even_source] add HANDLE 00000138	
events 0
0.000667][00006a44] libusb: debug [usbi_io_init] using timer for timeouts
0.000765]	[00006a44] libusb:debug [usbi_add_event_source] add HANDLE 0000013C	
events 0
[0.000885]	[00006a44] libusb:debug [get_windows_version] Windows 7 SP1 64-bit	
0.000980][00006a44] libusb:debug [htab_create] using 1021 entries hash table[0.005046] [00006a44] 1ibusb:info[winusbx_init] WinUSB DLL available (without isoch support)
0.005316] [00006a44] libusb:info [winusbx_init] libusbK DLL is not available
0.007330][00006a44] libusb:info [windows_init] UsbDk backend is not availabl Opening device 0483:5710...
0.007778][00006a44] libusb: debug [libusb_get_device_list]
[0.007919][0000602c] libusb:debug [windows_iocp_thread] I/0 completion thread started
[0.025449] [00006a44] libusb:debug [winusb_get_device_list] found new PnP enum erator string'NUSB3
0.026267] [00006a44] libusb:debug [winusb_get_device_list] allocating new dey ice for session [2150
0.027923] [00006a44] libusb:debug [winusb_get_devicelist] allocating new dev ice for session [2210
0.028664] [00006a44]	libusb:debug [winusb_get_device_list] allocating new dev	
ice for session [2248]
0.029487] [00006a44] libusb:debug [winusb_get_device_list] found new PnP enum erator stringUSBESTUB
0.030067] [00006a44] libusb:debug [winusb_get_device_list] allocating new dev ice for session [2284]
0.030869] [00006a44] libusb:debug [winusb_get_device_list] allocating new dev ice for session [22B4]
0.031950] [00006a44] libusb:debug [get_api_type] driver(s):HidUsb
0.032363][00006a44] libusb: debug [get_api_type] matched driver name against HID API
0.033037] [00006a44] libusb:debug [winusb_get_device_list] allocating new dev ice for session [1E04]
0.034106] [00006a44] 1ibusb:debug[enumerate_hd root_hub] assigning HCD’PCI VEN 1033&DEV 0194&SUBSYS 11001AF48REV_03\3&13COBC5&2&E’bus number 1
0.034819] [00006a44] 1ibusb:debug[enumeratehd roothub] assigning HCDPCI AVEN_80868DEV_24CD&SUBSYS_11001AF4&REV_10\3&13C0B0C5&2&D8’bus number 2
[0.035441] [00006a44] 1ibusb:debug [enumerate_hcd_root_hub] assigning HCD’PCI VEN 8086&DEV_7020&SUBSYS 11001AF48REV_01\3&13C0BC5&2&OA’bus number 3
0.036118] [00006a44] libusb:debug [enumerate_hed_root_hub] assigning HCDROO
T\USB\0000°	bus number 4	
[0.036706] [00006a44]1ibusb: warning [enumerate_hcd root_hub] could not infer VID/PID of HCD root hub fromROOT\USB\0000
10.039693][00006a44]ibusb:debug[init root_hub] root hubUSB\ROOT_HUB\4&1E B4B08D&o’reports 2 ports
[0.040353] [00006a44] libusb:debug[init_device] (bus:3，addr:0，depth:，p ort: 0): USB\ROOT_HUB\4&1EB4BO8D&O'
[0.042240] [00006a44] libusb:debug[init_root_hub] root hubUSB\ROOT_HUB20\4&26D2A416&0’reports 6_ports
[0.042796] [00006a44] libusb:debug [init_device](bus:2, addr:0,depth: 0, p ort:0):'USB\ROOT HUB20\4826D2A41680
[0.043890] [00006a44] 1ibusb:debug[winusb_get_device_list] found existing dev ice for session [1E04]
[0.045077] [00006a44] libusb: warning [init device] could not get node connecti on information for device’USB\VID_0483&PID_5710\00000000001B[87]参数错误。[0.046084] [00006a44] 1ibusb:warning [winusb_get_device_list] failed to initia lize deviceUSB\VID_0483&PID_5710\00000000001B
[0.049050] [00006a44] 1ibusb:debug [init_root_hub] root hub’NUSB3\ROOT_HUB30\4823676B3280reports 8 ports
[0.049698] [00006a44] 1ibusb:debug [init_device] (bus:1,addr:0,depth0 p ort:0):NUSB3\ROOT HUB30\4823676B32&0
[0.051255] [00006a44] 1ibusb:warning [winusb_get_device list] could not retrie ve port number for device USBESTUB\DEVICES\000[13]数据无效
[0.052175] [00006a44] libusb;debug [init_root_hub] root hub USBESTUB\DEVICES0000’reports 1 ports
[0.052776] [00006a44] 1ibusb: debug [init_device] (bus:4, addr:o, depth: 0, D ort: 0):’USBESTUB\DEVICES\0000’
[0.053659] [00006a44] 1ibusb:debug [winusb_get_device_list] unlisted ancestor for 'USBESTUB\DEVICES\0001’(non USB HID, newly connectedetc.)- ignoring
[0.054393] [00006a44] 1ibusb: debug[winusb_get_device_list] setting HID interf ace for [1E04]:
C0.055954][00006a44] 1ibusb:debug [set_hid_interface] interface[0]= \12\HID# VID 0483&PID_5710#38200F1EB3&1&0000#[4D1E55B2-F16F-11CF-88CB-001111000030)[0.057040][00006a44] 1ibusb: debug [libusb_unref_device] destroy device 0.0
0.057330]
0.057849]100006a44][00006a44] libusb:debug libusb: debug [libusb_unref_device] destroy device 4.0[1ibusb_get_device_descriptor
0.058435]	[00006a44]	libusb: debug (libusb_get_device_descriptor	
0.059064]
0.059609][00006a44][00006a44] libusb:debug libusb: debug [libusb_get_device_descriptor[libusb_get_device_descriptor]
0.060264
0.060883][00006a44] libusb: debug [libusb_unref device]destroy device 3.0
[00006a44] libusb:debug[libusb_unref_device] destroy device 2.0
0.061486][00006a44] libusb:debug [libusb_unref device] destroy device 1.0
0.062100][00006a44] 1ibusb:debug [1ibusb_unref_device] destroy device 4.0
Failed.	[0.064122]	0.063330]	[00006a44] libusb: debug [libusb_exit] destroying default context	[0000602c] libusb: debug [windows_iocpthread] I/0 completion thread	
exiting[0.065445] [00006a44] 1ibusb:debug [usbi_remove_event_source] remove HANDLE 00
00013C	000.066589] [0000a44] 1ibusb:debug [usbi_remove_eventsource] remove HANDLE 0O	
000138
D:\09 test\1iburbnaster\build\v142\Win32\Debug>

1369841815, 2023-09-21:
Now it seems that hidapi can send, but can it get the message length? I want to know whether it is full speed or low speed？

1369841815, 2023-09-21:
hidapi does not support the speed of obtaining usb, while libusb can obtain the speed of usb. I want to use the device across platforms. Is it impossible to know the transmission speed of usb with hidapi? I see that hidusb sets the size of input and output messages on window to 64. Is there any way to automatically get the speed mode supported by USB?

mcuee, 2023-09-21:
@1369841815 

It seems your Windows 7 machine has a USB 3.0 chipset not support by libusb Windows. Unfortunately Windows 7 does not have native USB driver from Microsoft and the vendor drivers may not be good.

I will suggest you upgrade your Windows machine to Windows 10/11, Windows 7 is no longer supported by Microsoft and it will not be easy to sort out issues if you encounter issues like this.

mcuee, 2023-09-21:
> Now it seems that hidapi can send, but can it get the message length? I want to know whether it is full speed or low speed？

You do not need to know the USB speed. HID devices talks in terms of Reports. You need to know the report length anyway.

Anyway, for hidapi related issues, please ask here. 
https://github.com/libusb/hidapi/issues

mcuee, 2023-09-21:
@1369841815 

Going back to the reported issue, as mentioned above, I will suggest you to upgrade to Windows 10/11. 

But if you want to stay with Windows 7, you  have to debug the issues of the USB 3.0 driver since it failed to properly respond to `get node connection information`, basically, the USB 3.0 hub driver failed to respond to `IOCTL_USB_GET_NODE_CONNECTION_INFORMATION` which is a bug of the USB 3.0 hub driver.

What is the USB 3.0 chipset you are using? You may want to check if they have a newer driver for Windows 7.

mcuee, 2023-09-21:
Similar issues reported before.
* https://github.com/libusb/libusb/issues/1103
* https://github.com/libusb/libusb/issues/918

1369841815, 2023-09-21:
Thank you for your answer, but this is a cloud desktop, I can not access the Internet, and I can identify in one of the ports, but change a port can not identify, I see the previous method seems to change the location has changed, and I have gone through all the virtual devices seem not to use.

mcuee, 2023-09-21:
> Thank you for your answer, but this is a cloud desktop, I can not access the Internet, and I can identify in one of the ports, but change a port can not identify, I see the previous method seems to change the location has changed, and I have gone through all the virtual devices seem not to use.

Sorry I do not quite unstand you here. But in any case, we can not help you here, report the issue to your "cloud desktop" vendor.

mcuee, 2023-09-21:
> When I test on the win10 virtual machine, it is not possible, is it because the parent node can not open?

What kind of Virtual Machine? Some virtual machine has bad USB support.

Again, please post the debug log of `xusb -d 0843:5710`.

mcuee, 2023-09-21:
> This is the log information of the port that works.

This port does not show error in terms of node connection information. However, there is another issue. Maybe your device also has an issue.
```
[0.049184] [00004af8] 1ibusb:debug [cache_config_descriptors] cached config descriptor 0(bConfigurationValue=1,41 bytes)
[0.049914] [00004af8] libusb:error[init_device] program assertion failed - device address overflow
```

Anyway, it seems to me your "cloud desktop" and the device are both broken. So I think I can not help you much here with regard to your Windows 7 issue.

1369841815, 2023-09-21:
Ok, thank you, but I still want to know why hidapi can communicate? Not this one.

mcuee, 2023-09-21:
> Ok, thank you, but I still want to know why hidapi can communicate? Not this one.

HIDAPI does not care about the USB connection information.

1369841815, 2023-09-22:
I seem to have found a problem, as if the parent node is a virtual port problem, this I saw before the change of the version has now changed, can you please give me the virtual port opening method, he reported an error is DeviceIoControl this place.

1369841815, 2023-09-22:
And his driver is not an official Microsoft driver.

1369841815, 2023-09-24:
Hello, my that should be the problem of virtual serial port, the driver of this serial port is not Microsoft, this if I want to identify the driver of this serial port, need to add what in the source code? Really trouble you!

mcuee, 2023-09-27:
> Hello, my that should be the problem of virtual serial port, the driver of this serial port is not Microsoft, this if I want to identify the driver of this serial port, need to add what in the source code? Really trouble you!

Sorry but I think you have to fix your virtual USB driver. I do not know how to fix it or whether it is possible to fix your issue within libusb.

mcuee, 2023-09-27:
I will consider not to be a libusb issue for now. And I will close this issue.

You can re-open this issue if you still have issues after you get the vendor to fix the virtual USB driver.

closed 2023-09-27 by mcuee