libusb/libusb #1315: gcc -m32 occur d000002.o:(.text+0x0): multiple definition of `libusb_claim_interface'

ziyuer2min, 2023-09-26:
Win32 compilation error， Win64 Compilation ok.

my code:
#include <stdio.h>
#include "./lib/libusb/libusb.h"
/* USB VID PID */
#define CJQT_QKEY_DEVICE_ID_VENDOR  0x2309
#define CJQT_QKEY_DEVICE_ID_PRODUCT 0x0706
unsigned int k_is_initm = 0;
struct libusb_context *k_ctxm;
struct libusb_device_handle *k_usb_device_handlem;
int main(void)
{
    int err;
    if (k_is_initm) {
        printf("not init again...\n");
        return 2;
    }
    err = libusb_init(&k_ctxm);
    if (err) {
        printf("libusb_init error:%s\n", libusb_error_name(err));
        return 2;
    }
    // 通过VID和PID连接设备
    k_usb_device_handlem = libusb_open_device_with_vid_pid(k_ctxm, CJQT_QKEY_DEVICE_ID_VENDOR, CJQT_QKEY_DEVICE_ID_PRODUCT);
    if (k_usb_device_handlem == NULL) {
        printf("usb open failed...\n");
        libusb_exit(k_ctxm);
        return 2;
    }
    // 设备打开成功
    printf("usb open success: %d\n", k_usb_device_handlem);

    err = libusb_kernel_driver_active(k_usb_device_handlem, 0);
    if (err) {
        // 移除原有驱动
        printf("libusb_detach_kernel_driver error:%s\n", libusb_error_name(err));
        libusb_detach_kernel_driver(k_usb_device_handlem, 0);
    }
    // 认领驱动
    err = libusb_claim_interface(k_usb_device_handlem, 0);
    if (err) {
        printf("libusb_claim_interface error:%s\n", libusb_error_name(err));
        return 2;
    }
    k_is_initm = 1;
    return 0;
}

compile：
gcc -m32 -o zm zm.c -lm -L./lib/win32 -lusb-1.0

problems：
gcc -m32 -o zm zm.c -lm -L./lib/win32 -lusb-1.0 
d000002.o:(.text+0x0): multiple definition of `libusb_claim_interface'
d000001.o:(.text+0x0): first defined here
d000002.o:(.idata$5+0x0): multiple definition of `_imp__libusb_claim_interface'
d000001.o:(.idata$5+0x0): first defined here
d000004.o:(.text+0x0): multiple definition of `libusb_detach_kernel_driver'
d000003.o:(.text+0x0): first defined here
d000004.o:(.idata$5+0x0): multiple definition of `_imp__libusb_detach_kernel_driver'
d000003.o:(.idata$5+0x0): first defined here
d000006.o:(.text+0x0): multiple definition of `libusb_error_name'
d000005.o:(.text+0x0): first defined here
d000006.o:(.idata$5+0x0): multiple definition of `_imp__libusb_error_name'
d000005.o:(.idata$5+0x0): first defined here
d000008.o:(.text+0x0): multiple definition of `libusb_exit'
d000007.o:(.text+0x0): first defined here
d000008.o:(.idata$5+0x0): multiple definition of `_imp__libusb_exit'
d000007.o:(.idata$5+0x0): first defined here
d000010.o:(.text+0x0): multiple definition of `libusb_init'
d000009.o:(.text+0x0): first defined here
d000010.o:(.idata$5+0x0): multiple definition of `_imp__libusb_init'
d000009.o:(.idata$5+0x0): first defined here
d000012.o:(.text+0x0): multiple definition of `libusb_kernel_driver_active'
d000011.o:(.text+0x0): first defined here
d000012.o:(.idata$5+0x0): multiple definition of `_imp__libusb_kernel_driver_active'
d000011.o:(.idata$5+0x0): first defined here
d000014.o:(.text+0x0): multiple definition of `libusb_open_device_with_vid_pid'
d000013.o:(.text+0x0): first defined here
d000014.o:(.idata$5+0x0): multiple definition of `_imp__libusb_open_device_with_vid_pid'
d000013.o:(.idata$5+0x0): first defined here
Warning: resolving _libusb_init@4 by linking to _libusb_init
Use --enable-stdcall-fixup to disable these warnings
Use --disable-stdcall-fixup to disable these fixups
Warning: resolving _libusb_error_name@4 by linking to _libusb_error_name
Warning: resolving _libusb_open_device_with_vid_pid@12 by linking to _libusb_open_device_with_vid_pid
Warning: resolving _libusb_exit@4 by linking to _libusb_exit
Warning: resolving _libusb_kernel_driver_active@8 by linking to _libusb_kernel_driver_active
Warning: resolving _libusb_detach_kernel_driver@8 by linking to _libusb_detach_kernel_driver
Warning: resolving _libusb_claim_interface@8 by linking to _libusb_claim_interface
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:8: zm] Error 1

tormodvolden, 2023-09-26:
What exactly are the library files you have under ./lib? Did you compile libusb yourself or is it from a downloaded archive? Which version of gcc is this?

ziyuer2min, 2023-09-26:
PS C:\Users\Admin\Desktop> gcc --version
gcc.exe (x86_64-win32-sjlj-rev0, Built by MinGW-W64 project) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tormodvolden, 2023-09-26:
Can you please try using the binaries under libusb-MinGW-Win32 instead of VS2015-Win32?

tormodvolden, 2023-09-26:
I tried building this from the libusb-MinGW-Win32 folder, changing the include line to:
`#include "./include/libusb-1.0/libusb.h"`
and it builds fine without warning (this is GCC 9.3):
`i686-w64-mingw32-gcc -m32 -o zm zm.c -lm -L./lib -lusb-1.0`

mcuee, 2023-09-27:
The original goal of libusb Windows backend was to have 32bit/64bit x86/x64 `libusb-1.0.dll` generated by MinGW-w64 x86/x64 toolchain and MSVC x86/x64 toolchain to be compatible.

It could also be the version specific issue from the user. It could also be that the 32bit x86 DLL is no longer compatible with each other (need to confirm). But I think that is inevitable in the end that we have different toolchains. Even if we can get MSYS2 mingw32 x86 toolchain to be compatible with MSVC x86 toolchain, there are other toolchain like clang32. Then we also have multiple version of MSYS2 mingw32 and MSVC version.

Even for 64bit x64 I am not 100% sure if different toolchain will build compatible DLLs now that we have clang64 as well.

Same for ARM64 now that we have clang-arm64.

I will say the bundled Windows binary in libusb release is provided as is. If it does not work, the user is adviced to build from source using his own toolchain.

ziyuer2min, 2023-09-27:
> I tried building this from the libusb-MinGW-Win32 folder, changing the include line to: `#include "./include/libusb-1.0/libusb.h"` and it builds fine without warning (this is GCC 9.3): `i686-w64-mingw32-gcc -m32 -o zm zm.c -lm -L./lib -lusb-1.0`

I follow your steps, it builds fine, too

ziyuer2min, 2023-09-27:
> I tried building this from the libusb-MinGW-Win32 folder, changing the include line to: `#include "./include/libusb-1.0/libusb.h"` and it builds fine without warning (this is GCC 9.3): `i686-w64-mingw32-gcc -m32 -o zm zm.c -lm -L./lib -lusb-1.0`

I follow your steps, it builds fine, too.
But, usb open success, libusb_detach_kernel_driver error.

closed 2023-09-27 by ziyuer2min

ziyuer2min, 2023-09-27:
Thanks everyone, I have solved the problem, good luck.

jenish-rudani, 2024-05-22:
how?

Can you provide your solution?

@ziyuer2min