libusb/libusb #1316: USB composite devices

Fly2023-HM, 2023-09-27:
USB composite devices have multiple sub devices, each with different interfaces. Can I use libusb_open the interface to open different sub devices separately,

operating system:windows1

mcuee, 2023-09-27:
> USB composite devices have multiple sub devices, each with different interfaces. Can I use libusb_open the interface to open different sub devices separately,
> 
> operating system:windows10

Yes you can as long as you have supported driver for the devices.

There are some issues though.
* https://github.com/libusb/libusb/issues/1177
* https://github.com/libusb/libusb/pull/1181 (which is supposed to fix the above issue)
* https://github.com/libusb/libusb/issues/1238

mcuee, 2023-09-27:
github is not for technical support. You can use libusb-devel if you need further support. I will close this issue.

closed 2023-09-27 by mcuee