libusb/libusb #1317: Ubuntu22.04 can't get usb-hid data

Iyiren, 2023-09-28:
hi, please help me. i use libusb-1.0.26 to get data from usb-hid dev(stm32), but can't get data from stm32 dev. 
the c code with libusb on windows works well, but on ubuntu(init, open, clam interface, all correct), libusb_interrupt_transfer() gets -1(io error).
i also tried this in vmware ubuntu, but get -7(timeout error).

tormodvolden, 2023-09-28:
We will need to see debug log files, and the simplest piece of code that can reproduce the issue.

Does it work if you try the usb-hid example from libopencm3-examples on your STM32? Otherwise I would suspect there is something wrong with your firmware. That it happens to work on Windows does not mean that it is USB compliant.

mcuee, 2023-10-03:
> when insert the usb-hid dev, hidrawX is under /dev. but libusb_set_auto_detach_kernel_driver(dev_handle, 1) will cause the hidrawX disappear. then libusb_interrupt_transfer() can't get data. i can get data by 'cat /dev/ hidrawX | hexdump -C'.

It is correct that `hidrawX` will disappear after `libusb_set_auto_detach_kernel_driver`  but it is not correct to say that is the cause of your problem. 

You should use HIDAPI library instead of libusb. It may make your life easier.
https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices

If you really want to use libusb, you'd better read the HID spec first before proceeding.
https://www.usb.org/sites/default/files/hid1_11.pdf

mcuee, 2023-10-03:
Close this issue as it is not a libusb issue.

github is not for technical support. You may want to study the USB HID spec and then ask questions in libusb-devel mailing list,