libusb/libusb #1319: I hit "may not be able to detach the kernel driver for this device" on MacOS

Boiethios, 2023-10-10:
Hello, I try and read the HID reports on MacOS, but I get the message saying that the kernel driver cannot be detached:

```
% sudo usbhid-dump --model=16c0:27db
Password:
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device
libusb: warning [darwin_transfer_status] transfer error: timed out
000:001:000:Failed to retrieve report descriptor: Operation timed out
libusb: error [darwin_reenumerate_device] darwin/reenumerate_device: timeout waiting for reenumerate
000:001:000:Failed to attach to the kernel driver: Operation timed out
libusb: warning [darwin_close] darwin_close device missing IOService
```

What does it mean? How can I debug my keyboard on MacOS?

Boiethios, 2023-10-10:
Alright, thank you for the answer! Never mind, I'll just boot my old linux laptop, I wanted a solution out-of-the-box: if it does not work directly, I'll not mess up with Apple stuff.

Have you considered adding a Github discussion? I feel bad writing an issue only for asking a question.

mcuee, 2023-10-10:
We recommend using libusb-devel mailing list for technical support.

There are too few libusb developers now to be able to support github discussion.

closed 2023-10-10 by mcuee

Boiethios, 2023-10-10:
Well, it does not work either on Linux, but for a different reason. First `lsusb` displays the name as something completely different (definitely not the one I put in the firmware).

Then I get:

```
$ sudo usbhid-dump --model=16c0:27db
Starting dumping interrupt transfer stream
with 1 minute timeout.

005:004:000:Failed to set report protocol: Operation timed out
005:004:000:Failed to attach to the kernel driver: Entity not found
```

And then my keyboard does not work anymore, I have to "reboot" it.

mcuee, 2023-10-10:
For Linux, when your use libusb and detach the kernel driver, your keyboad function will not work. That is why you need to re-attach the kernel driver once you finish the application.

Therefore you should use HIDAPI as well under Linux.

Are you developing a firmware for a USB keyboard? You should consider using debugging tool for your MCU and use debugger (eg: JTAG/SWD debugger or emulator), or use UART to dump the debugging info.

mcuee, 2023-10-10:
> We recommend using libusb-devel mailing list for technical support.
> 
> There are too few libusb developers now to be able to support github discussion.

In the end, I think we may add github discussion since some people may not like to use libusb-devel mailing list.