libusb/libusb #1328: libusb + WinUSB backend claim interface 1 and 2 failed for iPhone on Windows

Pivot2017, 2023-10-18:
I am using libusb + WinUSB backend to communicate with iPhone on Windows.
first, iPhone is recognized as a MTP device;
then i replace iPhone's driver with WinUSB(zadig/libwdi generate the driver);
libusb then can open iPhone and claim interface 0(submit control transfer);
but i can't claim interface 1 and interface 2 ( libusb_get_active_config_descriptor() can emun all the 5 altsettings in 3 interfaces ).
this is because WinUSB driver(.inf file) issue? what should i do?
thanks in advance.

mcuee, 2023-10-18:
How do you replace the driver of iPhone? 

What is the version of libusb you are using? Please post the debug log of `xusb -d xxxx:yyyy` where xxxx/yyyy is your iPhone's USB VID/PID.

Pivot2017, 2023-10-18:
Hi mcuee, 
i replace the driver with command "wdi-simple.exe -v 1452 -p 4776".
i am using libusb version 1.0.22.
below is "xusb -d 05ac:12a8" output(latest xusb from version 1.0.26):

===============================

Using libusb v1.0.26.11724
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000341] [00000b8c] libusb: debug [libusb_init] created default context
[ 0.000449] [00000b8c] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000570] [00000b8c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000A8 events 0
[ 0.000695] [00000b8c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000795] [00000b8c] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000AC events 0
[ 0.000915] [00000b8c] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000977] [00000b8c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.007578] [00000b8c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.007669] [00000b8c] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.008802] [00000b8c] libusb: info [windows_init] UsbDk backend is not available
Opening device 05AC:12A8...
[ 0.008875] [00000b8c] libusb: debug [libusb_get_device_list]
[ 0.009193] [000024cc] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.018083] [00000b8c] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.018248] [00000b8c] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.018680] [00000b8c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.019146] [00000b8c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.019225] [00000b8c] libusb: debug [winusb_get_device_list] allocating new device for session [33]
[ 0.019464] [00000b8c] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.019579] [00000b8c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.019594] [00000b8c] libusb: debug [winusb_get_device_list] allocating new device for session [37]
[ 0.019796] [00000b8c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.019842] [00000b8c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.019906] [00000b8c] libusb: debug [winusb_get_device_list] allocating new device for session [5C]
[ 0.020037] [00000b8c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.020119] [00000b8c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.020717] [00000b8c] libusb: debug [winusb_get_device_list] found existing device for session [5C]
[ 0.020852] [00000b8c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.020933] [00000b8c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 577 bytes)
[ 0.021021] [00000b8c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 5): 'USB\VID_0408&PID_A094\5&586B51A&0&5'
[ 0.021235] [00000b8c] libusb: debug [winusb_get_device_list] found existing device for session [33]
[ 0.021457] [00000b8c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.022091] [00000b8c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.022272] [00000b8c] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.022894] [00000b8c] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[ 0.022993] [00000b8c] libusb: debug [winusb_get_device_list] allocating new device for session [70]
[ 0.023412] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.073620] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.124764] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.175906] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.226654] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.277287] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.327634] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.378763] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.429808] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.480524] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.531175] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.581507] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.632714] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.683044] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.733751] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[ 0.784177] [00000b8c] libusb: debug [init_device] 0x1c7a:0x575 found 1 configurations (not configured)
[... cut ...]
[ 1.199204] [00000b8c] libusb: debug [libusb_unref_device] destroy device 1.19
[ 1.199301] [00000b8c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 1.199697] [00000b8c] libusb: debug [libusb_exit] destroying default context
[ 1.199855] [000024cc] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 1.201008] [00000b8c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000AC
[ 1.201080] [00000b8c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000A8

tormodvolden, 2023-10-18:
It looks like the device has 4 configurations. But WinUSB only allows accessing the first configuration, which has 1 interface as seen in the xusb log. https://github.com/libusb/libusb/wiki/Windows#known-restrictions

Pivot2017, 2023-10-18:
> It looks like the device has 4 configurations. But WinUSB only allows accessing the first configuration, which has 1 interface as seen in the xusb log. https://github.com/libusb/libusb/wiki/Windows#known-restrictions

any other way to switch configuration and access interface 1 and interface 2? for example install a different winusb.inf driver?
or use libusb-win32 instead?

mcuee, 2023-10-19:
> any other way to switch configuration and access interface 1 and interface 2? for example install a different winusb.inf driver? or use libusb-win32 instead?

You can try libusb-win32 device driver.

Still just wondering what is your end goal? I think you can not do much with libusb for iPhones.

Pivot2017, 2023-10-19:
> > any other way to switch configuration and access interface 1 and interface 2? for example install a different winusb.inf driver? or use libusb-win32 instead?
> 
> You can try libusb-win32 device driver.
> 
> Still just wondering what is your end goal. I think you can not do much with iPhones using libusb under Windows.
> 
> Not so sure if your needs are better served by usbmuxd project or not. https://github.com/libimobiledevice/usbmuxd

Yes. i want to use usbmuxd to communicate with iPhone. 
so "libusb+winusb backend" or "libusb-win32", which works?

mcuee, 2023-10-19:
Hmm, I do not use usbmuxd but I tend to believe that you do not need to use libusb at all, you just need to install iTunes under Windows. 

Maybe you want to ask in usbmuxd project.

Pivot2017, 2023-10-19:
> Hmm, I do not use usbmuxd but I tend to believe that you do not need to use libusb at all, you just need to install iTunes under Windows.
> 
> Maybe you want to ask in usbmuxd project.

Thanks very much. i will try to look into libusb-win32 first.

closed 2023-10-19 by Pivot2017