libusb/libusb #1333: Sync I/O blocks despite timeout

xairy, 2023-10-27:
`libusb_interrupt_transfer` does not respect its `timeout` argument. If called with `timeout` set to a low value, it still blocks block for 60 seconds. And `libusb_bulk_transfer` likely suffers from the same issue.

I think like the problem is that even though `do_sync_bulk_transfer` propagates `timeout` to `libusb_fill_bulk_transfer`, `sync_transfer_wait_for_completion` still blocks despite it, as it uses `libusb_handle_events_completed`.

I suspect the fix should be to use `libusb_handle_events_timeout_completed` instead of `libusb_handle_events_completed` and  stop the `while` loop in `sync_transfer_wait_for_completion` if the timer has fired.

xairy, 2023-10-27:
Just in case: I see this issue on Linux.

tormodvolden, 2023-11-05:
Thanks for your report. You say "still blocks despite it" above, but I suppose you mean "still time outs after 60 seconds"?

Did you try out your suggested fix? I guess sync_transfer_wait_for_completion() will need to pass on a timeout from its caller?

tormodvolden, 2023-11-05:
Could this work?
```
diff --git a/libusb/sync.c b/libusb/sync.c
index 146cce2..6619212 100644
--- a/libusb/sync.c
+++ b/libusb/sync.c
@@ -49,9 +49,18 @@ static void sync_transfer_wait_for_completion(struct libusb_transfer *transfer)
 {
        int r, *completed = transfer->user_data;
        struct libusb_context *ctx = HANDLE_CTX(transfer->dev_handle);
+       struct timeval tv;
+
+       if (transfer->timeout) {
+               tv.tv_sec = transfer->timeout / 1000;
+               tv.tv_usec = (transfer->timeout % 1000) * 1000;
+       } else {
+               tv.tv_sec = 60;
+               tv.tv_usec = 0;
+       }
 
        while (!*completed) {
-               r = libusb_handle_events_completed(ctx, completed);
+               r = libusb_handle_events_timeout_completed(ctx, &tv, completed);
                if (r < 0) {
                        if (r == LIBUSB_ERROR_INTERRUPTED)
                                continue;

```

xairy, 2023-11-07:
While testing your fix I realized that timeouts actually work as expected: I had an issue in my code. I'm not entirely sure how they work without `libusb_handle_events_timeout_completed`, but this fix is not required.

Sorry for the false positive report.

Thank you very much for looking into this!

closed 2023-11-07 by xairy