libusb/libusb #1338: Auto-claiming interface 0 may lead to an error "unable to match endpoint to an open interface"

dmitry-zakablukov, 2023-11-08:
Good day.

I have a device with the following endpoints:
```
config:1 iface:1 alt_setting:0 address:1 direction:OUT transfer_type:BULK
config:1 iface:1 alt_setting:0 address:129 direction:IN transfer_type:BULK
config:1 iface:0 alt_setting:0 address:131 direction:IN transfer_type:INTERRUPT
```

I have thought that I should claim only interface 1 to be able to call `libusb_bulk_transfer()`. But with libusb v1.0.26 I got the following error:
```
Claiming interface 1...
[Log] libusb: warning [winusbx_claim_interface] auto-claimed interface 0 (required to claim 1 with WinUSB)
Writing data...
[Log] libusb: error [winusbx_submit_bulk_transfer] unable to match endpoint to an open interface - cancelling transfer
libusb_bulk_transfer() call failed with return code: -5
```

If I claim interface 0 manually, there is no error:
```
Claiming interface 0...
Claiming interface 1...
Writing data...
Bytes written: 5
```

Is this an expected behavior? In my opinion, libusb should either return an error if I claim only interface 1 on such a device or auto-claim interface 0 properly by itself (no error should occur after auto-claim and `libusb_bulk_transfer()` call).

[libusb-auto-claim.txt](https://github.com/libusb/libusb/files/13298786/libusb-auto-claim.txt)

mcuee, 2023-11-09:
@dmitry-zakablukov 

There are quite some improvements in git main in dealing with USB Composite devices, please give git main a try. Thanks.

dmitry-zakablukov, 2023-11-09:
@mcuee , yes, there is no error with git main:
```
Claiming interface 1...
[Log] libusb: warning [winusbx_claim_interface] auto-claimed interface 0 (required to claim 1 with WinUSB)
Writing data...
Bytes written: 5
```

Looking forward for libusb v1.0.27 release!

mcuee, 2023-11-09:
Thanks a lot for reporting back. I will keep this issue open for now.

tormodvolden, 2023-12-08:
Any reason this ticket should be kept open, if it has been resolved?

mcuee, 2023-12-21:
I think we can close this issue.

closed 2023-12-21 by mcuee