libusb/libusb #1341: Binary distribution confusion

tikonen, 2023-11-18:
Main website libusb.info refers the binary distribution as "Latest Windows Binaries".  (libusb.info -> Download -> Latest Windows Binaries). This downloads file like `libusb-1.0.26-binaries.7z`

I found couple of issues with this.
* The package contains also MacOS binaries. Is this package also offical release of those?
* Package is missing include headers in VS2015-Win32 and VS2015-x64 folders.  I assume I can use libusb.h from any other folder in the package?

mcuee, 2023-11-18:
FYI, libusb binaries are provided as is. The binaries are not supported and may not work because of compiler versions. You are better off building from the source or getting from the distributions (Linux distro, macOS brew or macports, OpenBSD ports and NetBSD pkgsrc).

libusb-1.0.26 binaries are a bit different from the older release. Older releases only provide Windows binaries. 

Please also refer to the following issue.
* https://github.com/libusb/libusb/issues/1166

mcuee, 2023-11-18:
Please go to #1166 if you have further comments. I will close this one for now.

closed 2023-11-18 by mcuee