libusb/libusb #1344: libusb_error_timeout for Identiv reader connection with an ubuntu machine.

rubygoldbergmachineinc, 2023-11-22:
Libs:

libusb version - 1.0.0
libccid -1.4.31

Issue:

I did some more debugging and found out that communication via python script with reader gets stuck at this call for 110 seconds.
connection.transmit(READ_NDEF_DATA) ant returns with a timeout in pcscd[10529]: 99999999 ccid_usb.c:905:ReadUSB() read failed (1/2): -7 LIBUSB_ERROR_TIMEOUT. Would anyone be able to throw some light into this issue and potential root cause and fix? Some of the suggestions we are receiving are around changing
the cabling to active repeater cable for reader and ubuntu machine connectivity. Also, is there a way to override the timeout to say 2/3 seconds.

tormodvolden, 2023-11-22:
Please see https://github.com/libusb/libusb/wiki/Troubleshooting

LudovicRousseau, 2023-11-24:
The same user reported an issue at one of my projects https://github.com/LudovicRousseau/CCID/issues/121 but without providing much feedback.

I close this ticket.
@rubygoldbergmachineinc feel free to add your comments.

closed 2023-11-24 by LudovicRousseau