libusb/libusb #1346: [Unknown Error] Getting the serial number of the ios device always crashes

Amoystyle, 2023-11-24:
Good day.

I'm using the testlibusb example.

When I configure **libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG)**; after libusb_init(), getting the serial number of the ios device always crashes. I try to trace the code, but the reason for this error is unknown.

When I comment out **libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG),** it work well.

**version: libusb-1.0.26**
**platform: Windows 11**

```C++
  r = libusb_init(NULL);
  if (r < 0)
      return r;
  
  libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG);
```
[image]

### **\\?\USB#VID_05AC&PID_12A8#00008030000A31061188802E#{A5DCBF10-6530-11D2-901F-00C04FB951ED}**

```sh
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000280] [00004f24] libusb: debug [libusb_init] created default context
[ 0.000469] [00004f24] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000592] [00004f24] libusb: debug [usbi_add_event_source] add HANDLE 000000F0 events 0
[ 0.000724] [00004f24] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000816] [00004f24] libusb: debug [usbi_add_event_source] add HANDLE 000000F4 events 0
[ 0.000962] [00004f24] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001078] [00004f24] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003316] [00004f24] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005953] [00004f24] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.008051] [00004f24] libusb: info [windows_init] UsbDk backend is not available
[ 0.008210] [00004f24] libusb: debug [libusb_get_device_list]
[ 0.008602] [0000173c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.019173] [00004f24] libusb: debug [winusb_get_device_list] allocating new device for session [4]
[ 0.020438] [00004f24] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.020613] [00004f24] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.020747] [00004f24] libusb: debug [winusb_get_device_list] allocating new device for session [32]
[ 0.021310] [00004f24] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.021472] [00004f24] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.021597] [00004f24] libusb: debug [winusb_get_device_list] allocating new device for session [69]
[ 0.021853] [00004f24] libusb: debug [get_api_type] driver(s): libusb0
[ 0.022015] [00004f24] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.022140] [00004f24] libusb: debug [winusb_get_device_list] allocating new device for session [6A]
[ 0.022400] [00004f24] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.022563] [00004f24] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.022686] [00004f24] libusb: debug [winusb_get_device_list] allocating new device for session [6B]
[ 0.023238] [00004f24] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_86941043&REV_10\3&11583659&0&A0' bus number 1
[ 0.024326] [00004f24] libusb: debug [winusb_get_device_list] found existing device for session [32]
[ 0.024525] [00004f24] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.024659] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.024795] [00004f24] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 11): 'USB\VID_8087&PID_0025\5&38E97A59&0&11'
[ 0.025158] [00004f24] libusb: debug [winusb_get_device_list] found existing device for session [6B]
[ 0.025376] [00004f24] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.025518] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 66 bytes)
[ 0.025643] [00004f24] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 8): 'USB\VID_0C45&PID_6741\5&38E97A59&0&8'
[ 0.025848] [00004f24] libusb: debug [winusb_get_device_list] found existing device for session [6A]
[ 0.026014] [00004f24] libusb: debug [init_device] found 4 configurations (current config: 1)
[ 0.026130] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.027201] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=2, 149 bytes)
[ 0.027864] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=3, 62 bytes)
[ 0.028644] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=4, 117 bytes)
[ 0.028754] [00004f24] libusb: debug [init_device] (bus: 1, addr: 17, depth: 1, port: 2): 'USB\VID_05AC&PID_12A8\00008030000A31061188802E'
[ 0.028950] [00004f24] libusb: debug [winusb_get_device_list] found existing device for session [69]
[ 0.029097] [00004f24] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.029183] [00004f24] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.029269] [00004f24] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 7): 'USB\VID_046D&PID_C52F\5&38E97A59&0&7'
[... 31 lines cut ...]
[ 0.034358] [00004f24] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 0.034408] [00004f24] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 0.034460] [00004f24] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 83 to interface 0
[ 0.034523] [00004f24] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.034530] [0000173c] libusb: debug [windows_iocp_thread] ignoring overlapped 008FF1C0 for handle 00E5B3D8 (device 1.17)
[ 0.034580] [00004f24] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.034637] [0000173c] libusb: debug [windows_iocp_thread] ignoring overlapped 008FF1C0 for handle 00E5B3D8 (device 1.17)
[ 0.034740] [0000173c] libusb: debug [windows_iocp_thread] ignoring overlapped 008FF1C0 for handle 00E5B3D8 (device 1.17)
[ 0.034794] [0000173c] libusb: debug [windows_iocp_thread] ignoring overlapped 008FF1C0 for handle 00E5B3D8 (device 1.17)
[ 0.034848] [0000173c] libusb: debug [windows_iocp_thread] ignoring overlapped 008FF230 for handle 00E5B3D8 (device 1.17)
[ 0.035061] [00004f24] libusb: debug [windows_force_sync_completion] transfer 00E46A90, length 12150545
[ 0.035066] [0000173c] libusb: debug [windows_iocp_thread] transfer 00E46A90 completed, length 4
[ 0.035122] [00004f24] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.035174] [0000173c] libusb: debug [windows_iocp_thread] ignoring overlapped 00E469C8 for handle 00E5B3D8 (device 1.17)
[ 0.035226] [00004f24] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.035330] [00004f24] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.035384] [00004f24] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.035436] [00004f24] libusb: debug [handle_event_trigger] event triggered
[ 0.035485] [00004f24] libusb: debug [windows_handle_transfer_completion] handling transfer 00E46A90 completion with errcode 0, length 12150545
[ 0.035539] [00004f24] libusb: debug [libusb_release_interface] interface 0
[ 0.035591] [00004f24] libusb: debug [auto_release] auto-released interface 0
[ 0.035641] [00004f24] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.035699] [00004f24] libusb: debug [usbi_handle_transfer_completion] transfer 00E46A90 has callback 00B8F990
[ 0.035750] [00004f24] libusb: debug [sync_transfer_cb] actual_length=12150545
```

Youw, 2023-11-24:
Try if the issue persists with latest master. There has been changes to how options being handled.

Amoystyle, 2023-11-25:
> Try if the issue persists with latest master. There has been changes to how options being handled.

I tried the latest master and still have this problem.

[image]

mcuee, 2023-11-25:
Just wondering if you can test under other OS like Linux or macOS? If yes, does it happen under Linux or macOS?

mcuee, 2023-11-25:
Please also provide more details of the iOS device used for testing by posting the run log of testlibusb and xusb. Thanks.

For xusb, please post the output of both `xusb 05ac:12ab` and `xusb -d 05ac:12ab`. The second command may crash based on your reports.

Amoystyle, 2023-11-25:
> Please also provide more details of the iOS device used for testing by posting the run log of testlibusb and xusb. Thanks.
> 
> For xusb, please post the output of both `xusb 05ac:12ab` and `xusb -d 05ac:12ab`. The second command may crash based on your reports.

**> .\testlibusb.exe**
```sh
Dev (bus 1, device 3): 8087 - 0025 speed: 12M
Dev (bus 1, device 2): 0C45 - 6741 speed: 12M
  Manufacturer:              SONiX
  Product:                   USB DEVICE
Dev (bus 1, device 17): 05AC - 12A8 speed: 480M
  Manufacturer:              Apple Inc.
  Product:                   iPhone
Dev (bus 1, device 1): 046D - C52F speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 1, device 0): 8086 - A36D speed: 5G
```

**>.\xusb.exe 05ac:12ab**
```sh
Using libusb v1.0.26.11813

Opening device 05AC:12AB...
  Failed.
 
```
**>.\xusb.exe -d 05ac:12ab**

```sh
 Using libusb v1.0.26.11813

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000233] [000054d0] libusb: debug [libusb_init_context] created default context
[ 0.000373] [000054d0] libusb: debug [libusb_init_context] libusb v1.0.26.11813
[ 0.000499] [000054d0] libusb: debug [usbi_add_event_source] add HANDLE 000000F8 events 0
[ 0.000629] [000054d0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000742] [000054d0] libusb: debug [usbi_add_event_source] add HANDLE 000000FC events 0
[ 0.000902] [000054d0] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001022] [000054d0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003712] [000054d0] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.006447] [000054d0] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.008926] [000054d0] libusb: info [windows_init] UsbDk backend is not available
Opening device 05AC:12AB...
[ 0.009222] [000054d0] libusb: debug [libusb_get_device_list]
[ 0.009732] [00009f28] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.020196] [000054d0] libusb: debug [winusb_get_device_list] allocating new device for session [4]
[ 0.020476] [000054d0] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB30\4&5375334&0&0' bus number 1
[ 0.021809] [000054d0] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.021981] [000054d0] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.022107] [000054d0] libusb: debug [winusb_get_device_list] allocating new device for session [32]
[ 0.022663] [000054d0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.022831] [000054d0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.022951] [000054d0] libusb: debug [winusb_get_device_list] allocating new device for session [69]
[ 0.023202] [000054d0] libusb: debug [get_api_type] driver(s): libusb0
[ 0.023362] [000054d0] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.023482] [000054d0] libusb: debug [winusb_get_device_list] allocating new device for session [6A]
[ 0.023738] [000054d0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.023903] [000054d0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.024032] [000054d0] libusb: debug [winusb_get_device_list] allocating new device for session [6B]
[ 0.025543] [000054d0] libusb: debug [winusb_get_device_list] found existing device for session [32]
[ 0.025737] [000054d0] libusb: debug [init_device] found 1 configurations (current config: 1) for device 'USB\VID_8087&PID_0025\5&38E97A59&0&11'
[ 0.025884] [000054d0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.026074] [000054d0] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 11): 'USB\VID_8087&PID_0025\5&38E97A59&0&11'
[ 0.026481] [000054d0] libusb: debug [winusb_get_device_list] found existing device for session [6B]
[ 0.026649] [000054d0] libusb: debug [init_device] found 1 configurations (current config: 1) for device 'USB\VID_0C45&PID_6741\5&38E97A59&0&8'
[ 0.026768] [000054d0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 66 bytes)
[ 0.026897] [000054d0] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 8): 'USB\VID_0C45&PID_6741\5&38E97A59&0&8'
[ 0.027098] [000054d0] libusb: debug [winusb_get_device_list] found existing device for session [6A]
[... 21 lines cut ...]
[ 0.032908] [000054d0] libusb: debug [set_composite_interface] interface[1] = \\?\HID#VID_046D&PID_C52F&MI_01&COL01#7&3E4D41C&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.033048] [000054d0] libusb: debug [winusb_get_device_list] setting composite interface for [6B]:
[ 0.033119] [000054d0] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_0C45&PID_6741&MI_01&COL02\7&34D74ED&0&0001
[ 0.033236] [000054d0] libusb: debug [winusb_get_device_list] setting composite interface for [69]:
[ 0.033309] [000054d0] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_046D&PID_C52F&MI_01&COL02\7&3E4D41C&0&0001
[ 0.033426] [000054d0] libusb: debug [winusb_get_device_list] setting composite interface for [6B]:
[ 0.033501] [000054d0] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_0C45&PID_6741&MI_01&COL03\7&34D74ED&0&0002
[... cut ...]
[ 0.035336] [000054d0] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.035387] [000054d0] libusb: debug [libusb_unref_device] destroy device 1.0
  Failed.
[ 0.035489] [000054d0] libusb: debug [libusb_exit] destroying default context
[ 0.035547] [00009f28] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.036010] [000054d0] libusb: debug [usbi_remove_event_source] remove HANDLE 000000FC
[ 0.036060] [000054d0] libusb: debug [usbi_remove_event_source] remove HANDLE 000000F8
```

Amoystyle, 2023-11-25:
> Please also provide more details of the iOS device used for testing by posting the run log of testlibusb and xusb. Thanks.
> 
> For xusb, please post the output of both `xusb 05ac:12ab` and `xusb -d 05ac:12ab`. The second command may crash based on your reports.

**iPhone_Composite_Parent.cat
iPhone_Composite_Parent.inf**

**[iphone_info_cat.zip](https://github.com/libusb/libusb/files/13462998/iphone_info_cat.zip)**

mcuee, 2023-11-25:
> **iPhone_Composite_Parent.cat iPhone_Composite_Parent.inf**
> 
> **[iphone_info_cat.zip](https://github.com/libusb/libusb/files/13462998/iphone_info_cat.zip)**

Hmm, how do you generate the driver package? It is using an old version of libusb-win32 device (1.0.26).

Please use Zadig 2.8 to install latest release version of libusb-win32 (1.2.7.3) or WinUSB driver and then try again.
Zadig 2.8 download:
https://github.com/pbatard/libwdi/releases/tag/v1.5.0

Amoystyle, 2023-11-25:
> > **iPhone_Composite_Parent.cat iPhone_Composite_Parent.inf**
> > **[iphone_info_cat.zip](https://github.com/libusb/libusb/files/13462998/iphone_info_cat.zip)**
> 
> Hmm, how do you generate the driver package? It is using an old version of libusb-win32 device (1.0.26) and it is an unsigned package.
> 
> Please use Zadig 2.8 to install latest release version of libusb-win32 (1.2.7.3) or WinUSB driver and then try again. Zadig 2.8 download: https://github.com/pbatard/libwdi/releases/tag/v1.5.0

Using dpinst64 to install.

[driver.zip](https://github.com/libusb/libusb/files/13463219/driver.zip)

tormodvolden, 2023-11-25:
> > Try if the issue persists with latest master. There has been changes to how options being handled.
> 
> I tried the latest master and still have this problem.
> 

If the memcpy is crashing, can you please print out the transfer->actual_length just before it?

It would also be nice if you could copy-paste the text that shows up in the screenshots. That is way more searchable and accessible for everybody.

tormodvolden, 2023-11-25:
I notice this in your log:
```
[ 0.034215] [000048f0] libusb: debug [windows_force_sync_completion] transfer 0132C740, length 12955841
[ 0.034219] [00004d0c] libusb: debug [windows_iocp_thread] transfer 0132C740 completed, length 5
```
The first value looks like something uninitialized or corrupted.

Amoystyle, 2023-11-25:
> Hmm, how do you generate the driver package? It is using an old version of libusb-win32 device (1.0.26) and it is an unsigned package.
>
> Please use Zadig 2.8 to install latest release version of [...]

I installed the latest released version of libusb-win32 (1.2.7.3) and tried again. 

**It worked well.**

It seems not compatible with libusb-win32 (1.2.6.0). 

 **╮(╯▽╰)╭**

closed 2023-11-25 by Amoystyle