libusb/libusb #1348: test/stress{,_mt} currently always operate on empty list of devices on the CI

RReverser, 2023-11-25:
I noticed that those tests run under `ci-container-build.sh` instead of regular `ci-build.sh` that runs on all platforms.

`ci-container-build.sh` seems to be intended for `umockdev`-based emulation, as it runs the primary `umockdev` test, but `stress` and `stress_mt` run without any emulated device, so they only test that retrieving an empty list of devices works. This is from CI logs:

```
+ tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 discovered 0 devices
Thread 1 discovered 0 devices
Thread 2 discovered 0 devices
Thread 3 discovered 0 devices
Thread 4 discovered 0 devices
Thread 5 discovered 0 devices
Thread 6 discovered 0 devices
Thread 7 discovered 0 devices
All done, 0 errors
```

It's probably an oversight? It seems worth it to either move those tests into the general `ci-build.sh` so that they run on other platforms (e.g. Windows), or to actually hook up umockdev with at least one device before running those tests.

mcuee, 2023-11-25:
A PR is welcome. Thanks.

RReverser, 2023-11-25:
If you're referring to emulating a device for those tests, I'm afraid that one would be a bit beyond my skills, as I never used umockdev before and it looks... tricky.

Mostly I raised an issue to find out why those tests don't live in the cross-platform CI script but in the umockdev one.

mcuee, 2023-11-25:
> If you're referring to emulating a device for those tests, I'm afraid that one would be a bit beyond my skills, as I never used umockdev before and it looks... tricky.
> 
> Mostly I raised an issue to find out why those tests don't live in the cross-platform CI script but in the umockdev one.

I believe https://github.com/libusb/libusb/blob/master/tests/umockdev.c is only meant for testing under Linux. 

As for stress and stress_mt, they are meant to be cross-platform. If you (or others) can get them to test emulated device (or devices) under CI, that will be an welcome enhancement.

mcuee, 2023-11-25:
And you are not limited by umockdev if there are better choices.

RReverser, 2023-11-25:
> As for stress and stress_mt, they are meant to be cross-platform. If you (or others) can get them to test emulated device (or devices) under CI, that will be an welcome enhancement.

Right, that would be the ideal scenario, but I was rather thinking that as short-term solution they could be just moved to cross-platform testing script. Since they operate on empty list of devices right now anyway, for Linux it won't degrade testing coverage, but other platforms will at least run them with empty list on CI too rather than skip those tests. Already a small improvement.

> And you are not limited by umockdev if there are better choices.

Yeah I know. I was looking into other options and saw that thread too, but it's a mess as there's not that many USB emulation options, let alone cross-platform ones.

I think the easiest way to make those tests cross-platform is to move to umockdev's `.ioctl` and `.umockdev` files. Might even reuse the ones they have prepared for their own tests upstream: https://github.com/martinpitt/umockdev/blob/11d6d3fc2a4f344884363c7ec0dde9dfe8072367/devices/cameras/canon-powershot-sx200.umockdev

This would allow to use umockdev emulation easily in a different process / address space than the test itself. That, in turn, allows to enable things like emulation for WebUSB (emulate device on the actual host, but run tests inside WebAssembly) or even emulation for Windows (emulate device on WSL subsystem, run tests via USB/IP).

It's still not pretty as it still relies on Linux being the one to do emulation, but at least it would enable running tests on couple more platforms.

RReverser, 2023-11-25:
For the sake of reference, the only other appealing option I came across was emulation via USB/IP: https://breaking-the-system.blogspot.com/2014/08/emulating-usb-devices-in-python-with-no.html

A repo implementing it with a Python interface is here: https://github.com/lcgamboa/USBIP-Virtual-USB-Device

This would probably be more cross-platform too, as USB/IP can run on many platforms.

Still, emulation testing is a big undertaking and not something I can work on under my current contract. Maybe if I find spare time, but for now just looking into basic stress tests.

mcuee, 2023-11-25:
> Right, that would be the ideal scenario, but I was rather thinking that as short-term solution they could be just moved to cross-platform testing script. Since they operate on empty list of devices [...]

I agree. This is already an improvement. A PR like this will be welcome. Thanks.

mcuee, 2023-11-25:
> Yeah I know. I was looking into other options and saw that thread too, but it's a mess as there's not that many USB emulation options, let alone cross-platform ones.
>
> I think the easiest way to make [...]

I think using Linux as the one to do emulation is fine. This would be an even better improvement.

RReverser, 2023-11-25:
> I agree. This is already an improvement. A PR like this will be welcome. Thanks.

Trying this low-hanging fruit for now in https://github.com/libusb/libusb/pull/1349.