libusb/libusb #1354: Missing values in `libusb_speed` and `libusb_supported_speed` enums

fabiensanglard, 2023-11-29:
I have noticed that enum [libusb_speed](https://github.com/libusb/libusb/blob/dfed0d169508d630cce9e6579f7cd69ac5ed25bf/libusb/libusb.h#L1126) does not support speed beyong 10000Mbps.

```
enum libusb_speed {
	/** The OS doesn't report or know the device speed. */
	LIBUSB_SPEED_UNKNOWN = 0,

	/** The device is operating at low speed (1.5MBit/s). */
	LIBUSB_SPEED_LOW = 1,

	/** The device is operating at full speed (12MBit/s). */
	LIBUSB_SPEED_FULL = 2,

	/** The device is operating at high speed (480MBit/s). */
	LIBUSB_SPEED_HIGH = 3,

	/** The device is operating at super speed (5000MBit/s). */
	LIBUSB_SPEED_SUPER = 4,

	/** The device is operating at super speed plus (10000MBit/s). */
	LIBUSB_SPEED_SUPER_PLUS = 5
};
```

Likewise, enum [libusb_supported_speed)(https://github.com/libusb/libusb/blob/dfed0d169508d630cce9e6579f7cd69ac5ed25bf/libusb/libusb.h#L487C6-L487C28) has not support beyond 5000Mbps.

```
enum libusb_supported_speed {
	/** Low speed operation supported (1.5MBit/s). */
	LIBUSB_LOW_SPEED_OPERATION = (1 << 0),

	/** Full speed operation supported (12MBit/s). */
	LIBUSB_FULL_SPEED_OPERATION = (1 << 1),

	/** High speed operation supported (480MBit/s). */
	LIBUSB_HIGH_SPEED_OPERATION = (1 << 2),

	/** Superspeed operation supported (5000MBit/s). */
	LIBUSB_SUPER_SPEED_OPERATION = (1 << 3)
};
```

Is it an oversight. Or something libusb has no access to?

tormodvolden, 2023-11-29:
See also #680

fabiensanglard, 2023-12-01:
I have made more testing with another setting.  The configuration I used was:

Mac M1 -> Caldigit TS4+ -> Pixel 6

Reported speed (via `libusb_get_device_speed`): `LIBUSB_SPEED_SUPER_PLUS` = 10,000

Reported max supported (via libusb_ss_usb_device_capability_descriptor): `LIBUSB_SUPER_SPEED_OPERATION`  = 5,000

The numbers indicate a speed superior to the maximum supported speed. Which make no sense.