libusb/libusb #1365: Windows LIBUSB_TRANSFER_ADD_ZERO_PACKET

HM281099, 2023-12-07:
When I tested USB asynchronous sending on Windows, the same OUT endpoint sent 4 packets of data

1. The first send is not an integer multiple of 1024

2. The second send is not an integer multiple of 1024

3. The third send is not an integer multiple of 1024

4. Send 1024 integer times for the fourth time. This packet failed to send data

Message: cannot change ZERO_PACKET for endpoint 1 on Windows

As I understand it, each asynchronous send is assigned a separate structure through the ibusb_alloc_transfer(0) interface, and adding the short packet identifier should not affect other transfers, otherwise the short packet identifier will not be placed in the libusb_transfer structure

tormodvolden, 2023-12-07:
Did you set the LIBUSB_TRANSFER_ADD_ZERO_PACKET flag for all transfers?

HM281099, 2023-12-08:
Windows https://github.com/libusb/libusb/pull/312 added LIBUSB_TRANSFER_ADD_ZERO_PACKET function

1. The IBUSB_TRANSFER_ADD_ZERO_PACKET identifier is specific to the Endpoint, not to each Transfer Request object.

2. On Windows, each endpoint has a maximum Packet length (MaxPacketSize) and a Zero Packet setting. When using libusb for USB data transfer, you can control whether zero packets are automatically added by setting the options for the transfer request. The LIBUSB_TRANSFER_ADD_ZERO_PACKET option is used to instruct libusb to automatically add zero packets when transferring data.

When the LIBUSB_TRANSFER_ADD_ZERO_PACKET option is set, libusb automatically adds zero packets during data transfer to ensure data integrity. This option is specific to the endpoint, not to each transport request object. Therefore, if multiple transport request objects use the same endpoint, they are all affected by the LIBUSB_TRANSFER_ADD_ZERO_PACKET option.

Is the above understanding correct?

HM281099, 2023-12-08:
@tormodvolden

HM281099, 2023-12-11:
No, such as there are two transfers,

1. LIBUSB_TRANSFER_ADD_ZERO_PACKET is not set for the first Transfer

2. LIBUSB_TRANSFER_ADD_ZERO_PACKET is not set for the second Transfer

3. Set LIBUSB_TRANSFER_ADD_ZERO_PACKET for the third Transfer

HM281099, 2023-12-12:
Thank you for your reply

tormodvolden, 2023-12-12:
Can you please confirm that everything works fine if you set LIBUSB_TRANSFER_ADD_ZERO_PACKET flag for all transfers?

We should probably improve the wording in the documentation.

HM281099, 2023-12-13:
Yes, all transfers are set LIBUSB_TRANSFER_ADD_ZERO_PACKET. works fine

HM281099, 2023-12-13:
@tormodvolden

tormodvolden, 2023-12-13:
I'll keep the ticket open to remind us to improve the documentation.

closed 2023-12-14 by HM281099