libusb/libusb #1372: 1.0.27-rc1 test matrix

tormodvolden, 2023-12-10:
For as many platforms as possible we want to test building and a run of the RC. At least xusb or lsusb -v should be tested. Please use the libusb-1.0.27-rc1.tar.bz2 tarball, and not a git checkout. Use the included configure script, no autogen or bootstrap needed.

Also more application testing is welcome.

## Major platforms

### Linux x86_64
- [x] build and `make check`: gcc 9.3 (Ubuntu 20.04)
- [x] build and `make check`: gcc 12.2 (Debian 12)
- [ ] gcc 11.2 (Arch Linux)
- [ ] run: xusb, lsusb, dfu-util, libfreenect
### Linux arm64 (Raspbian buster)
- [ ] build: gcc 8.3 OK (with and without udev)
- [ ] run: xusb, lsusb, dfu-util
### macOS 13.6 (Intel)
- [x] build and `make check`

### Windows 10 64-bit
- [ ] build: (appveyor does this)
- [ ] run: xusb, libfreenect (crossbuild MinGW)

### macOS 12 Intel, macOS 12 ARM64
(mcuee, see following posts)

## Other platforms
### Windows 32-bit
- [ ] build: (appveyor cross-build on 64-bit)
- [ ] run: dfu-util, lsusb - at least test 32-bit builds on 64-bit OS
### macOS 11.6.1 Intel
- [ ] build: Apple clang version 13.0.0 (built both for 11.6 and 10.7)
- [ ] run: lsusb, dfu-util, pyusb
### NetBSD, OpenBSD, Haiku
(mcuee, see following posts)

## Applications
- [ ] isochronous streaming with libfreenect on XBox - with WinUSB (!)
- [ ] isochronous IN & OUT streaming with custom device (WinUSB, Linux)
- [ ] HackRF HS bulk streaming (WinUSB, Linux)
- [ ] gutenberg printer drivers

RReverser, 2023-12-10:
> ## Other platforms
> ### Windows 32-bit
> ...
> ### macOS 11.6.1 Intel
> ...
> ### NetBSD, OpenBSD, Haiku
> ...

I know I tested it as much as I could myself, but if you want to add WebUSB to the list and other people (@mcuee ?) have spare time to try it out with their local OS & devices too, I'm happy to provide instructions. More testing is always better after all.

Shouldn't be too difficult now that there is a Node.js-based shim added in #1351.

tormodvolden, 2023-12-11:
@RReverser Maybe you can list some of the programs (if anything standard) or e.g. transfer types and devices you have been testing with.

RReverser, 2023-12-11:
I've been testing with my port of gphoto2 (https://github.com/GoogleChromeLabs/web-gphoto2) as well as private commercial project and @goncalomb was testing with their port of SANE (https://github.com/goncalomb/sane-wasm).

But I rather meant you could also run regular tests against same devices you normally use to test other platforms just built for the Wasm target.

Those in `examples` folder should definitely work (e.g. `xusb` that you mentioned above; the only exception is `hotplugtest` since it's not yet supported by WebUSB backend). They should behave identically to native versions and, if they don't, that's a bug.

tormodvolden, 2023-12-11:
I am using a modified STM32 audio-example (made class-less) to run isochronous transfer tests, but otherwise it doesn't matter so much what devices you use with e.g. xusb. Actually the improved stress_mt does more or less the same as xusb on a generic device, however xusb also tries to detach any kernel driver, which is handy.

tormodvolden, 2023-12-12:
@mcuee or @hjelmn Can you do some testing on macOS ARM? Also Windows 11 is out of reach for me.

The tests I have done are generally on enumeration, control transfers and (low-bandwidth) isochronous transfers. More testing on other applications and bulk transfers would be great.

RReverser, 2023-12-12:
> Actually the improved stress_mt does more or less the same as xusb on a generic device, however xusb also tries to detach any kernel driver, which is handy.

Ah yeah WebUSB won't be able to detach kernel driver and I didn't implement isochronous transfers yet, so I guess that leaves only stress_mt to at least cover enumeration + control transfers. Still better than nothing.

tormodvolden, 2023-12-12:
I can confirm that stress_mt fails on Windows 10, like in https://github.com/libusb/libusb/pull/1360#issuecomment-1840056655  even if some devices have WinUSB driver and should be fine to open. But I suppose that is a problem with the test and not the library.

RReverser, 2023-12-12:
> even if some devices have WinUSB driver and should be fine to open. But I suppose that is a problem with the test and not the library.

In #1360 I think it was the library - at least it should be able to somehow detect and return access error early on devices it doesn't have write access to. Idk if it's possible though.

Btw, we should really merge that PR before further testing as it had a fix to the stress_mt test itself where it would otherwise quietly continue on error.

mcuee, 2023-12-12:
Yes I will be able to carry out some basic tests for macOS ARM64 and Windows 11 X64 over the weekend.

tormodvolden, 2023-12-12:
On Windows I have no luck with the libusbK driver, but I checked that it was the same with 1.0.26.

goncalomb, 2023-12-12:
I'm not a libusb expert by any stretch, but for what is worth, I can attest that `libusb-1.0.27-rc1.tar.bz2` works as expected on my [sane-wasm](https://github.com/goncalomb/sane-wasm/tree/libusb-threads) project (a port of the SANE scanning library) that makes use of the recently updated WebUSB backend of libusb, including the multithreading machinery. @RReverser's work on the WebUSB backend is the sole reason for sane-wasm to exist.

Currently I'm limited to manually testing on a single device, a CanoScan LiDE 400, but I can say that this version is **much more stable and complete** than the previous one and **I don't see any pending issues**.

My libusb testing is limited to the webusb backend + sane-wasm. Here's a test scan from my test page as "proof":

[image]

For completeness, I also ran the provided tests on my Linux machine (pass):

    Linux goncalomb-pc 5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

[image]

RReverser, 2023-12-12:
> For completeness, I also ran the provided tests on my Linux machine (pass):

Thank you, this is promising! Just in case, can you merge in #1360 locally and run `stress_mt` again? Without the fixes in said PR, `stress_mt` can report success even when an error actually occured.

goncalomb, 2023-12-12:
Just ran the tests again on current master + #1360, no issues, all 4 PASS.

goncalomb, 2023-12-12:
If it helps, I also ran `make check` on my MacBook Air M2 2022 (arm64), macOS 13.5:

    Darwin Goncalos-Air.lan 22.6.0 Darwin Kernel Version 22.6.0: Wed Jul  5 22:17:35 PDT 2023; root:xnu-8796.141.3~6/RELEASE_ARM64_T8112 arm64

`libusb-1.0.27-rc1.tar.bz2`: 5 PASS
current master 87d6686a + #1360: 5 PASS

Let me know if I can help test this platform in some other way.

[image]

IceMark, 2023-12-13:
MacBook Air M1 2020 (arm64), macOS 14.2:

Darwin LukedeMacBook-Air.local 23.2.0 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103 arm64

Fail when perform make check:

ld: archive member '/' not a mach-o file in '/Users/luke/Downloads/libusb-1.0.27-rc1/libusb/.libs/libusb-1.0.a'

[libusb-1.0.27-rc1_fails_mac_m1_air.log.txt](https://github.com/libusb/libusb/files/13657130/libusb-1.0.27-rc1_fails_mac_m1_air.log.txt)

tormodvolden, 2023-12-13:
@IceMark What does `ar -t /Users/luke/Downloads/libusb-1.0.27-rc1/libusb/.libs/libusb-1.0.a` show? Please add `V=1` to the make invocation so that we can see the details of compiling and linking commands.

IceMark, 2023-12-15:
@tormodvolden , command result as below :
ar -t /Users/luke/Downloads/libusb-1.0.27-rc1/libusb/.libs/libusb-1.0.a
core.o
descriptor.o
hotplug.o
io.o
strerror.o
sync.o
events_posix.o
threads_posix.o
darwin_usb.o

=====
➜  libusb-1.0.27-rc1 make V=1 check 
Making check in libusb
make[1]: Entering directory '/Users/luke/Downloads/libusb-1.0.27-rc1/libusb'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/Users/luke/Downloads/libusb-1.0.27-rc1/libusb'
Making check in examples
make[1]: Entering directory '/Users/luke/Downloads/libusb-1.0.27-rc1/examples'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/Users/luke/Downloads/libusb-1.0.27-rc1/examples'
Making check in tests
make[1]: Entering directory '/Users/luke/Downloads/libusb-1.0.27-rc1/tests'
/bin/sh ../libtool  --tag=CC   --mode=link gcc -std=gnu11  -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -falign-functions=8 -falign-functions=8  -static -L/opt/homebrew/opt/openblas/lib -o stress stress.o testlib.o ../libusb/libusb-1.0.la -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security
libtool: link: gcc -std=gnu11 -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -falign-functions=8 -falign-functions=8 -o stress stress.o testlib.o -Wl,-framework -Wl,IOKit -Wl,-framework -Wl,CoreFoundation -Wl,-framework -Wl,Security  -L/opt/homebrew/opt/openblas/lib ../libusb/.libs/libusb-1.0.a -lobjc -pthread
ld: archive member '/' not a mach-o file in '/Users/luke/Downloads/libusb-1.0.27-rc1/libusb/.libs/libusb-1.0.a'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:647: stress] Error 1
make[1]: Leaving directory '/Users/luke/Downloads/libusb-1.0.27-rc1/tests'
make: *** [Makefile:491: check-recursive] Error 1

tormodvolden, 2023-12-15:
@IceMark Can you please open a new issue and move it there? The ar -t output looks fine. I am unsure about the openblas entries in your linking command. Maybe you have some stray LD_FLAGS in your environment.

mcuee, 2023-12-15:
I plan to carry out some tests over the weekend and next week

## Test platforms

### Linux x86_64
- [] build and `make check`: (Ubuntu 20.04)
  - [] run: xusb, lsusb, hotplugtest, stress, stress_mt, libusbdotnet benchmark

### Linux ARM32 (Raspberry Pi OS on RPi 400)
- [] build and `make check` (with udev)
    - [] run: xusb, lsusb, hotplugtest, stress, stress_mt, libusbdotnet benchmark
- [] build and `make check` (with netlink)
    - [] run: xusb, lsusb, hotplugtest, stress, stress_mt, libusbdotnet benchmark

### macOS 14.2 (ARM64, mac Mini M1 2020)
- [] build and `make check`
    - [] run: xusb, lsusb, hotplugtest, stress, stress_mt, libusbdotnet benchmark

### Windows 11 64-bit
- [] build: (MSVC64, MSYS2 mingw64 and MSYS2 clang64)
    - [] run: xusb, lsusb, stress, stress_mt, libusbdotnet benchmark

## Other platforms

### NetBSD virtuL machine

### OpenBSD virtual machine

### Haiku virtual machine

## Applications
- [] high speed and super speed bulk transfer IN & OUT streaming with custom device - with WinUSB/libusb0,sys/libusbK.sys/Linux/macOS using libusbdotnet benchmark application
- [] high speed and super speed isochronous IN & OUT streaming with custom device - with WinUSB/libusb0,sys/libusbK.sys/Linux/macOS  using libusbdotnet benchmark
- [] pyusb simple tests
- [] avrdude simple tests using usbasp programmer

mcuee, 2023-12-15:
First thing is that there are quite some warnings for MYS2 mingw6 build. We can see that from github action build log. I can also see that from my local MSYS2 mingw64 build.
https://github.com/libusb/libusb/actions/runs/7159098003/job/19491987746

```
MINGW64 /c/work/libusb/libusb-1.0.27-rc1/mingw64
$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-13.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/include --libexecdir=/mingw64/lib --enable-bootstrap --enable-checking=release --with-arch=nocona --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --enable-lto --enable-libgomp --disable-libssp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev2, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --disable-libstdcxx-debug --with-boot-ldflags=-static-libstdc++ --with-stage1-ldflags=-static-libstdc++
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (Rev2, Built by MSYS2 project)

$ make
make  all-recursive
make[1]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64'
Making all in libusb
make[2]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_windows.lo
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
  CC       os/windows_usbdk.lo
  CC       os/windows_winusb.lo
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3750:11:
../../libusb/os/windows_winusb.c:3563:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 3563 |         id->bLength = LIBUSB_DT_INTERFACE_SIZE;
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3531:14: note: at offset 9 into object 'tmp' of size 41
 3531 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3750:11:
../../libusb/os/windows_winusb.c:3564:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 3564 |         id->bDescriptorType = LIBUSB_DT_INTERFACE;
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3531:14: note: at offset 9 into object 'tmp' of size 41
 3531 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3750:11:
../../libusb/os/windows_winusb.c:3565:30: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 3565 |         id->bInterfaceNumber = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3531:14: note: at offset 9 into object 'tmp' of size 41
 3531 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3750:11:
../../libusb/os/windows_winusb.c:3566:31: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 3566 |         id->bAlternateSetting = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3531:14: note: at offset 9 into object 'tmp' of size 41
 3531 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3750:11:
../../libusb/os/windows_winusb.c:3567:27: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
[... 166 lines cut ...]
  CC       listdevs.o
  CCLD     listdevs.exe
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark.exe
  CC       testlibusb.o
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
  CC       stress_mt-stress_mt.o
  CCLD     stress_mt.exe
  CC       set_option.o
  CCLD     set_option.exe
  CC       init_context.o
  CCLD     init_context.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
make[2]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64'
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64'
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64'
```

mcuee, 2023-12-15:
@tormodvolden 

`make check` failed under my Windows 11 laptop.

```
MINGW64 /c/work/libusb/libusb-1.0.27-rc1/mingw64
$ make check
Making check in libusb
make[1]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/libusb'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/libusb'
Making check in examples
make[1]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/examples'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/examples'
Making check in tests
make[1]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
make  check-TESTS
make[2]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
make[3]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
PASS: stress.exe
FAIL: stress_mt.exe
PASS: set_option.exe
PASS: init_context.exe
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
make[3]: *** [Makefile:800: test-suite.log] Error 1
make[3]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
make[2]: *** [Makefile:908: check-TESTS] Error 2
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
make[1]: *** [Makefile:1015: check-am] Error 2
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/mingw64/tests'
make: *** [Makefile:491: check-recursive] Error 1
```

tormodvolden, 2023-12-15:
> First thing is that there are quite some warnings for MYS2 mingw6 build.

Yes, we have had these array-bounds warning for a while, so it is nothing new and not to worry about for the release, but it would be nice to get rid of them later.

tormodvolden, 2023-12-15:
> `make check` failed under my Windows 11 laptop because `stress_mt` failed.

This is probably the same that we have seen in #1360.

mcuee, 2023-12-15:
> > `make check` failed under my Windows 11 laptop because `stress_mt` failed.
> 
> This is probably the same that we have seen in #1360.

Yes I believe this is the same as #1360.
* https://github.com/libusb/libusb/pull/1360#issuecomment-1857900471

> @mcuee is LIBUSB_ERROR_NOT_SUPPORTED something we can expect to see with some devices/drivers if libusb cannot use them?

Yes, I think so.

Just an example run log of `testlibusb` under Windows, you can see quite debug messages about `libusb_open` returns `-12` which is  `LIBUSB_ERROR_NOT_SUPPORTED`.

<details>
<summary> testlibusb debug log with LIBUSB_DEBUG=4 under Windows 11</summary>

```
 MINGW64 /c/work/libusb/libusb_pr1360
$ export LIBUSB_DEBUG=4

$ ./examples/testlibusb.exe -v
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001160] [00004608] libusb: debug [libusb_init_context] created default context
[ 0.001484] [00004608] libusb: debug [libusb_init_context] libusb v1.0.27.11850-rc1
[ 0.001784] [00004608] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b4 events 0
[ 0.002088] [00004608] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.002393] [00004608] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b8 events 0
[ 0.002514] [00004608] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.002628] [00004608] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004348] [00004608] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.006438] [00004608] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.008356] [00004608] libusb: info [windows_init] UsbDk backend is not available
[ 0.008569] [00004608] libusb: debug [libusb_get_device_list]
[ 0.008667] [000027fc] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.017605] [00004608] libusb: debug [winusb_get_device_list] ENUM pass HUB {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.017945] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_05E3&PID_0612\5&1A00B1C&0&2
[ 0.018172] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.018405] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB30\4&12C539F&0&0
[ 0.018575] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.018811] [00004608] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB30\4&12C539F&0&0' bus number 1
[ 0.018957] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB30\4&1A63DC7C&0&0
[ 0.019131] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.019376] [00004608] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB30\4&1A63DC7C&0&0' bus number 2
[ 0.019462] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_05E3&PID_0610\5&586B51A&0&2
[ 0.019620] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [8]
[ 0.020038] [00004608] libusb: debug [winusb_get_device_list] ENUM pass DEV {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.020284] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_1C7A&PID_0575\077E2F9A
[ 0.020432] [00004608] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.020511] [00004608] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.020560] [00004608] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.020623] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [20]
[ 0.020803] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_8087&PID_0026\5&586B51A&0&10
[ 0.020957] [00004608] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.021035] [00004608] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.021087] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [41]
[ 0.021263] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_1EA7&PID_0064\5&586B51A&0&3
[ 0.021389] [00004608] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.021471] [00004608] libusb: debug [get_api_type] matched driver name against HID API
[ 0.021522] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [6A]
[ 0.021583] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_04F2&PID_B6DD\0001
[ 0.021701] [00004608] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.021767] [00004608] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.021819] [00004608] libusb: debug [winusb_get_device_list] allocating new device for session [6B]
[ 0.021939] [00004608] libusb: debug [winusb_get_device_list] ENUM pass HCD {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.022156] [00004608] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0
[ 0.022232] [00004608] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68
[ 0.022448] [00004608] libusb: debug [winusb_get_device_list] ENUM pass GEN
[ 0.022905] [00004608] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_05E3&PID_0612\5&1A00B1C&0&2
[... cut ...]
[ 0.571934] [00004608] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.571982] [00004608] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.572028] [00004608] libusb: debug [libusb_exit] destroying default context
[ 0.572084] [000027fc] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.572466] [00004608] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b8
[ 0.572543] [00004608] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b4
```

</details>

tormodvolden, 2023-12-15:
LIBUSB_ERROR_ACCESS is fine, and we special-case it in stress_mt, but can we have LIBUSB_ERROR_NOT_FOUND other than in device (or driver) detachment situations?

mcuee, 2023-12-15:
> LIBUSB_ERROR_ACCESS is fine, and we special-case it in stress_mt, but can we have LIBUSB_ERROR_NOT_FOUND other than in device (or driver) detachment situations?
> 
> EDIT, sorry saw that now: [ 0.286946] [00004608] libusb: debug [libusb_open] open 1.2 returns -5
> 
> What kind of device is 04F2:B6DD ?

Indeed this is a bit strange. I tend to think it has something to do with USB Composite Device handling under Windows. There are a few pending PR and pending issues on that front.

It is the built-in USB Webcam of my Acer Swift 3 Windows 11 laptop.

Detailed info from USB Device Tree Viewer

<details>
<summary> detailed info from USB Device Tree Viewer for USB Webcam 04f2:b6dd".

```

    =========================== USB Port5 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 1-5

      ========================== Summary =========================
Vendor ID                : 0x04F2 (Chicony Electronics Co., Ltd.)
Product ID               : 0xB6DD
USB Version              : 2.0.1
Port maximum Speed       : High-Speed
Device maximum Speed     : High-Speed
Device Connection Speed  : High-Speed
Self powered             : no
Demanded Current         : 500 mA
Used Endpoints           : 2

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_04F2&PID_B6DD#0001#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-4
Device ID                : USB\VID_04F2&PID_B6DD\0001
Hardware IDs             : USB\VID_04F2&PID_B6DD&REV_2710 USB\VID_04F2&PID_B6DD
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0022 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.22621.2506  Date: 2023-11-15)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0005.Hub_#0002
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(5), ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS05)
Container ID             : {00000000-0000-0000-ffff-ffffffffffff} (GUID_CONTAINERID_INTERNALLY_CONNECTED_DEVICE)
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x90 (UniqueID, SurpriseRemovalOK)
Status                   : 0x0180200A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 5
Power State              : D3 (supported: D0, D3, wake from D0)
 Child Device 1          : HD User Facing (USB Video Device)
  Device Path 1          : \\?\USB#VID_04F2&PID_B6DD&MI_00#6&2f8d90e&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_CAPTURE)
  Device Path 2          : \\?\USB#VID_04F2&PID_B6DD&MI_00#6&2f8d90e&0&0000#{6994ad05-93ef-11d0-a3cc-00a0c9223196}\global (AM_KSCATEGORY_VIDEO)
  Device Path 3          : \\?\USB#VID_04F2&PID_B6DD&MI_00#6&2f8d90e&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global (STATIC_KSCATEGORY_VIDEO_CAMERA)
  Kernel Name            : \Device\000000b9
  Device ID              : USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000
  Class                  : Camera
  Driver KeyName         : {ca3e7ab9-b4c3-4ae6-8251-579ef933890f}\0000 (GUID_DEVCLASS_CAMERA)
  Service                : usbvideo
  Location               : 0000.0014.0000.005.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(5)#USBMI(0)  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(5)#USB(5)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS05)#ACPI(UCAM)

[... 693 lines cut ...]

    ----------------- Device Qualifier Descriptor -----------------
Error                    : ERROR_GEN_FAILURE  (because the device is in low power state)

      ---------- Binary Object Store (BOS) Descriptor -----------
bLength                  : 0x05 (5 bytes)
bDescriptorType          : 0x0F (Binary Object Store)
wTotalLength             : 0x0021 (33 bytes)
bNumDeviceCaps           : 0x01
Data (HexDump)           : 05 0F 21 00 01                                    ..!..

        ----------- Platform Capabilities Descriptor -----------
bLength                  : 0x1C (28 bytes)
bDescriptorType          : 0x10 (Device Capability Descriptor)
bDevCapabilityType       : 0x05 (Platform Capabilities)
bReserved                : 0x00
PlatformCapabilityUUID   : D8DD60DF-4589-4CC7-9CD2-659D9E648A9F
Data (HexDump)           : 1C 10 05 00 DF 60 DD D8 89 45 C7 4C 9C D2 65 9D   .....`...E.L..e.
                           9E 64 8A 9F 00 00 00 0A E8 00 5F 00               .d........_.

      -------------------- String Descriptors -------------------
String descriptors are not available  (because the device is in low power state)
```

</details>

mcuee, 2023-12-15:
I tend to think we should ignore devices which failed to be opened, like what `testlibusb` is doing.

tormodvolden, 2023-12-15:
> I tend to think we should ignore devices which fail to be opened, like what `testlibusb` is doing.

Yes, I agree this would make sense for stress_mt. But if we know it is limited to above errors on Windows we can also add these to the special-casing. It seems to work fine on other platforms.

RReverser, 2023-12-15:
Yeah it's rather unfortunate to have potentially multiple error codes for ~the same underlying issue, as it can means downstream consumers might need to deal with confusing error codes too (especially LIBUSB_ERROR_NOT_FOUND is pretty confusing for a device that's actually there).

But as an interim solution special-casing couple of error codes is still better than blanket ignore for any libusb_open errors in tests.

mcuee, 2023-12-15:
BTW, `-3 LIBUSB_ERROR_ACCESS` is kind of expected as it is a USB finger print sensor.

```
Dev (bus 1, device 3): 1C7A - 0575 speed: 480M
[ 0.427757] [00004608] libusb: debug [libusb_open] open 1.3
[ 0.428114] [00004608] libusb: error [winusbx_open] could not open device \\?\USB#VID_1C7A&PID_0575#077E2F9A#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [5] Access is denied.
[ 0.428229] [00004608] libusb: debug [libusb_open] open 1.3 returns -3
[ 0.428288] [00004608] libusb: debug [libusb_get_config_descriptor] index 0
```

USB Device Tree Viewer reports the same issue as the USB Webcam. Somehow libusb reports different error codes.

```
    ----------------- Device Qualifier Descriptor -----------------
Error                    : ERROR_GEN_FAILURE  (because the device is in low power state)

      -------------------- String Descriptors -------------------
String descriptors are not available  (because the device is in low power state)
```

RReverser, 2023-12-15:
> BTW, `-3 LIBUSB_ERROR_ACCESS` is kind of expected for `17ca:0575` as it is a USB finger print sensor.

Sure, LIBUSB_ERROR_ACCESS is perfectly reasonable and that's what we get on other platforms.

mcuee, 2023-12-17:
No issues with `make check` under macOS Sonoma 14.2, mac Mini M1 2020.

```
mcuee@mcuees-Mac-mini libusb_git % make check
Making check in libusb
make[1]: Nothing to be done for `check'.
Making check in examples
make[1]: Nothing to be done for `check'.
Making check in tests
/Library/Developer/CommandLineTools/usr/bin/make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
PASS: macos
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 5
# PASS:  5
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
make[1]: Nothing to be done for `check-am'.
```

mcuee, 2023-12-17:
@tormodvolden and @hjelmn 

However I encountered segmentation fault with the kernel detaching function under macOS 14.2 using sudo.
```
mcuee@mcuees-Mac-mini arm64 % sudo ./examples/xusb 0925:7001
Password:
Using libusb v1.0.27.11848

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x01): "Microchip Technology Inc."
   String (0x02): "Generic HID"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
zsh: segmentation fault  sudo ./examples/xusb 0925:7001
```

mcuee, 2023-12-17:
Full debug log.

<details>
<summary> Full debug log for xusb crash with sudo </summary>

```
mcuee@mcuees-Mac-mini x64 % sudo ./examples/xusb -d 0925:7001
Using libusb v1.0.27.11848

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000060] [00038ba6] libusb: debug [libusb_init_context] created default context
[ 0.000077] [00038ba6] libusb: debug [libusb_init_context] libusb v1.0.27.11848-rc1
[ 0.000089] [00038ba6] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000167] [00038ba9] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000414] [00038ba9] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000553] [00038ba6] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x682bc9b
[ 0.000574] [00038ba6] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x682bc9b
[ 0.002655] [00038ba6] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002661] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002663] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002665] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002666] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002667] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002668] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002669] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002671] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002672] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002673] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002698] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002702] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002704] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002721] [00038ba6] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002724] [00038ba6] libusb: debug [process_new_device] allocating new device in context 0x7fa303f05180 for with session 0x682bc9b
[ 0.002750] [00038ba6] libusb: debug [process_new_device] found device with address 1 port = 6 parent = 0x0 at 0x60000169863a
[ 0.002794] [00038ba6] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x699e7c7
[ 0.002809] [00038ba6] libusb: debug [darwin_get_cached_device] parent sessionID: 0x682bc9b
[ 0.002811] [00038ba6] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x699e7c7/0x2610000 against cached device with sessionID/locationID 0x682bc9b/0x2600000
[ 0.002813] [00038ba6] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x699e7c7
[ 0.002938] [00038ba6] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002942] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002943] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002945] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002946] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002947] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002949] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002950] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002951] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002952] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002953] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002968] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002970] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002972] [00038ba6] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002983] [00038ba6] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002985] [00038ba6] libusb: debug [process_new_device] allocating new device in context 0x7fa303f05180 for with session 0x699e7c7
[ 0.002987] [00038ba6] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x600001c94120 at 0x60000168c23a
[ 0.003023] [00038ba6] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x6a8a9e9
[ 0.003036] [00038ba6] libusb: debug [darwin_get_cached_device] parent sessionID: 0x699e7c7
[ 0.003038] [00038ba6] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x6a8a9e9/0x2611000 against cached device with sessionID/locationID 0x699e7c7/0x2610000
[... cut ...]
[ 0.093329] [00038ba9] libusb: debug [process_new_device] re-using existing device from context 0x7fa303f05180 for with session 0x124d6c1926 new session 0x1271b4abe4
[ 0.093332] [00038ba9] libusb: debug [process_new_device] found device with address 25 port = 2 parent = 0x600001c900c0 at 0x600001698a3a
[ 0.093336] [00038ba9] libusb: debug [darwin_devices_attached] cached device in reset state. reset complete...
[ 0.093338] [00038ba6] libusb: debug [darwin_reenumerate_device] darwin/reenumerate_device: checking whether descriptors changed
zsh: segmentation fault  sudo ./examples/xusb -d 0925:7001
```

</details>

mcuee, 2023-12-17:
No issues if I do not use sudo.

<details>
<summary> Full debug log for xusb without sudo - no crash </summary>

```
mcuee@mcuees-Mac-mini x64 % ./examples/xusb -d 0925:7001
Using libusb v1.0.27.11848

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000152] [00039167] libusb: debug [libusb_init_context] created default context
[ 0.000172] [00039167] libusb: debug [libusb_init_context] libusb v1.0.27.11848-rc1
[ 0.000244] [00039167] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000365] [0003916a] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000581] [0003916a] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000693] [00039167] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x682bc9b
[ 0.000716] [00039167] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x682bc9b
[ 0.003387] [00039167] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003393] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003395] [00039167] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003396] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003397] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003399] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003400] [00039167] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003401] [00039167] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003402] [00039167] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003403] [00039167] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003405] [00039167] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003406] [00039167] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003407] [00039167] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003408] [00039167] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003421] [00039167] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003423] [00039167] libusb: debug [process_new_device] allocating new device in context 0x7fb4b0705180 for with session 0x682bc9b
[ 0.003483] [00039167] libusb: debug [process_new_device] found device with address 1 port = 6 parent = 0x0 at 0x60000186913a
[ 0.003523] [00039167] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x699e7c7
[ 0.003539] [00039167] libusb: debug [darwin_get_cached_device] parent sessionID: 0x682bc9b
[ 0.003541] [00039167] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x699e7c7/0x2610000 against cached device with sessionID/locationID 0x682bc9b/0x2600000
[ 0.003543] [00039167] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x699e7c7
[ 0.003670] [00039167] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003673] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003675] [00039167] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003676] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003678] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003679] [00039167] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003680] [00039167] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003681] [00039167] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003683] [00039167] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003684] [00039167] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003685] [00039167] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003687] [00039167] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003688] [00039167] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003689] [00039167] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003697] [00039167] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003700] [00039167] libusb: debug [process_new_device] allocating new device in context 0x7fb4b0705180 for with session 0x699e7c7
[ 0.003702] [00039167] libusb: debug [process_new_device] found device with address 2 port = 1 parent = 0x6000012681e0 at 0x60000186c13a
[ 0.003736] [00039167] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x6a8a9e9
[ 0.003749] [00039167] libusb: debug [darwin_get_cached_device] parent sessionID: 0x699e7c7
[ 0.003751] [00039167] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x6a8a9e9/0x2611000 against cached device with sessionID/locationID 0x699e7c7/0x2610000
[... cut ...]
[ 0.007380] [0003919f] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.007382] [0003919f] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.007383] [0003919f] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.007385] [0003919f] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.007386] [0003919f] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.007399] [000391a1] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.007966] [0003919f] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

mcuee, 2023-12-17:
No issues with Ubuntu 20.04 x64.

RReverser, 2023-12-17:
> BTW, it is not as easy to reproduce under ARM64 build but easier to reproduce under x64 (Rosetta 2 emulation)

Have you tried building it with `-fsanitize=address` (for both library & the test itself)? It should make it easier to pinpoint the exact cause.

mcuee, 2023-12-17:
> Yeah it's rather unfortunate to have potentially multiple error codes for ~the same underlying issue, as it can means downstream consumers might need to deal with confusing error codes too [...]

@RReverser and @tormodvolden 

I agree that LIBUSB_ERROR_NOT_FOUND is confusing. I tend to think it is related to the Windows issue of handling certain USB composite devices.

As an interim solution and for 1.0.27 release, I think PR #1360 is good enough.

mcuee, 2023-12-17:
> > BTW, it is not as easy to reproduce under ARM64 build but easier to reproduce under x64 (Rosetta 2 emulation)
> 
> Have you tried building it with `-fsanitize=address` (for both library & the test itself)? It should make it easier to pinpoint the exact cause.

Sorry I am not familiar with this one. Please help to provide more details so that I can carry out the debugging. I think I need to use lldb under macOS to get the backtrace.

RReverser, 2023-12-17:
> Please help to provide more details so that I can carry out the debugging.

Just set `CFLAGS="-g - fsanitize=address"` when compiling the library. It enables AddressSanitizer, which automatically will tell you why the invalid address is accessed and will print a backtrace.

mcuee, 2023-12-17:
> > Please help to provide more details so that I can carry out the debugging.
> 
> Just set `CFLAGS="-g - fsanitize=address"` when compiling the library. It enables AddressSanitizer, which automatically will tell you why the invalid address is accessed and will print a backtrace.

Unfortunately this does not work. Apparently Apple clang does not support this feature.

<details>
<summary> Failed build log </summary>

```
mcuee@mcuees-Mac-mini libusb-1.0.27-rc1 % cd arm64_debug                 
mcuee@mcuees-Mac-mini arm64_debug % CFLAGS="-g - fsanitize=address" ../configure --enable-tests-build --enable-examples-build
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/Users/mcuee/build/libusb/libusb-1.0.27-rc1/arm64_debug':
configure: error: C compiler cannot create executables
See `config.log' for more details
mcuee@mcuees-Mac-mini arm64_debug % cat config.log 
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by libusb-1.0 configure 1.0.27-rc1, which was
generated by GNU Autoconf 2.71.  Invocation command line was

  $ ../configure --enable-tests-build --enable-examples-build

## --------- ##
## Platform. ##
## --------- ##

hostname = mcuees-Mac-mini
uname -m = arm64
uname -r = 23.2.0
uname -s = Darwin
uname -v = Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103

/usr/bin/uname -p = arm
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = Mach kernel version:
	 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103
Kernel configured for up to 8 processors.
8 processors are physically available.
8 processors are logically available.
Processor type: arm64e (ARM64E)
Processors active: 0 1 2 3 4 5 6 7
Primary memory available: 8.00 gigabytes
Default processor set: 623 tasks, 2359 threads, 8 processors
Load average: 1.65, Mach factor: 6.33
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /opt/homebrew/opt/qt@5/bin/
PATH: /Applications/ARM/bin/
PATH: /opt/homebrew/bin/
PATH: /opt/homebrew/sbin/
PATH: /usr/local/bin/
PATH: /System/Cryptexes/App/usr/bin/
PATH: /usr/bin/
PATH: /bin/
PATH: /usr/sbin/
PATH: /sbin/
PATH: /var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin/
PATH: /var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin/
PATH: /var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin/
PATH: /opt/X11/bin/
[... 293 lines cut ...]
prefix='NONE'
program_transform_name='s,x,x,'
psdir='${docdir}'
runstatedir='${localstatedir}/run'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "libusb-1.0"
#define PACKAGE_TARNAME "libusb-1.0"
#define PACKAGE_VERSION "1.0.27-rc1"
#define PACKAGE_STRING "libusb-1.0 1.0.27-rc1"
#define PACKAGE_BUGREPORT "libusb-devel@lists.sourceforge.net"
#define PACKAGE_URL "http://libusb.info"

configure: exit 77
```

</details>

RReverser, 2023-12-17:
It definitely does, I think they pioneered that feature before it went into other compilers.

Anyway, the error in your log says:

> clang: error: no such file or directory: 'fsanitize=address'; did you mean '-fsanitize=address'?

It's because I'm writing from my phone, that space between dash and parameter wasn't supposed to be there.

mcuee, 2023-12-17:
> It definitely does, I think they pioneered that feature before it went into other compilers.
>
> Anyway, the error in your log says:
>
>
> It's because I'm writing from my phone, that space between dash [...]

I see. I will carry out the test tomorrow.

mcuee, 2023-12-19:
> I see. I will carry out the test tomorrow.

Here is the log under ARM64. Now it seems to be easy to reproduce the crash everytime I run it.

```
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 0925:7001
Password:
Using libusb v1.0.27.11848

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x01): "Microchip Technology Inc."
   String (0x02): "Generic HID"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
=================================================================
==1066==ERROR: AddressSanitizer: heap-use-after-free on address 0x000106200360 at pc 0x0001033e4d40 bp 0x00016cf65e50 sp 0x00016cf65e48
READ of size 8 at 0x000106200360 thread T0
    #0 0x1033e4d3c in darwin_reenumerate_device darwin_usb.c:2118
    #1 0x1033d1a54 in darwin_attach_kernel_driver darwin_usb.c:2844
    #2 0x1033cefb4 in darwin_capture_release_interface darwin_usb.c:2869
    #3 0x1033ab0f4 in libusb_release_interface core.c:1832
    #4 0x102e9e99c in test_device xusb.c:1027
    #5 0x102e9af90 in main xusb.c:1185
    #6 0x1828e90dc  (<unknown module>)

0x000106200360 is located 32 bytes inside of 128-byte region [0x000106200340,0x0001062003c0)
freed by thread T1 here:
    #0 0x1038a5bcc in wrap__ZdlPv+0x74 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x61bcc)
    #1 0x102edb490 in IOUSBIUnknown::release()+0x48 (IOUSBLib:arm64e+0x3490)
    #2 0x1033d63c4 in darwin_devices_detached darwin_usb.c:656
    #3 0x1863a5210 in IODispatchCalloutFromCFMessage+0xc0 (IOKit:arm64e+0x6210)
    #4 0x182d77490 in __CFMachPortPerform+0x124 (CoreFoundation:arm64e+0xad490)
    #5 0x182d47fd8 in __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__+0x38 (CoreFoundation:arm64e+0x7dfd8)
[... 42 lines cut ...]
  0x000106200480: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000106200500: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000106200580: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1066==ABORTING
zsh: abort      sudo ./examples/xusb 0925:7001
```

mcuee, 2023-12-19:
@hjelmn and @tormodvolden,
With the suggestion from @RReverser, I can easily reproduce the issue under macOS (Mac Mini M1 2020, up to date macOS 14.2). Please take a look and see if you can reproduce the issue or not under your system. Thanks.

Another USB HID device -- same problem.

```
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 046d:c52b
Using libusb v1.0.27.11848

Opening device 046D:C52B...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 046D:C52B
         bcdDevice: 2410
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 84
         descriptor length: 9
             nb interfaces: 3
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 08
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 82
           max packet size: 0008
          polling interval: 02
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 83
           max packet size: 0020
          polling interval: 02

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Kernel driver attached for interface 1: none

Claiming interface 1...

Kernel driver attached for interface 2: none

Claiming interface 2...

Reading string descriptors:
   String (0x01): "Logitech"
   String (0x02): "USB Receiver"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
=================================================================
==1153==ERROR: AddressSanitizer: heap-use-after-free on address 0x0001061004e0 at pc 0x0001032e8d40 bp 0x00016d061e50 sp 0x00016d061e48
[... 59 lines cut ...]
  0x000106100600: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000106100680: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000106100700: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1153==ABORTING
zsh: abort      sudo ./examples/xusb 046d:c52b
```

mcuee, 2023-12-19:
Another device -- USB card reader and it has the same issue.

```
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.27.11848

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
  00000030  6e 00 64 00 45 00 6e 00 61 00 62 00 6c 00 65 00  n.d.E.n.a.b.l.e.
  00000040  64 00 00 00 04 00 00 00 00 00 00 00              d...........

Reading interface association descriptors (IADs) for first configuration:
[... 78 lines cut ...]
  0x000106000300: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000106000380: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000106000400: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1307==ABORTING
zsh: abort      sudo ./examples/xusb 05e3:0749
```

mcuee, 2023-12-19:
I have created a new issue for the macOS kernel driver detaching crashing issue.
* #1386

mcuee, 2023-12-20:
@tormodvolden 

Just wondering if we can downgrade the warning messages related to `no DeviceInterfaceGUID registered' to info.

```
MINGW64 /c/work/libusb/libusb-1.0.27-rc1/mingw64
$ ./examples/xusb.exe 0424:2740
Using libusb v1.0.27.11848

Opening device 0424:2740...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_5744\C&1936BD58&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\D&127B8A6A&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_2744\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&36020D6F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842\0123456789ABCD'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\9&AF4BBD4&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\7&2452366F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E&MI_00\D&E651DD6&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056\BB305534D79EBD418B6757528EAC0C19'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_00\6&1BD5FF13&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_8153\111000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06F\C&1F76A113&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_00\6&360CBCF3&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD\200901010001'

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0424:2740
         bcdDevice: 0200
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: 1 caps
    USB 2.0 extension:
      attributes             : A21E

Reading first configuration descriptor:
              total length: 18
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.FF.FF

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Microchip Tech"
   String (0x02): "Hub Controller"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
   Failed: Pipe error
Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
```

mcuee, 2023-12-20:
@tormodvolden 
`make check` issue due to `stress_mt` failure under Windows has been fixed in latest git main. Tested under Windows 10.

libusb-1.0.27-rc1
```
MINGW64 /c/work/libusb/libusb-1.0.27-rc1/mingw64
$ ./tests/stress_mt.exe
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 1: device 0: mismatch in field bcdUSB: 512 != 272
Thread 6: device 0: mismatch in field bcdUSB: 512 != 272
Thread 3: device 0: mismatch in field bcdUSB: 512 != 272
Thread 0: device 0: mismatch in field bcdUSB: 512 != 272
Thread 5: device 0: mismatch in field bcdUSB: 512 != 272
Thread 4: device 0: mismatch in field bcdUSB: 512 != 272
Thread 2: device 0: mismatch in field bcdUSB: 512 != 272
Thread 7: device 0: mismatch in field bcdUSB: 512 != 272
Thread 0 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 1 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 2 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 3 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 4 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 5 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 6 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 7 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
All threads discovered 0 devices (0 not opened)
All done, 8 errors
```

git main:
```
MINGW64 /c/work/libusb/libusb
$ ./tests/stress_mt.exe
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
No access to device 0424:5744, skipping transfer tests.
No access to device 0bda:0413, skipping transfer tests.
No access to device 0bda:5413, skipping transfer tests.
No access to device 0424:2740, skipping transfer tests.
No access to device 1a86:7523, skipping transfer tests.
No access to device 8087:0aaa, skipping transfer tests.
No access to device 0424:2744, skipping transfer tests.
No access to device 0bda:5487, skipping transfer tests.
No access to device 0bda:0487, skipping transfer tests.
No access to device 0bda:402e, skipping transfer tests.
No access to device 8086:a36d, skipping transfer tests.
No access to device 0a5c:5842, skipping transfer tests.
No access to device 8086:15f0, skipping transfer tests.
No access to device 8086:15db, skipping transfer tests.
No access to device 0bda:8153, skipping transfer tests.
No access to device 0bda:58fd, skipping transfer tests.
All threads discovered 23 devices (16 not opened)
All done, 0 errors
```

mcuee, 2023-12-20:
For Haiku OS, the build issue is still there, similar to 1.0.26 release.

```
~/Desktop/build/libusb-1.0.27-rc1> make -i
make  all-recursive
make[1]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1'
Making all in libusb
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/libusb'
  CXX      os/haiku_usb_backend.lo
cc1plus: warning: '-Werror=' argument '-Werror=implicit-function-declaration' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=missing-prototypes' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=strict-prototypes' is not valid for C++
In file included from ./libusbi.h:106,
                 from os/haiku_usb.h:25,
                 from os/haiku_usb_backend.cpp:27:
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:40:9: error: '_Atomic' does not name a type
   40 | typedef _Atomic _Bool atomic_bool;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:41:9: error: '_Atomic' does not name a type
   41 | typedef _Atomic char atomic_char;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:42:9: error: '_Atomic' does not name a type
   42 | typedef _Atomic signed char atomic_schar;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:43:9: error: '_Atomic' does not name a type
   43 | typedef _Atomic unsigned char atomic_uchar;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:44:9: error: '_Atomic' does not name a type
   44 | typedef _Atomic short atomic_short;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:45:9: error: '_Atomic' does not name a type
   45 | typedef _Atomic unsigned short atomic_ushort;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:46:9: error: '_Atomic' does not name a type
   46 | typedef _Atomic int atomic_int;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:47:9: error: '_Atomic' does not name a type
   47 | typedef _Atomic unsigned int atomic_uint;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:48:9: error: '_Atomic' does not name a type
   48 | typedef _Atomic long atomic_long;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:49:9: error: '_Atomic' does not name a type
   49 | typedef _Atomic unsigned long atomic_ulong;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:50:9: error: '_Atomic' does not name a type
   50 | typedef _Atomic long long atomic_llong;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:51:9: error: '_Atomic' does not name a type
   51 | typedef _Atomic unsigned long long atomic_ullong;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:52:9: error: '_Atomic' does not name a type
   52 | typedef _Atomic __CHAR16_TYPE__ atomic_char16_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:53:9: error: '_Atomic' does not name a type
   53 | typedef _Atomic __CHAR32_TYPE__ atomic_char32_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:54:9: error: '_Atomic' does not name a type
   54 | typedef _Atomic __WCHAR_TYPE__ atomic_wchar_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:55:9: error: '_Atomic' does not name a type
   55 | typedef _Atomic __INT_LEAST8_TYPE__ atomic_int_least8_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:56:9: error: '_Atomic' does not name a type
   56 | typedef _Atomic __UINT_LEAST8_TYPE__ atomic_uint_least8_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:57:9: error: '_Atomic' does not name a type
   57 | typedef _Atomic __INT_LEAST16_TYPE__ atomic_int_least16_t;
      |         ^~~~~~~
[... 417 lines cut ...]
      |         sig_atomic_t
./libusbi.h:513:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  513 |         usbi_atomic_t attached;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
Makefile:708: recipe for target 'os/haiku_usb_raw.lo' failed
make[2]: [os/haiku_usb_raw.lo] Error 1 (ignored)
  CXXLD    libusb_haiku.la
libtool:   error: 'os/haiku_usb_backend.lo' is not a valid libtool object
Makefile:625: recipe for target 'libusb_haiku.la' failed
make[2]: [libusb_haiku.la] Error 1 (ignored)
  CCLD     libusb-1.0.la
libtool:   error: cannot find the library 'libusb_haiku.la' or unhandled argument 'libusb_haiku.la'
Makefile:613: recipe for target 'libusb-1.0.la' failed
make[2]: [libusb-1.0.la] Error 1 (ignored)
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/libusb'
Making all in examples
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/examples'
  CC       dpfp.o
make[2]: *** No rule to make target '../libusb/libusb-1.0.la', needed by 'dpfp'.  Stop.
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/examples'
Makefile:491: recipe for target 'all-recursive' failed
make[1]: [all-recursive] Error 1 (ignored)
make[1]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1'
```

mcuee, 2023-12-20:
@tormodvolden 

`make check` failed under OpenBSD due to similar issues like Windows. It is just the error code is now `LIBUSB_ERROR_OTHER.
git HEAD will still fail `make check` for OpenBSD since we have not dealt with the error code.

```
openbsdvmn100$ sudo make check
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
FAIL: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 ..
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
*** Error 1 in tests (Makefile:900 'test-suite.log': @case $- in *e*) set +e;; esac; {  mgn= red= grn= lgn= blu= brg= std=;  am__color_tests...)
*** Error 2 in tests (Makefile:911 'check-TESTS': @set +e; bases='stress.log stress_mt.log set_option.log init_context.log';  bases=`for i i...)
*** Error 2 in tests (Makefile:1013 'check-am')
*** Error 1 in /home/mcuee/build/libusb (Makefile:514 'check-recursive': @fail=;  if (target_option=k; case ${target_option-} in  ?) ;;  *) ...)
openbsdvmn100$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 3): LIBUSB_ERROR_OTHER
Thread 1 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 2 failed (iteration 2): LIBUSB_ERROR_OTHER
Device count mismatch: Thread 3 discovered 3 devices instead of 0
Thread 4 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 5 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 6 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 7 failed (iteration 1): LIBUSB_ERROR_OTHER
All done, 8 errors
```

mcuee, 2023-12-20:
@tormodvolden 

`make check` failed under NetBSD due to similar issues like Windows. It is just the error code is now `LIBUSB_ERROR_IO`.
git HEAD will still fail `make check` under NetBSD since we have not dealt with the error code.

```
localhost$ sudo make check
Password:
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
FAIL: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
*** Error code 1

Stop.
make[3]: stopped in /home/mcuee/build/libusb/tests
*** Error code 1

Stop.
make[2]: stopped in /home/mcuee/build/libusb/tests
*** Error code 1

Stop.
make[1]: stopped in /home/mcuee/build/libusb/tests
*** Error code 1

Stop.
make: stopped in /home/mcuee/build/libusb
localhost$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 21): LIBUSB_ERROR_IO
Device count mismatch: Thread 1 discovered 1 devices instead of 0
All done, 2 errors
```

mcuee, 2023-12-21:
@braewoods and @jclulow

Just wondering if you can test the Solaris backend. Thanks.

mcuee, 2023-12-21:
> @tormodvolden `make check` issue due to `stress_mt` failure under Windows has been fixed in latest [git main](https://github.com/libusb/libusb/commit/37dee8f9dd429558e4d5bc4a804ab586abe7262a). Tested under Windows 10.
> 
> libusb-1.0.27-rc1
> 

This Windows laptop (with docks and external monitor) is a bit complicated with multiple level of hubs and USB devices connected.

```
MINGW64 /c/work/libusb/libusb
$ ./tests/stress_mt.exe
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
No access to device 16c0:05dc, skipping transfer tests.
No access to device 0424:5744, skipping transfer tests.
No access to device 0bda:0413, skipping transfer tests.
No access to device 0bda:5413, skipping transfer tests.
No access to device 0424:2740, skipping transfer tests.
No access to device 1a86:7523, skipping transfer tests.
No access to device 8087:0aaa, skipping transfer tests.
No access to device 0424:2744, skipping transfer tests.
No access to device 0bda:5487, skipping transfer tests.
No access to device 0bda:0487, skipping transfer tests.
No access to device 0bda:402e, skipping transfer tests.
No access to device 8086:a36d, skipping transfer tests.
No access to device 0a5c:5842, skipping transfer tests.
No access to device 8086:15f0, skipping transfer tests.
No access to device 0b95:772b, skipping transfer tests.
No access to device 8086:15db, skipping transfer tests.
No access to device 0bda:8153, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 0bda:58fd, skipping transfer tests.
All threads discovered 26 devices (19 not opened)
All done, 0 errors

$ ./examples/testlibusb.exe
Dev (bus 2, device 8): 16C0 - 05DC speed: 1.5M
  Manufacturer:              www.fischl.de
  Product:                   USBasp
Dev (bus 1, device 9): 24AE - 4057 speed: 1.5M
  Manufacturer:              SEMICO
  Product:                   USB Keyboard
Dev (bus 1, device 5): 0424 - 5744 speed: 5G
Dev (bus 1, device 6): 0BDA - 5413 speed: 480M
Dev (bus 1, device 3): 0BDA - 0413 speed: 5G
Dev (bus 1, device 27): 0424 - 2740 speed: 480M
  Manufacturer:              Microchip Tech
  Product:                   Hub Controller
Dev (bus 1, device 7): 413C - B06E speed: 480M
Dev (bus 1, device 25): 1A86 - 7523 speed: 12M
Dev (bus 2, device 5): 8087 - 0AAA speed: 12M
Dev (bus 1, device 8): 0424 - 2744 speed: 480M
Dev (bus 1, device 2): 0BDA - 5487 speed: 480M
Dev (bus 1, device 1): 0BDA - 0487 speed: 10G
Dev (bus 2, device 3): 2717 - 5013 speed: 12M
  Manufacturer:              MI
  Product:                   Mi Wireless Mouse
Dev (bus 1, device 26): 0BDA - 402E speed: 480M
Dev (bus 2, device 0): 8086 - A36D speed: 5G
Dev (bus 1, device 13): 413C - 2107 speed: 1.5M
  Manufacturer:              DELL
  Product:                   Dell USB Entry Keyboard
Dev (bus 2, device 4): 0A5C - 5842 speed: 480M
Dev (bus 1, device 0): 8086 - 15F0 speed: 10G
Dev (bus 2, device 7): 0B95 - 772B speed: 480M
Dev (bus 3, device 0): 8086 - 15DB speed: 5G
Dev (bus 2, device 12): 047F - C056 speed: 12M
  Manufacturer:              Plantronics
  Product:                   Plantronics Blackwire 3220 Series
Dev (bus 1, device 4): 0BDA - 8153 speed: 5G
Dev (bus 1, device 11): 413C - B06F speed: 480M
Dev (bus 1, device 14): 046D - C077 speed: 1.5M
  Manufacturer:              Logitech
  Product:                   USB Optical Mouse
Dev (bus 2, device 1): 05E3 - 0608 speed: 480M
Dev (bus 2, device 6): 0BDA - 58FD speed: 480M

$ ./examples/listdevs.exe
16c0:05dc (bus 2, device 8) path: 2.2
24ae:4057 (bus 1, device 9) path: 2.3.3
0424:5744 (bus 1, device 5) path: 4.3.2
0bda:5413 (bus 1, device 6) path: 2.3
0bda:0413 (bus 1, device 3) path: 4.3
0424:2740 (bus 1, device 27) path: 2.3.2.5
413c:b06e (bus 1, device 7) path: 2.5
1a86:7523 (bus 1, device 25) path: 2.3.2.2
8087:0aaa (bus 2, device 5) path: 14
0424:2744 (bus 1, device 8) path: 2.3.2
0bda:5487 (bus 1, device 2) path: 2
0bda:0487 (bus 1, device 1) path: 4
2717:5013 (bus 2, device 3) path: 3
0bda:402e (bus 1, device 26) path: 2.3.4
8086:a36d (bus 2, device 0)
413c:2107 (bus 1, device 13) path: 2.3.2.3
0a5c:5842 (bus 2, device 4) path: 10
8086:15f0 (bus 1, device 0)
0b95:772b (bus 2, device 7) path: 2.1
8086:15db (bus 3, device 0)
047f:c056 (bus 2, device 12) path: 1
0bda:8153 (bus 1, device 4) path: 4.4
413c:b06f (bus 1, device 11) path: 2.3.5
046d:c077 (bus 1, device 14) path: 2.3.2.4
05e3:0608 (bus 2, device 1) path: 2
0bda:58fd (bus 2, device 6) path: 11
```

mcuee, 2023-12-21:
@tormodvolden 

I am not so sure if this is a real bug or not but it seems to me that xusb has some issues to read the string descriptors for some devices under Windows.

```
C:\work\libusb\libusb-1.0.27-rc1\build\v142\x64\Release> .\xusb.exe 0456:b630
Using libusb v1.0.27.11848

Opening device 0456:B630...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_5744\C&1936BD58&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\D&127B8A6A&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_2744\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&36020D6F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842\0123456789ABCD'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\9&AF4BBD4&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0B95&PID_772B\C1B13D'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\7&2452366F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E&MI_00\D&E651DD6&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056\BB305534D79EBD418B6757528EAC0C19'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_00\6&1BD5FF13&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_8153\111000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06F\C&1F76A113&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_00\6&360CBCF3&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0608\5&E9F3E45&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD\200901010001'

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0456:B630
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.00.00
       endpoint[0].address: 85
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 06
           max packet size: 0200
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

[... 671 lines cut ...]
Language 0x0076          : "Blackfin USB Device"
Data (HexDump)           : 28 03 42 00 6C 00 61 00 63 00 6B 00 66 00 69 00   (.B.l.a.c.k.f.i.
                           6E 00 20 00 55 00 53 00 42 00 20 00 44 00 65 00   n. .U.S.B. .D.e.
                           76 00 69 00 63 00 65 00                           v.i.c.e.
bLength                  : 0x28 (40 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0069          : "Blackfin USB Device"
Data (HexDump)           : 28 03 42 00 6C 00 61 00 63 00 6B 00 66 00 69 00   (.B.l.a.c.k.f.i.
                           6E 00 20 00 55 00 53 00 42 00 20 00 44 00 65 00   n. .U.S.B. .D.e.
                           76 00 69 00 63 00 65 00                           v.i.c.e.
bLength                  : 0x28 (40 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0063          : "Blackfin USB Device"
[... cut ...]
                           76 00 69 00 63 00 65 00                           v.i.c.e.
bLength                  : 0x28 (40 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0065          : "Blackfin USB Device"
Data (HexDump)           : 28 03 42 00 6C 00 61 00 63 00 6B 00 66 00 69 00   (.B.l.a.c.k.f.i.
                           6E 00 20 00 55 00 53 00 42 00 20 00 44 00 65 00   n. .U.S.B. .D.e.
                           76 00 69 00 63 00 65 00                           v.i.c.e.
```

</details>

mcuee, 2023-12-21:
> > @tormodvolden `make check` issue due to `stress_mt` failure under Windows has been fixed in latest [git main](https://github.com/libusb/libusb/commit/37dee8f9dd429558e4d5bc4a804ab586abe7262a). Tested under Windows 10.
> 
> This Windows laptop (with docking station and external monitor with USB Hub) is a bit complicated with multiple level of hubs and USB devices connected. git HEAD is okay now.
> 
> ```
> MINGW64 /c/work/libusb/libusb
> $ ./tests/stress_mt.exe
> Running multithreaded init/exit test...
> Starting 8 threads
> Running multithreaded init/exit test with enumeration...
> Starting 8 threads
> No access to device 16c0:05dc, skipping transfer tests.
> No access to device 0424:5744, skipping transfer tests.
> No access to device 0bda:0413, skipping transfer tests.
> No access to device 0bda:5413, skipping transfer tests.
> No access to device 0424:2740, skipping transfer tests.
> No access to device 1a86:7523, skipping transfer tests.
> No access to device 8087:0aaa, skipping transfer tests.
> No access to device 0424:2744, skipping transfer tests.
> No access to device 0bda:5487, skipping transfer tests.
> No access to device 0bda:0487, skipping transfer tests.
> No access to device 0bda:402e, skipping transfer tests.
> No access to device 8086:a36d, skipping transfer tests.
> No access to device 0a5c:5842, skipping transfer tests.
> No access to device 8086:15f0, skipping transfer tests.
> No access to device 0b95:772b, skipping transfer tests.
> No access to device 8086:15db, skipping transfer tests.
> No access to device 0bda:8153, skipping transfer tests.
> No access to device 05e3:0608, skipping transfer tests.
> No access to device 0bda:58fd, skipping transfer tests.
> All threads discovered 26 devices (19 not opened)
> All done, 0 errors
> 
> $ ./examples/testlibusb.exe
> Dev (bus 2, device 8): 16C0 - 05DC speed: 1.5M
>   Manufacturer:              www.fischl.de
>   Product:                   USBasp
> Dev (bus 1, device 9): 24AE - 4057 speed: 1.5M
>   Manufacturer:              SEMICO
>   Product:                   USB Keyboard
> Dev (bus 1, device 5): 0424 - 5744 speed: 5G
> Dev (bus 1, device 6): 0BDA - 5413 speed: 480M
> Dev (bus 1, device 3): 0BDA - 0413 speed: 5G
> Dev (bus 1, device 27): 0424 - 2740 speed: 480M
>   Manufacturer:              Microchip Tech
>   Product:                   Hub Controller
> Dev (bus 1, device 7): 413C - B06E speed: 480M
> Dev (bus 1, device 25): 1A86 - 7523 speed: 12M
> Dev (bus 2, device 5): 8087 - 0AAA speed: 12M
> Dev (bus 1, device 8): 0424 - 2744 speed: 480M
> Dev (bus 1, device 2): 0BDA - 5487 speed: 480M
> Dev (bus 1, device 1): 0BDA - 0487 speed: 10G
> Dev (bus 2, device 3): 2717 - 5013 speed: 12M
>   Manufacturer:              MI
>   Product:                   Mi Wireless Mouse
> Dev (bus 1, device 26): 0BDA - 402E speed: 480M
> Dev (bus 2, device 0): 8086 - A36D speed: 5G
> Dev (bus 1, device 13): 413C - 2107 speed: 1.5M
>   Manufacturer:              DELL
>   Product:                   Dell USB Entry Keyboard
> Dev (bus 2, device 4): 0A5C - 5842 speed: 480M
> Dev (bus 1, device 0): 8086 - 15F0 speed: 10G
> Dev (bus 2, device 7): 0B95 - 772B speed: 480M
> Dev (bus 3, device 0): 8086 - 15DB speed: 5G
> Dev (bus 2, device 12): 047F - C056 speed: 12M
>   Manufacturer:              Plantronics
>   Product:                   Plantronics Blackwire 3220 Series
> Dev (bus 1, device 4): 0BDA - 8153 speed: 5G
> Dev (bus 1, device 11): 413C - B06F speed: 480M
> Dev (bus 1, device 14): 046D - C077 speed: 1.5M
[... 66 lines cut ...]
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
No access to device 16c0:05dc, skipping transfer tests.
No access to device 0424:5744, skipping transfer tests.
No access to device 0bda:0413, skipping transfer tests.
No access to device 0bda:5413, skipping transfer tests.
No access to device 1a86:7523, skipping transfer tests.
No access to device 8087:0aaa, skipping transfer tests.
No access to device 0424:2744, skipping transfer tests.
No access to device 0bda:5487, skipping transfer tests.
No access to device 8086:15db, skipping transfer tests.
No access to device 0bda:402e, skipping transfer tests.
No access to device 8086:a36d, skipping transfer tests.
No access to device 0a5c:5842, skipping transfer tests.
No access to device 8086:15f0, skipping transfer tests.
No access to device 0b95:772b, skipping transfer tests.
No access to device 0bda:0487, skipping transfer tests.
No access to device 0bda:8153, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 0bda:58fd, skipping transfer tests.
All threads discovered 26 devices (18 not opened)
All done, 0 errors
```

mcuee, 2023-12-21:
For VS2019 x64 build, it is quite consistent to show 18 devices not opened.
```
PS C:\work\libusb\libusb\build\v142\x64\Release> .\stress_mt.exe
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
No access to device 16c0:05dc, skipping transfer tests.
No access to device 0424:5744, skipping transfer tests.
No access to device 0bda:5413, skipping transfer tests.
No access to device 0bda:0413, skipping transfer tests.
No access to device 1a86:7523, skipping transfer tests.
No access to device 8087:0aaa, skipping transfer tests.
No access to device 0bda:0487, skipping transfer tests.
No access to device 0bda:5487, skipping transfer tests.
No access to device 0424:2744, skipping transfer tests.
No access to device 0bda:402e, skipping transfer tests.
No access to device 8086:a36d, skipping transfer tests.
No access to device 0a5c:5842, skipping transfer tests.
No access to device 8086:15f0, skipping transfer tests.
No access to device 0b95:772b, skipping transfer tests.
No access to device 8086:15db, skipping transfer tests.
No access to device 0bda:8153, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 0bda:58fd, skipping transfer tests.
All threads discovered 26 devices (18 not opened)
All done, 0 errors
```

mcuee, 2023-12-24:
[libusbdotnet v2 BenchmarkCon](https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2/stage/BenchmarkCon) seems to work with libusb-1.0.27-rc1 for SuperSpeed Bulk transfer, tested using Cypress FX3 bulkloop demo firmware, with either libusbK driver or WinUSB driver.

Example run log using libusbK driver
```
PS C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\BenchmarkCon\bin\Release\net6.0> .\BenchmarkCon.exe list notestselect driver=libusb10 verify
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 05E3:0612
2. 8087:0026
3. 1EA7:0064 2.4G Mouse
4. 04F2:B6DD
5. 1C7A:0575
6. 8086:A0ED
7. 04B4:00F0 Cypress - FX3
8. 05E3:0610
9. 8086:9A13

Select device (1-9) :7

Benchmark device 04B4:00F0 opened..
Loop Test Information
        Vid / Pid       : 04B4h / 00F0h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : On

Bulk Read (Ep81h) max packet size: 512
Bulk Write (Ep01h) max packet size: 512

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 6754884.60 Transfers: 1651 Bytes/s: 6754884.60
Avg. Bytes/s: 7018418.90 Transfers: 3445 Bytes/s: 7279793.22
Avg. Bytes/s: 7485090.03 Transfers: 5503 Bytes/s: 8422565.48
Avg. Bytes/s: 7862730.53 Transfers: 7703 Bytes/s: 8998315.31
Avg. Bytes/s: 7760199.75 Transfers: 9518 Bytes/s: 7353247.65
Avg. Bytes/s: 7812359.44 Transfers: 11491 Bytes/s: 8074164.67
Avg. Bytes/s: 8069166.41 Transfers: 13840 Bytes/s: 9615364.59
Avg. Bytes/s: 8075020.09 Transfers: 15824 Bytes/s: 8116091.63
Avg. Bytes/s: 8288501.48 Transfers: 18270 Bytes/s: 9998572.89
Avg. Bytes/s: 8301827.06 Transfers: 20332 Bytes/s: 8421794.92
Avg. Bytes/s: 8360026.74 Transfers: 22520 Bytes/s: 8942589.82
qAvg. Bytes/s: 8447320.32 Transfers: 24825 Bytes/s: 9406992.45
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
        Vid / Pid       : 04B4h / 00F0h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : On

Bulk Read (Ep81h) max packet size: 512
[... 171 lines cut ...]
        Alt Interface # : 00h
        Num Endpoints   : 2
        Priority        : 0
        Read Size       : 4096
        Write Size      : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : On

Bulk Read (Ep81h) Maximum Packet Size: 512
        Total Bytes     : 176209920
        Total Transfers : 43020
        Avg. Bytes/sec  : 11698972.25
        Elapsed Time    : 15.06 seconds

Bulk Write (Ep01h) Maximum Packet Size: 512
        Total Bytes     : 176209920
        Total Transfers : 43020
        Avg. Bytes/sec  : 11698972.25
        Elapsed Time    : 15.06 seconds

Press any key to exit..
```

mcuee, 2023-12-24:
I also tested with libusb0.sys 1.3.0.1.
```
PS C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\BenchmarkCon\bin\Release\net6.0> .\BenchmarkCon.exe list notestselect driver=libusb10 verify
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 05E3:0612
2. 8087:0026
3. 1EA7:0064 2.4G Mouse
4. 04F2:B6DD
5. 1C7A:0575
6. 8086:A0ED
7. 04B4:00F0 Cypress - FX3
8. 05E3:0610
9. 8086:9A13

Select device (1-9) :7

Benchmark device 04B4:00F0 opened..
Loop Test Information
        Vid / Pid       : 04B4h / 00F0h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : On

Bulk Read (Ep81h) max packet size: 512
Bulk Write (Ep01h) max packet size: 512

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 7619606.13 Transfers: 1850 Bytes/s: 7619606.13
Avg. Bytes/s: 8218831.84 Transfers: 4023 Bytes/s: 8808593.43
Avg. Bytes/s: 8558992.65 Transfers: 6301 Bytes/s: 9233919.30
Avg. Bytes/s: 8878686.08 Transfers: 8709 Bytes/s: 9840475.86
Avg. Bytes/s: 8985699.26 Transfers: 11036 Bytes/s: 9410180.41
Avg. Bytes/s: 8963868.91 Transfers: 13199 Bytes/s: 8854117.72
Avg. Bytes/s: 9202917.42 Transfers: 15804 Bytes/s: 10640703.15
Avg. Bytes/s: 9160317.82 Transfers: 17971 Bytes/s: 8861174.47
Avg. Bytes/s: 9112734.38 Transfers: 20104 Bytes/s: 8730638.22
Avg. Bytes/s: 9120804.22 Transfers: 22353 Bytes/s: 9193581.19
Avg. Bytes/s: 9139832.51 Transfers: 24660 Bytes/s: 9328397.79
Avg. Bytes/s: 9166292.55 Transfers: 26972 Bytes/s: 9458353.22
Avg. Bytes/s: 9136030.88 Transfers: 29116 Bytes/s: 8771719.86
Avg. Bytes/s: 9160575.63 Transfers: 31434 Bytes/s: 9480502.54
Avg. Bytes/s: 9218737.80 Transfers: 33889 Bytes/s: 10034496.18
qAvg. Bytes/s: 9233392.63 Transfers: 36200 Bytes/s: 9453773.13
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
        Vid / Pid       : 04B4h / 00F0h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : On

Bulk Read (Ep81h) max packet size: 512
        Total Bytes     : 148279296
        Total Transfers : 36201
        Avg. Bytes/sec  : 9233224.69
        Elapsed Time    : 16.06 seconds

Bulk Write (Ep01h) max packet size: 512
        Total Bytes     : 148279296
        Total Transfers : 36201
        Avg. Bytes/sec  : 9233230.32
        Elapsed Time    : 16.06 seconds

Press any key to exit..
```

mcuee, 2023-12-24:
Test with Cypress BulkSrcSink FW example (not loop back).

```
PS C:\work\libusb\libusbdotnet\libusbdotnet_v2\stage\BenchmarkCon\bin\Release\net6.0> .\BenchmarkCon.exe list notestselect driver=libusb-win32 buffercount=3 buffersize=393216 mode=async
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 04B4:00F1 Cypress - FX3

Select device (1-1) :1

Benchmark device 04B4:00F1 opened..
Loop Test Information
        Vid / Pid       : 04B4h / 00F1h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 393216
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
Bulk Write (Ep01h) max packet size: 1024

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 358953693.07 Transfers: 913 Bytes/s: 358953693.07
Avg. Bytes/s: 358739045.62 Transfers: 1834 Bytes/s: 358526515.87
Avg. Bytes/s: 359009794.80 Transfers: 2750 Bytes/s: 359553114.43
Avg. Bytes/s: 361168306.18 Transfers: 3688 Bytes/s: 367648844.88
Avg. Bytes/s: 360595584.76 Transfers: 4610 Bytes/s: 358322748.40
Avg. Bytes/s: 360261386.32 Transfers: 5526 Bytes/s: 358588809.02
Avg. Bytes/s: 359517694.40 Transfers: 6432 Bytes/s: 355047316.23
Avg. Bytes/s: 358925350.68 Transfers: 7347 Bytes/s: 354815920.93
Avg. Bytes/s: 358682578.28 Transfers: 8257 Bytes/s: 356734488.71
Avg. Bytes/s: 358642684.83 Transfers: 9170 Bytes/s: 358282298.62
Avg. Bytes/s: 358098601.34 Transfers: 10069 Bytes/s: 352641690.41
Avg. Bytes/s: 357813672.64 Transfers: 10974 Bytes/s: 354673883.06
Avg. Bytes/s: 357747083.78 Transfers: 11882 Bytes/s: 356944252.77
Avg. Bytes/s: 357720241.47 Transfers: 12792 Bytes/s: 357370126.87
Avg. Bytes/s: 357762652.96 Transfers: 13706 Bytes/s: 358357285.18
qAvg. Bytes/s: 357767055.83 Transfers: 14618 Bytes/s: 357833237.59
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
        Vid / Pid       : 04B4h / 00F1h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 393216
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep81h) max packet size: 1024
        Total Bytes     : 5748424704
        Total Transfers : 14619
        Avg. Bytes/sec  : 357767347.17
        Elapsed Time    : 16.07 seconds

[... 64 lines cut ...]
        Alt Interface # : 00h
        Num Endpoints   : 2
        Priority        : 0
        Read Size       : 393216
        Write Size      : 393216
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep81h) Maximum Packet Size:1024
        Total Bytes     : 5031591936
        Total Transfers : 12796
        Avg. Bytes/sec  : 357408150.02
        Elapsed Time    : 14.08 seconds

Bulk Write (Ep01h) Maximum Packet Size:1024
        Total Bytes     : 3302227968
        Total Transfers : 8398
        Avg. Bytes/sec  : 234566555.48
        Elapsed Time    : 14.08 seconds

Press any key to exit..
```

[13 more replies]

closed 2024-01-31 by tormodvolden