libusb/libusb #1373: stress_mt fails when there is no USB device

aurel32, 2023-12-10:
I have been trying to package the 1.0.27rc1 release in Debian and upload it to experimental. Unfortunately I have found that the new test `stress_mt` fails when there is no USB device (because of minimum `/dev` filesystem in the chroot):

```
Thread 0 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 1 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 2 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 3 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 4 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 5 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 6 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Thread 7 failed (iteration 1): LIBUSB_ERROR_NO_DEVICE
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 0 devices (0 not opened)
All done, 8 errors
FAIL stress_mt (exit status: 1)
```

Youw, 2023-12-10:
Is there a question or suggestion here?
This particualr test doesn't make sense if there is no devices in the system.

aurel32, 2023-12-10:
> Is there a question or suggestion here? This particualr test doesn't make sense if there is no devices in the system.

I think the test should exit without failure if it is not possible to run the tests. Basically catch the `LIBUSB_ERROR_NO_DEVICE` error in `init_and_exit`  to return no error and return 0 devices discovered instead.

I can have a look at improving the test in the next days.

Youw, 2023-12-10:
I'd rather suggest having a way to skip particular tests in some known environments explicitly on demand.
My oppinion: if the test successfull does nothing (even if only under some conditions) - that is way accidentally miss a bug or smth.

I imagine if you're trying to package the libusb, you have full access to its build scripts invocation, and since you know that a particular test is known to fail doe to specific reason (a minimum filesystem in the chroot of your env) - it would only make sense to disable the test explicitly (maybe specifying that exact reason in the comment).

@tormodvolden thoughts?

aurel32, 2023-12-10:
Ok, I have added a patch in the Debian packaging to disable the test.

closed 2023-12-10 by aurel32

tormodvolden, 2023-12-10:
I think it should be totally fine to run the test out-of-the-box on a system with no discoverable USB devices, and it shouldn't fail. Of course, that would not catch the error if there actually are devices that should be available but libusb fails to detect them. But that is not the focus of the stress_mt test.

Note that we do have a CI_WITHOUT_DEVICES flag that keeps test_no_discovery() in tests/set_option.c from failing if it is run on a host without discoverable USB devices. But that is a build flag, so it should not be used on a build chroot. Maybe it should be turned into a run-time flag. Then it could apply to stress_mt as well, and be used on the build chroot.

tormodvolden, 2023-12-10:
@RReverser What do you think about making CI_WITHOUT_DEVICES a run-time option (e.g. environment variable)?

RReverser, 2023-12-10:
> What do you think about making CI_WITHOUT_DEVICES a run-time option (e.g. environment variable)?

Sure, makes sense. I probably should've done so from the start but figured the CI is the only place where someone would want to run tests yet not have any devices attached.