libusb/libusb #1375: LIBUSB_ERROR_TIMEOUT where libusb seems to get stuck for roughly 168 seconds

azlamzn, 2023-12-12:
OS: 
```
Ubuntu 20.04.4 LTS
```

Libraries:
```
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64 [installed,automatic]
libccid/focal,now 1.4.31-1 amd64 [installed,automatic]
libpcsclite1/focal,now 1.8.26-3 amd64 [installed,automatic]
```

Background:
I have a Python application using `pyscard` to read data. There is a seemingly random issue where libusb gets stuck during a transmit call for about 168 seconds which results in a LIBUSB_ERROR_TIMEOUT.

I have attached libusb debug logs with converted timestamps in order to provide some background around the series of events [libusb_debug_logs_12_05_2023.txt](https://github.com/libusb/libusb/files/13642943/libusb_debug_logs_12_05_2023.txt).

The issue occurs near the bottom of the file. At timestamp `2023-12-05 09:45:56` the Python application starts the transmit that ultimately times out. The error I can observe is at `Dec-05 09:55:45`, which is a time period where the device is unresponsive. The logs end with unplugging the device and the issue is resolved once the device is replugged.

Question:
I wanted to confirm if this timeout behavior for 168 seconds is a known pattern and what the recommended recovery path should be since a usb reset from the Python code does not help. Thank you so much!

tormodvolden, 2023-12-12:
I don't know where the 168 seconds come from but I see in the log:
`[add_to_flying_list	arm	timerfd	for	timeout	in	167723ms	(first	in	line)`
and these show up long before the issue occurs. What is the timeout that you specify, if any?

Would it be possible for you try 1.0.27-rc1 and also updated pyusb?

azlamzn, 2023-12-13:
Thanks for the quick response! Which version of pyusb are you referring to here (I am using 1.2.1)?

tormodvolden, 2023-12-13:
So you are not using the stock python3-usb package then. Well, you can try latest pyusb git master, but I don't think anything relevant changed since 1.2.1.

OTOH a lot changed in libusb since 1.0.23. So if we are going to debug this further we need to use latest libusb git or RC.

rubygoldbergmachineinc, 2023-12-14:
Thanks @tormodvolden. Is the latest version for libusb backward compatible with python 3.8?

mcuee, 2023-12-18:
> Thanks @tormodvolden. Is the latest version for libusb backward compatible with python 3.8?

Yes.