libusb/libusb #1376: Windows error on closing while isoc transfers in flight

tormodvolden, 2023-12-12:
Testing a custom firmware (basically a modified audio example) with a modified sam3u_benchmark.c I see these warnings on Windows when I stop the test, and they do not show up on Linux or macOS.

Maybe the application needs fixing, or the error 87 needs special handling, anyway I am filing this here so I remember to look at it later.

```
$ ./stm32_isoc.exe
6387 transfers (total 140514 bytes) in 6408 milliseconds => 21927 bytes/sec
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing

 Releasing interface 0...
Releasing interface 1...
Closing device...
```

HM281099, 2023-12-19:
This problem exists in windows, when using asynchronous interfaces to send data, when close the device,                                                    it is very serious that the system must be restarted,                                                                                             otherwise the program cannot be killed by external means

HM281099, 2023-12-19:
[image]

HM281099, 2023-12-19:
There seems to be a deadlock, causing the program to freeze

tormodvolden, 2023-12-19:
I didn't see any deadlock. @HM281099 can you please attach a full debug log?

HM281099, 2023-12-20:
I added printing to core.c and confirmed that the program was stuck at usbi_backend.close(dev_handle) ，                                                             
using version: V1.0.26.11813.   
[image]
[image]
@tormodvolden

mcuee, 2023-12-20:
@HM281099 

Please do not use screenshot, just copy/paste the full debug log. Thanks.