libusb/libusb #1380: libusb_bulk_transfer error: Input/Output Error

ThanatosDan, 2023-12-14:
`
#include <iostream>
#include <cstring>
#include <csignal>
#include <libusb-1.0/libusb.h>

int main()
{
    libusb_context *context = NULL;
    libusb_device_handle *handle = NULL;
    int error;

    // 初始化libusb
    error = libusb_init(&context);
    libusb_set_option(nullptr, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG);
    if (error < 0) {
        fprintf(stderr, "libusb_init error: %s\n", libusb_strerror(error));
        return 1;
    }
    int vid = 0x2d37;
    int pid = 0x9e02;
    handle = libusb_open_device_with_vid_pid(NULL, vid, pid);
    if (handle == NULL) {
        fprintf(stderr, "libusb_open_device_with_vid_pid error\n");
        libusb_exit(context);
        return 1;
    }
    libusb_claim_interface(handle, 0);
    unsigned char data[] = "Hello, Printer!";
    int transferred;
    error = libusb_bulk_transfer(handle, 1, data, sizeof(data), &transferred, 5000);
    if (error == 0) {
        printf("Data sent to printer\n");
    } else {
        fprintf(stderr, "libusb_bulk_transfer error: %s\n", libusb_strerror(error));
    }
    libusb_release_interface(handle, 0);
    libusb_close(handle);
    libusb_exit(context);

    return 0;
}
`

[image]

`[ 0.003396] [0000420f] libusb: debug [libusb_get_device_list]  
[ 0.003409] [0000420f] libusb: debug [libusb_get_device_descriptor]  
[ 0.003643] [0000420f] libusb: debug [libusb_get_device_descriptor]  
[ 0.003699] [0000420f] libusb: debug [libusb_open] open 1.5
[ 0.003768] [0000420f] libusb: debug [usbi_add_event_source] add fd 7 events 4
[ 0.003824] [0000420f] libusb: debug [libusb_claim_interface] interface 0
[ 0.004448] [0000420f] libusb: debug [libusb_submit_transfer] transfer 0x5599db57b170
[ 0.004528] [0000420f] libusb: debug [add_to_flying_list] arm timer for timeout in 5000ms (first in line)
[ 0.004586] [0000420f] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 16
[ 0.004647] [0000420f] libusb: error [submit_bulk_transfer] submiturb failed, errno=16
[ 0.004699] [0000420f] libusb: debug [submit_bulk_transfer] first URB failed, easy peasy
[ 0.004750] [0000420f] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.004801] [0000420f] libusb: debug [libusb_free_transfer] transfer 0x5599db57b170
libusb_bulk_transfer error: Input/Output Error
[ 0.004906] [0000420f] libusb: debug [libusb_release_interface] interface 0
[ 0.004970] [0000420f] libusb: debug [libusb_close]  
[ 0.005023] [0000420f] libusb: debug [usbi_remove_event_source] remove fd 7
[ 0.005078] [0000420f] libusb: debug [libusb_exit]  
[ 0.005160] [00004212] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.005260] [0000420f] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.005319] [0000420f] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.005371] [0000420f] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.005421] [0000420f] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.005649] [0000420f] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.005704] [0000420f] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.005856] [0000420f] libusb: debug [usbi_remove_event_source] remove fd 3`

tormodvolden, 2023-12-14:
Please give as much information as you can. See https://github.com/libusb/libusb/wiki/Troubleshooting
As an absolute minimum: Which OS version is this? Drivers? What libusb version or checkout? What device is this? If you can reproduce the issue, please attach a complete debug log (LIBUSB_DEBUG=4).

ThanatosDan, 2023-12-14:
OS:Ubuntu 23.04
libusb version:1.0.26
device:VMware® Workstation 17 Pro

ThanatosDan, 2023-12-14:
Sorry!
OS:Ubuntu 23.04
libusb version:1.0.26
device:VMware® Workstation 17 Pro

I added the following code after "libusd_open_device_with_vid_id". The problem will be solved.

if(libusb_kernel_driver_active(handle, 0) == 1) {
    printf("Kernel Driver Active\n");
    if(libusb_detach_kernel_driver(handle, 0) == 0){
        printf("Kernel Driver Detached!\n");
    }
}

I don't understand why, hope there is a good explanation. Thanks.

mcuee, 2023-12-18:
This is a bit strange.

By right, if the kernel driver is still active, then `libusb_claim_interface` should fail.