libusb/libusb #1381: 1.0.27-rc1 compilation error on Windows

sl1200mk2, 2023-12-15:
Hi,
on Wiindows I have a build system based on clang 17.0.6.
When compiling libUSB 1.0.27-rc1 I got the following errors:

$ make  CFLAGS="-O2 -g0"
make  all-recursive
make[1]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
Making all in libusb
make[2]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_windows.lo
In file included from os/events_windows.c:24:
os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/events_windows.c:24:
os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
In file included from os/windows_common.c:30:
os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_common.c:30:
os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
os/windows_common.c:336:9: error: variable 'w' is uninitialized when used here
      [-Werror,-Wuninitialized]
  336 |         UNUSED(w); UNUSED(arch);
      |                ^
./libusbi.h:77:44: note: expanded from macro 'UNUSED'
   77 | #define UNUSED(var)     UNREFERENCED_PARAMETER(var)
      |                                                ^~~
C:/msys64/mingw64/include/winnt.h:1387:43: note: expanded from macro 'UNREFERENCED_PARAMETER'
 1387 | #define UNREFERENCED_PARAMETER(P) {(P) = (P);}
      |                                           ^
os/windows_common.c:332:15: note: initialize the variable 'w' to silence this warning
  332 |         const char *w, *arch;
      |                      ^
      |                       = NULL
os/windows_common.c:336:20: error: variable 'arch' is uninitialized when used here
      [-Werror,-Wuninitialized]
  336 |         UNUSED(w); UNUSED(arch);
      |                           ^~~~
./libusbi.h:77:44: note: expanded from macro 'UNUSED'
   77 | #define UNUSED(var)     UNREFERENCED_PARAMETER(var)
      |                                                ^~~
C:/msys64/mingw64/include/winnt.h:1387:43: note: expanded from macro 'UNREFERENCED_PARAMETER'
 1387 | #define UNREFERENCED_PARAMETER(P) {(P) = (P);}
      |                                           ^
os/windows_common.c:332:22: note: initialize the variable 'arch' to silence this warning
  332 |         const char *w, *arch;
      |                             ^
      |                              = NULL
2 warnings and 2 errors generated.
make[2]: *** [Makefile:684: os/windows_common.lo] Error 1
make[2]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master/libusb'
make[1]: *** [Makefile:491: all-recursive] Error 1
make[1]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
make: *** [Makefile:396: all] Error 2

best regards,
Nicolas

tormodvolden, 2023-12-15:
Thanks for the report. It looks like MinGW provides an UNREFERENCED_PARAMETER macro that doesn't go well with clang's uninitialized variable detection.

Please try to patch libusb/libusbi.h like this:
```
@@ -73,7 +73,7 @@
 #endif
 
 /* The following is used to silence warnings for unused variables */
-#if defined(UNREFERENCED_PARAMETER)
+#if defined(UNREFERENCED_PARAMETER) && !defined(__clang__)
 #define UNUSED(var)    UNREFERENCED_PARAMETER(var)
 #else
 #define UNUSED(var)    do { (void)(var); } while(0)
```

We don't see this error on our CI MSYS clang compilation, but I don't know which version of clang it has.

sl1200mk2, 2023-12-15:
it builds fin. thanks.
I still got pragma's errors:

  CC       os/events_windows.lo
In file included from os/events_windows.c:24:
os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/events_windows.c:24:
os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
In file included from os/windows_common.c:30:
os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_common.c:30:
os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
  CC       os/windows_usbdk.lo
In file included from os/windows_usbdk.c:30:
In file included from os/windows_usbdk.h:27:
os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_usbdk.c:30:
In file included from os/windows_usbdk.h:27:
os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
  CC       os/windows_winusb.lo
In file included from os/windows_winusb.c:34:
In file included from os/windows_winusb.h:30:
os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_winusb.c:34:
In file included from os/windows_winusb.h:30:
os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
In file included from os/windows_winusb.c:34:
os/windows_winusb.h:310:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  310 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_winusb.c:34:
os/windows_winusb.h:404:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  404 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
os/windows_winusb.h:446:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  446 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_winusb.c:34:
os/windows_winusb.h:456:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  456 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
os/windows_winusb.h:698:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  698 | #include <pshpack1.h>
      |          ^
C:/msys64/mingw64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies
      alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from os/windows_winusb.c:34:
os/windows_winusb.h:707:10: warning: the current #pragma pack alignment value is modified in the
      included file [-Wpragma-pack]
  707 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
8 warnings generated.

++

mcuee, 2023-12-15:
No errors but indeed there are quite some pragma related warnings under MSYS2 clang64.

```
 CLANG64 /c/work/libusb/libusb-1.0.27-rc1/clang64
$ clang -v
clang version 17.0.4
Target: x86_64-w64-windows-gnu
Thread model: posix
InstalledDir: C:/msys64/clang64/bin

$ gcc -v
clang version 17.0.4
Target: x86_64-w64-windows-gnu
Thread model: posix
InstalledDir: C:/msys64/clang64/bin
$ ../configure --enable-tests-build --enable-examples-build
configure: loading site script /etc/config.site
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of gcc... gcc3
checking dependency style of g++... gcc3
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/clang64/bin/ld
checking if the linker (C:/msys64/clang64/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /clang64/bin/nm -B
checking the name lister (/clang64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/clang64/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib_fallback
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /clang64/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
[... 180 lines cut ...]
  CC       listdevs.o
  CCLD     listdevs.exe
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark.exe
  CC       testlibusb.o
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/clang64/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/clang64/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
  CC       stress_mt-stress_mt.o
  CCLD     stress_mt.exe
  CC       set_option.o
  CCLD     set_option.exe
  CC       init_context.o
  CCLD     init_context.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/clang64/tests'
make[2]: Entering directory '/c/work/libusb/libusb-1.0.27-rc1/clang64'
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/clang64'
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.27-rc1/clang64'
```

mcuee, 2023-12-15:
> No errors but indeed there are quite some pragma related warnings under MSYS2 clang64 17.0.4.

Same thing after I upgraded to latest MSYS2 installation with clang64 17.0.6.

```
 CLANG64 /c/work/libusb/libusb-1.0.27-rc1/clang64
$ gcc -v
clang version 17.0.6
Target: x86_64-w64-windows-gnu
Thread model: posix
InstalledDir: C:/msys64/clang64/bin
```

@sl1200mk2 

Just wondering which build system you are using since I can not reproduce your originally reported issue (build error). Thanks.

tormodvolden, 2023-12-15:
> +#if defined(UNREFERENCED_PARAMETER) && !defined(__clang__)

At one point we may see this with GCC also so maybe use __GNUC__ instead of __clang__ already?

tormodvolden, 2023-12-15:
Also not sure I understand why it errors out on -Wuninitialized ?

tormodvolden, 2023-12-15:
@sl1200mk2 I suppose using __GNUC__ like in the PR #1382 work fine on your clang?

sl1200mk2, 2023-12-15:
@tormodvolden yes it also compiles fine with __GNUC__
@mcuee same as you except I need to clang --version

++

tormodvolden, 2023-12-18:
@sl1200mk2 Can you please build with make V=1 so that we see what flags are added? I still don't understand why this warning makes the build error out. I see "[-Werror,-Wuninitialized]" in your diagnosis output. I wonder if something like -Werror-uninitialized is default on this version of clang? Does clang -v display its build options?

sl1200mk2, 2023-12-18:
@tormodvolden Hi,
here it is and no, clang -v does not display the build options.

$ make  V=1 CFLAGS="-O2 -g0"
make  all-recursive
make[1]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
Making all in libusb
make[2]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master/libusb'
depbase=`echo core.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT core.lo -MD -MP -MF $depbase.Tpo -c -o core.lo core.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT core.lo -MD -MP -MF .deps/core.Tpo -c core.c  -DDLL_EXPORT -DPIC -o .libs/core.o
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT core.lo -MD -MP -MF .deps/core.Tpo -c core.c -o core.o >/dev/null 2>&1
depbase=`echo descriptor.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT descriptor.lo -MD -MP -MF $depbase.Tpo -c -o descriptor.lo descriptor.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT descriptor.lo -MD -MP -MF .deps/descriptor.Tpo -c descriptor.c  -DDLL_EXPORT -DPIC -o .libs/descriptor.o
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT descriptor.lo -MD -MP -MF .deps/descriptor.Tpo -c descriptor.c -o descriptor.o >/dev/null 2>&1
depbase=`echo hotplug.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT hotplug.lo -MD -MP -MF $depbase.Tpo -c -o hotplug.lo hotplug.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT hotplug.lo -MD -MP -MF .deps/hotplug.Tpo -c hotplug.c  -DDLL_EXPORT -DPIC -o .libs/hotplug.o
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT hotplug.lo -MD -MP -MF .deps/hotplug.Tpo -c hotplug.c -o hotplug.o >/dev/null 2>&1
depbase=`echo io.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT io.lo -M [... cut ...]
libtool: link: ( cd ".libs" && rm -f "libusb-1.0.la" && cp -pR "../libusb-1.0.la" "libusb-1.0.la" )
dlltool  --kill-at --input-def libusb-1.0.def --dllname libusb-1.0.dll --output-lib .libs/libusb-1.0.dll.a
make[2]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master/libusb'
make[2]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
make[2]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
make[1]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'

sl1200mk2, 2023-12-18:
yes,
without the __GNUC__ , make is failing:

$ make  V=1 CFLAGS="-O2 -g0"
make  all-recursive
make[1]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
Making all in libusb
make[2]: Entering directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master/libusb'
depbase=`echo core.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT core.lo -MD -MP -MF $depbase.Tpo -c -o core.lo core.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT core.lo -MD -MP -MF .deps/core.Tpo -c core.c  -DDLL_EXPORT -DPIC -o .libs/core.o
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT core.lo -MD -MP -MF .deps/core.Tpo -c core.c -o core.o >/dev/null 2>&1
depbase=`echo descriptor.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT descriptor.lo -MD -MP -MF $depbase.Tpo -c -o descriptor.lo descriptor.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT descriptor.lo -MD -MP -MF .deps/descriptor.Tpo -c descriptor.c  -DDLL_EXPORT -DPIC -o .libs/descriptor.o
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT descriptor.lo -MD -MP -MF .deps/descriptor.Tpo -c descriptor.c -o descriptor.o >/dev/null 2>&1
depbase=`echo hotplug.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT hotplug.lo -MD -MP -MF $depbase.Tpo -c -o hotplug.lo hotplug.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT hotplug.lo -MD -MP -MF .deps/hotplug.Tpo -c hotplug.c  -DDLL_EXPORT -DPIC -o .libs/hotplug.o
libtool: compile:  /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -O2 -g0 -MT hotplug.lo -MD -MP -MF .deps/hotplug.Tpo -c hotplug.c -o hotplug.o >/dev/null 2>&1
depbase=`echo io.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile /mingw64/bin/clang.exe -DHAVE_CONFIG_H -I. -I..    -std=gnu11 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -O2 -g0 -MT io.lo -MD -MP -MF $depbase.Tpo -c -o io.lo i [... cut ...]
os/windows_common.c:332:22: note: initialize the variable 'arch' to silence this warning
  332 |         const char *w, *arch;
      |                             ^
      |                              = NULL
2 warnings and 2 errors generated.
make[2]: *** [Makefile:684: os/windows_common.lo] Error 1
make[2]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master/libusb'
make[1]: *** [Makefile:491: all-recursive] Error 1
make[1]: Leaving directory '/c/Users/nico/dev/dlight/win64/clangLib64/libusb-master'
make: *** [Makefile:396: all] Error 2

tormodvolden, 2023-12-18:
Thank you both. I eventually realized that we have -Werror-uninitialized in our configure.ac, that's why this kind of warnings turn into errors. Which is good. So I will push the `__GNUC__` change, so that we'll fall back to the same UNUSED macro that is used already for all other (non-MinGW) GCC builds.

commit b2724706 (2023-12-19): libusb.h: Avoid UNREFERENCED_PARAMETER macro on GCC/clang

The UNREFERENCED_PARAMETER macro is provided by MinGW headers but
interferes with -Wuninitialized on recent clang. So instead fall back to
the same UNUSED macro already used elsewhere with GCC.

Relevant discussion:
https://discourse.llvm.org/t/rfc-winnt-hs-unreferenced-parameter-vs-clangs-wunused-value/38526

Fixes #1381
Closes #1382
closed 2023-12-19 by tormodvolden